crankyoldgit/IRremoteESP8266 #1119: Wrong codes OR extra data, with button push on remote. (IR receiving)

dogatonic, 2020-05-13:
I'm having problems with this IR Library while trying to receive IR codes. Using a Lolin (Wemos) D1 mini (the little blue one) and a typical Arduino IR remote "kit", I get really inconsistent results. Sometimes I get the expected code when pushing buttons (i.e. buttons 1 2 & 3) and other times I get unexpected code, and sometimes code just come in randomly.

### Version/revision of the library used
IRremoteESP8266@2.7.6 library
### Hardware
Lolin (Wemos) D1 mini (the little blue one)
IR receiver diode (Unknown, but looks like TSOP4838)
NEC Membrane remote (from typical Arduino kit)
_Receiver diode and remote came bundled together_
### Expected behavior
I expect that pressing the buttons on the remote will produce output in the serial monitor consistent with the values of those buttons pressed.
For comparison, I also have tested an Arduino Uno (SparkFun RedBoard) with the same sketch, same IR receiver and remote, but _with a different IR Library_. Under the exact same test conditions (lighting, remote, distance, button sequence) I get really GOOD results.
Here is some of the serial monitor output when I test the "IRremote library" on an Uno.
[ On SparkFun RedBoard ]
Pushing 1-2-3 on remote, as shown in parentheses:
```
FD08F7  (1)
FD8877  (2)
FD48B7  (3)
FD08F7  (1)
FD8877  (2)
FFFFFFFF
FD48B7  (3)
```
Those are the values that are expected.

### Actual behavior
Using the IRremoteESP8266@2.7.6 library and the Wemos D1 mini, I get very inconsistent results when pushing buttons on the remote. Sometimes I get the expected value, sometimes some seemingly random value.
Here is some of the Serial monitor output when I test the IRremoteESP8266@2.7.6 library.
[  On Lolin (Wemos) D1 mini (the little blue one) ]
Pushing 1-2-3 on remote, as shown in parentheses:
```
2331D92B  (1)
6C27673F  (2)
FD48B7  (3)
FFFFFFFFFFFFFFFF
6E1D760E  (random)
FD08F7  (1)
FD8877  (2)
FFFFFFFFFFFFFFFF
FD48B7  (3)
27D607E9  (1)
FD8877  (2)
DBC8CB72  (3)
FD48B7  (3)
FD48B7  (3)
E160984D  (3)
4CB0FADF  (random)
911C20CF  (random)
24AE7D4E  (random)
FCABFFBF  (random)
4AB0F7B5  (random)
FD08F7  (1)
FFFFFFFFFFFFFFFF
16D367EE  (2)
FD48B7  (3)
```
#### Testing Sketch
```
#include <IRremoteESP8266.h>
#include <IRrecv.h>
#include <IRutils.h>

int RECV_PIN = 14;
IRrecv irrecv(RECV_PIN);
decode_results results;

void setup(){
  Serial.begin(115200);
  irrecv.enableIRIn();
}

void loop() {
  if (irrecv.decode(&results)) {
    serialPrintUint64(results.value, HEX);
    Serial.println("");
    irrecv.resume();
  }
  delay(100);
}
```
#### Output of raw data from IRrecvDumpV2.ino 
```
Timestamp : 000059.835
Library   : v2.7.6
Protocol  : NEC
Code      : 0xFD08F7 (32 Bits)
uint16_t rawData[67] = {9002, 4506,  584, 548,  582, 548,  582, 548,  582, 548,  584, 546,  584, 548,  582, 548,  582, 548,  582, 1656,  584, 1656,  584, 1656,  582, 1658,  582, 1658,  582, 1652,  584, 548,  586, 1
658,  584, 548,  582, 548,  582, 548,  582, 550,  582, 1656,  584, 548,  582, 548,  584, 548,  582, 1656,  584, 1656,  584, 1656,  582, 1656,  582, 548,  582, 1656,  558, 1682,  582, 1656,  582};  // NEC FD08F7
uint32_t address = 0xBF00;
uint32_t command = 0x10;

uint64_t data = 0xFD08F7;

Timestamp : 000071.683
Library   : v2.7.6
Protocol  : NEC
Code      : 0xFD8877 (32 Bits)
uint16_t rawData[71] = {9006, 4504,  584, 550,  558, 572,  584, 546,  584, 542,  584, 548,  562, 572,  582, 548,  582, 548,  582, 1658,  556, 1682,  558, 1682,  582, 1656,  584, 1656,  582, 1658,  558, 572,  584, 1
658,  558, 1682,  582, 548,  584, 548,  558, 572,  582, 1656,  558, 572,  582, 548,  558, 572,  558, 570,  560, 1680,  582, 1656,  558, 1680,  582, 548,  558, 1680,  582, 1656,  582, 1658,  558, 41152,  9006, 2234,
  746};  // NEC FD8877
uint32_t address = 0xBF00;
uint32_t command = 0x11;
uint64_t data = 0xFD8877;

Timestamp : 000074.086
Library   : v2.7.6
Protocol  : UNKNOWN
Code      : 0x5419AA5D (38 Bits)
uint16_t rawData[75] = {146, 31792,  9014, 4504,  584, 546,  584, 546,  584, 546,  586, 546,  584, 546,  584, 546,  586, 544,  586, 546,  586, 1654,  584, 1654,  586, 1654,  584, 1656,  584, 1654,  584, 1656,  584,
 546,  584, 1656,  584, 546,  586, 1654,  586, 546,  586, 546,  584, 1654,  584, 546,  584, 546,  584, 546,  586, 1654,  584, 546,  584, 1656,  584, 1656,  584, 546,  586, 1654,  584, 1656,  584, 1654,  584, 5164,
 56, 35908,  9006, 2232,  586};  // UNKNOWN 5419AA5D
```

### So what's the problem?
The user experience is terrible. Even if I am filtering for just the codes I'm looking for, the User inputs are often useless, and it takes several button presses to get the right code. The experience on the Uno (non ESP8266) with the other IR library is quick and consistent.
I have also tried the IRremoteESP8266@2.7.6 library with a D1 Mini Pro, an additional D1 Mini, and also with a SparkFun Thing Dev. Same results on all the 8266 boards.

#### Circuit diagram and hardware used
_Pretty standard. I could supply if needed._

### Thank you!
Thanks for this library. I appreciate anyone that can help figure out how to make things go smoothly.

#### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
_Yes. As far as I can tell._

crankyoldgit, 2020-05-13:
There are a number of the issues you are hitting:
1. There is some minor IR noise in your environment. e.g. the `UNKNOWN 5419AA5D` has some small garbage at the start of the message.
1. The non-ESP IRremote library is not set up to capture LONG protocols with large gaps in it, e.g. some Air Conditioners. IRremoteESP8266 by default is tune to pick these up. This is controlled in `IRrecvDumpV2` etc via:
https://github.com/crankyoldgit/IRremoteESP8266/blob/09d9ed2a4c9e2d5ed4b18198b9ef7524be3e1379/examples/IRrecvDumpV2/IRrecvDumpV2.ino#L66-L73

There are a number of ways you can tune it to your specific needs.
You can:
* In your code `sample`, change `IRrecv irrecv(RECV_PIN);` to `IRrecv irrecv(RECV_PIN, kRawBuf, kTimeoutMs);`
As per:
https://github.com/crankyoldgit/IRremoteESP8266/blob/09d9ed2a4c9e2d5ed4b18198b9ef7524be3e1379/src/IRrecv.cpp#L131-L151
and
https://github.com/crankyoldgit/IRremoteESP8266/blob/09d9ed2a4c9e2d5ed4b18198b9ef7524be3e1379/src/IRrecv.h#L39-L47

* You can disable matching unknown protocols by either setting `DECODE_HASH` to `false` here:
https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRremoteESP8266.h#L90-L93
or
Call `irrecv.setUnknownThreshold(1000);` to effectively disable it at run-time.
Per:
https://github.com/crankyoldgit/IRremoteESP8266/blob/09d9ed2a4c9e2d5ed4b18198b9ef7524be3e1379/src/IRrecv.cpp#L295-L296

* You can set parameters to be more aggressive when trying to match protocols.
e.g. Change:
```
  if (irrecv.decode(&results)) {
```
to
```
  // decode the IR message, but skip up to 4 pieces of noise before the start of a message **and** 
  // consolidate any noise that's below 150 usecs.
  if (irrecv.decode(&results, NULL, 4, 150)) {
```
Per:
https://github.com/crankyoldgit/IRremoteESP8266/blob/09d9ed2a4c9e2d5ed4b18198b9ef7524be3e1379/src/IRrecv.cpp#L341-L384

* Disable all other protocols you are not expecting. That is, set `DECODE_NEC` to `true` and everything else to `false` per:
https://github.com/crankyoldgit/IRremoteESP8266/blob/09d9ed2a4c9e2d5ed4b18198b9ef7524be3e1379/src/IRremoteESP8266.h#L84-L88

* or, finally, you could check that the matched message is the protocol you actually wanted.
e.g. Change:
```
  if (irrecv.decode(&results)) {
    serialPrintUint64(results.value, HEX);
    Serial.println("");
    irrecv.resume();
  }
```
to
```
  if (irrecv.decode(&results)) {
    if (result.decode_type == NEC) {
      serialPrintUint64(results.value, HEX);
      Serial.println("");
    }
    irrecv.resume();
  }
```

FYI, the `FFFFFFFFFFFFFFFF`-like code is the special `NEC` repeat message.

dogatonic, 2020-05-13:
### SOLUTION!

A replacement IR diode of different source has cleared up my problems and the IRremoteESP8266@2.7.6 library is working just fine now.

### Implementation

When I first noticed the problems I was having with unexpected/inconsistent data coming in, the one thing I could NOT test/check was the IR receiver diode, because I only had 1 on hand. So, I immediately ordered a pack of 10 from Amazon. Then I proceeded my testing with other boards, other libraries, other breadboards & jumpers, other USB cables, anything I could.

The fact that a simple Uno with an _other_ IR library seemed to be working fine was in fact misleading, and pushed me to write a thorough issue report here.

Then this morning (thanks Amazon 1-day delivery!) the package arrived with the new receivers and I pulled one out and tested it with great success on the IRremoteESP8266 library.

Old receiver mishap rate: ~30%
New receiver mishap rate: <1%

[image]
_"Uncle Owen, this R2 unit has a bad motivator, look!"_

I don't really know if the "old" receiver was defective or not - perhaps just "different". It had never been used before, nor removed from it's package, but it has been in my box for over 5 years.

### Thank You crankyoldgit!
I _have_ learned a great deal about the IRremoteESP8266 library's "fine tuning" capabilites thanks to the rapid response from crankyoldgit - even though I have yet to try any of it.

crankyoldgit, 2020-05-13:
haha. The one thing I didn't list (but was going to) was the possibility of a bad/poor IR receiving hardware and/or circuit. Uno's `Vcc` is `5V`,  you may have connected the D1 mini's `3V3` instead of `5V`, and if the module is only a `5V` one, that could have been your problem.

closed 2020-05-13 by crankyoldgit

debagchi1978, 2021-09-26:
Hi,
This is due to the ripple effect created due to the other i/o devices to my wemos d1 mini. As a solution without changing the  IR sensor, I used a capacitor of value 470uf between positive and negative terminal of IR sensor. As a result, the ripple voltage came to a normal which caused the stable value (99%) of IR code.

Hope this helps.
Cheers!
Deb

crankyoldgit, 2021-09-26:
FYI, I've added this tip to https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide#advanced-1