crankyoldgit/IRremoteESP8266 #1121: GREE and other protocols reverting to C from F.

3ricj, 2020-05-14:
_(Please use this template for reporting issues. You can delete what ever is not relevant. Giving us this information will help us help you faster. Please also read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions) & [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide). Your problem may already have an answer there.)_

### Version/revision of the library used
_Typically located in the `library.json` & `src/IRremoteESP8266.h` files in the root directory of the library.
e.g. v2.0.0, or 'master' as at 1st of June, 2017. etc._
Tasmota 0.8.2 (not sure what specific version they have of this library)

### Expected behavior
_What steps did you do and what should it have done?_
1.  Set GREE in F. 
2.  Have the AC operate in F. 

### Actual behavior
_What steps did you do, and what did or didn't actually happen?_
1.  Set GREE in F
2.  Gree receives commands in C. 

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
_Include some raw dumps of what the device saw._

02:30:09 CMD: Group 0, Index 1, Command "IRHVAC", Data "{"Vendor":"GREE", "Power":"On","Mode":"Hot","FanSpeed":3,"Celsius":"Off","Temp":68}"
02:30:10 MQT: stat/tasmota_6A9E68/RESULT = {"IRHVAC":{"Vendor":"GREE","Model":1,"Power":"On","Mode":"Auto","Celsius":"Off","Temp":68,"FanSpeed":"Medium","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}
02:30:10 MQT: tele/tasmota_6A9E68/RESULT = {"IrReceived":{"Protocol":"GREE","Bits":64,"Data":"0x0x28094050002000D0","Repeat":0,"IRHVAC":{"Vendor":"GREE","Model":1,"Power":"On","Mode":"Auto","Celsius":"On","Temp":25,"FanSpeed":"Medium","SwingV":"Auto","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
02:30:25 WIF: Checking connection...

### Steps to reproduce the behavior
_What can we do to (pref. reliably) repeat what is happening?_

#### Example code used
_Include all relevant code snippets or links to the actual code files. Tip: [How to quote your code so it is still readable](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code)._

#### Circuit diagram and hardware used (if applicable)
_Link to an image of the circuit diagram used. Part number of the IR receiver module etc. ESP8266 or ESP32 board type._

nas-ir03w0, using the NAS-IR03W0 IR bridge template, using latest tasomoto-ircustom.bin : https://templates.blakadder.com/neo_coolcam_NAS-IR03W0.html

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
_Yes/No._
Yes. 

### Has this library/code previously worked as expected for you?
_Yes/No. If "Yes", which version last worked for you?_
No

### Other useful information
_More information is always welcome. Be verbose._

It seems the GREE protocol is able to understand F, as when I send commands from my remote in F it does convert them to C in the display:  

If I have my remote pointed at the Tasmota device AND my AC unit, and power it on, the AC unit displays 68F, but the tasmota device shows 20c.

18:01:56 MQT: tele/tasmota_6A9E68/RESULT = {"IrReceived":{"Protocol":"GREE","Bits":64,"Data":"0x0x0904205800200010","Repeat":0,"IRHVAC":{"Vendor":"GREE","Model":2,"Power":"On","Mode":"Cool","Celsius":"On","Temp":20,"FanSpeed":"Auto","SwingV":"Auto","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"On","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}

So unexpectedly, it's understanding F well enough to convert it to C... I guess?   

This is only really a problem because when the AC receives C, it changes the display, and in addition it's less effective resolution than in F as my unit (and likely the protocol) only supports ints.  (specifically, my target temp is 69F, which is 20.5C, but I don't think I can send 20.5C!)

I am able to send IRsend {"Protocol":"GREE","Bits":64,"Data":0x0904205800200010} and have it correctly operate in F, so I do suspect this is an encoding/decoding problem with this library. 

68F:  IRsend  {"Protocol":"GREE","Bits":64,"Data":0x0904205800200010}
69F: IRsend   {"Protocol":"GREE","Bits":64,"Data":"0x0905205800200020"}

A unrelated bug (maybe I should file something else on this) but there is some sort of bug in the IRreceived code  (note the 0x0x twice)

19:07:55 MQT: tele/tasmota_6A9E68/RESULT = {"IrReceived":{"Protocol":"GREE","Bits":64,"Data":"0x0x4904205001200090","Repeat":0,"IRHVAC":{"Vendor":"GREE","Model":2,"Power":"On","Mode":"Cool","Celsius":"On","Temp":20,"FanSpeed":"Auto","SwingV":"Auto","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"On","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}

See this original bug over here:  https://github.com/arendst/Tasmota/issues/8437

crankyoldgit, 2020-05-14:
Thanks for the report & data.
I'll start with the minor thing first, the `0x0x` issue. That is a Tasmota bug. Not this library.
I suggest you log an issue with the just on that minor problem. FYI/Heads-up @s-hadinger 

Now, the rest of it. Currently we don't have protocol level support for Fahrenheit for the GREE protocol. That's because no-one until now had supplied any data using the Fahrenheit settings & temps.

In order to add native support for it, I'm going to have to work out what bits change between Celsius and Fahrenheit, which I can probably deduce from the hex codes you provided. Plus, I'm going to need to know the minimum & maximum temp values in Fahrenheit the remote can produce, and copies of the corresponding hex codes. Provide those, and I can probably reverse engineer how it works etc.

But yes, you've correctly guessed that the library tries to do the right thing for users, it allows users to use F if they want, but converts it to C for A/C units that don't (yet?) support native temps in F.

FYI, as an aside. Only 10% or less of the protocols supported currently have native support for Fahrenheit. You North Americans are certainly in the minority! ;-) Also, when it comes to extra precision, don't hold your breath for it. Of the 3 or so protocols that do have native support, we found one that gave the placebo of sending F, but internally still sent it as integer C. The user swore blind that they honestly could feel the difference between the "half degree C" differences, but they too saw the protocol data and realised the remote was giving them a placebo effect. Any way. That's a digression, if your unit displays the set temp in F on it, then it probably does have that level of resolution.

To summarise, I'm going to need the min and max codes for F (and a description of what they are) in order in order to add support for it. Waiting on your data/response.

3ricj, 2020-05-14:
First of all, thank you for dealing with our strange units.  I'm sorry.  It's terrible.  We are more anti-science every day.   :(

The remote seems to be able to go from 61 to 86F.     This dump hopefully provides all of them to you; please let me know if I should capture them some other way?  

00:48:48 MQT: tele/tasmota_6A9E68/RESULT = {"IrReceived":{"Protocol":"GREE","Bits":64,"Data":"0x0x09002058002000D0","Repeat":0,"IRHVAC":{"Vendor":"GREE","Model":2,"Power":"On","Mode":"Cool","Celsius":"On","Temp":16,"FanSpeed":"Auto","SwingV":"Auto","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"On","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
00:48:51 MQT: tele/tasmota_6A9E68/RESULT = {"IrReceived":{"Protocol":"GREE","Bits":64,"Data":"0x0x09012058002000E0","Repeat":0,"IRHVAC":{"Vendor":"GREE","Model":2,"Power":"On","Mode":"Cool","Celsius":"On","Temp":17,"FanSpeed":"Auto","SwingV":"Auto","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"On","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
00:48:52 MQT: tele/tasmota_6A9E68/RESULT = {"IrReceived":{"Protocol":"GREE","Bits":64,"Data":"0x0x0901205C00200020","Repeat":0,"IRHVAC":{"Vendor":"GREE","Model":2,"Power":"On","Mode":"Cool","Celsius":"On","Temp":17,"FanSpeed":"Auto","SwingV":"Auto","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"On","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
00:48:53 MQT: tele/tasmota_6A9E68/RESULT = {"IrReceived":{"Protocol":"GREE","Bits":64,"Data":"0x0x09022058002000F0","Repeat":0,"IRHVAC":{"Vendor":"GREE","Model":2,"Power":"On","Mode":"Cool","Celsius":"On","Temp":18,"FanSpeed":"Auto","SwingV":"Auto","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"On","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
00:48:55 MQT: tele/tasmota_6A9E68/RESULT = {"IrReceived":{"Protocol":"GREE","Bits":64,"Data":"0x0x0902205C00200030","Repeat":0,"IRHVAC":{"Vendor":"GREE","Model":2,"Power":"On","Mode":"Cool","Celsius":"On","Temp":18,"FanSpeed":"Auto","SwingV":"Auto","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"On","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
00:48:56 MQT: tele/tasmota_6A9E68/RESULT = {"IrReceived":{"Protocol":"GREE","Bits":64,"Data":"0x0x0903205800200000","Repeat":0,"IRHVAC":{"Vendor":"GREE","Model":2,"Power":"On","Mode":"Cool","Celsius":"On","Temp":19,"FanSpeed":"Auto","SwingV":"Auto","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"On","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
00:48:57 MQT: tele/tasmota_6A9E68/RESULT = {"IrReceived":{"Protocol":"GREE","Bits":64,"Data":"0x0x0903205C00200040","Repeat":0,"IRHVAC":{"Vendor":"GREE","Model":2,"Power":"On","Mode":"Cool","Celsius":"On","Temp":19,"FanSpeed":"Auto","SwingV":"Auto","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"On","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
00:48:59 MQT: tele/tasmota_6A9E68/RESULT = {"IrReceived":{"Protocol":"GREE","Bits":64,"Data":"0x0x0904205800200010","Repeat":0,"IRHVAC":{"Vendor":"GREE","Model":2,"Power":"On","Mode":"Cool","Celsius":"On","Temp":20,"FanSpeed":"Auto","SwingV":"Auto","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"On","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
00:49:00 MQT: tele/tasmota_6A9E68/RESULT = {"IrReceived":{"Protocol":"GREE","Bits":64,"Data":"0x0x0905205800200020","Repeat":0,"IRHVAC":{"Vendor":"GREE","Model":2,"Power":"On","Mode":"Cool","Celsius":"On","Temp":21,"FanSpeed":"Auto","SwingV":"Auto","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"On","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
00:49:01 MQT: tele/tasmota_6A9E68/RESULT = {"IrReceived":{"Protocol":"GREE","Bits":64,"Data":"0x0x0905205C00200060","Repeat":0,"IRHVAC":{"Vendor":"GREE","Model":2,"Power":"On","Mode":"Cool","Celsius":"On","Temp":21,"FanSpeed":"Auto","SwingV":"Auto","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"On","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
00:49:03 MQT: tele/tasmota_6A9E68/RESULT = {"IrReceived":{"Protocol":"GREE","Bits":64,"Data":"0x0x0906205800200030","Repeat":0,"IRHVAC":{"Vendor":"GREE","Model":2,"Power":"On","Mode":"Cool","Celsius":"On","Temp":22,"FanSpeed":"Auto","SwingV":"Auto","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"On","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
00:49:04 MQT: tele/tasmota_6A9E68/RESULT = {"IrReceived":{"Protocol":"GREE","Bits":64,"Data":"0x0x0906205C00200070","Repeat":0,"IRHVAC":{"Vendor":"GREE","Model":2,"Power":"On","Mode":"Cool","Celsius":"On","Temp":22,"FanSpeed":"Auto","SwingV":"Auto","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"On","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
00:49:05 MQT: tele/tasmota_6A9E68/RESULT = {"IrReceived":{"Protocol":"GREE","Bits":64,"Data":"0x0x0907205800200040","Repeat":0,"IRHVAC":{"Vendor":"GREE","Model":2,"Power":"On","Mode":"Cool","Celsius":"On","Temp":23,"FanSpeed":"Auto","SwingV":"Auto","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"On","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
00:49:06 MQT: tele/tasmota_6A9E68/RESULT = {"IrReceived":{"Protocol":"GREE","Bits":64,"Data":"0x0x0907205C00200080","Repeat":0,"IRHVAC":{"Vendor":"GREE","Model":2,"Power":"On","Mode":"Cool","Celsius":"On","Temp":23,"FanSpeed":"Auto","SwingV":"Auto","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"On","Filter": [... cut ...] "SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"On","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
00:49:23 MQT: tele/tasmota_6A9E68/RESULT = {"IrReceived":{"Protocol":"GREE","Bits":64,"Data":"0x0x090E2058002000B0","Repeat":0,"IRHVAC":{"Vendor":"GREE","Model":2,"Power":"On","Mode":"Cool","Celsius":"On","Temp":30,"FanSpeed":"Auto","SwingV":"Auto","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"On","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}

3ricj, 2020-05-14:
Incidentally, it does look like something is able to decode them because it does seem to be able to correctly turn them into the accurate temp in C. Note in the log it's showing 16c to 30c - hey, look, that's the same as 61f to 86f!

crankyoldgit, 2020-05-15:
That's because \<drumroll> .. the unit really works in Celsius. After looking at it, it has this notion of "half degrees" elsewhere in the code to try to fudge up F.

I'm currently trying to map this devices F to C alg. because .. well, it's not the traditional/normal conversion.

crankyoldgit, 2020-05-15:
@3ricj & @s-hadinger I think I've correctly added Fahrenheit support to the Gree IR protocol in PR #1124 (Branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1121)

Please download and try it out, and let me know how it goes.

@3ricj You're on your own for integrating it with Tasmota. Perhaps @s-hadinger can help you there.
They (Tasmota) typically are very quick to include the latest version of this library when releases are made. That said, unless you "roll your own", you won't see this "fix" in standard Tasmota build for a while. i.e. You have to wait for this library to do a release (I tend to do one monthly), and then wait for Tasmota to pick it up, and then for them to do an official release.

Till then, you're probably stuck with the "real world" of Celsius, unless you can Engineer/Science it up a notch! :-P

P.S. ProTip: Don't inject or ingest bleach or other disinfectants. Just Sayin'! ;-)
P.P.S. This code is not to be pointed at Carbon Monoxide (CO) detectors. All warranties [(& cats)](https://github.com/crankyoldgit/IRremoteESP8266/issues/1123) are `NULL` & `void` if you do!
P.P.P.S. Same goes for cats. Don't give them bleach or disinfectants. They are dangerous enough as it is.

s-hadinger, 2020-05-15:
Thanks David, always super fast. That's not a tiny patch so I'd rather wait for your next release to update Tasmota.

@3ricj Can you please confirm this solves your issue? If you want to update the lib in Tasmota, that's super simple. Delete and replace all files in `lib/IRRemoteESP8266-vx.x.x` with the content of #1124, then recompile.

3ricj, 2020-05-15:
Wow, amazing.  Also, warm tips about bleach.  This is the greatest PR ever.  I'll compile it tomorrow and give it a shot.  Thanks so much for supporting my strange units.   @crankyoldgit  along the way here I also purchased a "universal AC remote" which claims to support 2000 different models.  (but not my other AC unit).  It does seem to implement GREE in both units.  If this is at all helpful to you I'm happy to mail it to you.  ericjohanson atty gmail. https://www.ebay.com/itm/Universal-A-C-Remote-Control-Fits-Samsung-Panasonic-Sanyo-Chigo-Fujitsu-more/221596252074?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

crankyoldgit, 2020-05-16:
> Thanks David, always super fast. That's not a tiny patch so I'd rather wait for your next release to update Tasmota.

I wasn't suggesting otherwise. ;-)

crankyoldgit, 2020-05-16:
> @crankyoldgit along the way here I also purchased a "universal AC remote" which claims to support 2000 different models. (but not my other AC unit). It does seem to implement GREE in both units. If this is at all helpful to you I'm happy to mail it to you.

@3ricj Thanks for the generous offer, but you can hang on to it for now. I'll keep it in mind though.

Having the remote is probably three quarters of the battle for adding support, but access to the controlled unit is very important. I could emulate this 2000-in-1 remote, but not know if it really works. i.e. The proof is in how & if it actually controls the A/C correctly. There is no way to know if I'm just emulating a bad implementation without the actual A/C unit.

Plus, with the number of A/Cs I've added to this library, it probably supports 2000 models by now already! I've also grown weary of doing full reverse engineering of protocols myself. I now push back onto the person with the remote to do the break-down etc.
e.g. It probably took me about 30+ mins to work out just the differences in temp control on the GREE protocol for Fahrenheit and I had Celsius was already done. All up, getting the code "just right" and passing all the tests & cases was another 4 hours easy.

NiKiZe, 2020-05-18:
Title says "and other protocols" any examples other than Gree?

crankyoldgit, 2020-05-18:
I _think_ he's referring to the linked Tasmota issue. It was one of the Mitsubishi ones.
The "issue" is if we only have native temp support in Celsius, and if the A/C unit displays the temp, and if the user tells the common AC api to use Fahrenheit, the library can't instruct the A/C in native Fahrenheit, so it converts the temp to Celsius, and sends it. Thus displaying the wrong units/numbers, but the correct temp. When the library supports sending "native" Fahrenheit, it does so, and this is not a problem. Hence, we added native Fahrenheit to Gree with his "codes".

TL;DR: No way to fix this without adding native Fahrenheit support for every A/C protocol. I would if I had the codes/protocol breakdown, but alas ... we live in reality. ;-)

closed 2020-05-18 by crankyoldgit

NiKiZe, 2020-05-18:
> TL;DR: No way to fix this without adding native Fahrenheit support for every A/C protocol. I would if I had the codes/protocol breakdown, but alas ... we live in reality. ;-)

Indeed we do

crankyoldgit, 2020-05-20:
@3ricj @s-hadinger FYI, the changes mentioned above have now been included in the latest release (v2.7.7) of the library.

PR #1124 (2020-05-18): Gree: Enable native support for Fahrenheit
* Add support for setting the A/C using degrees Fahrenheit.
* Add unit tests coverage for Fahrenheit.
* Clean up existing unit tests style a bit.
* Update supported devices.

X-Ref: https://github.com/arendst/Tasmota/issues/8437
Fixes #1121
commit 9859cf2a (2020-05-18): Gree: Enable native support for Fahrenheit (#1124)

* Add support for setting the A/C using degrees Fahrenheit.
* Add unit tests coverage for Fahrenheit.
* Clean up existing unit tests style a bit.
* Update supported devices.

X-Ref: https://github.com/arendst/Tasmota/issues/8437
Fixes #1121