crankyoldgit/IRremoteESP8266 #1123: IRremoteESP8266 triggering smoke detector.

3ricj, 2020-05-15:
### Version/revision of the library used

0.8.2 Tasmoto

### Expected behavior
1. Turn on my AC
2.  Smoke detector does not wake up the neighborhood. 

### Actual behavior
1.  Turn on my AC
2.   approximately 1/20 times I would turn on my AC, my smoke detector would go bonkers and go into self test mode, and waking up the cat.  And the neighbors.  "WARNING CARBON MONOXIDE detected in BASEMENT! WARNING!"

#### Output of raw data from IRrecvDumpV2.ino (if applicable)

### Steps to reproduce the behavior
1. Purchase first alert smoke detector with easy to self test mode. 
2.  Purchase high powered IR blaster. 
3.  Configure situation that the IR blaster is right on the edge of the threshold to blast enough IR light into the smoke detector to trigger it's self test.  
4.  Screw around with your AC and get really confused about why it's making CO all the time and setting off the alarm. 
5.  Finally figure out what's causing it -- "EASY SELF TEST, USE ANY TV REMOTE". 
6.  Grumble, and post a bug here to hopefully save other poor souls from this torment. 
7.  Move your IR blaster further away from the FOV of the smoke detector. Don't cover the detector in tape, that's just bad.

#### Example code used

AAAAAAGHHH. 

#### Circuit diagram and hardware used (if applicable)

IR BLASTER ->  SMOKE DETECTOR -> CAT

### I have followed the steps in the [Troubleshooting Guide]

Yes - Maybe this is a FAQ? Or maybe I'm the only one. 

### Has this library/code previously worked as expected for you?

No. 

### Other useful information

https://www.firstalertstore.com/store/products/first-alert-plug-in-digital-display-carbon-monoxide-detector.htm

" This unit features a digital display and exclusive, patented IR remote control test/silence capability which can be operated with most household TV/VCR remote units. "

crankyoldgit, 2020-05-15:
> Or maybe I'm the only one.

HAHA. I think you've nailed it there. Sadly this doesn't qualify as a "Frequent"ly Asked Question ... yet.

Wow, that's a really bad design of theirs. :-(

> Don't cover the detector in tape, that's just bad.

I disagree, partially. You should probably make a shield out of the opaque tape so that it blocks IR light from directly entering your CO detector's IR sensor.

> patented IR remote

Patently crap.

3ricj, 2020-05-15:
I also wanted to share this absurdity with you, because, well, it's patently absurd.

crankyoldgit, 2020-05-15:
Don't worry, it came across loud and clear! ;-)
I did <3 the circuit diagram!

NiKiZe, 2020-05-15:
The world need more of these kind of well described bug reports ❤

closed 2020-05-15 by crankyoldgit