crankyoldgit/IRremoteESP8266 #1129: Build errors with `_IR_ENABLE_DEFAULT_` set to false

NiKiZe, 2020-05-17:
When building tests with `_IR_ENABLE_DEFAULT_` set to `false` it generates build errors.
Do we want this kind of build to work? (in this case to simplify debugging of test)
Are non test builds causing issues as well?

### Version/revision of the library used
741b07141703f5d8038a353c1e94077aa65499ec

### Steps to reproduce the behavior
Set `_IR_ENABLE_DEFAULT_` to false in `src/IRremoteESP8266.h`
Enable specific protocol with `true`
`(cd test; make -j16 ir_Doshisha_test)`

### Expected behavior
build fine

### Actual behavior
getting build errors (warnings also, but that is not a real issue)
```
../src/IRac.cpp: In member function ‘bool IRac::sendAc(stdAc::state_t, const stdAc::state_t*)’:
../src/IRac.cpp:1391:9: warning: unused variable ‘degC’ [-Wunused-variable]
   float degC = desired.celsius ? desired.degrees
         ^~~~
../src/IRac.cpp: In function ‘bool IRAcUtils::decodeToState(const decode_results*, stdAc::state_t*, const stdAc::state_t*)’:
../src/IRac.cpp:2382:44: warning: unused parameter ‘prev’ [-Wunused-parameter]
                      const stdAc::state_t *prev) {
                                            ^~~~
../src/IRac.cpp: In static member function ‘static bool IRac::isProtocolSupported(decode_type_t)’:
../src/IRac.cpp:212:14: warning: statement will never be executed [-Wswitch-unreachable]
       return true;
              ^~~~
ir_LG.o: In function `IRLgAc::calcChecksum(unsigned int)':
../src/ir_LG.cpp:363: undefined reference to `calcLGChecksum(unsigned short)'
ir_Doshisha_test.o: In function `TestDecodeDoshisha_RealExample_Test::TestBody()':
ir_Doshisha_test.cpp:27: undefined reference to `IRsend::sendRaw(unsigned short const*, unsigned short, unsigned short)'
ir_Doshisha_test.cpp:50: undefined reference to `IRsend::sendRaw(unsigned short const*, unsigned short, unsigned short)'
ir_Doshisha_test.cpp:72: undefined reference to `IRsend::sendRaw(unsigned short const*, unsigned short, unsigned short)'
collect2: error: ld returned 1 exit status
make: *** [Makefile:159: ir_Doshisha_test] Error 1

```

#### Suggestion
For tests, we could always enable `SEND_RAW` via the makefile?
but It should be wrong of `ir_LG.cpp` to cause issues.
```
../src/ir_LG.cpp:363: undefined reference to `calcLGChecksum(unsigned short)'
```

crankyoldgit, 2020-05-17:
> When building tests with `_IR_ENABLE_DEFAULT_` set to `false` it generates build errors.
> Do we want this kind of build to work? (in this case to simplify debugging of test)

The "tests" assume everything is enabled. Turning everything off will cause the tests to fail. So, no, this is not something I think needs fixing.

While most `ir_Protocol_test.cpp` files probably only rely on their corresponding `ir_Protocol.cpp` etc files, and `sendRaw()` that is not the case for every file/set of tests. So, just enabling `sendRaw()` won't fix it. If we go down that approach, then we are back to custom Makefiles with build options for each test. Not going to happen. :-)

> Are non test builds causing issues as well?

I believe the Tasmota project turns of 90% of the protocols for its "normal" builds,  (@s-hadinger can confirm) I normally get a Issue lodged from them if there is a problem. I have run tests from time to time. So, I don't think this is a real problem.
Perhaps you should try it. i.e. A normal project, not the `test` ones.

FWIW, when I'm testing a new protocol and I need to turn on `DEBUG` and I don't want all the rest of the garbage, I change the unit test(s) to use `decodeProtocol(...)` rather than `decode()` which tries everything. Far far less debug info to wade through. I even accidentally left on of the unit tests using just the specific decoder by accident recently when I was debugging. You may have noticed. ;-)

NiKiZe, 2020-05-17:
Ok, closing this as won't fix, at least it is documented now :D

closed 2020-05-17 by NiKiZe

s-hadinger, 2020-05-17:
In Tasmota we have a flag to the entire lib if IR is not required. When included we never include zero protocol which would not make sense. I believe this is a minor issue not worth fixing.

NiKiZe, 2020-06-06:
It was decided that we should improve this to some extent in #1172

crankyoldgit, 2020-06-23:
FYI, the code & changes referenced above have been included in the [v2.7.8 release](https://github.com/crankyoldgit/IRremoteESP8266/releases/tag/v2.7.8) of the library.