crankyoldgit/IRremoteESP8266 #1134: Can help join HITACHI RAS series remote

leo12757, 2020-05-21:
My air conditioner is Hitachi RAS-22NK, which is unrecognizable, can help to join?
Thanks.
Version of the library : 2.7.2

###[Power On / cool / Celsius: 26 / Fan: min (min / low / mid / hi) ]###
Protocol  : UNKNOWN
Code      : 0x919B8582 (346 Bits)
uint16_t rawData[691] = {3410, 1624,  482, 1226,  458, 464,  458, 462,  456, 464,  460, 460,  462, 460,  460, 462,  460, 460,  460, 462,  460, 462,  456, 466,  458, 462,  456, 1228,  462, 460,  460, 462,  460, 462,  458, 462,  456, 466,  456, 464,  458, 464,  456, 466,  460, 460,  460, 462,  460, 460,  458, 464,  456, 464,  458, 464,  456, 464,  460, 462,  454, 466,  458, 1226,  460, 462,  456, 1226,  456, 1230,  456, 1228,  456, 1230,  458, 1228,  456, 1228,  456, 464,  456, 1230,  454, 1230,  458, 1228,  458, 1226,  454, 1230,  454, 1230,  454, 1230,  458, 1228,  458, 1230,  452, 468,  456, 464,  454, 468,  456, 464,  418, 502,  456, 466,  458, 464,  458, 462,  458, 464,  454, 468,  456, 1230,  454, 1230,  454, 466,  456, 464,  456, 1228,  456, 1230,  456, 1230,  456, 1230,  456, 464,  458, 464,  458, 1228,  456, 1230,  456, 464,  454, 466,  456, 466,  456, 464,  458, 462,  458, 1228,  458, 1228,  458, 464,  454, 468,  456, 1228,  458, 1228,  458, 1228,  456, 1228,  456, 464,  454, 468,  456, 1228,  458, 1226,  458, 464,  458, 1226,  456, 1230,  454, 468,  458, 462,  452, 1232,  460, 462,  458, 460,  460, 462,  456, 466,  458, 462,  456, 1230,  456, 1228,  460, 462,  460, 1226,  458, 1228,  458, 1226,  460, 460,  456, 468,  458, 464,  456, 1226,  460, 462,  458, 1228,  456, 1228,  458, 462,  462, 1224,  460, 1226,  458, 1226,  460, 464,  458, 1228,  456, 462,  460, 462,  460, 1226,  460, 460,  462, 460,  458, 462,  458, 462,  456, 464,  460, 460,  462, 460,  462, 460,  460, 1224,  462, 1224,  462, 1224,  458, 1226,  462, 1224,  460, 1224,  400, 1284,  446, 1240,  446, 476,  458, 462,  464, 458,  462, 460,  460, 460,  458, 462,  462, 462,  456, 464,  460, 1226,  460, 1226,  460, 1224,  462, 1224,  460, 1224,  460, 1224,  460, 1224,  462, 1224,  462, 460,  460, 462,  460, 460,  458, 464,  458, 462,  458, 464,  458, 462,  460, 460,  462, 1224,  460, 1226,  458, 1228,  456, 1226,  462, 1222,  460, 1228,  458, 1226,  460, 1226,  460, 460,  458, 462,  460, 462,  460, 460,  460, 462,  460, 462,  458, 464,  460, 458,  460, 1226,  456, 1228,  462, 1224,  460, 1224,  458, 1228,  458, 1226,  458, 1228,  462, 1224,  460, 462,  460, 462,  458, 464,  460, 460,  456, 466,  458, 462,  460, 460,  462, 458,  460, 1224,  460, 1224,  458, 1226,  460, 1224,  460, 1226,  460, 1226,  458, 1228,  456, 1230,  456, 1228,  462, 1224,  460, 460,  458, 462,  458, 1228,  456, 466,  458, 462,  454, 468,  458, 462,  458, 462,  460, 1226,  456, 1228,  458, 464,  420, 1264,  458, 1228,  458, 1228,  456, 1228,  454, 468,  456, 464,  456, 466,  456, 1228,  460, 1226,  456, 1230,  456, 1228,  456, 464,  456, 1230,  458, 1226,  458, 1226,  452, 468,  456, 466,  376, 546,  456, 466,  456, 464,  456, 466,  458, 464,  458, 464,  456, 466,  424, 496,  456, 464,  416, 504,  454, 1230,  454, 1232,  456, 1228,  456, 1228,  456, 1230,  456, 1230,  454, 1230,  460, 1226,  426, 496,  424, 496,  456, 466,  374, 546,  454, 468,  374, 544,  458, 464,  456, 464,  458, 1228,  424, 1262,  454, 1232,  426, 1258,  458, 1228,  426, 1260,  454, 1230,  456, 1228,  426, 496,  374, 546,  426, 494,  426, 496,  424, 496,  426, 496,  424, 496,  456, 1230,  456, 1230,  456, 1228,  458, 1228,  456, 1230,  456, 1230,  456, 1230,  424, 1260,  458, 464,  426, 1258,  456, 1230,  422, 500,  456, 466,  418, 504,  424, 496,  426, 496,  456, 464,  420, 500,  454, 468,  456, 1230,  424, 1260,  420, 1264,  422, 1264,  418, 1266,  420, 1264,  420, 500,  456, 466,  426, 494,  454, 468,  456, 464,  420, 1266,  452, 470,  454, 466,  374, 1310,  456, 1228,  456, 1230,  452, 1232,  420, 1266,  424, 498,  456, 1230,  456, 1230,  456, 466,  450, 470,  456, 464,  376, 546,  456, 466,  422, 498,  426, 496,  458, 464,  424, 1260,  454, 1232,  454, 1232,  454, 1232,  458, 1228,  422, 1262,  456, 1228,  454, 1230,  456, 468,  416, 504,  422, 498,  424, 498,  454, 466,  456, 466,  422, 500,  456, 466,  374, 1310,  418, 1266,  424, 1262,  422, 1262,  376, 1310,  376, 1308,  422, 1264,  424, 1262,  350};  // UNKNOWN 919B8582

###[Power On / cool / Celsius: 27 / Fan: min (min / low / mid / hi) ]###
Protocol  : UNKNOWN
Code      : 0x69C87D5C (346 Bits)
uint16_t rawData[691] = {3352, 1704,  404, 1284,  402, 518,  402, 518,  402, 518,  402, 520,  402, 518,  402, 518,  446, 474,  402, 518,  402, 522,  400, 518,  402, 520,  402, 1282,  444, 476,  402, 518,  404, 518,  404, 518,  402, 548,  372, 518,  402, 520,  402, 518,  402, 520,  402, 518,  404, 518,  402, 520,  402, 518,  402, 520,  400, 518,  404, 518,  402, 518,  402, 1282,  404, 518,  404, 1282,  402, 1284,  400, 1282,  402, 1284,  444, 1240,  450, 1238,  402, 518,  402, 1282,  402, 1282,  402, 1284,  402, 1282,  448, 1236,  404, 1282,  402, 1282,  402, 1282,  404, 1284,  402, 520,  402, 518,  404, 518,  402, 518,  404, 518,  402, 518,  404, 518,  402, 518,  402, 518,  442, 482,  400, 1282,  402, 1284,  402, 518,  402, 518,  402, 1282,  404, 1286,  398, 1282,  402, 1282,  402, 518,  402, 518,  402, 1282,  404, 1282,  404, 518,  404, 518,  404, 516,  404, 518,  402, 518,  404, 1284,  402, 1282,  404, 520,  402, 518,  404, 1282,  404, 1282,  402, 1282,  404, 1282,  404, 520,  402, 518,  402, 1284,  402, 1282,  402, 520,  400, 518,  404, 518,  404, 1282,  404, 520,  402, 518,  404, 520,  400, 1282,  404, 518,  404, 1282,  402, 1282,  404, 518,  404, 1282,  402, 1282,  404, 1282,  402, 518,  404, 1282,  402, 520,  402, 520,  402, 1284,  402, 1282,  404, 518,  402, 1282,  402, 1282,  404, 516,  404, 1282,  402, 1284,  402, 520,  400, 520,  400, 1282,  404, 518,  402, 518,  402, 1282,  402, 518,  402, 520,  402, 518,  404, 520,  402, 518,  404, 520,  402, 518,  404, 518,  404, 1282,  404, 1282,  402, 1282,  402, 1282,  402, 1282,  404, 1282,  404, 1284,  402, 1284,  402, 518,  404, 518,  402, 518,  404, 518,  402, 518,  408, 520,  396, 520,  400, 518,  404, 1282,  404, 1284,  402, 1282,  402, 1282,  402, 1280,  404, 1282,  404, 1282,  446, 1242,  402, 518,  402, 518,  404, 518,  404, 518,  404, 518,  402, 518,  404, 520,  402, 516,  404, 1282,  404, 1286,  400, 1286,  400, 1282,  404, 1280,  448, 1236,  404, 1282,  404, 1282,  444, 478,  402, 522,  400, 518,  452, 470,  404, 516,  452, 472,  402, 516,  450, 472,  402, 1280,  404, 1284,  446, 1236,  404, 1282,  444, 1248,  396, 1280,  404, 1280,  448, 1238,  452, 472,  444, 478,  400, 516,  404, 518,  444, 476,  446, 478,  402, 520,  450, 470,  448, 1236,  444, 1240,  446, 1238,  404, 1282,  446, 1238,  404, 1280,  448, 1238,  404, 1282,  404, 1282,  402, 1282,  404, 518,  402, 520,  402, 1282,  402, 518,  404, 518,  402, 520,  402, 518,  402, 518,  404, 1282,  402, 1284,  428, 518,  402, 1284,  402, 1284,  402, 1282,  404, 1284,  402, 518,  444, 480,  402, 518,  404, 1282,  404, 1282,  404, 1282,  404, 1282,  444, 480,  400, 1284,  402, 1284,  402, 1282,  402, 518,  442, 480,  448, 472,  452, 472,  442, 476,  450, 472,  402, 520,  402, 520,  400, 518,  448, 474,  444, 480,  456, 462,  448, 1236,  448, 1236,  448, 1238,  446, 1240,  444, 1240,  450, 1236,  446, 1240,  464, 1222,  402, 518,  450, 472,  450, 472,  458, 462,  450, 472,  444, 476,  462, 460,  462, 460,  462, 1222,  460, 1226,  460, 1224,  462, 1224,  462, 1224,  460, 1224,  462, 1224,  462, 1224,  448, 472,  456, 464,  460, 462,  462, 460,  458, 462,  462, 462,  458, 462,  464, 1222,  460, 1224,  450, 1236,  462, 1224,  462, 1222,  446, 1240,  402, 1284,  446, 1240,  462, 458,  462, 1224,  460, 1226,  460, 460,  488, 434,  460, 462,  462, 460,  462, 460,  460, 460,  464, 460,  460, 460,  456, 1228,  446, 1236,  464, 1226,  462, 1222,  458, 1228,  458, 1226,  460, 462,  460, 460,  460, 462,  462, 460,  460, 460,  460, 1226,  460, 460,  464, 458,  458, 1228,  458, 1226,  460, 1224,  460, 1226,  458, 1226,  460, 460,  462, 1224,  462, 1222,  462, 460,  458, 462,  462, 460,  456, 462,  462, 460,  458, 464,  460, 460,  462, 460,  460, 1226,  460, 1226,  460, 1224,  462, 1224,  460, 1224,  462, 1224,  460, 1224,  462, 1224,  462, 460,  456, 466,  458, 462,  460, 460,  460, 486,  460, 460,  460, 462,  458, 462,  462, 1224,  460, 1226,  458, 1226,  460, 1224,  454, 1230,  460, 1224,  456, 1228,  462, 1224,  372};  // UNKNOWN 69C87D5C

###[Power Off / cool / Celsius: 27 / Fan: min (min / low / mid / hi) ]###
Protocol  : UNKNOWN
[... cut ...] 404, 516,  404, 1280,  404, 1280,  406, 1280,  404, 1282,  404, 1282,  404, 520,  402, 1284,  400, 1284,  400, 516,  404, 516,  406, 518,  404, 518,  404, 522,  400, 518,  402, 518,  404, 518,  404, 1280,  404, 1280,  406, 1282,  404, 1282,  404, 1280,  404, 1286,  398, 1282,  404, 1282,  404, 516,  404, 516,  404, 518,  402, 518,  404, 522,  400, 516,  404, 518,  402, 518,  404, 1282,  402, 1282,  404, 1280,  406, 1282,  404, 1280,  404, 1280,  406, 1280,  404, 1280,  378};  // UNKNOWN 9CAE483A

leo12757, 2020-05-21:
I use auto_analyse_raw_data.py program in the tools directory to decode it.
What the next step?
Thanks.

### [Power off / Coll / Celsius:27 / Fan: min (min / low / mid / hi) ] ###
Found 691 timing entries.
Potential Mark Candidates:
[3324, 454]
Potential Space Candidates:
[1704, 1314, 550]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kHdrMark   = 3324
kHdrSpace  = 1704
kBitMark   = 409
kOneSpace  = 1276
kZeroSpace = 509

Decoding protocol based on analysis so far:

kHdrMark+kHdrSpace+10000000000010000000000000000010111111011111111100000000001100111100110000011001111001101100100000110111001101101100100100000000111111110000000011111111000000001111111100000000111111110000000011111111110010000011011110000111011110000000000011111111000000001111111100000001111111101100000000111111000001001111101100000000111111110000000011111111
  Bits: 344
  Hex:  0x80080002FDFF0033CC19E6C83736C900FF00FF00FF00FF00FFC837877800FF00FF01FEC03F04FB00FF00FF (MSB first)
        0xFF00FF00DF20FC037F80FF00FF001EE1EC13FF00FF00FF00FF00FF00936CEC13679833CC00FFBF40001001 (LSB first)
  Dec:  17922332463837785420140169386043796342223843319810404623017388634242109848230767105516879794782477418751 (MSB first)
        35696476512425258205595514046350053903166542491461064923571277426273739969478847223340843865540216754177 (LSB first)
  Bin:  0b10000000000010000000000000000010111111011111111100000000001100111100110000011001111001101100100000110111001101101100100100000000111111110000000011111111000000001111111100000000111111110000000011111111110010000011011110000111011110000000000011111111000000001111111100000001111111101100000000111111000001001111101100000000111111110000000011111111 (MSB first)
        0b11111111000000001111111100000000110111110010000011111100000000110111111110000000111111110000000011111111000000000001111011100001111011000001001111111111000000001111111100000000111111110000000011111111000000001111111100000000100100110110110011101100000100110110011110011000001100111100110000000000111111111011111101000000000000000001000000000001 (LSB first)

Total Nr. of suspected bits: 344

crankyoldgit, 2020-05-21:
> What the next step?

https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-IR-protocol
&
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol

crankyoldgit, 2020-05-23:
FYI, the latest `master` branch should now offer you basic send and receive ability for this protocol.

leo12757, 2020-05-23:
dear @crankyoldgit 
i creat this spreadsheets, hope it helps.
wind direction setting not been captured yet.

https://docs.google.com/spreadsheets/d/1LPd8K9V437oyEMZT6JDv5LlPXh61RPmgeoVcHLWWr7k/edit?usp=sharing

crankyoldgit, 2020-05-24:
Thanks. I've had a look at the spreadsheet.
This may feel horrible, but can you please collect the data using the latest `master` branch please?
That is, use the `uint8_t state[]` hex array etc data for your spreadsheet, rather than the `auto_analyse_raw_data.py` output.

This is as per: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#create-a-spreadsheet

leo12757, 2020-05-24:
ok, i will do this. Thanks.

leo12757, 2020-05-24:
> Thanks. I've had a look at the spreadsheet.
> This may feel horrible, but can you please collect the data using the latest `master` branch please?
> That is, use the `uint8_t state[]` hex array etc data [...]

Do you mean to use hexadecimal rather than binary?

crankyoldgit, 2020-05-24:
> I use newest 'auto_analyse_raw_data.py' in the tools directory to decode it.

No, I'm suggesting you use the latest `IRrecvDumpV2` (or V3) in conjunction with the latest version of the `master` branch. It should detect as `HITACHI_AC344` now, rather than `UNKNOWN`.

> Do you mean to use hexadecimal rather than binary?

Hexadecimal is probably easiest for you when using `state[]` etc from IRrecvDumpV(2 or 3). But you can use which ever you prefer. I can read both.

leo12757, 2020-05-25:
> No, I'm suggesting you use the latest `IRrecvDumpV2` (or V3) in conjunction with the latest version of the `master` branch. It should detect as `HITACHI_AC344` now, rather than `UNKNOWN`. [...]

As per our discussion, use IRrecvDumpV2 with latest version of th master branch to capture.
Use the uint8_t state[] hex array data for spreadsheet.
Thank you for your help.
Spreadsheet: https://docs.google.com/spreadsheets/d/1LPd8K9V437oyEMZT6JDv5LlPXh61RPmgeoVcHLWWr7k/edit?usp=sharing

crankyoldgit, 2020-05-25:
Thanks. Looks good. I'll try to start on this tomorrow.

leo12757, 2020-05-25:
> Thanks. Looks good. I'll try to start on this tomorrow.

Thank you for your help!

crankyoldgit, 2020-05-26:
Good news, this protocol appears to be just a shorter version of `HITACHI_AC424` so I could duplicate the code.

Bad news, is that in order to re-use the code, I had to change the bit ordering to LSB First order.
That means the data in your spreadsheet is invalid (again). However, hopefully you don't need to re-do it because, if we are lucky, it should "just work" now.

Can you please download PR #1147 / Branch https://github.com/crankyoldgit/IRremoteESP8266/tree/Hitachi344detailed and let us know if it works okay?

leo12757, 2020-05-26:
> Good news, this protocol appears to be just a shorter version of `HITACHI_AC424` so I could duplicate the code.
>
> Bad news, is that in order to re-use the code, I had to change the bit ordering to [...]

ok, I will test and report back immediately.

crankyoldgit, 2020-05-26:
> ok, I will test and report back immediately.

Looking forward to it.

crankyoldgit, 2020-05-26:
FYI, I've updated that branch for increased compatibility when using the class methods.

leo12757, 2020-05-26:
Hi, @crankyoldgit the test results below:

Protocol: [HITACHI_AC_344 | HITACHI_AC_424] - work
Model: select [Default]
Power: [On | Off] - no function.
Mode: [Off | Auto | Cool | Hot | Dry | fan_only] - work.
Temp: work.
Fan: [Auto, Min, Low, Medium, High, Max] - work(High is same to Max).
Swing(V): [Auto | Off] - work
Swing(H): no function.
Quiet, Turbo, Econo, Light, Beep: no function.
Filter, Clean, Force resend: not test

All in all, it is almost controllable.
Thank you for your help, grateful.

crankyoldgit, 2020-05-26:
Thanks for testing and coming back with feedback.

> Power: [On | Off] - no function.

Okay, I'll look into why that isn't working tomorrow.

> Swing(H): no function.
Quiet, Turbo, Econo, Light, Beep: no function.

Correct. They are not (yet) supported. If you want them to work, you'll need to supply data and analysis.

leo12757, 2020-05-26:
> Thanks for testing and coming back with feedback.
>
>
> Okay, I'll look into why that isn't working tomorrow.
>
>
> Correct. They are not (yet) supported. If you want them to work, you'll need to supply [...]

It already can control on / off, i can control modes off option to turn off.
As for swing control, I will keep updating.
Thanks again.

crankyoldgit, 2020-05-26:
> > > Power: [On | Off] - no function.
> > 
> > 
> > Okay, I'll look into why that isn't working tomorrow.
> 
> It already can control on / off, i can control modes off option to turn off.

I'm not sure what you're saying here.

Can you please explain the entire situation again? Your comments seem conflicting to me.

leo12757, 2020-05-26:
> > > > Power: [On | Off] - no function.
> > > 
> > > 
> > > Okay, I'll look into why that isn't working tomorrow.
> > 
> > 
> > It already can control on / off, i can control modes off option to turn off.
> 
> I'm not sure what you're saying here.
> 
> Can you please explain the entire situation again? Your comments seem conflicting to me.

On "Aircon" control web, there are these items:
Power: [On | Off]
Mode: [Off | Auto | Cool | Heat | Dry | fan_only]
In "Power" items, On | Off are no function, but "Mode" items can turn air conditioning to Off when select from Auto / Cool / Heat / Dry / fan_only to Off.

crankyoldgit, 2020-05-27:
Ahh.
You probably need/want to adjust this setting then. Home Assistant (HA) mode is on by default.
https://github.com/crankyoldgit/IRremoteESP8266/blob/232ec687291832d2bc4aa2e524bd2986f4dc7992/examples/IRMQTTServer/IRMQTTServer.h#L106-L120

closed 2020-05-27 by crankyoldgit

leo12757, 2020-05-27:
> Ahh.
> You probably need/want to adjust this setting then. Home Assistant (HA) mode is on by default.
> https://github.com/crankyoldgit/IRremoteESP8266/blob/232ec687291832d2bc4aa2e524bd2986f4dc7992/examples/IRMQTTServer/IRMQTTServer.h#L106-L120

Thank you very much.

BTW, i update the IR table for Swing(H&V), using master branch  "IRrecvDumpV3".
Can you help join it ?
Thank you again, and gratefully.
https://docs.google.com/spreadsheets/d/1LPd8K9V437oyEMZT6JDv5LlPXh61RPmgeoVcHLWWr7k/edit?usp=sharing

crankyoldgit, 2020-05-27:
Can you please capture a state with both SwingV & SwingH on at the same time?

leo12757, 2020-05-27:
> Can you please capture a state with both SwingV & SwingH on at the same time?

SwingV and SwingH are two button on air conditioner remote control.
Can you tell me how to do it, i will do it.

crankyoldgit, 2020-05-27:
> > Can you please capture a state with both SwingV & SwingH on at the same time?
> 
> SwingV and SwingH are two button on air conditioner remote control.
> Can you tell me how to do it, i will do it.

I'm not sure I understand. Can you turn on SwingV, and the press the SwingH button?
Or does the remote not allow both on at the same time?

leo12757, 2020-05-27:
> I'm not sure I understand. Can you turn on SwingV, and the press the SwingH button?
> Or does the remote not allow both on at the same time?

okey, i try and capture it.

leo12757, 2020-05-27:
I think i cannot complete this operation.
When i push up/down button, swing will up/down move(IRrecvDumpV3 get first command).
Next i push right/left button, swing will right/ left move, now swing V and H are moveing(IRrecvDumpV3 get second command).
Third step, i only can push one button (up/down or right/left) to control one of them to stop, but the other direction swing will still move(IRrecvDumpV3 get third command).
Four step i push the left button(up/down or right/left), swing will stop all(IRrecvDumpV3 get third command).
I try push up/down and right/ left button at the same time, air conditioner no no response, swing H and V are still move.
Is helpful...?

crankyoldgit, 2020-05-27:
@jamsinclair Does your `Hitachi424` remote track SwingV in `state[37]` like `Hitachi344` does?
e.g.
https://docs.google.com/spreadsheets/d/1LPd8K9V437oyEMZT6JDv5LlPXh61RPmgeoVcHLWWr7k/edit#gid=1144957158&range=AS84:AS87

crankyoldgit, 2020-05-27:
@leo12757 I think I've added SwingH & improved SwingV based on your data/feedback.
Can you please try out PR #1148 / Branch https://github.com/crankyoldgit/IRremoteESP8266/tree/Hitachi344Swing and let me know how it goes?

crankyoldgit, 2020-05-27:
Can you also give me the model number of the A/C wall unit too please? This is so I can add it to the list of supported models.

leo12757, 2020-05-27:
> @leo12757 I think I've added SwingH & improved SwingV based on your data/feedback.
> Can you please try out PR #1148 / Branch https://github.com/crankyoldgit/IRremoteESP8266/tree/Hitachi344Swing and let me know how it goes?

OK，I will test and report quickly.
Thanks a lot.

> Can you also give me the model number of the A/C wall unit too please? This is so I can add it to the list of supported models.

My A/C model number is HITACHI RAS-22NK(2018).

crankyoldgit, 2020-05-27:
What is the model number of your remote?

leo12757, 2020-05-27:
> What is the model number of your remote?

There are no description in the manual and remote.
According to the information I found on Internet, it is RF11T1.

crankyoldgit, 2020-05-28:
Thanks. I've updated the PR with that info.
How did your testing of the swing features go?

leo12757, 2020-05-28:
> Thanks. I've updated the PR with that info.
> How did your testing of the swing features go?

Sorry, These two days are not at home because of work.
Tomorrow will back, I will test and return as soon as I get home.

crankyoldgit, 2020-05-28:
Cool. Didn't mean to rush you. Looking forward to your feedback.

leo12757, 2020-05-28:
@crankyoldgit 
I test the PR #1148 / Branch results below:

Power, Mode, Temp and Fan are working at HITACHI_AC_344 & HITACHI_AC_424 protocol.

But the swinging behavior is as follows:
HITACHI_AC_344 - Swing(V) are all not working .
HITACHI_AC_344 - Swing(H) (Off | Auto | LeftMax | Left | Middle | Right | RightMax) are working, Swing(H) (wide)  no function(air conditioner controller haven't this option).

HITACHI_AC_344 - Swing(V) (Off | Auto) are working, Swing(V) (Highest | High | Middle | Low | Lowest) no function(air conditioner controller haven't this option).
HITACHI_AC_344 - Swing(H) are all not working .

Sorry to keep you waiting.

crankyoldgit, 2020-05-28:
> HITACHI_AC_344 - Swing(V) (Off | Auto) are working, Swing(V) (Highest | High | Middle | Low | Lowest) no function(air conditioner controller haven't this option).
> HITACHI_AC_344 - Swing(H) are all not working .

I assume you mean 424 for these?

leo12757, 2020-05-28:
> > HITACHI_AC_344 - Swing(V) (Off | Auto) are working, Swing(V) (Highest | High | Middle | Low | Lowest) no function(air conditioner controller haven't this option).
> > HITACHI_AC_344 - Swing(H) are all not working .
> 
> I assume you mean 424 for these?

oh, i'm sorry, i mean...

HITACHI_AC_344 - Swing(V) are all not working .
HITACHI_AC_344 - Swing(H) (Off | Auto | LeftMax | Left | Middle | Right | RightMax) are working, Swing(H) (wide) no function(air conditioner controller haven't this option).

HITACHI_AC_424 - Swing(V) (Off | Auto) are working, Swing(V) (Highest | High | Middle | Low | Lowest) no function(air conditioner controller haven't this option).
HITACHI_AC_424 - Swing(H) are all not working .

crankyoldgit, 2020-05-29:
> HITACHI_AC_344 - Swing(V) are all not working .

I'll look into that.
Can you capture me a sequence from the orig remote. (i.e. record what you did on the remote & the `state[]`) of:

(start with all swings "off"/middle etc)
1. Temp up one degree.
1. Swing V on.
1. Temp down one degree.
1. Swing V off.
1. Temp up one degree.

That should hopefully give me the data to make it work successfully. (Fingers crossed)

Re: The rest. That sounds correct as far as what is implemented.

[Q] You keep listing both the 344 & 424 versions. Do you have two AirCon units, one of each?!
i.e. Are you testing them both on separate A/C units, or are you trying both protocols on the same A/C unit?

leo12757, 2020-05-29:
`

> > HITACHI_AC_344 - Swing(V) are all not working .
> 
> I'll look into that.
> Can you capture me a sequence from the orig remote. (i.e. record what you did on the remote & the `state[]`) of:
> 
> (start with all swings "off"/middle etc)
> 
> 1. Temp up one degree.
> 2. Swing V on.
> 3. Temp down one degree.
> 4. Swing V off.
> 5. Temp up one degree.
> 
> That should hopefully give me the data to make it work successfully. (Fingers crossed)
> 
> Re: The rest. That sounds correct as far as what is implemented.

OK, i will capture that and report quickly.

> [Q] You keep listing both the 344 & 424 versions. Do you have two AirCon units, one of each?!
> i.e. Are you testing them both on separate A/C units, or are you trying both protocols on the same A/C unit?

No, i only have 1 AirCon units, i report 344 & 424 just because I found that 424 protocol option can also control my AirCon.
Do I need to test only one protocol (323)?
Follow your instructions.

leo12757, 2020-05-29:
@crankyoldgit, 
follow your instructions, update at spreadsheet "Code Table v4".
https://docs.google.com/spreadsheets/d/1LPd8K9V437oyEMZT6JDv5LlPXh61RPmgeoVcHLWWr7k/edit?usp=sharing

crankyoldgit, 2020-05-30:
@leo12757 I've had a look at the data, thanks for that!

There are a couple of possibilities on how to fix it. This is attempt number 1.

I've updated the PR & branch.
Can you please redownload it and test it out?
Things that *might* go wrong are that it will always toggle SwingV with each subsequent command. Watch out for that behaviour please.

leo12757, 2020-05-30:
> @leo12757 I've had a look at the data, thanks for that!
> 
> There are a couple of possibilities on how to fix it. This is attempt number 1.
> 
> I've updated the PR & branch.
> Can you please redownload it and test it out?

Okay, I will test it and still download the Hitachi344Swing branch, right? 

> Things that _might_ go wrong are that it will always toggle SwingV with each subsequent command. Watch out for that behaviour please.

Can you explain this part in detail, I am not as professional as you :’D
Which part should i pay attention to.
Thank you very much.

crankyoldgit, 2020-05-30:
> Okay, I will test it and still download the Hitachi344Swing branch, right?

Correct.

> Can you explain this part in detail, I am not as professional as you :’D
Which part should i pay attention to.

It may be the case that the way I've "fixed" it may cause the Vertical Swing to swap from on to off, and off to on with each each message IRMQTTServer sends to it.
i.e. Assume vertical swing is off at the start.
1. Turn unit on.
2. +1 to temp. (Swing V might turn on)
3. +1 to temp. (Swing V might turn off)
4. +1 to temp. (Swing V might turn on again etc etc)

There are two parts in the data that might cause swingV to be on. Depending how they actually control the unit, it could now accidentally keep triggering a change of swingV mode.

I've taken a 50/50 chance with how it works. I need you to confirm if the solution works, or if I need to go to "Plan B".

That clearer?

leo12757, 2020-05-30:
> Correct.
>
>
> It may be the case that the way I've "fixed" it may cause the Vertical Swing to swap from on to off, and off to on with each each message IRMQTTServer sends to it.
> i.e. Assume vertical [...]

I get it, thanks for your explanation, I start testing and report.

leo12757, 2020-05-30:
> @leo12757 I've had a look at the data, thanks for that!
>
> There are a couple of possibilities on how to fix it. This is attempt number 1.
>
> I've updated the PR & branch.
> Can you please redownload it [...]

@crankyoldgit ,
i'm tested, report the result(use HITACHI_AC_344).
Swing V was work, but need following operations below:
1. select Swing(V) option Off->Auto, swing V will START move.
2. select Swing(V) option Auto->Off, swing V will continue move.
3. select Swing(V) option Off->Auto, swing V will STOP.

crankyoldgit, 2020-05-30:
> Swing V was work, but need following operations below:
> 1). select Swing(V) option Off->Auto, swing V will START move.
> 2). select Swing(V) option Auto->Off, swing V will continue move.
> 3). select Swing(V) option Off->Auto, swing V will STOP move.
> or
> 1). select Swing(V) option Off->Auto, swing V will START move.
> 2). change any other seeting( E.g +1 temp), swing V will continue move.
> 3). select Seing(V) option Auto->Off, swing V will STOP move

Sorry, I don't get what you're saying here. Is this what it is doing, or is this what you think it should be doing? (I know English isn't your natural language, but unfortunately what your saying isn't clear)

Please remember, I can't see what your a/c is doing in response to what ever settings (and sequence) you've told it. You have to document everything, and make it clear.

crankyoldgit, 2020-05-30:
Ideally, when IRMQTTServer says swingV is NOT `off` the swing on the A/C unit should be swinging.
When it is `off` then the Swing on the A/C unit should not be moving etc.

leo12757, 2020-05-30:
> Sorry, I don't get what you're saying here. Is this what it is doing, or is this what you think it should be doing? (I know English isn't your natural language, but unfortunately what your saying [...]

Sorry, I texplain clearly, I hope:
1). when select Swing(V) Off -> Auto, Swing V will start move.
2). when select Swing (V) Auto -> Off, Swing V will stop move.

But in fact is below:
Select Swing(V) option Off->Auto, swing V START move. (This behavior is good.)
But when select Swing(V) Auto-< Off, swing V continue move, If want swing v stop, need:
1). Switch Swing(V) option to another option(Auto->Off), push send button (at this time, air conditioning will no respone, swing V still moving), then switch Swing(V) option again (Off->Auto), push send button again, swing V will stop.
2). Switch other setting (E.g +1 temp), push send button, at this time IRMQTT Web setting is  temp is +1, Swing (V) is Auto, air conditioning behavior is temp +1, swing V still moving, then switch Swing(V) Auto->Off, push send button again, air conditioning swing v will stop.

crankyoldgit, 2020-05-30:
Okay, I've uploaded another attempted fix to that branch, please download & try it now.

leo12757, 2020-05-30:
> Okay, I've uploaded another attempted fix to that branch, please download & try it now.

@crankyoldgit  i haved test, this report below:
When air conditioning is Off state, at IRMQTT Web, first send turn on controll code(operation method: web setting Mode: Off->Cool, Swing(V): Off(default value),  push send button), last Hitachi344Swing version's behavior was air conditioning will power on, at Cool mode, swing V won't move, but the newest Hitachi344Swing version, same operation method, air conditioning will power on, at Cool mode and swing V is moveing.

I think the right behavior is:
Web setting Mode: Off->Cool, Swing(V): Off(default value), air conditioning should power on, swing V won't move.
If need swing V, setting Swing(V) Off->Auto(other setting no change), then swing V willl move.
When swing V to the desired position, setting Swing(V) Auto -> Off, swing V will stop.

It look like that the lase Hitachi344Swing version was correct for the swing V when the air conditioner was Off->Cool, but the problem is can't turn off swing V reasonable.
When the air conditioner at Cool mode and the swing V need adjustment postion, set Swing(V) Off-> Auto, swing V will move correctly, but when the swing V reaches the desired position, set the Swing(V) Auto-> Off, the swing V continues move.

Thank you.

crankyoldgit, 2020-05-31:
Okay. New fix attempt uploaded. It should now try to use the same method as Hitachi424 which you indicated was working on your device.

Download it as usual and let me know how it goes.

leo12757, 2020-05-31:
> Okay. New fix attempt uploaded. It should now try to use the same method as Hitachi424 which you indicated was working on your device.
> 
> Download it as usual and let me know how it goes.

ok, i will test and report quickly.
Thank you for your help.

leo12757, 2020-05-31:
@crankyoldgit 
Good news, 
The newest Hitachi344Swing version is work perfect ! :D
Thank you very much for your help.

crankyoldgit, 2020-05-31:
Excellent (for you). It's ugly for me, I got it working doing some really horrible stuff. Oh well, it works. That's the main thing.

crankyoldgit, 2020-06-23:
FYI, the code & changes referenced above have been included in the [v2.7.8 release](https://github.com/crankyoldgit/IRremoteESP8266/releases/tag/v2.7.8) of the library.

commit 7638f381 (2020-05-26): HitachiAc344: Improve initial state.
* Fix the stateReset() so that we can recreate a message state correctly.
* More unit tests.

Fixes #1134
PR #1147 (2020-05-27): Hitachi344: Add detailed support and change bit ordering.
* **[Breaking change]** Moved `state[]` to LSB First ordering.
* Hitachi344 appears to just be a shorter version of Hitachi424, so using class inheritence to reduce code space.
* Update Hitachi424 class to allow virtual methods.
* Add unit test coverage.
* Add common IRac API support.

Fixes #1134 (Hopefully)
commit 9e232977 (2020-05-27): Hitachi344: Add detailed support and change bit ordering. (#1147)

* **[Breaking change]** Moved `state[]` to LSB First ordering.
* Hitachi344 appears to just be a shorter version of Hitachi424, so using class inheritence to reduce code space.
* Update Hitachi424 class to allow virtual methods.
* Add unit test coverage.
* Add common IRac API support.
* Confirmed working on real device.

Fixes #1134
PR #1148 (2020-05-31): Hitachi344: Add Swing(H) and improve Swing(V)
* Correct Swing(V) control
* Add Swing(H) support.
* Update common A/C support.
* Reduce duplicate code between 344 & 424 classes.
* add & update supporting Unit Tests.
* Fix typo

Fixes #1134