crankyoldgit/IRremoteESP8266 #1137: Question: getting Model no along with protocol

wahibmichael, 2020-05-22:
Hello,
is there a way to get the **int16_t model;** from the **decode_results**? 
I'm currently getting the protocol but technically is it feasible to get the model from the IR capture?

NiKiZe, 2020-05-23:
If the model was implemented for the protocol, My limited understanding so far is that this is only implemented for a few protocols, searching the code for where model is set should give you more information if you are up for that.

crankyoldgit, 2020-05-23:
@NiKiZe is correct. Only a small handful of protocols have supported "sub" models of a given protocol. Most different "models" use a substantially different protocol message structure.

This function is the universal way to get the model number (if any) from a decode result (ie. a capture)
https://github.com/crankyoldgit/IRremoteESP8266/blob/494c5957626d3c92bbfd87bb73d3f04569cbd023/src/IRac.h#L392-L393

e.g. (untested code, just for idea purposes)
```
IRrecv irrecv(RECV_PIN);
decode_results capture;
if (irrecv.decode(&capture)) {
  stdAc::state_t ac_state;
  if (IRAcUtils::decodeToState(&, &ac_state, NULL)) {
    Serial.print("Model number detected was: ");
    Serial.println(ac.model);  // -1, for no model. >= 1 for there is a model number used/supported.
  }
}
```
Or, if you can do it at a lower level with the protocol specific class object if you know what protocol you are dealing with AND it supports sub-models.
```
IRrecv irrecv(RECV_PIN);
decode_results result;
if (irrecv.decode(&capture) && result.decode_type == PANASONIC_AC) {
  IRPanasonicAc ac_class(0);
  ac_class.setRaw(result.state);
  Serial.print("Model number detected was: ");
  Serial.println(ac_state.getModel());
}
```

wahibmichael, 2020-05-23:
It working fine thank you, Minor correction below for future reader 

IRrecv irrecv(RECV_PIN);
decode_results capture;
if (irrecv.decode(&capture)) {
  stdAc::state_t ac_state;
  if (IRAcUtils::decodeToState(&capture, &ac_state, NULL)) {
    Serial.print("Model number detected was: ");
    Serial.println(ac_state.model);  // -1, for no model. >= 1 for there is a model number used/supported.
  }
}

closed 2020-05-23 by wahibmichael

crankyoldgit, 2020-05-24:
Thanks for posting the "correct" code. I've updated my earlier version. Glad it's working for you.