crankyoldgit/IRremoteESP8266 #1145: AC Protocol with On/Off Timer that counts down

NiKiZe, 2020-05-25:
I have a lovely little AC over here that has the nice feature of having it's timer data being represented by an `uint16_t` that is time left (in 2 sec intervals) and counting down.

Setting it to off in 1 hour means that it sends 1800 (*2 = 3600)
If I say, lower the intended temp 2 seconds later it sends 1799
At 30 minutes left it sends 900. and so on.

This means that in the AC class I need to record when a timer was set and to which value.
And when something is sent, these timer values needs to be recalculated.

Is there any protocol that has this kind of timer? (my search returned no results)
Any suggestions or considerations?

crankyoldgit, 2020-05-25:
I think there are a few protocols that send a "minutes remaining" type message. I've never seen anything in the seconds level of precision. So that's new. I'd suggest only accepting minutes as the level of resolution to match others but that's your call really.

I tend to not support timers well, or at a priority. External (home) automation can do that better IMHO.
e.g. turn on/off the unit on different triggers including time etc.

My suggestion is include the ability to set/get the timer value and timer on/off control in the class.
Let the user of the class worry about changing it according to a wall clock etc.

This is similar to clock/time setting on an a/c in protocols that support it.
We don't track the time inside a class, we just send what we are told to send.
Keeping it that low level and simple allows maximum flexibility, least complication, and minimal code footprint.

The a/c classes are not meant to be perfect emulations/simulations of real IR remotes.
They should however only produce valid codes/messages for the a/c unit.
e.g. It's okay to enforce "auto mode always has the temp at 25C" etc.
I'm often reluctant to have the class remember temperatures when a user cycles through modes etc. That said, I often acquiesce to those requests as they are simple. I definitely draw the line at tracking the passage of time etc. ;-)

TL;DR: Give people the tools/interface to be able to fill out the structure of a message, and make the ac do the right thing. Leave it to the user/caller to write the code to emulate the hardware to what ever level they desire.

NiKiZe, 2020-05-25:
Sounds good, thanks!

closed 2020-05-25 by NiKiZe