crankyoldgit/IRremoteESP8266 #1146: Aiwa IR decoder and encoder does not work properly.

kdschlosser, 2020-05-25:
### Version/revision of the library used
master

### Expected behavior
proper decoding and sending of Aiwa IR

### Actual behavior
doesn't work properly

### Steps to reproduce the behavior
send and decode an Aiwa IR code

### Other useful information

The code below is Python is what properly decodes Awia IR 
I am sure you can port this to c++ and get it running in your library without issue.

I do want to note that in the ir_Awia.cpp there is a comment about using the Arduino IRremote library as a baseline. In the ir_Awia.cpp file in the IRremote library it states

> THIS CODE LOOKS LIKE IT MIGHT BE WRONG - CHECK!

the Awia IR protocol has device, sub device and function. device and function are 8 bits each and sub device is 5 bits.

The code breaks down like this.

```
burst pair 1 = [HEADER_MARK, HEADER_SPACE]
burst pairs 2 - 9 = device check sum. They have the values of the bits reversed from the device
1 1 1 0 1 1 0 0

burst pairs 10 - 14 = sub device checksum. again the bit values are reversed from the sub device
1 0 1 1 1

burst pairs 15 - 22 = device
0 0 0 1 0 0 1 1

burst pairs 23 - 27 = sub device
0 1 0 0 0 

burst pairs 28 - 35 = function checksum, same deal as above
1 1 0 1 1 0 0 1 

burst pairs 36 - 43 = function
0 0 1 0 0 1 1 0 

burst pair 44 = [MARK, ?]
burst pair 45 = [HEADER_MARK, HEADER_SPACE]
burst pair 46 = [MARK, ?]
```

The correct timings are also in the code below.

```python
NUM_BITS = 46
DATA_BITS = 42
HEADER_MARK = 8782
HEADER_SPACE = -4391
MARK = 552
SPACE_0 = -1656
SPACE_1 = -552

rlc = [8782, -4391, 552, -552, 552, -552, 552, -552, 552, -1656, 552, -552, 552, -552, 552, -1656, 552, -1656, 552, -552, 552, -1656, 552, -552, 552, -552, 552, -552, 552, -1656, 552, -1656, 552, -1656, 552, -552, 552, -1656, 552, -1656, 552, -552, 552, -552, 552, -1656, 552, -552, 552, -1656, 552, -1656, 552, -1656, 552, -552, 552, -552, 552, -1656, 552, -552, 552, -552, 552, -1656, 552, -1656, 552, -552, 552, -1656, 552, -1656, 552, -552, 552, -1656, 552, -1656, 552, -552, 552, -552, 552, -1656, 552, -23087, 8782, -4391, 552, -90694]

if len(rlc) != NUM_BITS * 2:
    raise RuntimeError

if rlc[-2] != MARK or rlc[-6] != MARK:
    raise RuntimeError

if rlc[-4] != HEADER_MARK or rlc[-3] != HEADER_SPACE:
    raise RuntimeError

data = []
for i in range(3, len(rlc) - 6, 2):
    item = rlc[i]
    if item == SPACE_1:
        data += [1]
    else:
        data += [0]

if len(data) != DATA_BITS:
    raise RuntimeError

device_checksum = data[:8]
sub_device_checksum = data[8:13]
device = data[13:21]
sub_device = data[21:26]
function_checksum = data[26:34]
function = data[34:42]

dev = 0
dev_check = 0
sub = 0
sub_check = 0
func = 0
func_check = 0

def set_bit(val, bit_num, flag):
    if flag:
        return val | (1 << bit_num)
    else:
        return val & ~(1 << bit_num)

for i, item in enumerate(device_checksum):
    dev_check = set_bit(dev_check, i, item)

for i, item in enumerate(sub_device_checksum):
    sub_check = set_bit(sub_check, i, item)

for i, item in enumerate(device):
    dev = set_bit(dev, i, item)

for i, item in enumerate(sub_device):
    sub = set_bit(sub, i, item)

for i, item in enumerate(function_checksum):
    func_check = set_bit(func_check, i, item)

for i, item in enumerate(function):
    func = set_bit(func, i, item)

if dev | dev_check != dev + dev_check:
    raise RuntimeError

if sub | sub_check != sub + sub_check:
    raise RuntimeError

if func | func_check != func + func_check:
    raise RuntimeError

print 'device:', dev
print 'sub device:', sub
print 'function:', func
```

the code above outputs

device: 200
sub device: 2
function: 100

crankyoldgit, 2020-05-26:
The "raw" data you provided (i.e. `rlc`) appears to be the `SANYO_LC7461` protocol when decoded by the library.

See:
https://github.com/crankyoldgit/IRremoteESP8266/blob/232ec687291832d2bc4aa2e524bd2986f4dc7992/src/ir_Sanyo.cpp#L89-L94

You should be able to reproduce the exact `rlc` message with:
```
irsend.sendSanyoLC7461(0x4D1D9726D9, kSanyoLC7461Bits, 1);  // Includes one repeat.
```

The subsequent `device`, `sub_device`, & `function` decoding is not supported. You will need to do the bit maths for that yourself.

crankyoldgit, 2020-05-26:
Can you let me know if that works for you? When doing the testing for this, I noticed a very subtle timing difference in the "repeat" section of the message. If it works as is, I won't change it, but I can if it doesn't work.

kdschlosser, 2020-05-26:
The important part about the decode is the xor of the device sub device and function to make sure the code is valid. I would have to go and look at the NEC protocol to see what the decoding arrangement is there. I believe that is what you are using because the protocols are so similiar bit I think there is an issue with the number of bits between the NEC protocol and the Aiwa protocol. How close are the timings? I think that Sanyo is really close to NEC also It might be best to tighten up somewhat on the tolerance for the timings.

I thought your library has support for device sub device and function. Well if it does then you can add them if you want. You can use the code example above to help.

crankyoldgit, 2020-05-26:
> The important part about the decode is the xor of the device sub device and function to make sure the code is valid.

It does those checks. See:
https://github.com/crankyoldgit/IRremoteESP8266/blob/232ec687291832d2bc4aa2e524bd2986f4dc7992/src/ir_Sanyo.cpp#L142-L155

> How close are the timings?

Exactly the same pretty much, except for the Header Space in the repeat portion.
i.e. `4391` vs `2240`

See https://github.com/crankyoldgit/IRremoteESP8266/commit/c0925be1d1d21eacc73a2f52cee43cecb0cffb9f

crankyoldgit, 2020-05-30:
How did you go?

crankyoldgit, 2020-06-03:
@kdschlosser Friendly chase up.

crankyoldgit, 2020-06-14:
No response in over 14 days. Marking this closed. Please re-open or reply if you have a response/update/etc.

closed 2020-06-14 by crankyoldgit

mtdcr, 2020-09-15:
I tried to control an Aiwa device using a recent Tasmota firmware.

Sending `{"Bits": 42, "DataLSB": "0x01C78FC00FF", "Protocol": "SANYO_LC7461"}` triggers `{"IrReceived":{"Protocol":"SANYO_LC7461","Bits":42,"Data":"0x0381E3F00FF","DataLSB":"0x01C78FC00FF","Repeat":0}}`.

Pressing the power button on the remote (Aiwa RC-AAT11) results in the exact same decoded message: `{"IrReceived":{"Protocol":"SANYO_LC7461","Bits":42,"Data":"0x0381E3F00FF","DataLSB":"0x01C78FC00FF","Repeat":0}}`

Unfortunately, the device (Aiwa LCX-108) does not react at all to the signal generated by Tasmota.

I could try to debug it further, but I'd need somebody to point me into the right direction.

crankyoldgit, 2020-09-15:
@mtdcr please create a new issue about this. Also, please capture the original remote with IRrecvDumpV2 or V3 and include the full output in the new issue.

crankyoldgit, 2020-09-15:
Also, how sure are you that you're physical IR transmission is working? I.e. is it working for other devices?

mtdcr, 2020-09-18:
> Also, include how sure are you that you're physical IR transmission is working? I.e. is it working for other devices?

To test it, I used two IR bridges in the same room. They are able to receive each other's transmissions.

I'll create a new issue as soon as I have a device I can safely flash IRrecvDumpV2 or V3 on.