crankyoldgit/IRremoteESP8266 #1153: update of library maintainers guide wiki

crankyoldgit, 2020-05-29:
@NiKiZe Can you cast any eye over this please:
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Library-Maintainers-Guide/_compare/38264cbc1436416cf35ac17c90254b7b4981471f

I think it's correct, but just want a second set of eyes to review it's easy enough to follow.

Ta.

NiKiZe, 2020-05-31:
It looks good, I only have comments about git usage, feel free to ignore them, as what is there now works.

`git pull` always does a `git fetch` as well, so no need for both, the last `git pull` can in default configuration be replaced with `git merge` but normally you don't want a merge there, so
I would recommend `git fetch origin; git checkout master; git reset --hard origin/master`
That would get you a clean state, but needs due warning about any local changes being lost.

later optimize

      $ git fetch; git checkout master; git pull
      $ git checkout gh-pages
      $ git pull
      $ git merge -s subtree master
      $ git push

to:
      $ git fetch
      $ git checkout gh-pages
      $ git reset --hard origin/gh-pages
      $ git merge -s subtree origin/master
      $ git push

there is no need to have an up to date local master, only care about origin/master

crankyoldgit, 2020-05-31:
Yeah, that's a much better/safer approach.  I've be so used to being the only one committing to master that my master branch would always be a clean update.

Please check over my update at: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Library-Maintainers-Guide#new-version-release-checklist to check I got what you said correct.

If so, we can mark this resolved.

NiKiZe, 2020-05-31:
Look good

closed 2020-05-31 by NiKiZe