crankyoldgit/IRremoteESP8266 #1172: TODO: Make the library compile cleanly with pretty much every protocol turned off.

crankyoldgit, 2020-06-05:
@NiKiZe Okay, you were right. We should fix this.
There are way to many Compiler Warnings when it builds with everything off. We should get around to fixing/cleaning that up/handling it better etc.

Test environment can crash and burn as previously discussed, but it'd be nice Joe User didn't get compiler warnings when compiling the library with protocols disabled.

This is a LOW priority. As it works, just isn't _nice_.

NiKiZe, 2020-06-06:
Adding a test case for no protos, but enabling one AC state, and for full coverage then another one.

crankyoldgit, 2020-06-06:
Not sure what you mean

NiKiZe, 2020-06-07:
It was a reminder to myself to check the build error from #1129 (and written when I was more asleep than awake)

In that issue I got build errors when I only tried to enable Corona, and quick fix for that was to enable LG as well.

In current PR checks are done for all disabled, and all enabled, but not only one enabled, I wanted to investigate that.

crankyoldgit, 2020-06-07:
> It was a reminder to myself to check the build error from #1129 (and written when I was more asleep than awake)
>
> In that issue I got build errors when I only tried to enable Corona, and quick fix [...]

Ahh .. gotcha

Yeah, I often get the too asleep but still at keyboard syndrome aswell. ;-)

PR #1179 (2020-06-08): Suppress more potential compiler warnings.
Ref #1129
Fixes #1172 (I think)
commit 373afa5c (2020-06-08): Suppress more potential compiler warnings. (#1179)

Ref #1129
Fixes #1172 (I think)
closed 2020-06-08 by crankyoldgit