crankyoldgit/IRremoteESP8266 #1181: Can I use the library to detect the received IR signal's protocol, without decoding?

Faberdash, 2020-06-08:
Hi David,
I am a computer science undergrad currently working on my bachelor project, in which I am supposed to build a smart IR receiver that  first outputs information about the signal received, for example: Its protocol, on an embedded web server on the ESP8266. And so my question: Using your library, is there a way to get the received signal's protocol name?

crankyoldgit, 2020-06-08:
First, let me answer the question in the topic.
> Can I use the library to detect the received IR signal's protocol, without decoding?

Technically, no .. and yes. You have to effectively decode a protocol to ensure that it is the correct. Now, you don't have to work out what the data is in the message or what it means (e.g. air con protocols), so there the "yes" part of the answer.

> Using your library, is there a way to get the received signal's protocol name?

Yes, you can get _our_ name for the protocol, but it might not be the "official" name of the protocol.
There is no "name" save in the protocol messages etc.

See the function [typeToString()](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/IRutils_8cpp.html#a9e98a1b929f36dfa75c2e325bf281cd1)
https://github.com/crankyoldgit/IRremoteESP8266/blob/bb527d964608964485cceff1257a0308248d08ce/src/IRutils.cpp#L101-L105

To get _our_ name/label for the protocol, you can do effectively the following:
(Note: Code not tested or compiled, I'm just writing it off the top of my head.)

Place this in the `loop()` section of [IRrecvDemo.ino](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDemo/IRrecvDemo.ino)
```
#include <IRutils.h>  // Needed for `typeToString()`

void loop() {
  if (irrecv.decode(&results)) {
    // We got "a" message, now lets print out what protocol it is.
    Serial.printf("Got a '%s' message\n", typeToString(results.decode_type));
    irrecv.resume();  // Receive the next value
  }
  delay(10);
}
```

Faberdash, 2020-06-08:
It worked perfectly, thank you so much for your quick response.
 It only needed a little addition to change the returned string to a char array to be readable in C.
I have attached the code for anyone with the same issue.

`#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRrecv.h>
#include <IRutils.h>

uint8_t kRecvPin = D1;

IRrecv irrecv(kRecvPin);

decode_results results;

void setup() {
  Serial.begin(115200);
  irrecv.enableIRIn();  // Start the receiver
  while (!Serial)  // Wait for the serial connection to be establised.
    delay(50);
  Serial.println();
  Serial.print("IRrecvDemo is now running and waiting for IR message on Pin ");
  Serial.println(kRecvPin);
}

void loop() {
  if (irrecv.decode(&results)) {
    // print() & println() can't handle printing long longs. (uint64_t)
    Serial.printf("Got a '%s' message\n", typeToString(results.decode_type).c_str());
    Serial.println("");
    serialPrintUint64(results.value, HEX);
    Serial.println("");
    irrecv.resume();  // Receive the next value
  }
  delay(100);
}`

closed 2020-06-08 by Faberdash