crankyoldgit/IRremoteESP8266 #1182: TODO: Move from SPIFFS to LittleFS or similar.

crankyoldgit, 2020-06-09:
Apparently SPIFFS is depreciated.
```
warning: 'SPIFFS' is deprecated (declared at /home/travis/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/FS.h:269): SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems. [-Wdeprecated-declarations]
```

Low priority.

crankyoldgit, 2020-06-12:
See also: https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html#spiffs-deprecation-warning

Caffreyfans, 2020-07-29:
Can I come to do this job？

NiKiZe, 2020-07-29:
> Can I come to do this job？

Absolutely, marked as help-wanted for that reason. ;)
Let us know if you need anything or if something is unclear.

NiKiZe, 2020-07-29:
ESP32 support for LittleFS still seems less then stable, I wonder if `#define FS LittleFS` and `#define FS SPIFFS` could be used.
Otherwise I think the ESP32 build needs to pull in other dependencies.

Also one other thing to note (for the release) is that any existing SPIFFS data will be lost.

Caffreyfans, 2020-07-29:
So, do I only move the ESP8266 part now?

crankyoldgit, 2020-07-30:
> So, do I only move the ESP8266 part now?

That's a reasonable first step.

> I wonder if `#define FS LittleFS` and `#define FS SPIFFS` could be used.

Agreed. Having one spot for the `#if defined(ESP8266)` control is a better solution.

> Otherwise I think the ESP32 build needs to pull in other dependencies.

I ran into that when I first gave it a quick attempt (at a full conversion). ESP8266 first is a good step in the right direction. Baby steps. ;-)

> Also one other thing to note (for the release) is that any existing SPIFFS data will be lost.

Yeah, I couldn't see an easy/obvious way to do a conversion. Hence giving the user some compile-time option to control what filesystem is used is probably a good idea.
e.g.
```c++
// Uncomment one of the following to manually override what type of persistent storage is used.
// Warning: Changing filesystems will cause all previous locally saved configuration data to be lost.
// #define FILESYSTEM SPIFFS
// #define FILESYSTEM LittleFS

#ifndef FILESYSTEM
// Set the default filesystem if none was specified.
#if defined(ESP8266)
#define FILESYSTEM LittleFS
#else
#define FILESYSTEM SPIFFS
#endif  // defined(ESP8266)
#endif  // FILESYSTEM
```

Caffreyfans, 2020-07-31:
I have on idea what makes warning as below:
***warning: "FILESYSTEM" redefined
#define FILESYSTEM SPIFFS
note: this is the location of the previous definition
 #define FILESYSTEM LittleFS***

@crankyoldgit @crankyoldgit

crankyoldgit, 2020-07-31:
> I have on idea what makes warning as below:
> _**warning: "FILESYSTEM" redefined #define FILESYSTEM SPIFFS note: this is the location of the previous definition #define FILESYSTEM LittleFS**_
> 
> @crankyoldgit @NiKiZe

Appears this is generated in:
```
/home/travis/build/crankyoldgit/IRremoteESP8266/examples/Web-AC-control/Web-AC-control.ino:48:0: warning: "FILESYSTEM" redefined
 #define FILESYSTEM SPIFFS
 ^
/home/travis/build/crankyoldgit/IRremoteESP8266/examples/Web-AC-control/Web-AC-control.ino:46:0: note: this is the location of the previous definition
 #define FILESYSTEM LittleFS
```
and
```
/home/travis/build/crankyoldgit/IRremoteESP8266/examples/IRMQTTServer/IRMQTTServer.ino:399:0: warning: "FILESYSTEM" redefined [enabled by default]
 #define FILESYSTEM SPIFFS
 ^
/home/travis/build/crankyoldgit/IRremoteESP8266/examples/IRMQTTServer/IRMQTTServer.ino:397:0: note: this is the location of the previous definition
 #define FILESYSTEM LittleFS
```

Looks like the `#ifndef FILESYSTEM` line isn't working as expected.

Try:
```c++
#if !defined(FILESYSTEM)
// Set the default filesystem if none was specified.
#if defined(ESP8266)
#define FILESYSTEM LittleFS
#else
#define FILESYSTEM SPIFFS
#endif  // defined(ESP8266)
#endif  // !defined(FILESYSTEM)
```

closed 2020-08-12 by crankyoldgit

crankyoldgit, 2020-08-31:
The code changes mention have now been included in the newly released v2.7.10 of the library.

commit a6cd2acb (2020-08-12): move SPIFFS to LittleFS for ESP8266 (#1226)

* move SPIFFS to LittleFS for ESP8266
* Make use of FILESYSTEM to make it more generic/expandable later.
* Use common/better strings in serial messages.

Fixes #1182