crankyoldgit/IRremoteESP8266 #1185: errors in fujistu timings

JoachimF, 2020-06-09:
Using the tasmota-ir firmware to control my fujistu HVAC, I found that it can't decode fuji remote.
I tried on two ESP and multiples remote. Then used debug mode to read the timing and found that timing was not correct.
I changed the timing, and now tasmota is able de decode the frame
Is it normal or only from my french remotes ?

IRremoteESP8266-2.7.7
Sonoff T1EU 1CH & Generic ESP8266
Fujitsu remote AR-REB1E

Correction in "ir_Fujitsu.cpp" 
const uint16_t kFujitsuAcBitMark = 394; //was 448
const uint16_t kFujitsuAcZeroSpace = 432; // was 390

crankyoldgit, 2020-06-09:
> Is it normal or only from my french remotes ?

Good question. I have a Fuji A/C, I'll (re-)check it in the morning.

The other alternative is to adjust the tolerance at run-time. See [IRrecv::setTolerance()](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRrecv.html#aa091c449db70c65fd0221669df7438ea)

No idea if Tasmota exposes that setting or not (Try looking at their docs etc ... Paging @s-hadinger )

FWIW, it is not uncommon to see subtle differences in timings from different remotes (even of the same brand), also the hardware IR demodulator could be a factor, and what ever else is running at the same time in Tasmota.

Do you happen to have a `rawData` capture from `IRrecvDumpV2` or V3 for it handy for me to look at?

JoachimF, 2020-06-09:
I had also increased the tolerance from 25 to 50 because tasmota didn't decode what it sends. But in all case, the Hvac accepts the commands, the problem is only during decoding. I'll try to do a dump soon

crankyoldgit, 2020-06-09:
50% is a lot. :-(

Also take a read of #703 The user/reporter there indicated model `1` was what they used and it worked.
https://github.com/crankyoldgit/IRremoteESP8266/issues/703#issuecomment-493659069

crankyoldgit, 2020-06-09:
and https://github.com/crankyoldgit/IRremoteESP8266/commit/1534c47db9ca02ea6451aa1f1b86700a6e66d711

crankyoldgit, 2020-06-10:
> Correction in "ir_Fujitsu.cpp"
> const uint16_t kFujitsuAcBitMark = 394; //was 448
> const uint16_t kFujitsuAcZeroSpace = 432; // was 390

As promised, here are the values I got from my Fujitsu A/C remote (averaged) via `auto_analyse_raw_data.py`

```
kHdrMark   = 3316
kHdrSpace  = 1632
kBitMark   = 442
kOneSpace  = 1206
kZeroSpace = 384
```

Here's the raw data from my capture of my AR-DB1 remote:
```
Timestamp : 000340.129
Library   : v2.7.7

Protocol  : FUJITSU_AC
Code      : 0x1463001010FC083061040400000067 (120 Bits)
Mesg Desc.: Model: 2 (ARDB1), Power: On, Mode: 4 (Heat), Temp: 22C, Fan: 4 (Quiet), Command: N/A
uint16_t rawData[243] = {3316, 1632,  446, 380,  446, 408,  418, 1206,  438, 388,  438, 1212,  440, 386,  440, 384,  442, 384,  442, 1210,  444, 1208,  444, 380,  446, 380,  446, 380,  448, 1204,  438, 1222,  420, 388,  438, 396,  440, 384,  442, 384,  442, 382,  444, 382,  444, 382,  444, 380,  446, 380,  446, 380,  446, 378,  448, 378,  438, 388,  438, 1116,  526, 388,  438, 396,  442, 384,  442, 384,  442, 382,  444, 382,  444, 382,  444, 1206,  446, 378,  448, 378,  436, 388,  438, 388,  438, 388,  440, 1212,  442, 1210,  442, 1210,  444, 1208,  444, 1206,  446, 1206,  438, 388,  440, 386,  438, 386,  440, 1212,  442, 384,  442, 384,  442, 382,  444, 382,  444, 382,  444, 382,  444, 380,  446, 380,  448, 1204,  438, 1214,  440, 386,  440, 386,  440, 1210,  442, 384,  442, 384,  442, 382,  444, 382,  444, 1206,  446, 1204,  438, 388,  438, 386,  440, 386,  440, 1212,  442, 384,  442, 384,  444, 382,  444, 382,  444, 382,  444, 380,  446, 380,  446, 1206,  438, 388,  438, 388,  438, 386,  440, 386,  440, 386,  440, 384,  442, 384,  442, 382,  444, 382,  444, 382,  446, 380,  444, 380,  446, 380,  446, 378,  448, 386,  418, 390,  446, 386,  440, 394,  422, 386,  440, 394,  442, 384,  442, 384,  442, 382,  444, 382,  444, 382,  446, 380,  446, 378,  446, 380,  448, 378,  438, 1214,  438, 1222,  422, 1212,  440, 394,  442, 384,  442, 1208,  446, 1208,  446, 380,  446};  // FUJITSU_AC
uint8_t state[15] = {0x14, 0x63, 0x00, 0x10, 0x10, 0xFC, 0x08, 0x30, 0x61, 0x04, 0x04, 0x00, 0x00, 0x00, 0x67};
```

Without seeing the raw data from your remote, it looks like you've got atypical timings for your remote, thus I'm not likely to change the default values. Maybe try fresh batteries?

Looking forward to your rawData.

crankyoldgit, 2020-06-14:
@JoachimF Have you been able to collect any rawData from `IRrecvDumpV2` (or newer) at all?
i.e. Friendly chase up.

crankyoldgit, 2020-06-25:
@JoachimF Another friendly ping/chase up for more data. I'd hate to have to close this issue due to inactivity.

crankyoldgit, 2020-07-06:
Closing due to no response/inactivity. Ping me and I'll reopen.

closed 2020-07-06 by crankyoldgit