crankyoldgit/IRremoteESP8266 #1187: Electra Israeli HVAC codes seems not to work.

OfferLa, 2020-06-09:
_(Please use this template for reporting issues. You can delete what ever is not relevant. Giving us this information will help us help you faster. Please also read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions) & [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide). Your problem may already have an answer there.)_

### Version/revision of the library used
_Typically located in the `library.json` & `src/IRremoteESP8266.h` files in the root directory of the library.
e.g. v2.0.0, or 'master' as at 1st of June, 2017. etc._

2.7.7

### Expected behavior
_What steps did you do and what should it have done?_

e.g.
1. Initialise the IRsend class.
2. IRsend.sendFoobar(0xdeadbeef);
3. Foobar branded BBQ turns on and cooks me some ribs.

Run the protocol example - protocol number 48 (electra AC)

### Actual behavior
_What steps did you do, and what did or didn't actually happen?_

e.g.
1. Initialise the IRsend class.
2. IRsend.sendFoobar(0xdeadbeef);
3. Foobar BBQ went into Cow(er)-saving mode and fried me a couple of eggs instead.

nothing happens.
IR led is blinking, but AC doesn't get any command.

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
_Include some raw dumps of what the device saw._

It's the output only example. No IR sensor connected.

### Steps to reproduce the behavior
_What can we do to (pref. reliably) repeat what is happening?_

Run the CommonACcontrol example, 'lock' the loop to protocol number 48 (Electra AC, which is supported), Serial output is working, but the commands sent doesn't do anything.

#### Example code used
_Include all relevant code snippets or links to the actual code files. Tip: [How to quote your code so it is still readable](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code)._

#### Circuit diagram and hardware used (if applicable)
_Link to an image of the circuit diagram used. Part number of the IR receiver module etc. ESP8266 or ESP32 board type._

IR led connected to NodeMCU D2 (GPIO 4) via 100 ohm resistor, or 2 IR Leds connected to that pin (in Serial connection) with 10 ohm resistor.

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
_Yes/No._

Partly.

### Has this library/code previously worked as expected for you?
_Yes/No. If "Yes", which version last worked for you?_

No.
I'm new to this amazing idea.

### Other useful information
_More information is always welcome. Be verbose._

I put the circuit right near the AC eye, so I don't think LED power is the issue.
I really hope to see this amazing idea working.

Many thanks!

NiKiZe, 2020-06-09:
What is the exact brand and model of your A/C and your original remote?
One possibility is that your A/C isn't supported, but to know and debug this easily, we will need the output from your original remote, which means that you will want to attach a sensor and run `IRrecvDumpV2`or `V3`

crankyoldgit, 2020-06-09:
> I put the circuit right near the AC eye, so I don't think LED power is the issue.

See point 5 of https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide#receiving-an-ir-signal

> 5. Use an appropriate distance between the transmitter and the receiver. A receiver module can easily be overpowered and produce a corrupt signal if you have the IR source too close. A range of less than ~30mm is often enough to confuse the receiver.

crankyoldgit, 2020-06-09:
Also, some brands often use another brands protocol on some models. If you don't have an IR receiver, try running through them all to see if you can get a response at least (if it is supported after all)

OfferLa, 2020-06-09:
Thank you so much for that!

 

My AC is Elco/Electra INV 17, original RC is RC-7 (but I think many Electra/Elco RCs use same codes).

 

Will really appreciate any debug and help.

crankyoldgit, 2020-06-10:
> IR led connected to NodeMCU D2 (GPIO 4) via 100 ohm resistor, or 2 IR Leds connected to that pin (in Serial connection) with 10 ohm resistor.

I just notice this comment.
You **really** want to power the LEDs via a transistor (BJT or MOSFET) per https://github.com/crankyoldgit/IRremoteESP8266/wiki#ir-sending

A GPIO does not provide enough current to an IR led to produce a useful signal.

crankyoldgit, 2020-06-11:
Have you been able to successfully control something simple like a TV yet?

As stated earlier, I think we are going to need a capture of your remote to see if it is something that is supported already, or if we have to add something for it.

crankyoldgit, 2020-06-14:
@OfferLa Friendly chase up.

crankyoldgit, 2020-06-25:
Ping @OfferLa

OfferLa, 2020-06-29:
I'm really sorry for not updating.
Working on re-assembling my IR led circuit, and hope to update back soon.
(btw - personally, I don't have any other ir-operated equipment at home, so my only option is to test this on my Elco/Electra AC).
Many thanks again for looking into this issue for me!

crankyoldgit, 2020-07-06:
Any luck?

yurir, 2020-07-06:
I also have Electra AC. It works for me with airwell codes from fort distance, about one meter. @OfferLa , have you tried that?

yurir, 2020-07-06:
> I just notice this comment.
> You **really** want to power the LEDs via a transistor (BJT or MOSFET) per https://github.com/crankyoldgit/IRremoteESP8266/wiki#ir-sending
>
> A GPIO does not provide enough [...]

@crankyoldgit in the link you gave, appears image of circuit for signal dump. Sending ir signal circuit isn't available (at least I can't see it)

crankyoldgit, 2020-07-07:
> @crankyoldgit in the link you gave, appears image of circuit for signal dump. Sending ir signal circuit isn't available (at least I can't see it)

@yurir Thanks for the note that the image URL no longer working. Looks like where it was pointing to has gone away. I think I found a copy of the image and it's now hosted on Github, so it should be (more) permanent.

crankyoldgit, 2020-07-14:
@OfferLa Ping ... we haven't heard anything from you for weeks now. Have you debug'ed your transmit circuit? Or gotten a IR receiver/demodulator so we can check what signal your original remote produces?

crankyoldgit, 2020-07-23:
@OfferLa Last chance to get back to us before I close this issue due to unresponsiveness.

OfferLa, 2020-07-23:
Dear David and everyone here.

 

I'm really sorry and apologize for my delays.

 

Really appreciate your investments into this so important code and function. I'm sure it changed the life of many.

 

Due to some issues (related to the crazy and challenging times we live now) I wasn't able to build and test the circuit as I wanted to.

 

If you agree  - let's do like this: I'll get more 48-72 hours (up to you to decide) to re-build my circuit and try the tests you suggested – and if I'm not returning with a productive, detailed answer – please close my issue for unresponsiveness, and again – I apologize for the efforts you put into my issue.

 

Many thanks again,

Hope to update soon,

Offer.

crankyoldgit, 2020-07-23:
Sounds reasonable to me. Plus you can always create a new issue and refer back to this one.

anyway, fingers crossed.

OfferLa, 2020-07-27:
> I also have Electra AC. It works for me with airwell codes from short distance, about one meter. @OfferLa , have you tried that?

Hello again!

Just tried it now again. Re-built the circuit (with the transistor) and can see the IR LED blinking when looking through my phone's camera.

For testing, I'm using the example code called CommonAcControl.
It loops through many models, but Airwell is not one of them, I think. Electra is in the loop (model code 48), but unfortunately - no any success yet.

Maybe there is another code I can use to test my AC?

Many thanks again!

crankyoldgit, 2020-07-27:
Have you tried capturing your existing remote with IRrecvDumpV2 or V3 yet?

crankyoldgit, 2020-08-07:
Repeat: Have you tried capturing your existing remote with IRrecvDumpV2 or V3 yet?

yurir, 2020-08-07:
@OfferLa  can you post photo of your remote? If it's like mine, I will share you code I use.

OfferLa, 2020-08-12:
> @OfferLa can you post photo of your remote? If it's like mine, I will share you code I use.

Thanks for that!
[image]

yurir, 2020-08-12:
I have another one, looks I can't help. Mine is:
[image: image.png]

crankyoldgit, 2020-08-18:
> Have you tried capturing your existing remote with IRrecvDumpV2 or V3 yet?

Repeating this question as it has been unanswered. We'll need some data to help you out.

crankyoldgit, 2020-08-29:
Marking this issue state/inactive, and closing it. If I see a response here, I'll reopen it.

closed 2020-08-29 by crankyoldgit

zevele, 2023-01-27:
> Meanwhile - tried to google for code differences between Electra RC-3 and my RC-7, and didn't find usable information.
> 
> Again - I appreciate everyone's help, and really hope we'll be able to 'make it happen', at last.

Sorry to wake an old thread, but I wanted to share my experience regarding this. 
I managed to control an old Electra AC (remote controller RC-3). Apparently these air-conditioners uses the "Airwell" protocol. I managed to control the unit using this protocol (turn on/off change modes fan and set temperatures). 
One thing to note is that setPowerToggle(const bool on), does not sets the power, but rather adds the power toggle message to the data (just like the name implies :-) ). i.e. setPowerToggle(true) toggles the power state of the unit. setPowerToggle(false) maintains the current power state.

Hope this helps anyone, and thanks for the great work!

mbrevda, 2023-08-27:
Would be great to try and revive this issue. I'm using a similar remote (I belive its the RC3), here is some of the output from dumpv3

```
115200 baud
Timestamp : 000130.662
Library   : v2.8.6

Protocol  : AIRWELL
Code      : 0x276C80002 (34 Bits)
Mesg Desc.: Power Toggle: On, Mode: 1 (Cool), Fan: 3 (Auto), Temp: 24C
uint16_t rawData[175] = {3020, 3752,  1946, 1028,  892, 1924,  914, 936,  944, 976,  1860, 1964,  914, 936,  1860, 1962,  1006, 936,  1810, 1010,  870, 1946,  1900, 976,  868, 1054,  890, 1030,  852, 1070,  868, 962,  892, 1030,  892, 1030,  868, 1054,  892, 936,  870, 1052,  894, 1028,  892, 1030,  868, 962,  868, 1052,  892, 1010,  876, 1048,  888, 1834,  1918, 1054,  2960, 3794,  1894, 1056,  866, 1930,  896, 976,  946, 974,  1862, 1960,  876, 976,  1862, 1962,  966, 974,  1782, 1054,  888, 1928,  1882, 956,  884, 1056,  866, 1056,  866, 1056,  866, 964,  866, 1056,  866, 1056,  866, 1056,  866, 964,  866, 1056,  866, 1054,  866, 1056,  890, 940,  890, 1012,  910, 1010,  912, 1010,  912, 1832,  1880, 1048,  2980, 3770,  1944, 1008,  914, 1902,  896, 956,  966, 956,  1880, 1964,  914, 916,  1858, 1986,  942, 980,  1832, 1004,  914, 1900,  1912, 928,  928, 992,  926, 1018,  904, 996,  922, 930,  878, 1044,  876, 1046,  876, 1044,  878, 952,  876, 1042,  880, 1064,  844, 1058,  862, 986,  844, 1078,  844, 1076,  844, 1078,  844, 1902,  1850, 1078,  3866};  // AIRWELL 276C80002
uint64_t data = 0x276C80002;

Timestamp : 000136.325
Library   : v2.8.6

Protocol  : AIRWELL
Code      : 0x76D00002 (34 Bits)
Mesg Desc.: Power Toggle: Off, Mode: 1 (Cool), Fan: 3 (Auto), Temp: 25C
uint16_t rawData[181] = {2962, 2894,  858, 1086,  836, 1086,  836, 1958,  880, 992,  998, 924,  1868, 1958,  924, 926,  1866, 1956,  1018, 924,  1820, 1912,  1886, 1064,  904, 926,  906, 1016,  904, 1018,  904, 1018,  904, 926,  904, 1018,  902, 1018,  904, 1018,  904, 926,  902, 1020,  902, 1018,  902, 1020,  902, 928,  902, 1018,  902, 1020,  900, 1020,  902, 1824,  1886, 1064,  2964, 2848,  880, 1086,  902, 1020,  902, 1914,  922, 928,  924, 998,  1840, 1984,  920, 910,  1858, 1984,  942, 980,  1834, 1918,  1858, 1070,  918, 912,  918, 1004,  918, 1004,  916, 1004,  918, 912,  918, 1004,  918, 1004,  918, 1004,  918, 910,  852, 1070,  920, 1000,  874, 1048,  874, 930,  946, 1002,  874, 1048,  874, 1048,  874, 1872,  1882, 1046,  3014, 2844,  912, 1010,  910, 1012,  910, 1928,  904, 926,  994, 926,  1910, 1934,  904, 926,  1908, 1954,  936, 986,  1758, 1994,  1850, 1078,  844, 986,  844, 1078,  844, 1078,  842, 1078,  842, 988,  844, 1078,  842, 1078,  844, 1078,  842, 970,  860, 1078,  834, 1088,  842, 1080,  842, 988,  842, 1078,  842, 1080,  842, 1080,  842, 1904,  1848, 1080,  3862};  // AIRWELL 76D00002
uint64_t data = 0x76D00002;

Timestamp : 000136.777
Library   : v2.8.6

Protocol  : AIRWELL
Code      : 0x76D00002 (34 Bits)
Mesg Desc.: Power Toggle: Off, Mode: 1 (Cool), Fan: 3 (Auto), Temp: 25C
uint16_t rawData[181] = {3026, 2808,  876, 1070,  852, 1070,  852, 1986,  920, 910,  1010, 912,  1880, 1962,  874, 956,  1926, 1918,  966, 956,  1834, 1918,  1882, 1046,  916, 914,  904, 1018,  906, 1016,  908, 1014,  906, 924,  902, 1018,  904, 1018,  906, 1016,  904, 926,  904, 1018,  902, 1020,  902, 1018,  902, 928,  902, 1018,  904, 1018,  902, 1020,  902, 1862,  1888, 1040,  2988, 2850,  862, 1078,  844, 1078,  844, 1974,  862, 986,  936, 986,  1850, 1994,  844, 986,  1850, 1974,  954, 986,  1760, 1992,  1852, 1078,  844, 986,  844, 1078,  844, 1078,  844, 1078,  844, 986,  844, 1078,  844, 1078,  844, 1078,  844, 986,  844, 1058,  862, 1078,  842, 1080,  842, 988,  844, 1078,  844, 1078,  844, 1078,  844, 1904,  1850, 1078,  2948, 2908,  842, 1080,  844, 1078,  842, 1972,  866, 988,  934, 988,  1850, 1994,  842, 988,  1850, 1994,  936, 988,  1758, 1994,  1850, 1080,  842, 988,  842, 1078,  842, 1080,  842, 1080,  842, 988,  842, 1080,  842, 1080,  842, 1078,  842, 988,  840, 1082,  840, 1080,  842, 1080,  842, 988,  842, 1080,  840, 1080,  842, 1080,  840, 1904,  1826, 1104,  3862};  // AIRWELL 76D00002
uint64_t data = 0x76D00002;

Timestamp : 000138.572
Library   : v2.8.6

Protocol  : AIRWELL
Code      : 0x76C80002 (34 Bits)
Mesg Desc.: Power Toggle: Off, Mode: 1 (Cool), Fan: 3 (Auto), Temp: 24C
uint16_t rawData[181] = {3024, 2860,  894, 1048,  846, 1076,  872, 1920,  874, 1002,  966, 956,  1836, 1964,  890, 984,  1836, 1986,  942, 1000,  1792, 1046,  830, 1988,  1856, 1000,  876, 1046,  874, 1048,  876, 1046,  876, 954,  828, 1094,  902, 1020,  900, 1022,  898, 930,  876, 1046,  876, 1046,  876, 1044,  898, 932,  876, 1046,  898, 1022,  898, 1024,  876, 1828,  1950, 1022,  2938, 2876,  876, 1090,  902, 1020,  900, 1916,  920, 932,  922, 998,  1840, 1984,  922, 928,  1840, 1960,  1038, 928,  1818, 1020,  900, 1916,  1860, 998,  902, 1020,  902, 1020,  900, 1022,  902, 928,  900, 1022,  900, 1004,  918, 1002,  918, 912,  916, 1004,  916, 1008,  914, 984,  874, 980,  916, 1006,  914, 1008,  914, 1008,  914, 1832,  1858, 1070,  2956, 2902,  850, 1050,  872, 1072,  918, 1918,  918, 912,  942, 978,  1880, 1964,  920, 912,  1880, 1964,  966, 956,  1794, 1042,  880, 1954,  1912, 928,  906, 1016,  904, 1018,  902, 1020,  904, 926,  878, 1044,  878, 1064,  844, 1078,  844, 986,  844, 1078,  844, 1078,  844, 1078,  844, 986,  844, 1078,  844, 1078,  844, 1078,  844, 1902,  1850, 1078,  3864};  // AIRWELL 76C80002
uint64_t data = 0x76C80002;

```

Unfortunately, trying to use Airwell (or the CommonAcControll example) didn't work ( via camera test Ir seems to be transmitting, but the a/c didn't receive/react to anything). Happy to provide anything else necessary.

One random bit of data I picked up hinted at a frequency of 33kHz (not sure if this is true(. Is there some way to set the frequency (at least for testing)?

crankyoldgit, 2023-08-28:
Please don't try to revive an old issue. It's better to create a new issue and reference it using "Related to issue #1187" or "see also #1187" etc.

> One random bit of data I picked up hinted at a frequency of 33kHz (not sure if this is true(. Is there some way to set the frequency (at least for testing)?

Yes, try modifying the library here:
https://github.com/crankyoldgit/IRremoteESP8266/blob/2bfdf9719250471f004b720e81c14ab00a244836/src/ir_Electra.cpp#L46
to be `33` or `33000`, that should let you try it with a different frequency for that protocol.

Let us know how it goes.