crankyoldgit/IRremoteESP8266 #1199: AIRWELL Support

juanbaldovi, 2020-06-18:
Hello
I have an Airwell HVAC with the RC-4 remote.
Using the current support included on this thread I have captured some data and created an [spreadsheet](https://docs.google.com/spreadsheets/d/1MeqVsgXQAMSiAx7zNunG8qbFcIT8tqZtcjLmMa0GEng/edit#gid=0) with my finings.
Could you please have a look and let me know if that is enough for supporting the protocol within the library?

Also. When I capture the data the AIRWELL protocol is only recognized when the on/off button is pressed. When the fan or temp button is pressed, it is always recognized as UNKNOWN protocol.
Is that because the protocol is not fully supported in the current version?
Regarding the unit.

Remote conrol is [Airwell RC4](https://i.ebayimg.com/images/g/s2MAAOSwPcVVhZFv/s-l300.jpg)
Wall IR receiver is [[1](https://www.amazon.es/Genuine-Airwell-Receptor-distancia-acondicionado/dp/B07B4H1B3G/ref=pd_lpo_107_t_1/260-0231940-9645161?_encoding=UTF8&pd_rd_i=B07B4H1B3G&pd_rd_r=00269d38-c8da-427a-873d-ac765821d500&pd_rd_w=KY2EJ&pd_rd_wg=oUREH&pf_rd_p=4221015a-01c7-4a3d-a84d-985d938e9995&pf_rd_r=5YE0RX0EDE3CV6KKRXV6&psc=1&refRID=5YE0RX0EDE3CV6KKRXV6)] [[2](https://user-images.githubusercontent.com/29886403/28319721-61d651ea-6bd7-11e7-9141-1afc63ade853.png)]
The AC unit -> Airwell DLS 21 DCI R410 AW

juanbaldovi, 2020-06-19:
Here is some Dumpv3 data

**cool. 19. Fan=3**
01:53:43.071 -> Timestamp : 001133.228
01:53:43.071 -> Library   : v2.7.7
01:53:43.071 -> 
01:53:43.071 -> Protocol  : UNKNOWN
01:53:43.071 -> Code      : 0xA3B3B8FC (97 Bits)
01:53:43.071 -> uint16_t rawData[193] = {2998, 2864,  888, 1032,  890, 1060,  862, 1974,  862, 940,  1898, 1032,  888, 1032,  890, 968,  862, 1060,  862, 1058,  862, 1060,  862, 1856,  1898, 1058,  862, 1032,  890, 968,  862, 1032,  890, 1032,  890, 1060,  862, 968,  862, 1030,  892, 1032,  890, 1060,  862, 968,  862, 1058,  862, 1032,  890, 1058,  864, 940,  890, 1060,  862, 1032,  888, 1060,  862, 1882,  1870, 1058,  2970, 2888,  862, 1032,  888, 1034,  888, 1974,  862, 968,  1870, 1032,  888, 1032,  890, 968,  862, 1060,  862, 1058,  862, 1060,  862, 1882,  1870, 1058,  862, 1058,  862, 968,  862, 1060,  862, 1032,  890, 1032,  890, 942,  888, 1058,  864, 1034,  888, 1060,  862, 940,  890, 1060,  862, 1058,  864, 1032,  890, 942,  888, 1060,  862, 1060,  862, 1060,  862, 1882,  1870, 1058,  2970, 2860,  892, 1032,  890, 1032,  890, 1974,  862, 968,  1870, 1032,  890, 1060,  862, 940,  890, 1058,  862, 1032,  890, 1036,  886, 1854,  1898, 1058,  862, 1060,  862, 942,  888, 1032,  888, 1060,  862, 1058,  864, 966,  864, 1058,  862, 1034,  888, 1058,  862, 968,  862, 1032,  890, 1058,  862, 1032,  890, 940,  890, 1058,  862, 1032,  888, 1060,  862, 1856,  1898, 1032,  3912};  // UNKNOWN A3B3B8FC

**cool. 20. Fan=3**
01:54:27.182 -> Timestamp : 001177.348
01:54:27.182 -> Library   : v2.7.7
01:54:27.182 -> 
01:54:27.182 -> Protocol  : UNKNOWN
01:54:27.182 -> Code      : 0xC277276E (94 Bits)
01:54:27.216 -> uint16_t rawData[187] = {2974, 2858,  896, 1054,  868, 1052,  862, 1946,  890, 966,  1872, 1056,  864, 1058,  862, 966,  862, 1060,  862, 1058,  864, 1058,  862, 1882,  1870, 1974,  1872, 966,  864, 1058,  862, 1058,  862, 1058,  862, 968,  862, 1060,  864, 1058,  862, 1058,  862, 968,  862, 1060,  862, 1060,  862, 1060,  862, 968,  862, 1058,  862, 1060,  862, 1060,  864, 1882,  1870, 1058,  2968, 2890,  862, 1058,  864, 1058,  862, 1974,  862, 968,  1872, 1058,  864, 1058,  862, 968,  862, 1060,  862, 1060,  862, 1060,  862, 1882,  1868, 1976,  1870, 968,  862, 1058,  862, 1060,  862, 1060,  862, 968,  862, 1060,  864, 1058,  862, 1058,  862, 968,  862, 1060,  862, 1060,  862, 1060,  862, 968,  862, 1060,  862, 1060,  862, 1060,  862, 1884,  1870, 1058,  2968, 2892,  862, 1058,  862, 1058,  862, 1976,  862, 968,  1870, 1058,  862, 1058,  862, 968,  862, 1060,  862, 1060,  862, 1060,  862, 1884,  1868, 1976,  1870, 968,  862, 1058,  862, 1060,  860, 1060,  862, 970,  862, 1060,  862, 1060,  862, 1060,  862, 968,  860, 1062,  862, 1060,  862, 1060,  862, 968,  860, 1060,  860, 1062,  860, 1062,  862, 1884,  1868, 1060,  3882};  // UNKNOWN C277276E

**cool. 20. Fan=auto**
01:55:17.294 -> Timestamp : 001227.465
01:55:17.294 -> Library   : v2.7.7
01:55:17.294 -> 
01:55:17.328 -> Protocol  : UNKNOWN
01:55:17.328 -> Code      : 0xFF4091A2 (94 Bits)
01:55:17.328 -> uint16_t rawData[187] = {3006, 2854,  898, 1052,  868, 1052,  864, 1948,  924, 934,  960, 962,  1872, 1084,  832, 968,  862, 1060,  862, 1058,  864, 1058,  862, 1884,  1870, 1974,  1870, 968,  862, 1060,  862, 1060,  862, 1060,  862, 968,  862, 1060,  862, 1060,  862, 1060,  860, 970,  862, 1060,  862, 1060,  862, 1058,  862, 968,  862, 1060,  862, 1060,  862, 1060,  862, 1882,  1868, 1060,  2968, 2890,  862, 1060,  860, 1062,  860, 1976,  862, 968,  954, 966,  1870, 1060,  860, 970,  862, 1060,  862, 1060,  862, 1060,  860, 1884,  1870, 1974,  1870, 968,  862, 1060,  860, 1060,  862, 1060,  862, 968,  862, 1060,  862, 1060,  860, 1062,  860, 968,  862, 1060,  862, 1060,  862, 1060,  860, 970,  860, 1060,  862, 1060,  862, 1060,  862, 1884,  1868, 1060,  2968, 2890,  862, 1060,  860, 1062,  860, 1976,  862, 968,  954, 968,  1868, 1060,  860, 970,  862, 1060,  862, 1060,  862, 1060,  860, 1886,  1868, 1976,  1870, 968,  860, 1062,  860, 1062,  860, 1062,  862, 968,  862, 1060,  862, 1060,  860, 1062,  860, 970,  860, 1060,  862, 1060,  860, 1062,  860, 970,  860, 1062,  860, 1062,  860, 1060,  860, 1886,  1866, 1062,  3882};  // UNKNOWN FF4091A2

**cool. 20. Fan=1**
01:56:15.862 -> Timestamp : 001286.015
01:56:15.862 -> Library   : v2.7.7
01:56:15.862 -> 
01:56:15.862 -> Protocol  : UNKNOWN
01:56:15.862 -> Code      : 0xE3832232 (94 Bits)
01:56:15.862 -> uint16_t rawData[187] = {3002, 2854,  898, 1052,  870, 1052,  870, 1966,  1812, 1026,  862, 1058,  862, 1060,  862, 968,  862, 1058,  862, 1060,  860, 1062,  860, 1884,  1870, 1974,  1868, 968,  862, 1060,  862, 1060,  862, 1058,  862, 968,  860, 1060,  862, 1060,  862, 1060,  862, 968,  862, 1058,  862, 1062,  860, 1060,  862, 968,  862, 1058,  862, 1060,  860, 1060,  860, 1886,  1870, 1058,  2968, 2892,  860, 1060,  862, 1060,  862, 1976,  1776, 1060,  860, 1062,  862, 1060,  862, 968,  862, 1060,  860, 1062,  860, 1062,  862, 1884,  1870, 1976,  1866, 970,  860, 1060,  862, 1060,  860, 1060,  860, 970,  858, 1062,  860, 1060,  862, 1060,  862, 968,  860, 1062,  860, 1062,  860, 1062,  860, 968,  862, 1060,  860, 1062,  860, 1062,  858, 1886,  1868, 1060,  2966, 2892,  860, 1062,  862, 1060,  860, 1976,  1774, 1064,  858, 1062,  860, 1062,  860, 970,  860, 1090,  832, 1090,  832, 1062,  860, 1886,  1868, 2004,  1840, 972,  860, 1062,  860, 1062,  860, 1088,  832, 998,  832, 1090,  832, 1062,  860, 1062,  860, 998,  832, 1090,  832, 1090,  832, 1064,  858, 970,  860, 1062,   [... cut ...] 1030,  890, 1034,  888, 1032,  890, 968,  862, 1060,  862, 1032,  890, 1032,  888, 1884,  1870, 1058,  2970, 2862,  890, 1030,  890, 1974,  1870, 1854,  1898, 1030,  890, 1060,  862, 966,  864, 1032,  890, 1034,  888, 1058,  862, 968,  862, 1974,  954, 942,  1896, 940,  890, 1058,  862, 1058,  864, 1034,  888, 942,  888, 1036,  886, 1058,  862, 1032,  890, 968,  862, 1032,  890, 1034,  888, 1032,  890, 968,  862, 1058,  864, 1034,  888, 1032,  890, 1854,  1898, 1030,  3912};  // UNKNOWN 2B04D06E

crankyoldgit, 2020-06-19:
Thanks for the report. As discussed in the other issue, I'm looking into it. It's definitely a bug.

crankyoldgit, 2020-06-19:
Note: This may take a few days. I'm re-writing the [Manchester Code](https://en.wikipedia.org/wiki/Manchester_code) matching from scratch to fix this.

crankyoldgit, 2020-06-21:
@juanbaldovi Can you please downloading the code from PR #1200 / Branch https://github.com/crankyoldgit/IRremoteESP8266/tree/manchester_redo and try it out for me?

I think it should solve it being incorrectly detected as `UNKNOWN` and now be detected as `AIRWELL`.
Note: The bit size has changed, and so will all previous codes you've recorded. i.e. You'll need to recapture them again to covert them to the new code/length etc.

It's still experimental, so it still might change. But, ... fingers crossed.

juanbaldovi, 2020-06-21:
@crankyoldgit now Airwell it's being detected! :-)
**cool. 19. Fan=3**
20:45:21.173 -> Timestamp : 000148.520
20:45:21.173 -> Library   : v2.7.7
20:45:21.173 -> 
20:45:21.173 -> Protocol  : AIRWELL
20:45:21.173 -> Code      : 0x60200002 (34 Bits)
20:45:21.173 -> uint16_t rawData[193] = {2998, 2860,  892, 1060,  862, 1058,  864, 1946,  892, 968,  1870, 1058,  862, 1060,  862, 968,  862, 1058,  864, 1058,  864, 1058,  864, 1882,  1870, 1058,  864, 1058,  864, 966,  864, 1058,  864, 1058,  864, 1058,  864, 966,  864, 1058,  864, 1058,  862, 1060,  862, 966,  864, 1058,  864, 1058,  864, 1058,  864, 968,  862, 1058,  864, 1058,  864, 1058,  864, 1882,  1870, 1058,  2970, 2888,  864, 1058,  864, 1058,  864, 1976,  862, 968,  1870, 1058,  864, 1058,  862, 968,  862, 1060,  862, 1058,  862, 1058,  864, 1882,  1870, 1058,  864, 1058,  864, 966,  862, 1058,  864, 1058,  864, 1058,  862, 966,  864, 1058,  862, 1060,  862, 1058,  864, 966,  862, 1060,  862, 1058,  864, 1058,  862, 968,  862, 1058,  862, 1060,  862, 1058,  862, 1884,  1870, 1058,  2970, 2890,  862, 1060,  862, 1058,  864, 1974,  862, 968,  1870, 1058,  862, 1060,  862, 968,  862, 1060,  864, 1058,  862, 1060,  862, 1884,  1870, 1058,  862, 1060,  862, 968,  862, 1060,  862, 1060,  862, 1060,  862, 968,  864, 1058,  862, 1060,  864, 1058,  862, 968,  862, 1060,  862, 1060,  862, 1058,  862, 968,  862, 1060,  862, 1060,  862, 1060,  862, 1884,  1868, 1060,  3884};  // AIRWELL 60200002
20:45:21.275 -> uint64_t data = 0x60200002;

**cool. 20. Fan=3**
20:45:54.133 -> Timestamp : 000181.506
20:45:54.133 -> Library   : v2.7.7
20:45:54.167 -> 
20:45:54.167 -> Protocol  : AIRWELL
20:45:54.167 -> Code      : 0x60280002 (34 Bits)
20:45:54.167 -> uint16_t rawData[149] = {2968, 2892,  862, 1060,  862, 1060,  862, 1976,  862, 968,  1868, 1060,  862, 1060,  862, 968,  862, 1060,  860, 1060,  862, 1060,  860, 1884,  1868, 1976,  1868, 968,  860, 1060,  860, 1062,  860, 1062,  860, 970,  860, 1062,  860, 1062,  860, 1062,  860, 970,  860, 1062,  860, 1062,  860, 1062,  860, 970,  860, 1062,  860, 1062,  860, 1062,  858, 1888,  1866, 1090,  2938, 2894,  858, 1090,  832, 1090,  832, 2006,  832, 998,  1840, 1090,  832, 1090,  832, 1000,  830, 1090,  832, 1090,  830, 1090,  832, 1914,  1838, 2006,  1838, 998,  830, 1092,  830, 1090,  830, 1092,  830, 1000,  830, 1092,  830, 1092,  830, 1092,  830, 1000,  830, 1092,  830, 1092,  830, 1092,  830, 1000,  830, 1092,  830, 1092,  828, 1094,  828, 1918,  1834, 1122,  2904, 2926,  826, 1096,  826, 1096,  826, 2010,  826, 1032,  1772, 1156,  796, 1126,  764, 1066,  518, 1402,  278, 1644,  364, 1782,  144, 2426,  256};  // AIRWELL 60280002
20:45:54.234 -> uint64_t data = 0x60280002;

**cool. 21. Fan=3**
20:46:17.700 -> Timestamp : 000205.068
20:46:17.700 -> Library   : v2.7.7
20:46:17.700 -> 
20:46:17.700 -> Protocol  : AIRWELL
20:46:17.700 -> Code      : 0x60300002 (34 Bits)
20:46:17.734 -> uint16_t rawData[193] = {3002, 2880,  870, 1084,  832, 1060,  860, 1974,  862, 968,  1870, 1058,  862, 1060,  862, 968,  862, 1060,  862, 1060,  862, 1060,  862, 1884,  954, 968,  1870, 1060,  862, 968,  862, 1060,  862, 1060,  862, 1060,  862, 968,  862, 1060,  862, 1060,  862, 1060,  860, 968,  862, 1060,  862, 1060,  860, 1060,  862, 968,  862, 1060,  862, 1060,  862, 1060,  860, 1884,  1868, 1060,  2966, 2892,  860, 1062,  860, 1062,  860, 1976,  862, 968,  1870, 1060,  860, 1062,  860, 970,  860, 1062,  860, 1062,  860, 1062,  860, 1886,  952, 970,  1868, 1062,  858, 970,  860, 1062,  860, 1062,  860, 1062,  860, 970,  858, 1062,  860, 1062,  858, 1064,  858, 972,  858, 1090,  832, 1064,  858, 1090,  832, 972,  858, 1090,  832, 1090,  832, 1088,  832, 1914,  1838, 1090,  2938, 2920,  832, 1090,  830, 1090,  832, 2006,  832, 998,  1838, 1090,  832, 1090,  832, 998,  832, 1090,  830, 1090,  832, 1090,  830, 1914,  924, 998,  1838, 1090,  832, 998,  832, 1090,  830, 1092,  832, 1090,  830, 1000,  832, 1090,  830, 1090,  830, 1092,  830, 1000,  830, 1090,  830, 1092,  830, 1092,  830, 1000,  830, 1092,  830, 1092,  830, 1090,  830, 1916,  1838, 1092,  3852};  // AIRWELL 60300002
20:46:17.836 -> uint64_t data = 0x60300002;

**cool. 21. Fan=auto**
20:46:49.886 -> Timestamp : 000237.254
20:46:49.886 -> Library   : v2.7.7
20:46:49.886 -> 
20:46:49.886 -> Protocol  : AIRWELL
20:46:49.886 -> Code      : 0x70300002 (34 Bits)
20:46:49.920 -> uint16_t rawData[193] = {3004, 2854,  864, 1088,  832, 1060,  862, 1976,  862, 968,  954, 968,  1868, 1060,  862, 968,  862, 1060,  862, 1060,  862, 1060,  860, 1884,  952, 970,  1868, 1060,  862, 968,  862, 1060,  860, 1060,  862, 1060,  860, 970,  860, 1060,  862, 1060,  860, 1060,  862, 970,  860, 1060,  860, 1062,  860, 1060,  860, 970,  860, 1062,  860, 1060,  860, 1062,  860, 1884,  1868, 1060,  2966, 2892,  860, 1062,  860, 1060,  860, 1976,  860, 970,  952, 970,  1868, 1062,  860, 970,  860, 1062,  858, 1062,  860, 1062,  860, 1886,  950, 970,  1868, 1062,  858, 972,  858, 1064,  858, 1064,  858, 1064,  858, 998,  832, 1090,  832, 1090,  832, 1090,  832, 998,  832, 1090,  832, 1090,  832, 1090,  832, 998,  832, 1090,  830, 1090,  830, 1090,  832, 1914,  1838, 1090,  2938, 2922,  830, 1090,  832, 1090,  832, 2006,  830, 1000,  922, 1000,  1838, 1092,  830, 998,  830, 1092,  830, 1090,  830, 1092,  830, 1914,  922, 1000,  1838, 1092,  830, 1000,  830, 1090,  830, 1092,  830, 1092,  830, 1000,  830, 1092,  830, 1092,  830, 1092,  830, 1000,  830, 1092,  828, 1094,  828, 1092,  828, 1002,  828, 1092, [... cut ...] 1916,  1838, 1092,  2936, 2922,  830, 1092,  830, 1092,  828, 2008,  1744, 1094,  828, 1094,  828, 1094,  828, 1002,  828, 1094,  828, 1094,  828, 1094,  828, 1918,  920, 1002,  1836, 1094,  828, 1004,  826, 1096,  828, 1094,  826, 1096,  826, 1032,  798, 1122,  798, 1124,  798, 1124,  798, 1032,  798, 1124,  798, 1124,  798, 1124,  796, 1034,  462};  // AIRWELL 40300002
20:47:20.580 -> uint64_t data = 0x40300002;

Please confirm if next step is to record on spreadshit and decode codes.
Thanks!!

NiKiZe, 2020-06-22:
Looks good, it would now be good to get the code (raw data no longer needed) for full range of temp without anything else change, the same for timers if they exist, reason is to be sure the code does increase withe the temp

crankyoldgit, 2020-06-22:
@juanbaldovi Thanks for confirming it works. I'm very very glad it is. That's the third time I've taken an axe to that code. Hopefully it's fixed for good now!

As PR #1200 will close this issue out when it's merged, when you've gathered more data per https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol
can you please create another new issue for that. i.e. adding detailed support.
Thanks in advance.

closed 2020-06-22 by crankyoldgit

crankyoldgit, 2020-06-23:
FYI, the code & changes referenced above have been included in the [v2.7.8 release](https://github.com/crankyoldgit/IRremoteESP8266/releases/tag/v2.7.8) of the library.

PR #1200 (2020-06-22): [BREAKING BUG] Fix Manchester code handling; Increase Airwell to `34` bits.
* [BREAKING] Airwell changing from  `32` to `34` bits. **_All previous codes are invalid._**
* [BREAKING] Remove sync bits from `sendManchester` & `IRrecv::matchManchester()` etal.
* Add `IRrecv::matchManchesterData()` method.
  - Try a different (better?) approach to handle matching Manchester code data portions.
* Rewrite of `decodeManchester()`
* Update unit tests accordingly.
* Add more unit tests to fix/find/identify bugs.

Fixes #1199 (Hopefully)
commit 1d211fd0 (2020-06-22): [BREAKING BUG] Fix Manchester code handling; Increase Airwell to `34` bits. (#1200)

* [BREAKING] Airwell changing from  `32` to `34` bits. **_All previous codes are invalid._**
* [BREAKING] Remove sync bits from `sendManchester` & `IRrecv::matchManchester()` etal.
* Add `IRrecv::matchManchesterData()` method.
  - Try a different (better?) approach to handle matching Manchester code data portions.
* Rewrite of `decodeManchester()`
* Update unit tests accordingly.
* Add more unit tests to fix/find/identify bugs.

Fixes #1199