crankyoldgit/IRremoteESP8266 #1202: Add detailed support for Airwell AC

juanbaldovi, 2020-06-22:
Hello
data has been captured and interpreted in [online spreadsheet](https://docs.google.com/spreadsheets/d/1MeqVsgXQAMSiAx7zNunG8qbFcIT8tqZtcjLmMa0GEng/edit?usp=sharing)

Also a second tab on the excel you'll find a command encoder which generates the same commands as captured

What I've found out is

On/off toggle | state[0] | 0b00000010
Mode | state[0] | 0b00000001
  | state[1] | 0b11000000
Fan | state[1] | 0b00110000
Temp | state[2] | 0b01111000
| state[3] | always 00
| state[4] | always 02

No checksum detected on command but commands are received twice on irDumpv3 each time a button is pressed so I guess the remote repeats the messages as the FAQ suggests.

Let me know if further capturing or analysis is required.

Regards

NiKiZe, 2020-06-23:
Nice work!
The AC switches on/off state, each time that bit is sent?
Lowest temp is 16 and highest 30?

juanbaldovi, 2020-06-23:
Hello @NiKiZe 
in any of the modes you can't below 16 or above 30 so yes, those are the lowest and highest.
And yes, just that bit changes when the on/off button is pressed.
It is a toggle command because the remote is stateless. Same command for on and off.

crankyoldgit, 2020-06-24:
FYI, I've started working on this now.

crankyoldgit, 2020-06-24:
@juanbaldovi Can you please download & test branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/DetailedAirwell / PR #1204 and let us know how it goes etc?
It should be fully supported now (i.e. IRrecvDumpV2+ & IRMQTTServer etc etc)

closed 2020-06-25 by crankyoldgit

crankyoldgit, 2020-07-30:
FYI, the aforementioned changes have been included in the latest release (v2.7.9) of the library.

PR #1204 (2020-06-25): Add detailed support for Airwell A/C protocol.
* Add support for controling:
  - Operation Mode
  - Power Toggle
  - Fan Speed
  - Temperature
* Common A/C API support added.
* Unit tests

Fixes #1202
commit 0fb37a3a (2020-06-25): Add detailed support for Airwell A/C protocol. (#1204)

* Add support for controlling:
  - Operation Mode
  - Power Toggle
  - Fan Speed
  - Temperature
* Enforce a fan speed of Low when in Dry mode.
* Common A/C API support added.
* Unit tests

Fixes #1202