crankyoldgit/IRremoteESP8266 #1205: Adding Support for Carrier AC 144 bit

casasale, 2020-06-25:
Hi,

I have checked version 2.7.7 of IRremoteESP8266, that is included in the latest development version of tasmota_ir, which has the just added basic support for Carrier 40 and 64 bit, but it did not work for my Carrier X-Power Gold Inverter ACs.

Consequently I looked at issue #1127 and went on to do the same, so I extracted the raw data from my Tasmota Tuya IR bridge, which was then processed and analyzed:

- The protocol seems to be using mostly 144 bits

- At the first glance there seem to be hints, that the message is being sent potentially more than once, but there is only partial data available in the raw data

- Nevertheless the AC accepts the first ON message replay, if it includes the following at its end: '0' followed by a DoubleLengthMARK, and then the same MARKs from the start followed by a '0'

The Google Sheet with all the results, raw data, and details can be found here: [https://docs.google.com/spreadsheets/d/1dKMio_OyL5GJKlCluuyghA2mIfKFLx9iMn5D38pcYig/edit?usp=drivesdk](https://docs.google.com/spreadsheets/d/1dKMio_OyL5GJKlCluuyghA2mIfKFLx9iMn5D38pcYig/edit?usp=drivesdk)

The 144 bit message format covers the normal different operating modes, device on/off, fan, and temperature settings. These are details:

`START(4406 4335); Header_80bit(0x55040051000555500001); Temp_8bit; 0_8bit; Fan_8bit; Mode_8bit; CheckSum1_8bit(Fan^Temp); CheckSum2_8bit(Mode^1); STOP(556 7441); STOP2_like_START_followed_by_0(4411 4335 565)`

Nevertheless the less frequently used commands seem to have diverging bit lengths, e.g.:

- Air direction is 32 bits shorter: 112 bits

- While the commands that enable Economy and Hi-Power seem to be even 16 bits longer: 160 bits  

Considering these bit length oddities for the rarely used commands, the idea is to get support for the most important 144 bit length commands, which would cover all my needs.

The binary number encoding uses '0' fillers to avoid subsequent '1's, e.g.:
0=00000000
1=00000001
2=00000100
3=00000101
4=00010000
5=00010001
6=00010100
7=00010101
8=01000000
9=01000001
…

Finally the remote control works with many models from the Carrier 42NQV series ACs, e.g. 42NQV060M2, 42NQV035M2, etc.

Thanks for the great IRremoteESP8266

NiKiZe, 2020-06-25:
Could you try to use the latest version, preferably git master
Several changes was made to Carrier in the last release, see https://github.com/crankyoldgit/IRremoteESP8266/releases/tag/v2.7.8

casasale, 2020-06-25:
I can try it, but the existing supported Carrier versions seem to look different .

Thanks

crankyoldgit, 2020-06-25:
@NiKiZe / @casasale I'm pretty sure this isn't an existing Carrier protocol. i.e. It's a new/not-yet-supported one.

@casasale 
Taking the raw data for your first raw tasmota entry in the spreadsheet (e.g. https://docs.google.com/spreadsheets/d/1dKMio_OyL5GJKlCluuyghA2mIfKFLx9iMn5D38pcYig/edit#gid=520670848&range=A2)

I've massaged it a bit, and used the `auto_analyse_raw_data.py` tool (as per https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-IR-protocol)

It gives the following output:
```shell
$ cat /tmp/data
uint16_t rawData[] = {4386,4330,574,1588,574,1612,546,1610,552,1612,548,532,548,530,548,1588,574,530,550,530,548,532,546,532,550,532,550,1586,548,1614,560,544,548,1614,548,508,572,530,550,530,550,530,550,532,548,532,548,1610,550,1612,548,1612,548,1588,572,1612,550,1584,576,1612,550,1588,574,530,550,530,548,510,572,530,548,530,552,526,554,530,550,532,548,530,550,1612,550,530,550,1588,572,532,550,1610,550,532,548,532,548,530,550,530,550,530,550,528,552,532,550,530,548,530,550,530,550,532,548,532,550,534,546,530,550,530,550,528,552,506,548,558,548,530,548,528,554,526,552,1612,548,530,550,1608,552,530,548,532,548,532,548,1586,574,7456,4388,4330,574,1588,572,1612,550,1612,548,1612,550,530,550,530,550,1612,548,532,550,530,550,528,552,530,550,530,548,1612,550,1608,554,530,550,1610,550,530,552,504,548,558,548,530,548,530,552,526,552,1612,548,1610,552,1612,548,1612,550,1586,572,1608,554,1606,554,1610,550,528,552,528,552,526,552,530};
$ tools/auto_analyse_raw_data.py -f /tmp/data
Found 218 timing entries.
Potential Mark Candidates:
[4388, 576]
Potential Space Candidates:
[7456, 4330, 1614, 558]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kHdrMark   = 4387
kHdrSpace  = 4330
kBitMark   = 552
kOneSpace  = 1604
kZeroSpace = 529
kSpaceGap = 7456

Decoding protocol based on analysis so far:

kHdrMark+kHdrSpace+111100100000110100000011111111000000000101010000000000000000000001010001GAP(7456)
  Bits: 72
  Hex:  0xF20D03FC0150000051 (MSB first)
        0x8A00000A803FC0B04F (LSB first)
  Dec:  4465049936067702030417 (MSB first)
        2545650693717859807311 (LSB first)
  Bin:  0b111100100000110100000011111111000000000101010000000000000000000001010001 (MSB first)
        0b100010100000000000000000000010101000000000111111110000001011000001001111 (LSB first)
kHdrMark+kHdrSpace+1111001000001101000000111111110000
  Bits: 34
  Hex:  0x3C8340FF0 (MSB first)
        0x3FC0B04F (LSB first)
  Dec:  16243757040 (MSB first)
        1069592655 (LSB first)
  Bin:  0b1111001000001101000000111111110000 (MSB first)
        0b0000111111110000001011000001001111 (LSB first)

Total Nr. of suspected bits: 106
```

It doesn't look 144 bit to me.
The data/analysis seems to indicate 72 + 34 = `106` bits .. or less, as the trailing `34` bits in the second section of data appear to be a repeat of part of the first section of data.
e.g. (When looking at it in LSB first order)
0x8A00000A80**3FC0B04F** & 0x**3FC0B04F**.

i.e. The protocol could actually only have <= 72 bits of data in it.

I'm also confused at where the `112`, `144`, or `160` bits you mention fit this example data.

Can I suggest you convert the data you've collected (or use `IRrecvDumpV2` or V3) and follow the guides in the wiki so we have a clearer picture of what this protocol really looks like?

While your spreadsheet is a great bit of work, I'm worried it might have some fundamental issues that need to be resolved before we can use/trust the analysis/data etc. That is, my quick analysis disagrees with yours significantly.

casasale, 2020-06-25:
@crankyoldgit Thank you very much for analyzing and confirming, that it is a new Carrier protocol. I will do the suggested items during the next days and come back.

casasale, 2020-06-25:
@crankyoldgit Do I need to obtain data via IRrecvDumpV2 or V3, or would it be enough to convert the data that I get from Tasmota?

casasale, 2020-06-25:
So far, looking at the 3 different length messages it seems like most of the commands are 72 bits, a few (AirDir/Swing) are 56 bits and the long ones (Eco/Hi) are 80 bits.

casasale, 2020-06-26:
Here is the latest spreadsheet based on the converted data which was then processed by auto_analyse_raw_data.py:
[https://docs.google.com/spreadsheets/d/1euu5TNvqWq7vJoAYTrwJs-gor1jJHC0PhEWY8c-0q-A/edit?usp=drivesdk](https://docs.google.com/spreadsheets/d/1euu5TNvqWq7vJoAYTrwJs-gor1jJHC0PhEWY8c-0q-A/edit?usp=drivesdk)

My previous fundamental error was to assume something like a pulse length encoding instead of the space encoding, and now things seem to look more reasonable.

Looking for Carrier 80 bit I found a Carrier decoder file for 40, 64 and 80 bit on Github:
[https://github.com/ToniA/Raw-IR-decoder-for-Arduino](https://github.com/ToniA/Raw-IR-decoder-for-Arduino) 
The file that contains the Carrier decoders is:
[https://github.com/ToniA/Raw-IR-decoder-for-Arduino/raw/master/Carrier.cpp](https://github.com/ToniA/Raw-IR-decoder-for-Arduino/raw/master/Carrier.cpp)

It checks for 0x4F and 0xB0 at the beginning of the bytes to decide if it is a carrier protocol which would match the LSB HEX in my case, if the bytes are looked at from the back: 0x8A00000A803FC0**B04F** 

But, what strikes me a bit is the unexpected order of the temperatures the LSB approach that decodeCarrier3() uses:
`    const byte temperatures[]  = { 17, 25, 21, 29, 19, 27, 23, 00, 18, 26, 22, 30, 20, 28, 24 };`

In the spreadsheet I have analyzed the MSB version, and there the temperatures end up in a correct order assigned to numbers from 0 to 13, so my conclusion is that the MSB version seems to look nicer, and hence should be correct.
@crankyoldgit and @NiKiZe What do you think which version is the correct one: LSB or MSB? 

Nevertheless these different message lengths (72, 56, 80) seem somehow off.

NiKiZe, 2020-06-26:
If temp goes up with MSB one step at a time then use that, also you will find "the same bytes" at the beginning.

My approach for this would be to update the test file for Carrier with the raw data you collect, and then add a decoder, when that is verified to work, add a send function and verify that on the AC

When this is in place continue with the analysis

crankyoldgit, 2020-06-26:
@casasale I'll try to take a look tomorrow night/day after that and see if we can make a simple decoder/sender for it.
I'm with you all, MSB sounds like the way to do it to me.

crankyoldgit, 2020-06-26:
@casasale I've had a quick look at your raw data collected via Tasmota. I think it's being truncated. Can you please capture one each of the 56, 72, 80 bit messages using IRrecvDumpV2 (or better)? (that would also explains the odd bit size of 34bits etc for the "second" section of data. eg. 72 + 34,  80 + 26, & 56 + 50) I just want to confirm it is a "repeat" of exactly the same size & data for each bit size. e.g. 72 + 72, 80 + 80, & 56 + 56 etc.

Looking at your captured data, it seems tasmota's buffer is probably around 220. We probably need to extend it out to 340 or so to be safe. IRrecvDumpV2 etc uses `1000`.

Tasmota probably has some feature to increase the capture buffer for it's IR stuff, increasing that would be fine too.
(Paging @s-hadinger who probably knows what/where to change that value in Tasmota)

casasale, 2020-06-26:
@crankyoldgit Yes, the default buffer size value is low in tasmota-ir, and there seems to be no other way to change it, but to compile:

#define IR_RCV_BUFFER_SIZE 100
#define IR_RCV_TIMEOUT 15
#define IR_RCV_MIN_UNKNOWN_SIZE 6

casasale, 2020-06-27:
72 bit full message example for ON:

uint16_t rawData_ON[295] = {4426, 4314,  588, 1572,  590, 1572,  588, 1572,  588, 1572,  588, 492,  588, 492,  590, 1572,  588, 492,  588, 492,  588, 492,  590, 490,  588, 492,  590, 1572,  588, 1572,  588, 492,  588, 1572,  590, 492,  588, 492,  588, 492,  588, 492,  588, 492,  588, 490,  588, 1572,  588, 1572,  588, 1572,  588, 1572,  588, 1572,  588, 1572,  588, 1572,  590, 1570,  590, 490,  588, 492,  588, 492,  588, 492,  588, 494,  588, 492,  586, 492,  588, 492,  588, 492,  588, 1572,  588, 492,  588, 1572,  588, 492,  588, 1572,  588, 492,  586, 492,  588, 492,  590, 490,  588, 492,  588, 494,  586, 492,  588, 492,  586, 492,  588, 492,  588, 490,  588, 492,  586, 492,  590, 492,  586, 494,  588, 492,  588, 492,  588, 496,  584, 492,  588, 492,  588, 492,  588, 1572,  588, 492,  586, 1574,  588, 492,  588, 492,  588, 492,  590, 1570,  588, 7420,  4424, 4318,  588, 1572,  588, 1572,  590, 1572,  588, 1570,  588, 492,  588, 492,  588, 1572,  588, 492,  588, 492,  590, 492,  586, 492,  588, 490,  588, 1572,  588, 1572,  588, 492,  588, 1576,  584, 492,  590, 490,  588, 492,  588, 492,  588, 492,  588, 492,  588, 1574,  588, 1572,  588, 1576,  584, 1574,  586, 1574,  588, 1574,  584, 1574,  586, 1574,  588, 494,  584, 494,  588, 492,  588, 492,  588, 492,  590, 494,  584, 494,  586, 494,  588, 492,  586, 1572,  588, 492,  590, 1572,  588, 490,  588, 1574,  590, 490,  588, 490,  590, 490,  590, 490,  588, 490,  588, 490,  590, 490,  590, 492,  588, 492,  588, 492,  590, 494,  584, 490,  590, 498,  584, 490,  588, 492,  590, 490,  592, 488,  592, 490,  590, 494,  584, 490,  592, 494,  586, 1574,  586, 490,  590, 1570,  590, 492,  588, 490,  590, 488,  590, 1570,  590};  // TOSHIBA_AC
Found 295 timing entries.
Potential Mark Candidates:
[4426, 592]
Potential Space Candidates:
[7420, 4318, 1576, 498]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kHdrMark   = 4425
kHdrSpace  = 4316
kBitMark   = 588
kOneSpace  = 1572
kZeroSpace = 491
kSpaceGap = 7420

Decoding protocol based on analysis so far:

kHdrMark+kHdrSpace+111100100000110100000011111111000000000101010000000000000000000001010001GAP(7420)
  Bits: 72
  Hex:  0xF20D03FC0150000051 (MSB first)
        0x8A00000A803FC0B04F (LSB first)
  Dec:  4465049936067702030417 (MSB first)
        2545650693717859807311 (LSB first)
  Bin:  0b111100100000110100000011111111000000000101010000000000000000000001010001 (MSB first)
        0b100010100000000000000000000010101000000000111111110000001011000001001111 (LSB first)
kHdrMark+kHdrSpace+111100100000110100000011111111000000000101010000000000000000000001010001
  Bits: 72
  Hex:  0xF20D03FC0150000051 (MSB first)
        0x8A00000A803FC0B04F (LSB first)
  Dec:  4465049936067702030417 (MSB first)
        2545650693717859807311 (LSB first)
  Bin:  0b111100100000110100000011111111000000000101010000000000000000000001010001 (MSB first)
        0b100010100000000000000000000010101000000000111111110000001011000001001111 (LSB first)

Total Nr. of suspected bits: 144

casasale, 2020-06-27:
56 bits full message example for SWING_ON and SWING_OFF:
uint16_t rawData_SWING_ON[231] = {4426, 4316,  590, 1572,  590, 1578,  584, 1570,  590, 1570,  590, 490,  588, 490,  590, 1570,  590, 490,  588, 492,  588, 492,  590, 490,  590, 490,  588, 1572,  590, 1570,  592, 490,  590, 1570,  590, 490,  588, 492,  590, 490,  590, 490,  590, 492,  588, 488,  590, 490,  590, 1572,  588, 1572,  590, 1570,  590, 1570,  590, 1574,  586, 1570,  590, 1570,  590, 1570,  590, 490,  590, 490,  590, 490,  590, 1570,  590, 490,  590, 490,  590, 490,  590, 490,  588, 1570,  590, 490,  590, 490,  588, 490,  590, 490,  590, 490,  590, 492,  588, 488,  590, 1570,  592, 488,  590, 496,  584, 1572,  590, 494,  588, 488,  590, 494,  586, 494,  586, 490,  588, 7420,  4428, 4314,  590, 1570,  590, 1570,  590, 1570,  590, 1570,  590, 490,  590, 494,  586, 1572,  590, 490,  590, 490,  590, 490,  590, 490,  590, 490,  590, 1570,  590, 1570,  590, 492,  590, 1570,  590, 490,  590, 492,  586, 494,  588, 488,  590, 490,  590, 490,  590, 490,  590, 1574,  586, 1570,  592, 1570,  590, 1574,  588, 1570,  590, 1570,  590, 1572,  590, 1570,  590, 490,  590, 494,  586, 494,  586, 1570,  590, 490,  590, 490,  588, 494,  586, 496,  584, 1572,  590, 492,  586, 490,  590, 494,  588, 492,  584, 494,  586, 496,  584, 496,  582, 1574,  588, 492,  588, 494,  584, 1572,  590, 494,  586, 494,  586, 494,  588, 488,  590, 494,  586};  // UNKNOWN 9C9F3C37
Found 231 timing entries.
Potential Mark Candidates:
[4428, 592]
Potential Space Candidates:
[7420, 4316, 1578, 496]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kHdrMark   = 4427
kHdrSpace  = 4315
kBitMark   = 588
kOneSpace  = 1571
kZeroSpace = 491
kSpaceGap = 7420

Decoding protocol based on analysis so far:

kHdrMark+kHdrSpace+11110010000011010000000111111110001000010000000100100000GAP(7420)
  Bits: 56
  Hex:  0xF20D01FE210120 (MSB first)
        0x0480847F80B04F (LSB first)
  Dec:  68131246573682976 (MSB first)
        1267206470021199 (LSB first)
  Bin:  0b11110010000011010000000111111110001000010000000100100000 (MSB first)
        0b00000100100000001000010001111111100000001011000001001111 (LSB first)
kHdrMark+kHdrSpace+11110010000011010000000111111110001000010000000100100000
  Bits: 56
  Hex:  0xF20D01FE210120 (MSB first)
        0x0480847F80B04F (LSB first)
  Dec:  68131246573682976 (MSB first)
        1267206470021199 (LSB first)
  Bin:  0b11110010000011010000000111111110001000010000000100100000 (MSB first)
        0b00000100100000001000010001111111100000001011000001001111 (LSB first)

Total Nr. of suspected bits: 112

---------------------------------------------------------------------------------------------------------------------------------------

uint16_t rawData_SWING_OFF[231] = {4424, 4318,  586, 1576,  586, 1572,  588, 1574,  588, 1572,  588, 494,  586, 498,  584, 1574,  586, 494,  588, 492,  588, 492,  586, 492,  588, 494,  586, 1574,  586, 1576,  584, 492,  588, 1578,  582, 494,  588, 490,  586, 494,  588, 498,  580, 496,  586, 494,  586, 492,  588, 1576,  582, 1574,  586, 1578,  584, 1572,  588, 1574,  588, 1574,  586, 1574,  586, 1574,  588, 494,  586, 494,  586, 494,  590, 1570,  588, 492,  586, 498,  584, 494,  586, 494,  588, 1576,  582, 494,  586, 494,  586, 494,  586, 492,  586, 494,  586, 492,  588, 1572,  588, 492,  586, 494,  586, 494,  586, 1574,  588, 492,  588, 492,  588, 494,  588, 1578,  582, 1574,  588, 7424,  4420, 4318,  588, 1574,  586, 1576,  584, 1580,  582, 1574,  588, 492,  588, 494,  584, 1574,  588, 492,  586, 496,  584, 494,  586, 492,  588, 494,  586, 1574,  588, 1574,  584, 494,  588, 1574,  588, 492,  586, 494,  586, 494,  588, 492,  586, 494,  588, 492,  586, 494,  584, 1580,  584, 1574,  588, 1572,  588, 1572,  588, 1576,  584, 1578,  584, 1572,  588, 1572,  588, 494,  586, 494,  584, 494,  588, 1574,  586, 496,  584, 494,  586, 492,  588, 492,  588, 1572,  588, 494,  588, 492,  586, 496,  586, 494,  586, 492,  588, 492,  586, 1574,  588, 492,  586, 492,  588, 496,  584, 1576,  584, 492,  586, 494,  588, 492,  588, 1574,  586, 1574,  588};  // UNKNOWN 2BC29FFB
Found 231 timing entries.
Potential Mark Candidates:
[4424, 590]
Potential Space Candidates:
[7424, 4318, 1580, 498]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kHdrMark   = 4422
kHdrSpace  = 4318
kBitMark   = 586
kOneSpace  = 1574
kZeroSpace = 493
kSpaceGap = 7424

Decoding protocol based on analysis so far:

kHdrMark+kHdrSpace+11110010000011010000000111111110001000010000001000100011GAP(7424)
  Bits: 56
  Hex:  0xF20D01FE210223 (MSB first)
        0xC440847F80B04F (LSB first)
  Dec:  68131246573683235 (MSB first)
        55240033254289487 (LSB first)
  Bin:  0b11110010000011010000000111111110001000010000001000100011 (MSB first)
        0b11000100010000001000010001111111100000001011000001001111 (LSB first)
kHdrMark+kHdrSpace+11110010000011010000000111111110001000010000001000100011
  Bits: 56
  Hex:  0xF20D01FE210223 (MSB first)
        0xC440847F80B04F (LSB first)
  Dec:  68131246573683235 (MSB first)
        55240033254289487 (LSB first)
  Bin:  0b11110010000011010000000111111110001000010000001000100011 (MSB first)
        0b11000100010000001000010001111111100000001011000001001111 (LSB first)

Total Nr. of suspected bits: 112

casasale, 2020-06-27:
56 bits full message example for AIR_DIRECTION:

uint16_t rawData_AIR_DIRECTION[115] = {4424, 4318,  588, 1574,  588, 1572,  588, 1574,  586, 1572,  590, 490,  586, 494,  586, 1574,  588, 492,  586, 494,  586, 494,  586, 496,  584, 492,  588, 1572,  588, 1572,  590, 492,  588, 1572,  590, 490,  588, 492,  586, 494,  588, 492,  588, 492,  588, 494,  584, 492,  588, 1572,  588, 1574,  588, 1574,  588, 1572,  588, 1572,  588, 1572,  588, 1574,  590, 1570,  588, 494,  586, 496,  584, 494,  588, 1572,  588, 492,  588, 492,  588, 490,  588, 492,  590, 1572,  588, 492,  588, 496,  586, 492,  588, 492,  586, 492,  588, 492,  588, 490,  590, 490,  588, 492,  588, 490,  588, 1572,  588, 492,  588, 492,  588, 490,  588, 492,  588, 1572,  586};  // UNKNOWN DEB8845C
Found 115 timing entries.
Potential Mark Candidates:
[4424, 590]
Potential Space Candidates:
[4318, 1574, 496]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kHdrMark   = 4424
kHdrSpace  = 4318
kBitMark   = 587
kOneSpace  = 1572
kZeroSpace = 492

Decoding protocol based on analysis so far:

kHdrMark+kHdrSpace+11110010000011010000000111111110001000010000000000100001
  Bits: 56
  Hex:  0xF20D01FE210021 (MSB first)
        0x8400847F80B04F (LSB first)
  Dec:  68131246573682721 (MSB first)
        37155266000629839 (LSB first)
  Bin:  0b11110010000011010000000111111110001000010000000000100001 (MSB first)
        0b10000100000000001000010001111111100000001011000001001111 (LSB first)

Total Nr. of suspected bits: 56

casasale, 2020-06-27:
80 bits full message example for HIGH_POWER ON:

uint16_t rawData_HIGH_POWER_ON[327] = {4424, 4320,  582, 1574,  588, 1578,  582, 1574,  586, 1578,  586, 496,  582, 492,  586, 1576,  586, 492,  586, 492,  588, 496,  584, 496,  584, 496,  584, 1626,  534, 1626,  534, 494,  586, 1578,  582, 494,  586, 494,  586, 494,  588, 492,  586, 492,  586, 1576,  586, 494,  588, 492,  588, 1574,  588, 1576,  584, 1578,  584, 1574,  588, 1574,  588, 492,  588, 1572,  590, 1570,  590, 492,  588, 492,  590, 488,  590, 494,  584, 1570,  592, 492,  586, 490,  590, 1572,  590, 490,  590, 1570,  590, 490,  590, 1570,  590, 492,  588, 490,  588, 492,  588, 492,  590, 490,  590, 494,  586, 490,  590, 490,  588, 490,  590, 490,  588, 492,  590, 490,  588, 492,  590, 490,  590, 490,  590, 494,  584, 490,  590, 490,  590, 490,  590, 490,  588, 490,  588, 492,  588, 492,  586, 492,  588, 490,  588, 492,  588, 490,  590, 1572,  588, 494,  586, 1574,  588, 492,  588, 1572,  590, 1572,  588, 492,  588, 492,  586, 494,  588, 7422,  4424, 4320,  586, 1572,  588, 1572,  588, 1576,  584, 1574,  588, 494,  586, 492,  588, 1572,  588, 492,  588, 492,  588, 492,  588, 494,  586, 496,  584, 1574,  586, 1578,  582, 494,  586, 1578,  584, 494,  586, 492,  588, 492,  586, 496,  584, 494,  586, 1578,  584, 494,  586, 494,  584, 1574,  588, 1572,  586, 1574,  588, 1574,  588, 1572,  588, 494,  590, 1572,  588, 1574,  588, 492,  588, 492,  588, 492,  586, 492,  588, 1572,  588, 498,  582, 492,  588, 1576,  586, 492,  588, 1572,  588, 494,  588, 1572,  588, 492,  586, 492,  588, 492,  590, 490,  588, 492,  586, 492,  588, 492,  590, 490,  588, 490,  588, 492,  590, 490,  588, 492,  590, 490,  588, 490,  590, 490,  590, 490,  592, 488,  592, 494,  584, 494,  586, 490,  590, 494,  586, 494,  588, 488,  592, 490,  588, 492,  586, 490,  592, 490,  588, 1576,  584, 494,  586, 1570,  590, 494,  586, 1576,  582, 1572,  590, 490,  590, 490,  588, 490,  590};  // UNKNOWN 54926187

casasale, 2020-06-27:
80 bit full message example for ECO ON:

uint16_t rawData_ECO_ON[327] = {4426, 4316,  590, 1572,  590, 1572,  590, 1570,  588, 1572,  588, 490,  590, 490,  590, 1574,  586, 490,  590, 490,  592, 494,  586, 488,  590, 490,  588, 1572,  590, 1572,  588, 490,  590, 1570,  588, 492,  588, 492,  588, 490,  592, 490,  588, 494,  586, 1570,  590, 492,  588, 492,  588, 1572,  590, 1572,  590, 1570,  588, 1572,  590, 1570,  590, 492,  588, 1572,  590, 1576,  584, 494,  586, 494,  586, 490,  590, 490,  590, 1572,  590, 490,  590, 490,  590, 1572,  592, 488,  590, 1572,  588, 496,  584, 1570,  590, 492,  590, 490,  592, 492,  586, 492,  590, 494,  588, 494,  582, 496,  584, 490,  590, 490,  590, 494,  584, 496,  584, 492,  588, 496,  584, 492,  590, 490,  590, 490,  588, 490,  590, 496,  582, 492,  590, 488,  592, 492,  586, 490,  588, 492,  590, 490,  590, 494,  586, 490,  590, 1572,  590, 1570,  590, 490,  590, 1570,  588, 492,  590, 1570,  590, 1570,  588, 492,  588, 1572,  588, 492,  588, 7422,  4424, 4316,  590, 1572,  590, 1576,  584, 1570,  590, 1572,  588, 492,  588, 494,  588, 1570,  588, 496,  586, 490,  590, 490,  590, 490,  590, 496,  584, 1576,  584, 1572,  590, 490,  590, 1570,  590, 490,  588, 496,  584, 494,  586, 494,  586, 496,  584, 1576,  586, 490,  590, 492,  588, 1572,  590, 1570,  590, 1570,  590, 1570,  592, 1570,  590, 490,  588, 1572,  590, 1570,  590, 490,  588, 492,  588, 496,  584, 496,  584, 1570,  592, 490,  590, 490,  588, 1572,  590, 492,  586, 1570,  590, 496,  584, 1570,  590, 494,  588, 490,  590, 490,  588, 494,  586, 494,  586, 490,  590, 496,  584, 490,  590, 492,  590, 494,  586, 492,  588, 490,  590, 490,  590, 488,  590, 496,  584, 494,  586, 490,  590, 494,  586, 490,  590, 494,  586, 490,  590, 492,  590, 490,  590, 488,  590, 494,  586, 494,  586, 1570,  590, 1570,  590, 494,  588, 1570,  590, 490,  590, 1570,  590, 1576,  584, 496,  586, 1570,  590, 490,  588};  // UNKNOWN 58D214DF
Found 327 timing entries.
Potential Mark Candidates:
[4426, 592]
Potential Space Candidates:
[7422, 4316, 1576, 496]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kHdrMark   = 4425
kHdrSpace  = 4316
kBitMark   = 588
kOneSpace  = 1571
kZeroSpace = 491
kSpaceGap = 7422

Decoding protocol based on analysis so far:

kHdrMark+kHdrSpace+11110010000011010000010011111011000010010101000000000000000000000000001101011010GAP(7422)
  Bits: 80
  Hex:  0xF20D04FB09500000035A (MSB first)
        0x5AC000000A90DF20B04F (LSB first)
  Dec:  1143052855418246874006362 (MSB first)
        428554758332037638172751 (LSB first)
  Bin:  0b11110010000011010000010011111011000010010101000000000000000000000000001101011010 (MSB first)
        0b01011010110000000000000000000000000010101001000011011111001000001011000001001111 (LSB first)
kHdrMark+kHdrSpace+11110010000011010000010011111011000010010101000000000000000000000000001101011010
  Bits: 80
  Hex:  0xF20D04FB09500000035A (MSB first)
        0x5AC000000A90DF20B04F (LSB first)
  Dec:  1143052855418246874006362 (MSB first)
        428554758332037638172751 (LSB first)
  Bin:  0b11110010000011010000010011111011000010010101000000000000000000000000001101011010 (MSB first)
        0b01011010110000000000000000000000000010101001000011011111001000001011000001001111 (LSB first)

Total Nr. of suspected bits: 160

casasale, 2020-06-27:
80 bit message example for HI_POWER + TEMP23:
uint16_t rawData_HIGH_POWER_ON_TEMP23[327] = {4420, 4322,  584, 1576,  584, 1576,  584, 1576,  584, 1576,  584, 500,  580, 494,  586, 1574,  584, 496,  584, 494,  584, 500,  582, 494,  584, 496,  582, 1576,  586, 1576,  584, 500,  582, 1578,  582, 494,  584, 494,  586, 494,  586, 494,  586, 494,  584, 1574,  586, 496,  584, 494,  586, 1574,  586, 1576,  584, 1576,  586, 1574,  586, 1576,  586, 498,  582, 1576,  588, 1574,  586, 500,  580, 494,  586, 496,  582, 496,  586, 1574,  586, 496,  584, 496,  582, 1578,  582, 500,  582, 1574,  586, 1576,  584, 498,  582, 502,  578, 498,  580, 498,  584, 494,  586, 494,  584, 500,  578, 498,  586, 500,  580, 498,  582, 496,  586, 500,  578, 498,  582, 496,  584, 494,  584, 498,  584, 498,  580, 496,  586, 496,  582, 496,  582, 498,  584, 496,  584, 494,  586, 494,  586, 500,  580, 496,  584, 498,  582, 496,  584, 1576,  586, 500,  580, 1576,  582, 1576,  586, 496,  584, 1576,  586, 496,  582, 498,  582, 496,  586, 7430,  4414, 4322,  584, 1578,  584, 1576,  582, 1582,  580, 1580,  582, 498,  582, 496,  584, 1576,  584, 496,  584, 496,  584, 496,  584, 496,  584, 496,  586, 1574,  586, 1576,  586, 492,  586, 1576,  586, 494,  584, 496,  584, 496,  584, 498,  582, 494,  586, 1576,  582, 504,  576, 498,  584, 1574,  584, 1576,  584, 1580,  582, 1578,  584, 1576,  584, 496,  584, 1576,  582, 1578,  584, 496,  584, 494,  584, 498,  580, 498,  582, 1576,  586, 496,  582, 498,  584, 1576,  586, 496,  584, 1576,  584, 1576,  586, 496,  584, 496,  584, 496,  584, 496,  584, 496,  584, 496,  586, 494,  584, 496,  586, 496,  584, 498,  582, 496,  584, 496,  584, 496,  586, 494,  586, 494,  586, 496,  584, 496,  584, 496,  584, 500,  580, 496,  586, 494,  584, 496,  584, 496,  582, 554,  526, 502,  578, 496,  586, 496,  584, 494,  584, 1578,  584, 496,  584, 1576,  586, 1576,  584, 496,  582, 1578,  584, 494,  584, 496,  584, 496,  584};  // UNKNOWN 35DF3237
Found 327 timing entries.
Potential Mark Candidates:
[4420, 588]
Potential Space Candidates:
[7430, 4322, 1582, 554]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kHdrMark   = 4417
kHdrSpace  = 4322
kBitMark   = 583
kOneSpace  = 1576
kZeroSpace = 496
kSpaceGap = 7430

Decoding protocol based on analysis so far:

kHdrMark+kHdrSpace+11110010000011010000010011111011000010010110000000000000000000000000000101101000GAP(7430)
  Bits: 80
  Hex:  0xF20D04FB096000000168 (MSB first)
        0x168000000690DF20B04F (LSB first)
  Dec:  1143052855418315593482600 (MSB first)
        106253245871786305826895 (LSB first)
  Bin:  0b11110010000011010000010011111011000010010110000000000000000000000000000101101000 (MSB first)
        0b00010110100000000000000000000000000001101001000011011111001000001011000001001111 (LSB first)
kHdrMark+kHdrSpace+11110010000011010000010011111011000010010110000000000000000000000000000101101000
  Bits: 80
  Hex:  0xF20D04FB096000000168 (MSB first)
        0x168000000690DF20B04F (LSB first)
  Dec:  1143052855418315593482600 (MSB first)
        106253245871786305826895 (LSB first)
  Bin:  0b11110010000011010000010011111011000010010110000000000000000000000000000101101000 (MSB first)
        0b00010110100000000000000000000000000001101001000011011111001000001011000001001111 (LSB first)

Total Nr. of suspected bits: 160

casasale, 2020-06-27:
80 bit message example for ECO_ON and TEMP24:
uint16_t rawData_ECO_ON_TEMP24[327] = {4420, 4318,  586, 1576,  558, 1600,  556, 1606,  554, 1606,  556, 524,  556, 526,  552, 1606,  554, 530,  548, 526,  552, 528,  554, 526,  552, 526,  554, 1612,  548, 1606,  554, 526,  554, 1606,  554, 528,  552, 526,  554, 526,  554, 526,  556, 524,  556, 1608,  550, 526,  554, 526,  552, 1608,  554, 1606,  556, 1606,  554, 1610,  550, 1608,  552, 528,  550, 1606,  556, 1604,  554, 528,  554, 526,  556, 522,  558, 526,  552, 1610,  548, 530,  552, 524,  556, 1610,  550, 524,  554, 1610,  550, 1606,  554, 1606,  556, 530,  550, 526,  554, 524,  554, 526,  556, 526,  554, 524,  556, 526,  554, 528,  554, 526,  556, 526,  552, 526,  554, 528,  552, 526,  554, 526,  554, 530,  550, 526,  554, 528,  552, 524,  554, 526,  556, 524,  556, 524,  556, 524,  556, 524,  556, 524,  554, 526,  556, 524,  554, 1664,  496, 1614,  548, 526,  552, 1608,  554, 1606,  556, 1606,  558, 1604,  558, 524,  552, 1610,  550, 528,  552, 7452,  4394, 4348,  558, 1602,  556, 1604,  558, 1604,  556, 1610,  550, 526,  552, 524,  556, 1608,  554, 530,  550, 526,  554, 526,  554, 528,  550, 524,  558, 1604,  556, 1602,  558, 524,  556, 1608,  552, 522,  556, 526,  554, 524,  556, 524,  558, 522,  556, 1608,  552, 524,  556, 522,  556, 1604,  556, 1604,  558, 1604,  556, 1604,  556, 1606,  556, 526,  554, 1606,  554, 1608,  552, 524,  556, 530,  548, 530,  552, 524,  554, 1608,  552, 526,  556, 526,  554, 1606,  556, 522,  556, 1608,  554, 1604,  556, 1604,  556, 526,  554, 526,  554, 524,  554, 526,  554, 524,  556, 526,  554, 524,  558, 524,  552, 528,  554, 522,  556, 524,  556, 532,  546, 530,  552, 524,  556, 526,  552, 526,  554, 524,  554, 524,  554, 532,  550, 524,  554, 524,  556, 524,  556, 526,  556, 526,  552, 524,  556, 524,  556, 1610,  552, 1608,  554, 528,  552, 1606,  554, 1610,  552, 1604,  556, 1604,  556, 524,  558, 1604,  554, 530,  552};  // UNKNOWN 3008239F
Found 327 timing entries.
Potential Mark Candidates:
[4420, 586]
Potential Space Candidates:
[7452, 4348, 1664, 532]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kHdrMark   = 4407
kHdrSpace  = 4333
kBitMark   = 553
kOneSpace  = 1606
kZeroSpace = 525
kSpaceGap = 7452

Decoding protocol based on analysis so far:

kHdrMark+kHdrSpace+11110010000011010000010011111011000010010111000000000000000000000000001101111010GAP(7452)
  Bits: 80
  Hex:  0xF20D04FB09700000037A (MSB first)
        0x5EC000000E90DF20B04F (LSB first)
  Dec:  1143052855418384312959866 (MSB first)
        447444224267914265538639 (LSB first)
  Bin:  0b11110010000011010000010011111011000010010111000000000000000000000000001101111010 (MSB first)
        0b01011110110000000000000000000000000011101001000011011111001000001011000001001111 (LSB first)
kHdrMark+kHdrSpace+11110010000011010000010011111011000010010111000000000000000000000000001101111010
  Bits: 80
  Hex:  0xF20D04FB09700000037A (MSB first)
        0x5EC000000E90DF20B04F (LSB first)
  Dec:  1143052855418384312959866 (MSB first)
        447444224267914265538639 (LSB first)
  Bin:  0b11110010000011010000010011111011000010010111000000000000000000000000001101111010 (MSB first)
        0b01011110110000000000000000000000000011101001000011011111001000001011000001001111 (LSB first)

Total Nr. of suspected bits: 160

casasale, 2020-06-27:
Protocol summary:
- Normal operation (On/Off, Mode, Fan, temp): 72 bits message repeated twice with GAP in-between
- Swing On/Off: 56 bits message repeated twice with GAP in-between
- Air direction: 56 bits message without repeating and without GAP
-If Hi-Power or Eco is set on, and any normal operation command used with Hi-Power/Eco : 80 bits  message repeated twice with GAP in-between

crankyoldgit, 2020-06-27:
> 72 bit full message example for ON:
> 
> ```
> uint16_t rawData_ON[295] = {4426, 4314,  [SNIP],  590};  // TOSHIBA_AC
> ```

TOSHIBA_AC? Is IRrecvDumpV2 etc detecting this as the `TOSHIBA_AC` protocol?

casasale, 2020-06-27:
Sometimes, e.g. if you repeat the commands, then it appears again as Unknown, so it looks like a bogus detection.

I am uploading the latest spreadsheet.

casasale, 2020-06-27:
@crankyoldgit Here is the latest spreadsheet:
[https://docs.google.com/spreadsheets/d/1yidE2fvaO9kpCHfKafIdH31q4uaskYR1OwwrkyOxbp0/edit?usp=drivesdk](https://docs.google.com/spreadsheets/d/1yidE2fvaO9kpCHfKafIdH31q4uaskYR1OwwrkyOxbp0/edit?usp=drivesdk)

crankyoldgit, 2020-06-28:
> Yes, IRrecvDumpV2 is sometimes detecting it as Toshiba

Can you supply the full output of the dump when it does? And if it decodes it into detailed settings, are those settings close/correct?

It's unlikely that the library is detecting it as Toshiba incorrectly, as it does have a check as part of the decode.
e.g.
https://github.com/crankyoldgit/IRremoteESP8266/blob/0fb37a3afa329487a03af8d333f5f869201009c0/src/ir_Toshiba.cpp#L336-L340
&
https://github.com/crankyoldgit/IRremoteESP8266/blob/0fb37a3afa329487a03af8d333f5f869201009c0/src/ir_Toshiba.cpp#L97-L110

casasale, 2020-06-28:
In the following dump I repeated the 72 bits ON/OFF (22C, Mode =Auto, Fan = AUTO), and as you can see it got detected only once as TOSHIBA_AC and it had the Mode wrong as Heat:  Mesg Desc.: Power: Off, Mode: 3 (**Heat**), Temp: 22C, Fan: 0 (Auto:

DUMP:

Timestamp : 000031.438
                      Library   : v2.7.8

Protocol  : UNKNOWN
                   Code      : 0xE2D990EB (148 Bits)
                                                    uint16_t rawData[295] = {4424, 4316,  590, 1572,  590, 1570,  590, 1572,  588, 1572,  588, 490,  590, 490,  590, 1570,  590, 496,  584, 490,  592, 498,  580, 492,  590, 494,  584, 1572,  588, 1572,  588, 490,  590, 1576,  584, 496,  584, 496,  584, 490,  592, 544,  534, 488,  590, 490,  590, 1570,  592, 1570,  590, 1572,  588, 1570,  590, 1570,  590, 1570,  590, 1570,  592, 1570,  588, 492,  588, 496,  584, 488,  592, 498,  580, 496,  584, 494,  584, 490,  590, 490,  590, 490,  590, 1570,  590, 490,  590, 1570,  590, 490,  590, 1572,  590, 490,  588, 490,  590, 498,  580, 492,  588, 490,  590, 490,  590, 492,  590, 488,  590, 490,  588, 490,  588, 490,  590, 490,  590, 494,  584, 490,  590, 488,  590, 488,  590, 490,  592, 488,  590, 490,  590, 490,  588, 490,  590, 1572,  588, 492,  588, 1570,  590, 490,  588, 490,  590, 490,  590, 1570,  590, 7418,  4424, 4316,  590, 1572,  590, 1570,  590, 1570,  590, 1570,  590, 488,  592, 494,  584, 1572,  592, 488,  590, 490,  588, 490,  590, 490,  588, 490,  590, 1572,  590, 1572,  590, 488,  592, 1570,  588, 490,  592, 488,  592, 488,  590, 490,  590, 490,  590, 490,  590, 1570,  592, 1568,  592, 1570,  592, 1570,  590, 1570,  592, 1570,  590, 1570,  592, 1568,  590, 490,  590, 490,  590, 488,  592, 488,  592, 488,  592, 488,  590, 490,  590, 488,  592, 488,  590, 1570,  590, 492,  586, 1576,  584, 490,  588, 1572,  590, 488,  592, 492,  586, 490,  592, 488,  590, 490,  590, 488,  592, 488,  590, 490,  588, 490,  590, 490,  590, 490,  590, 490,  590, 490,  590, 488,  592, 490,  590, 496,  584, 490,  590, 488,  590, 490,  590, 490,  590, 490,  590, 1568,  590, 490,  588, 1572,  588, 492,  590, 490,  590, 490,  590, 1570,  590};  // UNKNOWN E2D990EB

Timestamp : 000034.813
                      Library   : v2.7.8

Protocol  : UNKNOWN
                   Code      : 0x1B6696D7 (148 Bits)
                                                    uint16_t rawData[295] = {4428, 4312,  590, 1570,  590, 1574,  588, 1568,  590, 1574,  588, 488,  592, 486,  592, 1568,  590, 490,  590, 490,  590, 488,  594, 488,  588, 490,  590, 1570,  592, 1568,  592, 492,  588, 1570,  590, 494,  586, 490,  592, 488,  592, 488,  592, 490,  592, 488,  590, 1570,  590, 1570,  592, 1568,  590, 1570,  590, 1570,  590, 1572,  588, 1570,  592, 1568,  592, 490,  590, 494,  586, 490,  590, 490,  592, 488,  590, 490,  588, 492,  588, 492,  592, 488,  590, 1570,  590, 488,  592, 1568,  592, 490,  590, 1570,  590, 544,  538, 488,  590, 490,  590, 490,  590, 490,  590, 488,  592, 488,  592, 488,  592, 488,  590, 1570,  592, 1568,  592, 1568,  592, 490,  588, 490,  592, 488,  590, 496,  584, 488,  592, 490,  592, 488,  588, 492,  590, 488,  590, 1570,  590, 490,  590, 1570,  592, 488,  588, 1572,  592, 1570,  590, 490,  590, 7424,  4422, 4312,  592, 1568,  592, 1570,  590, 1572,  588, 1570,  590, 490,  590, 490,  592, 1568,  590, 490,  590, 488,  592, 488,  588, 496,  586, 488,  592, 1570,  590, 1570,  590, 488,  592, 1570,  590, 490,  588, 490,  590, 490,  590, 488,  592, 488,  590, 490,  588, 1570,  590, 1570,  590, 1570,  590, 1572,  592, 1568,  590, 1574,  586, 1570,  592, 1570,  590, 494,  586, 488,  592, 488,  592, 492,  586, 490,  590, 488,  590, 490,  590, 494,  588, 494,  586, 1570,  590, 494,  586, 1574,  586, 488,  592, 1574,  584, 488,  592, 488,  592, 492,  588, 488,  590, 490,  592, 490,  590, 488,  592, 488,  590, 494,  586, 1570,  592, 1574,  588, 1568,  590, 490,  592, 488,  592, 488,  592, 492,  586, 492,  590, 494,  584, 490,  590, 488,  592, 488,  592, 1568,  592, 488,  592, 1568,  592, 488,  588, 1572,  592, 1568,  592, 492,  588};  // UNKNOWN 1B6696D7

Timestamp : 000039.946
                      Library   : v2.7.8

Protocol  : UNKNOWN
                   Code      : 0xE2D990EB (148 Bits)
                                                    uint16_t rawData[295] = {4414, 4322,  582, 1578,  582, 1580,  554, 1604,  584, 1576,  582, 498,  580, 500,  554, 1610,  576, 502,  578, 504,  548, 526,  578, 502,  582, 498,  554, 1618,  542, 1608,  562, 522,  552, 1606,  580, 498,  582, 498,  554, 528,  552, 528,  582, 498,  580, 498,  582, 1578,  568, 1594,  580, 1580,  582, 1578,  584, 1578,  554, 1606,  554, 1606,  582, 1578,  584, 498,  584, 496,  580, 502,  578, 500,  554, 524,  582, 496,  584, 496,  584, 496,  580, 500,  580, 1580,  554, 526,  554, 1606,  582, 498,  554, 1606,  582, 504,  548, 526,  582, 498,  582, 496,  584, 498,  582, 496,  584, 496,  586, 494,  584, 496,  584, 498,  582, 494,  586, 496,  584, 498,  584, 494,  584, 496,  584, 496,  584, 502,  576, 508,  572, 552,  528, 496,  586, 494,  586, 1576,  582, 498,  584, 1576,  586, 494,  586, 500,  576, 498,  584, 1584,  576, 7424,  4422, 4318,  588, 1574,  588, 1572,  586, 1574,  586, 1576,  586, 498,  582, 496,  584, 1572,  588, 500,  580, 492,  588, 496,  582, 498,  584, 496,  584, 1570,  590, 1572,  588, 498,  584, 1572,  588, 494,  584, 498,  582, 498,  584, 492,  586, 492,  588, 498,  582, 1574,  586, 1572,  590, 1570,  590, 1572,  588, 1572,  588, 1570,  590, 1572,  588, 1578,  582, 5 [... cut ...]   584, 492,  586, 492,  588, 496,  584, 492,  588, 498,  580, 496,  586, 496,  582, 494,  586, 496,  582, 1576,  586, 498,  582, 1574,  586, 494,  586, 1574,  584, 496,  584, 496,  586, 494,  586, 494,  584, 496,  586, 496,  582, 494,  586, 492,  588, 496,  582, 1580,  580, 1574,  586, 1574,  584, 498,  582, 502,  578, 496,  584, 494,  586, 494,  586, 498,  582, 496,  584, 496,  584, 494,  586, 1572,  586, 496,  586, 1574,  586, 494,  586, 1574,  586, 1574,  588, 492,  586};  // UNKNOWN 1B6696D7

crankyoldgit, 2020-06-28:
Thanks for the date, that's helpful. I'll see what I can do.

casasale, 2020-06-28:
Looking for Toshiba and Carrier on Google I found the following, which would explain things :) 
https://www.toshiba-carrier.co.jp/global/about/index.htm

casasale, 2020-06-29:
If I adjust the values for matchGeneric(), then all the 72 bit version messages so far are recognized as Toshiba.

```
bool IRrecv::decodeToshibaAC(decode_results* results, uint16_t offset,
                             const uint16_t nbits, const bool strict) {
  // Compliance
  if (strict && nbits != kToshibaACBits)
    return false;  // Must be called with the correct nr. of bytes.

  // Match Header + Data + Footer
  if (!matchGeneric(results->rawbuf + offset, results->state,
                    results->rawlen - offset, nbits,
/*
                    kToshibaAcHdrMark, kToshibaAcHdrSpace,
                    kToshibaAcBitMark, kToshibaAcOneSpace,
                    kToshibaAcBitMark, kToshibaAcZeroSpace,
                    kToshibaAcBitMark, kToshibaAcMinGap, true,
*/
                    4406, 4337,
                    554, 1604,
                    554, 526,
                    554, 7441, true,
                    _tolerance, kMarkExcess)) return false;
...
```

casasale, 2020-06-29:
With the following small changes (does not include the different timing values for decodeToshibaAC/matchGeneric) the 72 bit messages get recognized correctly:

diff ir_Toshiba.h ir_Toshiba_.h
```
32c32
< const uint8_t kToshibaAcModeSize = 2;  // Nr. of bits
---
> const uint8_t kToshibaAcModeSize = 3;  // Nr. of bits
36a37,38
> const uint8_t kToshibaAcFan = 4;
> const uint8_t kToshibaAcOff = 7;

```

diff ir_Toshiba.cpp ir_Toshiba_.cpp
```
28a29,38
>
> // Toshiba Carrier A/C
> const uint16_t kToshibaCarrierAcHdrMark = 4406;
> const uint16_t kToshibaCarrierAcHdrSpace = 4337;
> const uint16_t kToshibaCarrierAcBitMark = 554;
> const uint16_t kToshibaCarrierAcOneSpace = 1604;
> const uint16_t kToshibaCarrierAcZeroSpace = 526;
> const uint16_t kToshibaCarrierAcMinGap = 7441;
>
>
144c154
<             kToshibaAcHeat);
---
>             kToshibaAcOff);
208a219,220
>     case kToshibaAcFan:
>     case kToshibaAcOff:
227c239,240
<     // No Fan mode.
---
>     case stdAc::opmode_t::kFan:   return kToshibaAcFan;
>     case stdAc::opmode_t::kOff:   return kToshibaAcOff;
253a267,268
>     case kToshibaAcFan:  return stdAc::opmode_t::kFan;
>     case kToshibaAcOff:  return stdAc::opmode_t::kOff;
305c320
<                             kToshibaAcHeat, kToshibaAcDry, kToshibaAcAuto);
---
>                             kToshibaAcHeat, kToshibaAcDry, kToshibaAcFan);
```

casasale, 2020-06-29:
OK, I checked the IRsendDemo and used sendToshibaAC() to send a command: Basically, both the original timing values, as well as the ones I averaged over a few dozen of messages mentioned above worked, but only with my values does IRrecvDumpV2 recognize all the commands as Toshiba.

casasale, 2020-06-29:
@crankyoldgit Tomorrow I will check your basic support code

crankyoldgit, 2020-06-29:
Thanks for the testing above. So, yeah, it looks like it's the same protocol. I'll look at adjusting timings & tolerances. I'll try to see if there is a way I can tell from the data if it is the Toshiba or the Carrier model, so we can get the mode/swing values correct etc

casasale, 2020-06-29:
By the way the last byte of all different length messages is the XOR result over all previous bytes. The swing binary values are mixed up in the spreadsheet (hex is correct), and if they are corrected, then the XOR value is also correct.

crankyoldgit, 2020-06-29:
Yep. That's the same as Toshiba.

crankyoldgit, 2020-06-29:
BTW, nice work on working out that the 3rd byte is the length. That's cool/helpful. I'm still trying to work out how I can tell the difference between this and Toshiba from the decoded hex data. :-/

crankyoldgit, 2020-06-30:
For the record, looking back at the history of Toshiba AC support, it looks like someone found a link to some existing code and I imported it based on me reading/re-coding it. So we don't have a lot of analysis data for it to verify against. :-/

crankyoldgit, 2020-06-30:
So, I think the original code I looked at had poor analysis. Looking at your data & analysis, it looks a better fit. I'll thus update Toshiba to match your results.

crankyoldgit, 2020-06-30:
Hey @casasale can you please download and try out the following branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Carrier72

I've tried to improve the Toshiba matching and protocol decoding. I think it will probably match the 72bit versions of your messages better.

Let me know how that goes, and I can then try to work on the shorter and longer versions of the messages.

casasale, 2020-06-30:
Hi @crankyoldgit, 

Great, thank you. 
I will download it and test it.

casasale, 2020-06-30:
Hi @crankyoldgit,

This is the current testing status of the **receiving** side:
- The command recognition rate as TOSHIBA_AC is great and very high >99%
- All different length commands get recognized with the following lengths: 72, 56, 56_no_repeat, and 80
- There was 1 reporting issue in 72 bit mode regarding the mode string when Mode Fan is enabled (potential code fixes in code diff below)
- 56 bit messages report wrong mode values (Auto, Cool, Heat), which seems to indicate that the last XOR byte is being used to get the mode value, and additionally it reports also wrong temperature, i.e. in case of 56 bit messages no Power/Mode/Temp/Fan states should be read or set, while 80 bit should be fine (needs to be fixed)

By the way, another behavior information is that in the special 80 bit High/Eco power command modes the 80 bit command length stays only in 80 bit when **no** Mode (Auto/Cool/Dry/Heat/Fan) is changed, i.e. if the Mode button is pressed, then the 80 bit Hi_Power/Eco mode is reset to the normal 72 bit mode, and vice versa changing temperature, swing on/off and air direction maintain the current bit length of 80 bit.

**Code diff:**
```
*** ir_Toshiba.cpp_org  2020-06-30 05:04:09.000000000 -0400
--- ir_Toshiba.cpp      2020-06-30 13:33:37.640796200 -0400
***************
*** 225,230 ****
--- 225,231 ----
      case stdAc::opmode_t::kHeat: return kToshibaAcHeat;
      case stdAc::opmode_t::kDry:  return kToshibaAcDry;
      case stdAc::opmode_t::kFan:  return kToshibaAcFan;
+     //case stdAc::opmode_t::kOff:  return kToshibaAcOff;   //Don't know if this is necessary
      default:                     return kToshibaAcAuto;
    }
  }
***************
*** 251,256 ****
--- 252,259 ----
      case kToshibaAcCool: return stdAc::opmode_t::kCool;
      case kToshibaAcHeat: return stdAc::opmode_t::kHeat;
      case kToshibaAcDry:  return stdAc::opmode_t::kDry;
+     case kToshibaAcFan:  return stdAc::opmode_t::kFan;   //Missing case
+     //case kToshibaAcOff:  return stdAc::opmode_t::kOff;    //Don't know if this is necessary
      default:             return stdAc::opmode_t::kAuto;
    }
  }
***************
*** 303,309 ****
    result += addBoolToString(getPower(), kPowerStr, false);
    if (getPower())
      result += addModeToString(getMode(), kToshibaAcAuto, kToshibaAcCool,
!                               kToshibaAcHeat, kToshibaAcDry, kToshibaAcAuto);
    result += addTempToString(getTemp());
    result += addFanToString(getFan(), kToshibaAcFanMax, kToshibaAcFanMin,
                             kToshibaAcFanAuto, kToshibaAcFanAuto,
--- 306,312 ----
    result += addBoolToString(getPower(), kPowerStr, false);
    if (getPower())
      result += addModeToString(getMode(), kToshibaAcAuto, kToshibaAcCool,
!                             kToshibaAcHeat, kToshibaAcDry, kToshibaAcFan);    //Fix Mode Fan reporting
    result += addTempToString(getTemp());
    result += addFanToString(getFan(), kToshibaAcFanMax, kToshibaAcFanMin,
                             kToshibaAcFanAuto, kToshibaAcFanAuto,
```

Here are the reports with the above code changes:
```
Protocol : TOSHIBA_AC
 Code : 0xF20D03FC0150070056 (72 Bits)
 Mesg Desc.: Power: Off,Temp: 22C,Fan: 0 (Auto)
uint16_t rawData_OFF[295] = {4424,4322,584,1572,588,1576,582,1576,586,1572,590,496,584,490,588,1574,586,492,588,494,586,496,584,492,588,494,
  586,1576,584,1574,586,490,588,1572,586,496,584,496,584,494,586,496,582,492,590,494,584,1572,588,1578,582,1572,588,1572,588,1574,588,1580,580,
  1574,586,1576,586,496,584,490,590,496,584,490,588,496,584,498,582,496,586,496,582,500,584,1572,586,534,544,1572,588,498,582,1576,584,492,588,490,
  588,492,588,494,586,496,584,498,584,490,588,496,582,498,584,1570,588,1576,586,1576,586,490,588,498,584,494,586,496,584,492,590,490,588,490,590,494,
  586,492,588,1576,584,496,584,1576,584,492,586,1574,588,1572,590,494,586,7420,4424,4316,590,1574,586,1572,588,1574,586,1574,588,492,588,496,584,1576,
  586,494,586,496,584,492,588,496,586,492,588,1572,588,1576,586,496,582,1576,586,496,584,496,582,496,582,492,590,496,584,498,584,1576,584,1572,588,1572,
  588,1572,588,1576,584,1576,586,1576,582,1574,590,496,582,496,584,496,584,490,588,496,584,496,582,494,588,494,586,494,586,1578,582,498,582,1574,586,496,
  584,1574,586,496,584,492,586,494,584,496,584,498,582,496,584,490,590,496,584,496,584,1576,586,1576,586,1572,588,492,590,496,584,498,584,548,530,496,584,
  494,586,496,582,492,588,496,584,1576,584,492,590,1572,588,494,586,1576,586,1570,590,490,588}; // TOSHIBA_AC
 uint8_t state[9] = {0xF2,0x0D,0x03,0xFC,0x01,0x50,0x07,0x00,0x56};

[... 53 lines cut ...]
  582,498,582,496,586,494,586,494,586,492,588,496,584,492,588,498,582,494,584,498,584,496,582,490,590,494,586,496,580,498,584,490,590,496,582,498,584,
  496,582,494,586,492,588,498,582,496,584,496,584,494,586,1572,590,496,584,1576,584,498,582,1572,588,1576,586,498,582,498,582,496,584,7420,4424,4322,
  584,1572,590,1570,590,1578,582,1572,590,496,582,492,588,1576,584,500,582,496,584,490,590,496,582,496,586,1572,588,1572,590,496,582,1572,590,496,584,
  500,580,492,588,494,586,496,584,1574,586,494,584,492,588,1576,586,1576,584,1572,588,1576,586,1572,590,490,588,1578,584,1572,588,494,586,500,582,496,
  584,494,584,1572,588,498,582,496,586,1570,588,496,584,1572,588,494,586,1574,584,498,582,492,588,492,588,494,586,498,582,496,584,492,590,494,584,496,
[... cut ...]
  492,588,494,584,496,584,496,582,1578,584,496,584,496,582,1574,588,1574,588,1576,584,1576,584,1576,584,496,582,1578,584,1576,584,494,586,496,584,494,
  586,496,582,1574,586,494,586,490,590,1576,584,494,586,1574,586,494,584,1576,586,494,586,492,588,494,586,490,588,498,584,496,582,498,584,496,582,498,
  584,496,584,492,590,494,586,496,584,494,584,494,586,490,588,492,586,498,584,490,588,496,584,496,584,494,586,492,588,492,588,496,586,496,584,1572,588,
  1572,588,494,586,1572,588,492,588,1578,582,1572,590,494,586,1570,588,492,590}; // TOSHIBA_AC
 uint8_t state[10] = {0xF2,0x0D,0x04,0xFB,0x09,0x50,0x00,0x00,0x03,0x5A};

```

NiKiZe, 2020-06-30:
If the code is changed to be correct for your model, it would be wrong for previous supported models (assuming they are/where correct), so more logic is probably needed(?)

casasale, 2020-06-30:
Previous old Toshiba models potentially seem to have had just 2 bits for Mode, so extending it to 3 bits is still "somewhat" backwards compatible, e.g. I imagine the AC will either ignore the Mode Fan command (0b100), or read the lower 2 bits which would default to Auto mode (0b00). Nevertheless I imagine that newer Toshiba models have the same 3 bit value.

crankyoldgit, 2020-07-01:
@casasale  Thanks for the feedback and data.

> * There was 1 reporting issue in 72 bit mode regarding the mode string when Mode Fan is enabled (potential code fixes in code diff below)

Yep. That's something I missed. I'll add it shortly.

> 56 bit messages report wrong mode values (Auto, Cool, Heat), which seems to indicate that the last XOR byte is being used to get the mode value, and additionally it reports also wrong temperature, i.e. in case of 56 bit messages no Power/Mode/Temp/Fan states should be read or set, while 80 bit should be fine (needs to be enhanced)

As I said, the non-72 bits messages are not expected to decode to settings yet. If they do, that's just luck at this point.

My main goal is to get the 72 bit stuff working then fix up the 56/80 stuff later.

@NiKiZe 
> If the code is changed to be correct for your model, it would be wrong for previous supported models (assuming they are/where correct), so more logic is probably needed(?)

I think it's a case that the code I borrowed from was slightly incorrect/incomplete. Seeing the data that @casasale has collected, I can see how "that" code had evolved that way. The data is/format is largely compatible with the previous implementation, and I think still decodes "real" messages correctly. Previously people had guessed there was a single power on/off bit as used in most A/C implementations, however, it looks like it really is just a special "mode" that turns the unit off, which just happened to flip that bit as well. This is highlighted/proved by the "fan" mode. Which changed that bit, but didn't affect power operation. It also better fits the weird "the mode has to be set to `heat` in addition to turning the power off" previous implementation.

In general, we relied on someone else's implementation, and I think they didn't (or couldn't) do as good a data collection/analysis as we can with this make/model etc.

crankyoldgit, 2020-07-01:
@casasale That branch has been updated and should have better "fan" mode support.

I'll try working on the different bit length stuff tonight.

casasale, 2020-07-01:
Thank you very much.

casasale, 2020-07-02:
@crankyoldgit FYI [https://stackoverflow.com/questions/628526/is-short-circuiting-logical-operators-mandated-and-evaluation-order](https://stackoverflow.com/questions/628526/is-short-circuiting-logical-operators-mandated-and-evaluation-order)
The above can be taken advantage of by setting the faster to evaluate, or more probable to fail(&&)/hit(||), conditional expression first within e.g. a logical operator evaluation.

crankyoldgit, 2020-07-02:
> @crankyoldgit FYI https://stackoverflow.com/questions/628526/is-short-circuiting-logical-operators-mandated-and-evaluation-order
> The above can be taken advantage of by setting the faster to [...]

Yep. I know this well. The most likely case is once it's made it through the data collection match (i.e. timings sequence, and length) successfully which is more than half and the most expensive battle, the "validation" step is almost always likely to pass and thus require all of the sub-checks to be run.
The checksum (Xor) is the most likely to fail (single bit error anywhere in the entire message) and it is checked first. It's not the fastest of the checks, but it's only a few instructions longer. i.e. it's _O(n)_, the "length" check is _O(1)_ but it's probably the least likely to be wrong/fail.

crankyoldgit, 2020-07-02:
@casasale Can you please download and test the latest version of that branch?

I think it should support everything _except_ for the `Swing Dir` function. I'll look at adding that tomorrow.
i.e. swing on/off, turbo on/off, and econo on/off.

Let me know what doesn't work etc etc etc.

casasale, 2020-07-03:
Thanks @crankyoldgit, I will test it and come back to you.

casasale, 2020-07-03:
Reception test:
- 72 bit: OK
- 80 bit: Hi Power: OK
- 80 bit: Eco Power: OK
- 56 bit: Swing On/Off: Works but the states need to be exchanged (my error in the spreadsheet), i.e. On is Off, and Off is On, e.g. On == {0xF2,0x0D,0x01,0xFE,0x21,0x01,0x20}, and Off == {0xF2,0x0D,0x01,0xFE,0x21,0x02,0x23}
-56 bit no repeat: Swing Dir: Reception recognized as 56 bit, but not explicitly reported as Swing Dir

Additionally for both 56 bit Swing messages, maybe it would make sense not to report the dummy constant Temp 17?

```
Protocol  : TOSHIBA_AC
  Code      : 0xF20D03FC0150600031 (72 Bits)
    Mesg Desc.: Temp: 22C,Power: On,Mode: 0 (Auto),Fan: 2 (UNKNOWN),Turbo: Off,Econo: Off
      uint8_t state[9] = {0xF2,0x0D,0x03,0xFC,0x01,0x50,0x60,0x00,0x31};

Protocol  : TOSHIBA_AC
  Code      : 0xF20D04FB095060000138 (80 Bits)
    Mesg Desc.: Temp: 22C,Power: On,Mode: 0 (Auto),Fan: 2 (UNKNOWN),Turbo: On,Econo: Off
      uint8_t state[10] = {0xF2,0x0D,0x04,0xFB,0x09,0x50,0x60,0x00,0x01,0x38};

Protocol  : TOSHIBA_AC
  Code      : 0xF20D03FC0150600031 (72 Bits)
    Mesg Desc.: Temp: 22C,Power: On,Mode: 0 (Auto),Fan: 2 (UNKNOWN),Turbo: Off,Econo: Off
      uint8_t state[9] = {0xF2,0x0D,0x03,0xFC,0x01,0x50,0x60,0x00,0x31};

Protocol  : TOSHIBA_AC
  Code      : 0xF20D04FB09506000033A (80 Bits)
    Mesg Desc.: Temp: 22C,Power: On,Mode: 0 (Auto),Fan: 2 (UNKNOWN),Turbo: Off,Econo: On
      uint8_t state[10] = {0xF2,0x0D,0x04,0xFB,0x09,0x50,0x60,0x00,0x03,0x3A};

Protocol  : TOSHIBA_AC
  Code      : 0xF20D03FC0150600031 (72 Bits)
    Mesg Desc.: Temp: 22C,Power: On,Mode: 0 (Auto),Fan: 2 (UNKNOWN),Turbo: Off,Econo: Off
      uint8_t state[9] = {0xF2,0x0D,0x03,0xFC,0x01,0x50,0x60,0x00,0x31};

Protocol  : TOSHIBA_AC
  Code      : 0xF20D01FE210120 (56 Bits)
    Mesg Desc.: Temp: 17C,Swing(V): Off
      uint8_t state[7] = {0xF2,0x0D,0x01,0xFE,0x21,0x01,0x20};

Protocol  : TOSHIBA_AC
  Code      : 0xF20D01FE210223 (56 Bits)
    Mesg Desc.: Temp: 17C,Swing(V): On
      uint8_t state[7] = {0xF2,0x0D,0x01,0xFE,0x21,0x02,0x23};

Protocol  : TOSHIBA_AC
  Code      : 0xF20D01FE210021 (56 Bits)
    Mesg Desc.: Temp: 17C,Swing(V): Off
      uint8_t state[7] = {0xF2,0x0D,0x01,0xFE,0x21,0x00,0x21};

```

casasale, 2020-07-03:
Transmission test:

- Default temp should be 22 instead of 17

- 72 bit: OK
```
  IRToshibaAC ac(kIrLed);
  ac.on();
  ac.setFan(2);
  ac.setMode(0);
  ac.setTemp(22);
  printState();
  ac.send();
```
```
Toshiba A/C remote is in the following state:
	Temp: 22C, Power: On, Mode: 0 (Auto), Fan: 2 (UNKNOWN), Turbo: Off, Econo: Off
		IR Code: 0xF20D03FC0150600031

```
- 80 bit Hi Power On/Off: OK, but in the ac.send() report only 72 bits  instead of 80 bits are reported, i.e. the checksum byte is missing in the IR Code
```
ac.setTurbo(true);
printState();
ac.send();
delay(2000); 
ac.setTurbo(false);
printState();
ac.send();
```

```
Toshiba A/C remote is in the following state:
	Temp: 22C, Power: On, Mode: 0 (Auto), Fan: 2 (UNKNOWN), Turbo: On, Econo: Off
		IR Code: 0xF20D04FB0950600001

Toshiba A/C remote is in the following state:
	Temp: 22C, Power: On, Mode: 0 (Auto), Fan: 2 (UNKNOWN), Turbo: Off, Econo: Off
		IR Code: 0xF20D03FC0150600031

```
- 80 bit Eco Power On/Off: Both commands accepted, but in the ac.send() report only 72 bits instead of 80 bits are reported, i.e. the checksum byte is missing in the IR Code
```
  ac.setEcono(true);
  printState();
  ac.send();
  delay(20000); 
  ac.setEcono(false);
  printState();
  ac.send();
```
```
Toshiba A/C remote is in the following state:
	Temp: 22C, Power: On, Mode: 0 (Auto), Fan: 2 (UNKNOWN), Turbo: Off, Econo: On
		IR Code: 0xF20D04FB0950600003

Toshiba A/C remote is in the following state:
	Temp: 22C, Power: On, Mode: 0 (Auto), Fan: 2 (UNKNOWN), Turbo: Off, Econo: Off
		IR Code: 0xF20D03FC0150600031
```

- 56 bits Swing On/Off: Commands accepted, but Swing does not work, and IR Code report looks like 72 bit report instead of 56 bits
```
  ac.setSwing(false);
  printState();
  ac.send();
  delay(5000);
  ac.setSwing(true);
  printState();
  ac.send();
``` 
```
Toshiba A/C remote is in the following state:
	Temp: 22C, Power: On, Mode: 0 (Auto), Fan: 2 (UNKNOWN), Turbo: Off, Econo: Off
		IR Code: 0xF20D03FC0150600031

Toshiba A/C remote is in the following state:
	Temp: 22C, Power: On, Mode: 0 (Auto), Fan: 2 (UNKNOWN), Turbo: Off, Econo: Off
		IR Code: 0xF20D03FC0150600031

```

casasale, 2020-07-03:
Transmission raw test:

- 72 bits: OK
```
IRsend irsend(kIrLed);  // Set the GPIO to be used to sending the message.
uint8_t state[9] = {0xF2,0x0D,0x03,0xFC,0x01,0x50,0x60,0x00,0x31};
irsend.sendToshibaAC(state, 9, true);
```

- 80 bits: OK
```
uint8_t state2[10] = {0xF2,0x0D,0x04,0xFB,0x09,0x50,0x60,0x00,0x01,0x38};
irsend.sendToshibaAC(state2, 10, true);
```

- 56 bits Swing On/Off: OK
```
 uint8_t state3[7] = {0xF2,0x0D,0x01,0xFE,0x21,0x01,0x20}; 
 irsend.sendToshibaAC(state3, 7, true);
 delay(5000);
 uint8_t state4[7] = {0xF2,0x0D,0x01,0xFE,0x21,0x02,0x23};
 irsend.sendToshibaAC(state4, 7, true);

``` 

- 56 Swing dir: OK
```
 uint8_t state5[7] = {0xF2,0x0D,0x01,0xFE,0x21,0x00,0x21};
 irsend.sendToshibaAC(state5, 7, true);
```

crankyoldgit, 2020-07-05:
> * 56 bit no repeat: Swing Dir: Reception recognized as 56 bit, but not explicitly reported as Swing Dir

As I said, it hasn't been added/supported yet. So that's expected.

> 56 bit: Swing On/Off: Works but the states need to be exchanged (my error in the spreadsheet), i.e. On is Off, and Off is On

(Being) fixed.

> 80 bit Hi Power On/Off: OK, but in the ac.send() report only 72 bits instead of 80 bits are reported, i.e. the checksum byte is missing in the IR Code

(Being) Fixed. It's a bug in the Example code, not the `IRToshibaAc` class.

> 56 bits Swing On/Off: Commands accepted, _**but Swing does not work**_, and IR Code report looks like 72 bit report instead of 56 bits

Size should be fixed. Can you give me the code it now produces and the equivalent code from actual remote for the same settings?

crankyoldgit, 2020-07-05:
@casasale I've created PR #1212 and updated the existing branch. Fingers crossed, it should now detect/support the swing direction/step/move/whatever command. I may have also fixed the swing problem too.

As usual, can you please download and test it out for me and let me know if there are any problems.

casasale, 2020-07-05:
Thanks @crankyoldgit .

Reception test: Everything OK with all bit lengths (72, 80, 56, and 56 no repeat)
```
Code      : 0xF20D03FC0150400011 (72 Bits)
	Mesg Desc.: Temp: 22C, Power: On, Mode: 0 (Auto), Fan: 1 (Low), Turbo: Off, Econo: Off

Code      : 0xF20D04FB095040000118 (80 Bits)
	Mesg Desc.: Temp: 22C, Power: On, Mode: 0 (Auto), Fan: 1 (Low), Turbo: On, Econo: Off

Code      : 0xF20D04FB09504000031A (80 Bits)
	Mesg Desc.: Temp: 22C, Power: On, Mode: 0 (Auto), Fan: 1 (Low), Turbo: Off, Econo: On

Code      : 0xF20D01FE210120 (56 Bits)
	Mesg Desc.: Temp: 17C, Swing(V): 1 (On)

Code      : 0xF20D01FE210223 (56 Bits)
	Mesg Desc.: Temp: 17C, Swing(V): 2 (Off)

Code      : 0xF20D01FE210021 (56 Bits)
	Mesg Desc.: Temp: 17C, Swing(V): 0 (Step)
```
Transmission test:
- 80 bits OK
```
ac.setTurbo(true);
printState();
ac.send();
```
```
Temp: 22C, Power: On, Mode: 0 (Auto), Fan: 1 (Low), Turbo: On, Econo: Off
	IR Code: 0xF20D04FB095040000118
```
- 56 bit does not work, e.g. commands get accepted, but there is no effect on swing, and all the serial prints are for 72 bits and equal, i.e. it looks like as if the 56 bit mode was not detected
```
ac.setSwing(kToshibaAcSwingOn);
printState();
ac.send();
delay(10000);
ac.setSwing(kToshibaAcSwingOff);
printState();
ac.send();
delay(5000);
ac.setSwing(kToshibaAcSwingStep);
printState();
ac.send();
```
```
Toshiba A/C remote is in the following state:
	Temp: 22C, Power: On, Mode: 0 (Auto), Fan: 1 (Low), Turbo: Off, Econo: Off
		IR Code: 0xF20D03FC0150400011
```

casasale, 2020-07-05:
I added some debug messages to see why 56 bit swing send does not work, and so far I confirmed, that the 56 bit command does not get accepted, and at the first look the IR Code seems to be wrong. I will update here once I have more details.

crankyoldgit, 2020-07-06:
@casasale Ah yes, that makes sense. I didn't notice/know it was a requirement for the 56 bit messages.
I'm using your suggested solution. Can you please re-download and confirm it's all working now? If so, we can merge PR #1212 which will close this issue.

The Toshiba A/C example code ... well, it probably won't show/display the short codes too well. I'm not to worried about that. It's something I can live with.

casasale, 2020-07-06:
@crankyoldgit I will test it today

casasale, 2020-07-06:
All sent commands are accepted, i.e. all the previously reported issues are resolved.

The only thing that is different compared to how the remote control works is that Turbo/Econo is not reset on Power On/Off or Mode changes, e.g. if Turbo or Econo is ON, then it gets set to OFF on AC restart or mode change.

crankyoldgit, 2020-07-07:
@casasale I just updated the PR/branch. I think it covers all of your recent points/suggests. Can you please confirm it works as you would expect?

casasale, 2020-07-07:
@crankyoldgit Yes, it works as expected. Thank you very much.

[2 more replies]

commit 765a9c2d (2020-07-06): Force the temp to be 17C for Toshiba "Short" msgs.

Ref: https://github.com/crankyoldgit/IRremoteESP8266/issues/1205#issuecomment-653922374

Fixes #1205
PR #1212 (2020-07-07): ToshibaAc: Rework to support Carrier models and add more settings.
* Turns out some Toshiba & Carrier A/Cs are the same.
* With more data we have adjusted protocol timings.
* Also the protocol has more than the 72bit/9 byte message size.
  - Support 80 bit/10 byte messages which control Econo and Turbo modes.
  - Support 56 bit/7 byte messages which control Swing settings.
* Rework `toString()` output.
* Mode & power settings re-done after more data collection & analysis.
* Lots of unit tests update and adjusted.
* Common A/C API support updated.

Fixes #1205
commit 9d4731f1 (2020-07-07): ToshibaAc: Rework to support Carrier models and add more settings. (#1212)

* Turns out some Toshiba & Carrier A/Cs are the same.
* With more data we have adjusted protocol timings.
* Also the protocol has more than the 72bit/9 byte message size.
  - Support 80 bit/10 byte messages which control Econo and Turbo modes.
  - Support 56 bit/7 byte messages which control Swing settings.
* Rework `toString()` output.
* Mode & power settings re-done after more data collection & analysis.
* Add Swing, Econo, & Turbo support.
* Lots of unit tests update and adjusted.
* Common A/C API support updated.

 [... cut]
closed 2020-07-07 by crankyoldgit