crankyoldgit/IRremoteESP8266 #1209: Home Assistant MQTT cant send global cache Panasonic TV

geniusalexey, 2020-07-01:
Hello, thank you for this great project. This is can be not an issue. I cant send throug Home Assistant MQTT this command:
topic: "ir_server/send"
payload: "31,37000,1,1,128,63,16,16,16,48,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,48,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,48,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,48,16,48,16,48,16,48,16,48,16,48,16,16,16,16,16,48,16,48,16,48,16,48,16,48,16,48,16,16,16,48,16,2712" and nothig happens, (from web gui this command works well),

but if i send 
topic: "ir_server/send"
payload: "31,38000,1,1,170,170,20,63,20,63,20,63,20,20,20,20,20,20,20,20,20,20,20,63,20,63,20,63,20,20,20,20,20,20,20,20,20,20,20,20,20,63,20,20,20,20,20,20,20,20,20,20,20,20,20,63,20,20,20,63,20,63,20,63,20,63,20,63,20,63,20,1798" this command works well.

This behaviour from version 2.7.4. What i am doing wrong?
i set #define MQTT_MAX_PACKET_SIZE 1024 and 2048 and this not helped.

Version/revision of the library used 2.7.6 - 2.7.8

crankyoldgit, 2020-07-01:
What does the info page say for "Last MQTT command seen:" for the failing mqtt request?

geniusalexey, 2020-07-01:
When i send panasonic TV command, nothing happens, no errors, no new commands on the system info page (look like esp no coomand receive over mqtt, but command is present on the broker), only the last successful command over mqtt (for example climate) on the system info page is displayed.

crankyoldgit, 2020-07-01:
Can you cut and paste the entire info page here? I'm guessing there's a mismatch in the mqtt topic name your using (typo) or it's not connecting, or something. Have you tried subscribing to the send and sent etc topics to see the traffic?

crankyoldgit, 2020-07-05:
Ping

crankyoldgit, 2020-07-15:
@geniusalexey Chasing a reply to:

> Can you cut and paste the entire info page here? I'm guessing there's a mismatch in the mqtt topic name your using (typo) or it's not connecting, or something. Have you tried subscribing to the send and sent etc topics to see the traffic?

crankyoldgit, 2020-07-23:
Last chance before I close this issue due to no response/inactivity etc.

geniusalexey, 2020-07-23:
Sorry for long response.
First picture successfully sent from web gui.
Second pic. fail from mqtt.
Third successful send another command.
[image]
[image]
[image]

crankyoldgit, 2020-07-23:
If you look at the `Last MQTT command seen:` line in the info page, you can see that the IRMQTTServer program is not seeing the MQTT message from the 2nd image.
As you can manually send other successful commands via MQTT to it (as shown in the 3rd image) then you know how to do it and it can see it sometimes.
You are using QoS of `0`, try using a QoS of 1 or mote in Home Assistant. Or try repeating it. There is something at the MQTT level (or lower level) that is stopping the IRMQTTServer from seeing the message. That could be a poor MQTT server or bad wifi/traffic, or something odd with the host that is running the MQTT server.

IRMQTTServer is connecting to your MQTT Server fine according to the info page.

See: https://www.hivemq.com/blog/mqtt-essentials-part-6-mqtt-quality-of-service-levels/

geniusalexey, 2020-07-24:
Setting QoS to level 0,1,2 not helped. Other commands like air conditioneer (toshiba), TV (5,0x40040100BCBD,48,8) works well. Problem only happens in global cache Panasonic. I need global cache because it has separate commands to turn on and off and  separate input command.

crankyoldgit, 2020-07-24:
I just tried reproducing what you're doing, with an older copy and the latest version of the library & example code.
It appears to work fine for me.
e.g.
Sending the message (My MQTT server is using `10.0.0.4` and the default MQTT port)
```shell
$ mosquitto_pub -h 10.0.0.4 -t ir_server/send -m '31,37000,1,1,128,63,16,16,16,48,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,48,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,48,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,48,16,48,16,48,16,48,16,48,16,48,16,16,16,16,16,48,16,48,16,48,16,48,16,48,16,48,16,16,16,48,16,2712'
```

Here is the output from `mosquitto_sub -h 10.0.0.4 -v -t \#` which dumps everything MQTT sees/sends etc.
```
ir_server/send 31,37000,1,1,128,63,16,16,16,48,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,48,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,48,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,48,16,48,16,48,16,48,16,48,16,48,16,16,16,16,16,48,16,48,16,48,16,48,16,48,16,48,16,16,16,48,16,2712
ir_server/sent 31,37000,1,1,128,63,16,16,16,48,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,48,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,48,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,48,16,48,16,48,16,48,16,48,16,48,16,16,16,16,16,48,16,48,16,48,16,48,16,48,16,48,16,16,16,48,16,2712
```
Note: It acknowledges what it has action-ed correctly to the `ir_server/sent` topic.

The info page on IRMQTTServer displays:
```
Last MQTT command seen: (topic) 'ir_server/send' (payload) '31,37000,1,1,128,63,16,16,16,48,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,48,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,48,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,48,16,48,16,48,16,48,16,48,16,48,16,16,16,16,16,48,16,48,16,48,16,48,16,48,16,48,16,16,16,48,16,2712' (8 Minutes 23 Seconds ago)
```

Full output:
```
General
Hostname: ir_server
IP address: 10.0.0.122
MAC address: REDACTED
Booted: 10 Minutes 39 Seconds ago
Version: v1.5.0
Built: Jul 24 2020 20:00:24
Period Offset: -5us
IR Lib Version: 2.7.8
ESP8266 Core Version: 2_7_1
Free Sketch Space: 2540k
Cpu Freq: 80MHz
IR Send GPIO(s): 4 (default)
Inverting GPIO output: Off
Total send requests: 1
Last message sent: Ok (9 Minutes 23 Seconds ago)
IR Recv GPIO: UNUSED
Total IR Received: 0
Last IR Received: None (Never)
Duplicate WiFi networks: Show
Min WiFi signal required: 8%
Serial debugging: Off

MQTT Information
Server: 10.0.0.4:1883 (Connected 10 Minutes 23 Seconds ago)
Disconnections: 0
Max Packet Size: 768 bytes
Client id: ir_server15f9d1
Command topic(s): ir_server/send, ir_server/send_0
Acknowledgements topic: ir_server/sent
IR Received topic: ir_server/received
Log topic: ir_server/log
LWT topic: ir_server/status
QoS: 1
Last MQTT command seen: (topic) 'ir_server/send' (payload) '31,37000,1,1,128,63,16,16,16,48,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,48,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,48,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,48,16,48,16,48,16,48,16,48,16,48,16,16,16,16,16,48,16,48,16,48,16,48,16,48,16,48,16,16,16,48,16,2712' (9 Minutes 23 Seconds ago)
Total published: 30
Total received: 11

Climate Information
IR Send GPIO: 4
Last update source: MQTT (via retain)
Total sent: 0
Last send: Never
State listen period: 5 Seconds
State broadcast period: 10 Minutes
Last state broadcast: 26 Seconds ago
Last discovery sent: Never
Discovery topic: homeassistant/climate/ir_server/config
Command topics: ir_server/ac/cmnd/(protocol|model|power|mode|temp|fanspeed|swingv|swingh|quiet|turbo|light|beep|econo|sleep|filter|clean|use_celsius|resend)
State topics: ir_server/ac/stat/(protocol|model|power|mode|temp|fanspeed|swingv|swingh|quiet|turbo|light|beep|econo|sleep|filter|clean|use_celsius|resend)
```

So, in a normal network setup with everything being reliable, it just works.
FYI, you should be able to drop `MQTT_MAX_PACKET_SIZE` to `768` like I have, and it will work.

The data you've provided strongly indicates that the problem is the MQTT connection between IRMQTTServer and your MQTT server. The fact it is getting most, but not those particular messages makes me think if you've got some odd character in it, a bad cut & paste, so some MQTT server/network problem.

I'd recommend you connect your ESP up to a serial terminal and turn on serial debugging:
https://github.com/crankyoldgit/IRremoteESP8266/blob/ad436f76ba8e102bb54c89c6ce15e6cde95b9054/examples/IRMQTTServer/IRMQTTServer.h#L243-L249
And post the TEXT output (cut & paste) of that, not a screen shot. It might provide some more clues as to what is going on.

On a side note, you can use the tool `gc_decode` in the `tools` directory to convert global cache into raw format and/or see if the library recognises it as something else, i.e. The Panasonic hex code.:
```
$ cd git/IRremoteESP8266/tools/
git/IRremoteESP8266/tools$ make gc_decode
<make output removed>
git/IRremoteESP8266/tools$ ./gc_decode 
Usage: ./gc_decode [-gc] [-rawdump] <global_code>
Usage: ./gc_decode -prontohex [-rawdump] [-repeats num] <prontohex_code>
Usage: ./gc_decode -raw [-rawdump] <raw_code>
git/IRremoteESP8266/tools$ ./gc_decode -gc 37000,1,1,128,63,16,16,16,48,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,48,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,48,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,48,16,48,16,48,16,48,16,48,16,48,16,16,16,16,16,48,16,48,16,48,16,48,16,48,16,48,16,16,16,48,16,2712
Code length 103
Code type      5 (PANASONIC)
Code bits      48
Code value     0x40040100fcfd
Code address   0x4004
Code command   0x100fcfd
```
i.e. `5,40040100fcfd` via MQTT to `ir_server/send`

FYI, screen shots are not really super helpful as I can't cut and paste from them to make sure I'm using the same data as you using/getting.

crankyoldgit, 2020-07-24:
I've even tested sending it via Home Assistant (0.112.4) Developer's tools page using the 'MQTT.publish' service.
e.g.
```
topic: "ir_server/send"
payload: "31,37000,1,1,128,63,16,16,16,48,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,48,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,48,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,16,48,16,48,16,48,16,48,16,48,16,48,16,16,16,16,16,48,16,48,16,48,16,48,16,48,16,48,16,16,16,48,16,2712"
```

It works as expected, as IRMQTTServer sees the message and does what is asked, and ackknowledges back on the `ir_server/sent` MQTT topic.

crankyoldgit, 2020-07-30:
Did you try any of the suggestions/values?

crankyoldgit, 2020-08-07:
Ping ... It's been a week.

crankyoldgit, 2020-08-18:
Last chase up before I mark this stale/closed.