crankyoldgit/IRremoteESP8266 #1211: Adding support for  Sanyo AC 72bit protocol

terrapinlogo, 2020-07-05:
I have two Sanyo AC units that I have decoded that are not currently supported.

SAP-K121AHA       Remote RCS-2HS4E   Testing was done with this remote mostly, quick check with the other remote and it appears to be identical protocol
SAP-K242AH         Remote RCS-2S4E

Manual [https://www.manualslib.com/manual/957988/Sanyo-Sap-K91aha.html#product-SAP%E2%80%93K121AHA](url)

Raw data captured and exported from analysIR

Flap 6/Top
uint16_t rawData[148] = {8492,4148,608,452,584,1512,600,468,584,1492,628,452,604,1488,628,1468,560,508,608,1508,612,444,612,444,612,444,580,1492,628,1488,628,1504,580,476,576,1512,588,476,608,444,612,1504,580,476,520,516,604,1492,624,448,608,452,604,448,580,460,516,540,580,476,608,448,608,452,604,448,580,476,528,512,608,448,580,476,608,1508,580,472,580,456,548,508,584,1512,628,1488,596,1492,544,532,608,452,604,452,580,476,608,1480,548,508,612,448,608,448,584,476,580,476,580,476,576,480,524,508,608,452,580,476,584,472,612,448,580,476,524,508,580,480,604,452,584,1512,624,448,576,1512,612,452,580,1512,628,1504,528,508,524,532,584}; 

Flap 5
uint16_t rawData[148] = {8516,4112,612,480,520,1568,588,472,612,1484,628,448,576,1496,624,1492,624,448,608,1484,544,512,608,448,608,452,604,1492,600,1488,560,1552,628,448,608,1508,580,472,580,460,604,1492,624,448,608,448,608,1468,560,512,608,452,604,452,584,472,608,448,584,468,584,456,524,532,608,452,604,452,608,448,584,468,584,476,524,1548,628,448,608,452,604,452,584,468,580,1516,580,1512,628,452,580,476,576,476,528,512,520,1572,628,448,608,452,576,476,580,480,524,508,580,476,584,476,580,480,576,480,576,476,528,508,540,516,584,472,608,452,580,476,584,472,580,480,520,512,580,1532,608,452,584,1524,580,1496,628,448,580,480,580}; 

Heating
uint16_t rawData[148] = {8520,4112,612,484,520,1548,624,452,608,1488,600,468,580,1500,624,1488,624,452,584,1492,560,512,604,452,584,472,608,1488,628,1480,544,1568,588,472,584,1512,600,472,524,512,608,1484,628,452,584,472,608,1484,544,512,608,448,608,452,604,452,584,472,580,472,580,460,524,532,580,476,608,448,608,452,608,444,584,472,528,1568,580,476,608,448,608,452,604,1468,560,516,604,1488,600,476,608,452,576,476,576,460,520,1576,600,476,580,480,604,448,580,476,528,508,604,452,584,472,584,472,612,452,576,472,580,460,540,516,576,480,584,472,584,476,604,452,580,1512,604,1492,596,480,604,452,576,1496,564,1548,600,472,584,480,576}; 

Flap 4
uint16_t rawData[148] = {8492,4128,628,452,580,1496,612,480,524,1548,628,448,608,1488,596,1496,540,532,612,1484,628,452,580,472,580,476,528,1548,628,1484,628,1500,580,460,540,1552,628,448,608,452,604,1468,616,460,604,452,580,1512,628,452,604,448,580,476,524,516,604,452,604,452,580,476,608,448,584,472,580,456,520,536,608,452,604,452,584,472,608,1484,544,512,580,476,580,1536,580,476,608,1464,564,512,580,476,580,480,604,452,580,1528,528,508,524,532,612,444,612,452,576,480,580,472,580,480,524,508,584,472,612,444,612,452,604,452,576,476,528,508,540,516,584,472,584,476,608,452,576,1532,528,508,576,1520,600,1532,584,468,584,476,524}; 

flap 3
uint16_t rawData[148] = {8516,4112,612,456,524,1572,604,476,608,1496,588,476,576,1512,584,1512,628,452,584,1488,560,516,604,452,608,448,608,1508,604,1468,536,1576,628,444,608,1472,616,476,524,512,580,1512,628,452,580,476,580,1492,540,532,612,444,612,448,608,452,580,476,576,480,524,508,584,472,608,448,612,448,608,448,584,468,584,480,520,512,584,1512,624,456,600,452,580,472,580,460,524,1568,628,452,604,452,584,468,584,476,524,1548,628,448,608,452,604,452,580,472,584,476,524,508,584,476,608,448,608,452,580,476,580,476,520,512,524,532,584,472,608,452,584,476,580,1508,544,1548,628,452,580,476,580,1496,612,1496,628,452,576,480,580}; 

Flap 2
uint16_t rawData[148] = {8492,4132,628,452,584,1488,564,512,580,1512,628,452,580,1496,612,1496,624,448,608,1492,624,448,584,472,576,460,608,1504,584,1512,628,1468,556,516,604,1488,628,448,608,452,580,1492,536,540,604,452,580,1512,628,448,580,476,528,508,604,452,584,472,612,444,612,452,576,476,576,460,540,516,576,480,584,472,608,452,608,448,580,1492,544,532,584,472,580,1536,584,1488,564,512,576,480,580,476,580,480,604,452,576,1496,560,512,584,472,584,476,608,452,604,452,576,476,528,508,604,456,580,472,584,472,612,452,604,448,576,484,524,508,580,476,584,472,584,480,604,1500,528,512,520,536,576,1520,600,1496,616,476,524,512,576}; 

Flap 1 / Bottom
uint16_t rawData[148] = {8516,4124,524,532,608,1492,628,444,608,1468,612,460,584,1512,628,1488,628,444,580,1492,628,448,608,452,580,476,608,1468,612,1496,624,1492,628,448,608,1464,564,512,604,452,608,1484,628,452,580,468,584,1492,628,448,608,452,584,472,584,472,576,480,524,512,580,476,608,452,604,452,580,476,580,472,580,480,524,508,612,448,604,1492,628,452,576,472,580,484,496,1592,580,476,608,448,608,448,584,472,532,508,604,1488,628,448,608,452,580,472,576,484,520,512,580,476,584,476,604,452,584,472,580,476,524,512,520,536,580,480,604,452,580,480,576,472,580,1496,600,476,604,452,584,472,584,1492,616,1516,576,480,580,476,580}; 

On
uint16_t rawData[148] = {8456,4192,624,448,584,1508,608,452,580,1512,628,452,604,1468,560,1552,600,472,584,1532,580,472,528,516,512,540,576,1516,628,1472,612,1508,612,452,580,1512,628,1468,612,1496,632,444,580,480,580,476,580,1496,564,508,576,480,576,480,580,476,584,472,584,468,584,480,520,512,580,480,576,480,580,476,584,472,584,472,528,508,524,1568,600,480,576,480,584,1492,560,512,580,1536,576,480,580,476,580,476,528,528,524,1568,580,476,584,476,580,476,580,472,528,512,520,536,576,480,580,480,576,480,576,476,532,528,520,512,576,480,584,476,580,476,580,480,576,472,528,1548,600,480,576,480,576,1520,592,1496,600,476,580,480,576}; 

Using the following settings for analysIR I can reliably decode the remotes. The timings seem to vary a lot with temperature and these have yet to fail.
Header Mark = 8500
Header Space = 4100
Mark = 625
Space0 = 500
Space1 = 1500
Delta = 200
Bits = 72
Carrier = 38000? guessing

I have worked out how each button corresponds to the bits here: 
[https://docs.google.com/spreadsheets/d/1dYfLsnYvpjV-SgO8pdinpfuBIpSzm8Q1R5SabrLeskw/edit?usp=sharing](url)

The only bits I am unsure of are the address bits, the remotes support addresses but changing them requires breaking a tab and the property is a rental so I would rather not permanently damage anything.

crankyoldgit, 2020-07-05:
Hey ya @terrapinlogo I've left some questions on the spreadsheet. It looks like you've done excellent work on the analysis.
I'm currently trying to finish off a Carrier/Toshiba protocol and I'll try to get to this shortly for you.

terrapinlogo, 2020-07-05:
No worries, I have responded to your questions and posted screenshots of the manual pages in sheet 2, they may explain better than I have.

crankyoldgit, 2020-07-10:
@terrapinlogo Hey ya, Can you please download and test the basic support for this protocol I just added in branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/SanyoAc

Please let me know how it goes capturing the real remote signals.

If it seems to work fine and match up with your collected data, I'll start on the "detailed" support for the protocol.

terrapinlogo, 2020-07-10:
I have tested it out and could only get it to detect if I turned the tolerance up to about 50%. I figured the timings were wrong so I took some more recordings with the remote both hot and cold. The data is in the 3rd sheet. I did some analysis and the timings vary by less than 20% so the default 25% should have been plenty.
Even with the tweaked timings the only way to get it to detect was with a tolerance of 40%

crankyoldgit, 2020-07-10:
Thanks for the feedback. I'll try to look into it shortly.

Can you please provide the full `uint16_t rawData[] = {...};` line for the captures it's failing on?

crankyoldgit, 2020-07-10:
@terrapinlogo How's the latest update fair? It's based on your spreadsheet averages.

crankyoldgit, 2020-07-10:
I've also added some of the detailed support now as well to that branch.
i.e.
* Checksum validation.
* Power
* Operation Mode
* Temp
* Fan Speed
* Sleep (Night Setback)

crankyoldgit, 2020-07-10:
Vertical Swing support added as well.

crankyoldgit, 2020-07-11:
Now with Sensor (location) & the Sensor's Temp....

crankyoldgit, 2020-07-11:
@terrapinlogo I'm having difficulty locating where the "on timer" time value is being stored/set in the message. Can you please do some tests/collect some data/analysis to find where it is stored/set/passed on etc? I can't see where it is being transmitted in your spreadsheet.

terrapinlogo, 2020-07-12:
As far as I could see it was never sent. I suspect the remote just sends an on signal when it needs to turn on

terrapinlogo, 2020-07-12:
I'll do a bit of testing today to find out

terrapinlogo, 2020-07-12:
> @terrapinlogo How's the latest update fair? It's based on your spreadsheet averages.

Just tested it and it seems to be receiving everything correctly. 

> @terrapinlogo I'm having difficulty locating where the "on timer" time value is being stored/set in the message. Can you please do some tests/collect some data/analysis to find where it is stored/set/passed on etc? I can't see where it is being transmitted in your spreadsheet.

I just left it recording for an hour with the on timer set. The remote sets the bits to enable the timer and enter standby mode (00 on the two power bits) when the button is pressed, then an hour later it sends an on command and clears the timer bits (line 101,102 in the spreadsheet). It also sets another bit that I have never seen before in the 4th byte. Some sort of On timer expiry flag I suspect.

crankyoldgit, 2020-07-12:
> > @terrapinlogo How's the latest update fair? It's based on your spreadsheet averages.
> 
> Just tested it and it seems to be receiving everything correctly.

Great.
Thanks for confirming.

> I just left it recording for an hour with the on timer set. The remote sets the bits to enable the timer and enter standby mode (00 on the two power bits) when the button is pressed, then an hour [...]

Okay. If that is the case, I might leave support for the "On Timer" out.

crankyoldgit, 2020-07-12:
Beep control added, per conversation in the Spreadsheet.

I think that's it for it now, unless you find an (new) issue. The PR will be merged (and close this issue) once the code review is done.

closed 2020-07-13 by crankyoldgit

crankyoldgit, 2020-07-30:
FYI, the aforementioned changes have been included in the latest release (v2.7.9) of the library.

commit 569e891f (2020-07-12): SanyoAc: Add support for controling the Beep setting.

Fixes #1211
PR #1218 (2020-07-13): Add Sanyo A/C (72 bit) protocol with detailed support.
* Add basic send and decode unit tests for new protocol.
* Add various protocol settings:
  - Power
  - Desired Temp
  - Operating Mode
  - Fan Speed
  - Vertical Swing
  - Sleep / Night setback
  - Off Timer
  - Temp Sensor Location & Value
  - Checksum
* Add to Common A/C API.
* Unit tests for Sanyo A/C

Fixes #1211
commit 9546b078 (2020-07-13): Add Sanyo A/C (72 bit) protocol with detailed support. (#1218)

* Add basic send and decode unit tests for new protocol.
* Add various protocol settings:
  - Power
  - Desired Temp
  - Operating Mode
  - Fan Speed
  - Vertical Swing
  - Sleep / Night setback
  - Off Timer
  - Temp Sensor Location & Value
  - Beep
  - Checksum
* Add to Common A/C API.
* Unit tests for Sanyo A/C

Fixes #1211