crankyoldgit/IRremoteESP8266 #1214: LG AC missing fan speed levels

SomeoneCares, 2020-07-08:
i used the MQTT example and it works fine with my LG AC, however it does not set the highest 2 levels of fan speed, when i send MQTT with fanspeed HIGH or MAX it actually falls back to AUTO, everything else is working fine
i still can decode the received IR codes, it can detect the LG2 protocol and all the attributes including fanspeed HIGH and MAX, it just can not send it 
any suggestion

NiKiZe, 2020-07-09:
We need to know the data that you expect it to send,
As described in the template.

Please use the DumpV2 or DumpV3 utils to collect that data.

SomeoneCares, 2020-07-09:
Thanks for your fast response below is what i receive on DumpV3 all are same settings except for fan speed

On Remote this is FAN 1

```
Protocol  : LG2
Code      : 0x880870F (28 Bits)
Mesg Desc.: Model: 2 (AKB75215403), Power: On, Mode: 0 (Cool), Temp: 22C, Fan: 0 (Low)
uint16_t rawData[59] = {3260, 9844,  568, 1514,  542, 520,  522, 512,  522, 512,  520, 1548,  522, 520,  520, 498,  542, 522,  520, 514,  520, 512,  522, 512,  522, 520,  520, 1548,  520, 514,  520, 498,  544, 506,  542, 530,  520, 1548,  520, 1556,  520, 1556,  520, 514,  520, 514,  520, 514,  520, 500,  542, 1542,  542, 1564,  520, 1548,  520, 1564,  520};  // LG2 880870F
uint32_t address = 0x88;
uint32_t command = 0x870;
uint64_t data = 0x880870F;
```

On Remote this is FAN 2

```
Protocol  : LG2
Code      : 0x8808798 (28 Bits)
Mesg Desc.: Model: 2 (AKB75215403), Power: On, Mode: 0 (Cool), Temp: 22C, Fan: 1 (UNKNOWN)
uint16_t rawData[59] = {3286, 9864,  544, 1514,  544, 520,  522, 512,  520, 512,  522, 1546,  520, 520,  520, 514,  520, 506,  542, 512,  522, 512,  522, 512,  522, 498,  542, 1546,  522, 514,  520, 506,  542, 514,  520, 512,  520, 1548,  522, 1564,  520, 1540,  544, 1540,  544, 512,  522, 498,  544, 1556,  544, 1548,  542, 512,  520, 514,  520, 514,  520};  // LG2 8808798
uint32_t address = 0x88;
uint32_t command = 0x879;
uint64_t data = 0x8808798;
```

On Remote this is FAN 3

```
Protocol  : LG2
Code      : 0x8808721 (28 Bits)
Mesg Desc.: Model: 2 (AKB75215403), Power: On, Mode: 0 (Cool), Temp: 22C, Fan: 2 (Medium)
uint16_t rawData[59] = {3290, 9868,  546, 1522,  542, 498,  542, 500,  542, 512,  522, 1546,  520, 512,  522, 512,  520, 514,  522, 512,  522, 512,  520, 514,  520, 520,  522, 1546,  522, 520,  520, 512,  520, 514,  520, 512,  522, 1554,  522, 1560,  516, 1556,  520, 498,  542, 512,  522, 1548,  520, 500,  542, 512,  520, 512,  522, 512,  522, 1546,  522};  // LG2 8808721
uint32_t address = 0x88;
uint32_t command = 0x872;
uint64_t data = 0x8808721;
```

On Remote this is FAN 4

```
Protocol  : LG2
Code      : 0x88087A9 (28 Bits)
Mesg Desc.: Model: 2 (AKB75215403), Power: On, Mode: 0 (Cool), Temp: 22C, Fan: 2 (Medium)
uint16_t rawData[59] = {3264, 9844,  546, 1514,  542, 506,  542, 520,  522, 512,  520, 1564,  520, 498,  544, 512,  522, 512,  520, 514,  520, 512,  522, 512,  520, 500,  542, 1542,  542, 512,  520, 514,  522, 512,  520, 514,  520, 1556,  520, 1534,  542, 1556,  520, 1540,  544, 512,  520, 1556,  520, 508,  542, 1556,  520, 498,  542, 512,  522, 1556,  520};  // LG2 88087A9
uint32_t address = 0x88;
uint32_t command = 0x87A;
uint64_t data = 0x88087A9;
```

On Remote this is FAN 5

```
Protocol  : LG2
Code      : 0x8808743 (28 Bits)
Mesg Desc.: Model: 2 (AKB75215403), Power: On, Mode: 0 (Cool), Temp: 22C, Fan: 4 (High)
uint16_t rawData[59] = {3306, 9856,  520, 1564,  542, 520,  520, 520,  522, 512,  520, 1548,  522, 512,  520, 514,  522, 512,  522, 512,  520, 512,  522, 514,  520, 500,  540, 1548,  520, 520,  522, 512,  520, 512,  520, 514,  522, 1554,  520, 1548,  520, 1556,  520, 522,  520, 1542,  542, 514,  520, 512,  520, 514,  520, 512,  522, 1554,  520, 1556,  520};  // LG2 8808743
uint32_t address = 0x88;
uint32_t command = 0x874;
uint64_t data = 0x8808743;

```

On Remote this is FAN AUTO

```
Protocol  : LG2
Code      : 0x8808754 (28 Bits)
Mesg Desc.: Model: 2 (AKB75215403), Power: On, Mode: 0 (Cool), Temp: 22C, Fan: 5 (Auto)
uint16_t rawData[59] = {3276, 9862,  544, 1546,  544, 514,  518, 514,  494, 540,  520, 1572,  494, 522,  572, 486,  494, 540,  494, 540,  520, 512,  520, 514,  494, 550,  518, 1556,  520, 506,  570, 488,  494, 540,  494, 540,  520, 1556,  520, 1554,  548, 1530,  520, 520,  520, 1548,  546, 470,  518, 1574,  522, 496,  546, 1538,  572, 468,  518, 540,  546};  // LG2 8808754
uint32_t address = 0x88;
uint32_t command = 0x875;
uint64_t data = 0x8808754;
```

crankyoldgit, 2020-07-09:
Hey @SomeoneCares , Thanks for reporting the issue. I think I've found it and have a fix for it.

Please download branch https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1214 / PR #1215 and confirm if it fixes it. My tests show it should.

closed 2020-07-09 by crankyoldgit

crankyoldgit, 2020-07-30:
FYI, the aforementioned changes have been included in the latest release (v2.7.9) of the library.

commit eacada81 (2020-07-09): [Bug] Fix mistake in `IRLGAc::convertFan()`.
* Incorrect use of `kHitachiAcFan` instead of `kLgAcFan`
* Based on user provided data, make old "low" value the lowest, and support a new intermediate "low" value.

* Minor clean up of LG unit test cases.

Note: This lowest value will display as "Quiet" in the fan speed readout in `toString()`.

Fixes #1214
PR #1215 (2020-07-09): [Bug] Fix mistake in `IRLgAc::convertFan()`.
* Incorrect use of `kHitachiAcFan` instead of `kLgAcFan`
* Based on user provided data, make old "low" value the lowest, and support a new intermediate "low" value.

* Minor clean up of LG unit test cases.

Note: The lowest value will display as "Quiet" in the fan speed readout in `toString()`.

Fixes #1214
commit 5a007a81 (2020-07-09): [Bug] Fix mistake in `IRLGAc::convertFan()`. (#1215)

* Incorrect use of `kHitachiAcFan` instead of `kLgAcFan`
* Based on user provided data, make old "low" value the lowest, and support a new intermediate "low" value.

* Minor clean up of LG unit test cases.

Note: This lowest value will display as "Quiet" in the fan speed readout in `toString()`.

Fixes #1214