crankyoldgit/IRremoteESP8266 #1216: Compilation error FS. H: no such file or directory

1043717432, 2020-07-10:
Hi, thank you for sharing the firmware. I'm happy to use your firmware all the time.

Seeing the update, I started compiling the firmware using arduino-1.813,

Load Arduino, libraries, irremotesp8266, examples, irmqttserver

When compiling, the following error message is prompted:
IRMQTTServer:336:10: fatal error: FS.h: No such file or directory
 #include <FS.h>
          ^~~~~~
compilation terminated.
exit status 1
FS.h: No such file or directory
Please help me. I haven't done anything well.

crankyoldgit, 2020-07-10:
The library and all examples get compiled/checked with every update of the library, so it is extremely unlikely it's a problem with the library itself. It's most likely something in your setup/environment that is causing a problem.
I'm going to ask a bunch of generic questions so I can get an idea of what your problem is. `FS.h` is included in the ESP8266 core library (i.e. Board Manager)

- What destination board type did you use?
- What ESP hardware are you using?
- What version of the ESP core library are you using?
- Have you tried some of the other examples in the library? Did they work as expected?

1043717432, 2020-07-11:
> The library and all examples get compiled/checked with every update of the library, so it is extremely unlikely it's a problem with the library itself. It's most likely something in your [...]

thank you!! The problem is solved!

closed 2020-07-11 by crankyoldgit