crankyoldgit/IRremoteESP8266 #1217: How to force SEND_DAIKIN128?

gpteacher, 2020-07-11:
Hi:

Love your library!

NOOBIE HERE!

I did a raw dump of my Daikin AC remote and it says I have a DAIKIN128 remote profile

But when I use the TurnOnDaikinAC sketch, it always uses the DAIKIN profile.

How can I have the sketch use the DAIKIN128 profile for sending data?

I've read your FAQs and have spent a lot of time in the <IRremoteESP8266.h>
<IRsend.h>
<ir_Daikin.h>  and associated .cpp files and I can't figure out how to get the sketch to use the DAIKIN128 format.

Your assistance is appreciated.

Thank you.

Gord

crankyoldgit, 2020-07-11:
You need to use the [IRDaikin128 class](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRDaikin128.html) for the `DAIKIN128` protocol.
That example code is set up for `IRDaikinESP`, which is used when the protocol is detected as just `DAIKIN`.

Alternatively, look at [CommonAcControl.ino](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/CommonAcControl/CommonAcControl.ino)

crankyoldgit, 2020-07-11:
i.e. Change `IRDaikinESP` to `IRDaikin128` and you'll probably be fine. There might be some other differences, but it will probably work with just that change.

gpteacher, 2020-07-11:
Thanks very kindly for your help and fast reply.

I figured it out about 10 minutes after I sent it. 

You've built a FANTASTIC library!

Much appreciated.

Gord

closed 2020-07-11 by gpteacher