crankyoldgit/IRremoteESP8266 #1220: Tasmota IRReceive to IRSend not the same >32 bits

t3h-n00b, 2020-07-20:
_(Please use this template for reporting issues. You can delete what ever is not relevant. Giving us this information will help us help you faster. Please also read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions) & [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide). Your problem may already have an answer there.)_

### Version/revision of the library used
_Typically located in the `library.json` & `src/IRremoteESP8266.h` files in the root directory of the library.
e.g. v2.0.0, or 'master' as at 1st of June, 2017. etc._
Tasmota version 8.3.1
### Expected behavior
_What steps did you do and what should it have done?_

1. IRReceive to receive message
2. IRSend to send the same message as received
3. IRReceive / Device to respond to sent command

### Actual behavior
_What steps did you do, and what did or didn't actually happen?_

Example below is the MUTE command from a Samsung Sound Bar. Also experience similar with Pioneer devices.

1. Capture IR command from Samsung/Pioneer remote
Example capture from original remote:
`{"IrReceived":{"Protocol":"SAMSUNG36","Bits":36,"Data":"0x0CF008E71","DataLSB":"0x0F300718E","Repeat":0,"RawData":[4544,4458,500,498,500,500,496,496,500,498,500,1464,530,1464,522,502,494,502,496,1470,524,1470,504,1464,522,1476,520,500,496,500,498,500,498,498,500,4460,530,492,494,504,496,500,496,500,496,1472,524,500,498,498,500,498,468,1474,522,1472,524,1470,526,498,500,498,500,1468,528,1466,530,1464,500,524,474,498,500,500,498,1466,530],"RawDataInfo":[77,77,0]}}`

2. Send IR command via IRSend:
`IRSend {"Protocol":"SAMSUNG36","Bits":36,"Data":"0x0CF008E71","DataLSB":"0x0F300718E","Repeat":0}`

3. Note the device does not interpret the message (although another Tasmota device **will**). Note the RAW data is different to what is received from the initial capture.
Tasmota Device receives the following:
`tele/tasmota_4EEF64/RESULT = {"IrReceived":{"Protocol":"SAMSUNG36","Bits":36,"Data":"0x0CF008E71","DataLSB":"0x0F300718E","Repeat":0,"RawData":[4456,4508,568,550,602,532,576,562,578,542,632,1616,578,1674,620,500,596,530,600,1654,630,1618,622,1626,568,1678,628,494,602,524,594,532,598,534,596,4480,596,530,600,528,602,526,604,524,596,1652,600,528,602,526,604,524,596,1656,626,1616,602,1646,604,522,596,532,598,1648,602,1652,630,1610,594,532,598,530,600,528,602,1644,604],"RawDataInfo":[77,77,0]}}`

4. Send IR Raw Command (that works from initial capture):
`Irsend 0,4544,4458,500,498,500,500,496,496,500,498,500,1464,530,1464,522,502,494,502,496,1470,524,1470,504,1464,522,1476,520,500,496,500,498,500,498,498,500,4460,530,492,494,504,496,500,496,500,496,1472,524,500,498,498,500,498,468,1474,522,1472,524,1470,526,498,500,498,500,1468,528,1466,530,1464,500,524,474,498,500,500,498,1466,530`

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
_Include some raw dumps of what the device saw._
N/A
### Steps to reproduce the behavior
_What can we do to (pref. reliably) repeat what is happening?_
t is worth noting that all 32 bit commands captured/resent appear to work fine. IRHvac commands also seem fine - just IRSend messsages >32 bits that dont.

Sample known bad commands:
```
Samsung_Soundbar_PWR {"Protocol":"SAMSUNG36","Bits":36,"Data":"0x0CF000EF1","DataLSB":"0x0F300708F","Repeat":0}
Samsung_Soundbar_DIN {"Protocol":"SAMSUNG36","Bits":36,"Data":"0x0CF00C936","DataLSB":"0x0F300936C","Repeat":0}
Samsung_Soundbar_Up {"Protocol":"SAMSUNG36","Bits":36,"Data":"0x0CF00EE11","DataLSB":"0x0F3007788","Repeat":0}
Samsung_Soundbar_Down {"Protocol":"SAMSUNG36","Bits":36,"Data":"0x0CF001EE1","DataLSB":"0x0F3007887","Repeat":0}
Samsung_Soundbar_Mute {"Protocol":"SAMSUNG36","Bits":36,"Data":"0x0CF008E71","DataLSB":"0x0F300718E","Repeat":0}
AV_Power_On {"Protocol":"PIONEER","Bits":64,"Data":"0xA55A58A7A55A58A7","DataLSB":"0xA55A1AE5A55A1AE5","Repeat":0}
AV_Power_Off {"Protocol":"PIONEER","Bits":64,"Data":"0xA55AD827A55AD827","DataLSB":"0xA55A1BE4A55A1BE4","Repeat":0}
AV_Sat_Cable {"Protocol":"PIONEER","Bits":64,"Data":"0xA55A08F7A55A08F7","DataLSB":"0xA55A10EFA55A10EF","Repeat":0}
AV_Vol_Up {"Protocol":"PIONEER","Bits":64,"Data":"0xA55A50AFA55A50AF","DataLSB":"0xA55A0AF5A55A0AF5","Repeat":0}
AV_Vol_Down {"Protocol":"PIONEER","Bits":64,"Data":"0xA55AD02FA55AD02F","DataLSB":"0xA55A0BF4A55A0BF4","Repeat":0}
AV_Bass_Up {"Protocol":"PIONEER","Bits":64,"Data":"0xA55AEA15A55A00FF","DataLSB":"0xA55A57A8A55A00FF","Repeat":0}
AV_Bass_Down {"Protocol":"PIONEER","Bits":64,"Data":"0xA55AEA15A55A807F","DataLSB":"0xA55A57A8A55A01FE","Repeat":0}

```
#### Example code used
_Include all relevant code snippets or links to the actual code files. Tip: [How to quote your code so it is still readable](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code)._

#### Circuit diagram and hardware used (if applicable)
_Link to an image of the circuit diagram used. Part number of the IR receiver module etc. ESP8266 or ESP32 board type._
IR LED: TSAL6400 (x2)
IR Receiver: TSOP38536
Board: Wemos D1 Mini
### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
_Yes/No._

### Has this library/code previously worked as expected for you?
_Yes/No. If "Yes", which version last worked for you?_
No
### Other useful information
_More information is always welcome. Be verbose._
Created an initial issue with Tasmota here: https://github.com/arendst/Tasmota/issues/8938

t3h-n00b, 2020-07-20:
For Pioneer, example commands:
Pioneer Power On (From Remote):
`{"IrReceived":{"Protocol":"PIONEER","Bits":64,"Data":"0xA55A58A7A55A58A7","DataLSB":"0xA55A1AE5A55A1AE5","Repeat":0,"RawData":[8504,4196,588,1536,562,464,578,1544,586,464,578,470,584,1540,580,470,588,1536,580,470,584,1514,582,470,582,1538,582,1542,588,462,580,1542,554,496,558,466,588,1536,582,466,586,1538,582,1542,588,462,580,468,584,466,588,1512,584,464,578,1546,586,464,580,470,586,1538,584,1540,560,1538,582,25190,8508,4190,586,1538,584,466,576,1546,552,496,560,466,578,1546,588,462,582,1542,580,468,586,1538,584,464,580,1518,580,1544,578,472,582,1542,582,468,586,462,582,1542,556,492,564,1536,586,1538,584,466,578,472,586,462,578,1546,586,462,582,1518,582 ...`

From Tasmota-sent command:
`{"IrReceived":{"Protocol":"PIONEER","Bits":64,"Data":"0xA55A58A7A55A58A7","DataLSB":"0xA55A1AE5A55A1AE5","Repeat":0,"RawData":[8520,4282,564,1590,556,520,558,1586,556,518,580,496,560,1584,560,514,564,1580,552,522,554,1590,554,520,556,1588,556,1588,558,518,560,1584,560,512,554,520,558,1586,560,510,556,1588,556,1588,556,514,564,512,554,522,554,1590,560,516,562,1582,558,512,554,522,554,1584,558,1580,564,1580,564,25210,8580,4242,558,1586,560,516,562,1582,564,512,554,520,556,1588,556,518,558,1586,558,518,560,1584,562,514,562,1582,562,1580,564,506,560,1582,562,510,556,520,558,1586,558,514,562,1576,558,1588,558,514,564,512,554,524,564,1576,556,518,558,1586,560 ...`

crankyoldgit, 2020-07-20:
For the Samsung36 stuff, we might need to adjust the timings on the protocol. We've only had one capture by one user to base the data on previously. See #621 

Can you try replacing these lines/values:
https://github.com/crankyoldgit/IRremoteESP8266/blob/b2c36ca56fb82fc39da1ca66b0c24c74a62e2401/src/ir_Samsung.cpp#L159-L167
with:
```
    sendGeneric(kSamsungHdrMark, kSamsungHdrSpace,
                506, 1468,
                506, 500,
                506, kSamsungHdrSpace,
                data >> (nbits - 16), 16, 38, true, 0, kDutyDefault);
    // Block #2 (The rest, typically 20 bits)
    sendGeneric(0, 0,  // No header
                506, 1468,
                506, 500,
```

And let me know how you go with that? Those changes/timings are based on the data you provided.

t3h-n00b, 2020-07-20:
Thanks @crankyoldgit 

Example Samsung command from above (Mute) works with those changes.
Also tested with Power, Vol Up and Down and all are working as expected

Sample IRReceive (working!):
`{"IrReceived":{"Protocol":"SAMSUNG36","Bits":36,"Data":"0x0CF008E71","DataLSB":"0x0F300718E","Repeat":0,"RawData":[4490,4456,578,440,570,442,580,434,576,438,572,1410,580,1402,574,440,570,442,580,1402,574,1406,570,1410,578,1404,574,440,572,442,580,434,578,440,570,4456,578,434,574,438,572,442,570,444,578,1404,574,440,570,442,580,434,576,1404,574,1408,580,1402,576,438,574,440,580,1404,572,1408,578,1404,576,438,570,444,574,438,572,1410,630],"RawDataInfo":[77,77,0]}}`

Let me know if you need any more info/data for the Pioneer issue, and thanks again!

crankyoldgit, 2020-07-20:
Several raw samples of each protocol would be good, so we have more data to calculate the values from.

t3h-n00b, 2020-07-20:
No worries,

**Pioneer Raw Commands**
Model is: Pioneer VSX-324
```
Pwr Off {"IrReceived":{"Protocol":"PIONEER","Bits":64,"Data":"0xA55AD827A55AD827","DataLSB":"0xA55A1BE4A55A1BE4","Repeat":0,"RawData":[8510,4188,578,1546,562,462,578,1572,560,488,554,496,560,1538,584,490,554,1546,576,498,556,1518,582,492,552,1572,550,1548,586,488,554,1544,556,518,536,1562,560,1538,584,492,552,1546,588,1536,552,498,556,494,550,498,556,492,552,472,582,1540,582,468,576,498,556,1540,582,1542,556,1544,576,25196,8512,4186,580,1544,578,498,558,1566,532,516,528,498,558,1540,580,494,556,1542,586,490,552,1544,582,466,586,1512,582,1542,578,496,558,1540,582,492,552,1546,586,1512,586,488,556,1542,580,1544,578,498,558,492,552,498,556,492,552,498,558,1514,586,490,554,496,560,1538,584,1540,582,1542,580,25168,8506,4192,586,1540,582,468,576,1548,586,488,556,494,560,1538,552,522,532,1540,580,494,562,1536,586,490,554,1544,578,1520,580,496,558,1564,562,488,554,1544,588,1538,580,492,558,1540,560,1540,584,492,552,496,558,492,552,498,558,492,552,1544,578,498,558,492,552,1520,578,1546,588,1536,586],"RawDataInfo":[203,203,0]}}
Pwr On {"IrReceived":{"Protocol":"PIONEER","Bits":64,"Data":"0xA55A58A7A55A58A7","DataLSB":"0xA55A1AE5A55A1AE5","Repeat":0,"RawData":[8506,4194,588,1536,560,490,552,1572,558,466,586,488,554,1544,586,490,552,1570,558,492,550,1522,586,490,552,1572,558,1540,578,498,554,1568,528,522,530,494,560,1538,580,494,558,1564,552,1546,582,494,558,490,552,498,556,1542,554,496,556,1566,552,498,554,494,558,1540,582,1542,550,1548,582,25192,8512,4188,582,1544,586,490,552,1570,536,488,554,470,584,1564,554,496,556,1542,576,498,556,1568,562,488,554,1544,552,1548,582,494,560,1538,578,472,582,468,586,1538,558,516,526,1548,582,1542,586,488,554,496,558,492,550,1548,582,468,584,1514,582,494,560,464,578,1546,580,1544,586,1538,580,25168,8514,4186,584,1542,588,462,580,1568,560,490,552,498,556,1542,554,520,532,1566,552,498,556,1568,562,488,554,1570,560,1512,582,468,584,1564,554,496,556,492,552,1572,558,468,586,1562,556,1518,580,472,582,492,560,488,554,1544,584,490,552,1572,558,466,588,488,554,1544,552,1546,584,1540,578],"RawDataInfo":[203,203,0]}}
Pwr Toggle {"IrReceived":{"Protocol":"PIONEER","Bits":64,"Data":"0xA55A38C7A55A38C7","DataLSB":"0xA55A1CE3A55A1CE3","Repeat":0,"RawData":[8510,4188,580,1544,556,470,582,1566,556,494,560,488,556,1542,580,470,584,1540,582,492,552,1522,578,496,558,1564,558,1542,580,470,586,1564,536,512,532,494,560,464,582,1542,580,1544,578,1544,578,498,558,492,562,488,556,1518,582,1542,582,492,552,498,556,494,550,1572,560,1538,562,1536,586,25188,8512,4186,584,1540,584,468,576,1572,528,522,532,492,552,1546,584,492,550,1548,582,494,560,1538,580,468,586,1514,586,1538,584,490,554,1546,586,488,554,470,584,490,556,1542,556,1542,580,1544,578,496,560,466,578,496,560,1538,584,1516,584,490,554,472,582,492,552,1546,586,1536,586,1538,586,25162,8514,4184,582,1542,580,496,562,1536,584,490,554,496,560,1538,562,514,530,1542,580,496,560,1538,584,466,578,1546,576,1524,578,496,558,1542,586,488,554,496,558,466,576,1546,576,1548,586,1512,578,498,558,492,552,496,558,1540,580,1542,578,498,556,494,550,498,556,1516,584,1540,584,1540,580],"RawDataInfo":[203,203,0]}}
FM1 {"IrReceived":{"Protocol":"PIONEER","Bits":64,"Data":"0x25DAB04F25DAB04F","DataLSB":"0xA45B0DF2A45B0DF2","Repeat":0,"RawData":[8506,4192,586,490,556,494,562,1536,552,498,556,492,552,1572,562,488,556,1542,580,1544,580,1520,580,496,558,1540,582,1540,582,494,560,1538,562,514,530,1542,580,494,560,1538,584,1540,584,492,552,498,556,492,552,498,556,492,532,1566,552,496,558,492,552,1570,552,1546,586,1538,562,1540,582,25188,8508,4192,588,488,556,492,560,1538,584,492,552,496,558,1514,584,490,554,1570,552,1546,578,1546,588,462,580,1544,556,1542,582,496,558,1564,558,490,554,1570,552,498,536,1564,558,1540,582,468,576,498,556,492,552,498,558,490,552,1544,578,498,558,492,528,1568,554,1546,588,1536,590,1534,584,25162,8506,4194,588,488,554,496,558,1540,578,472,580,496,556,1540,578,496,556,1542,552,1546,582,1542,588,488,554,1570,558,1514,582,492,560,1564,556,468,584,1564,552,472,580,1544,584,1514,582,492,558,492,550,498,554,496,558,492,560,1562,556,494,558,492,550,1572,556,1518,578,1546,582,1546,582],"RawDataInfo":[203,203,0]}}
AM {"IrReceived":{"Protocol":"PIONEER","Bits":64,"Data":"0x25DA708F25DA708F","DataLSB":"0xA45B0EF1A45B0EF1","Repeat":0,"RawData":[8506,4192,580,496,556,494,560,1538,558,490,552,498,554,1568,562,488,554,1544,586,1538,580,1520,576,498,554,1568,560,1538,580,494,558,1564,532,518,536,490,552,1570,562,1538,582,1542,588,462,580,496,558,490,552,498,556,1542,556,494,558,490,552,472,580,1542,576,1546,584,1540,556,1542,588,25186,8504,4194,580,496,558,492,560,1536,582,492,562,488,554,1518,578,498,556,1568,562,1536,582,1542,586,464,578,1520,576,1548,582,494,560,1538,580,494,558,490,552,1570,536,1538,580,1544,586,488,554,494,558,492,552,498,556,1540,578,498,556,492,560,490,530,1566,552,1546,584,1540,578,1546,582,25166,8512,4186,586,490,552,498,556,1566,552,498,556,494,560,1538,580,494,560,1514,582,1542,576,1548,582,494,560,1538,580,1518,578,472,582,1542,578,472,580,468,584,1564,556,1544,586,1514,584,490,552,498,554,494,558,490,552,1546,584,492,550,498,554,494,558,1566,554,1520,586,1538,580,1544,576],"RawDataInfo":[203,203,0]}}
Input DVD {"IrReceived":{"Protocol":"PIONEER","Bits":64,"Data":"0xA55AA15EA55AA15E","DataLSB":"0xA55A857AA55A857A","Repeat":0,"RawData":[8510,4188,584,1540,556,494,560,1538,580,494,560,490,552,1570,558,492,552,1572,558,492,560,1538,558,492,560,1562,556,1542,588,464,578 [... cut ...] RawDataInfo":[77,77,0]}}
Next {"IrReceived":{"Protocol":"SAMSUNG36","Bits":36,"Data":"0x0CF0031CE","DataLSB":"0x0F3008C73","Repeat":0,"RawData":[4512,4428,524,498,490,504,496,500,498,498,500,1464,524,1466,524,498,502,494,494,1472,518,1474,526,1466,524,1468,522,500,500,494,494,502,496,498,502,4438,520,502,498,498,502,494,494,502,498,496,492,504,496,1468,520,1470,518,504,496,498,502,494,494,1470,518,1474,526,1464,524,498,502,494,494,1470,518,1474,526,1464,524,498,502],"RawDataInfo":[77,77,0]}}
```

crankyoldgit, 2020-07-20:
@t3h-n00b Can you please download/try out PR #1221 and let me know how it goes? It should hopefully fix your `SAMSUNG36` issue.

I'll work on/look at the Pioneer one after we've got the Samsung one confirmed/merged etc,

crankyoldgit, 2020-07-21:
FYI, the Samsung36 timings (averaged) are:
```c++
// Data from https://github.com/crankyoldgit/IRremoteESP8266/issues/1220
// Values calculated based on the average of ten messages.
const uint16_t kSamsung36HdrMark = 4515;  /// < uSeconds
const uint16_t kSamsung36HdrSpace = 4438;  /// < uSeconds
const uint16_t kSamsung36BitMark = 512;  /// < uSeconds
const uint16_t kSamsung36OneSpace = 1468;  /// < uSeconds
const uint16_t kSamsung36ZeroSpace = 490;  /// < uSeconds
```

crankyoldgit, 2020-07-21:
And the Pioneer averaged data is:
```c++
const uint16_t kPioneerTick = 534;
const uint16_t kPioneerHdrMark = 8506;
const uint16_t kPioneerHdrSpace = 4191;
const uint16_t kPioneerBitMark = 568;
const uint16_t kPioneerOneSpace = 1542;
const uint16_t kPioneerZeroSpace = 487;
const uint32_t kPioneerMinCommandLength = 84906;
const uint32_t kPioneerMinGap = 25181;
```

crankyoldgit, 2020-07-21:
@t3h-n00b Can you please supply the brand and model numbers etc for the remotes & the devices controlled by them?

t3h-n00b, 2020-07-21:
Thanks @crankyoldgit  - Just building now and updating the Pioneer values in ir_Pioneer.cpp as well. Will let you know how it goes ASAP

Samsung Remote Model: AH59-02692E
Samsung Soundbar Model: HW-J551

Pioneer Remote Number: AXD7690
Pioneer Model: VSX-324

crankyoldgit, 2020-07-21:
@t3h-n00b I've created PR #1222 which has (just) the Pioneer timings changed. Can you let me know how that goes as well?

crankyoldgit, 2020-07-21:
> Thanks @crankyoldgit - Just building now and updating the Pioneer values in ir_Pioneer.cpp as well. Will let you know how it goes ASAP
> 
> Samsung Remote Model: AH59-02692E
> Samsung Soundbar Model: HW-J551
> 
> Pioneer Remote Number: AXD7690
> Pioneer Model: VSX-324

Thanks for that info. I've added that info to the respective PRs.

t3h-n00b, 2020-07-21:
Alrighty, Samsung looks good - thanks.

Pioneer doesn't appear to work with Power On/Power Toggle; but does work for Volume change...

crankyoldgit, 2020-07-21:
> Pioneer doesn't appear to work with Power On/Power Toggle; but does work for Volume change...

Can you please capture the "Power On/Power Toggle" messages with `IRrecvDumpV2` or `V3` (in the examples dir), and post the full serial output here please. Tasmota's buffer isn't big enough to capture the entire message that is being sent. i.e. It stops at around 200-ish. I think your remote is sending a repeated message.

e.g. `irsend.sendPioneer(0xA55A38C7A55A38C7, 64, 1);  // Power Toggle with one repeat.`

t3h-n00b, 2020-07-21:
No worries - I'll try and give it a shot this evening and get back to you

t3h-n00b, 2020-07-21:
Ok, managed to get it working. Did each command twice, below.

**Power Off:**
```
Timestamp : 000027.490
Library   : v2.7.8

Protocol  : PIONEER
Code      : 0xA55AD827A55AD827 (64 Bits)
uint16_t rawData[203] = {8504, 4220,  556, 1544,  578, 472,  582, 1540,  580, 470,  584, 464,  578, 1544,  576, 474,  582, 1516,  582, 468,  586, 1536,  584, 466,  578, 1544,  588, 1536,  584, 466,  556, 1542,  578, 472,  582, 1540,  580, 1544,  578, 472,  582, 1540,  558, 1540,  580, 470,  584, 464,  580, 470,  586, 464,  580, 494,  562, 1536,  584, 492,  552, 472,  584, 1540,  560, 1540,  582, 1542,  578, 25196,  8510, 4188,  578, 1548,  584, 466,  556, 1542,  580, 470,  584, 466,  578, 1544,  588, 462,  580, 1542,  578, 496,  558, 1514,  584, 466,  578, 1546,  586, 1538,  584, 492,  552, 1546,  586, 464,  578, 1520,  580, 1544,  576, 474,  580, 1542,  580, 1544,  588, 464,  558, 490,  554, 470,  584, 466,  578, 470,  584, 1540,  582, 468,  586, 462,  580, 1544,  578, 1546,  586, 1512,  586, 25186,  8510, 4190,  586, 1540,  582, 466,  588, 1536,  584, 464,  578, 470,  562, 1536,  584, 466,  578, 1544,  588, 462,  580, 1544,  578, 472,  584, 1514,  584, 1540,  580, 468,  586, 1538,  584, 466,  578, 1546,  586, 1512,  586, 464,  580, 1544,  588, 1536,  586, 488,  554, 496,  560, 488,  556, 494,  560, 490,  554, 1520,  580, 494,  560, 490,  554, 1544,  588, 1536,  584, 1540,  582};  // PIONEER A55AD827A55AD827
uint32_t address = 0xA51B;
uint32_t command = 0xA51B;
uint64_t data = 0xA55AD827A55AD827;
```

```
Timestamp : 000034.628
Library   : v2.7.8

Protocol  : PIONEER
Code      : 0xA55AD827A55AD827 (64 Bits)
uint16_t rawData[203] = {8512, 4212,  554, 1546,  586, 462,  580, 1542,  578, 470,  584, 466,  578, 1544,  586, 462,  582, 1518,  582, 468,  586, 1538,  584, 466,  576, 1548,  584, 1538,  584, 466,  554, 1544,  576, 472,  582, 1542,  580, 1544,  578, 472,  582, 1542,  558, 1542,  580, 470,  584, 464,  578, 470,  586, 464,  578, 470,  584, 1538,  582, 466,  576, 474,  582, 1542,  558, 1542,  580, 1544,  588, 25186,  8508, 4190,  586, 1536,  584, 464,  556, 1542,  578, 472,  584, 466,  578, 1546,  586, 464,  580, 1542,  578, 472,  582, 1516,  582, 466,  576, 1548,  584, 1538,  582, 468,  576, 1546,  586, 464,  580, 1518,  580, 1544,  576, 474,  582, 1542,  580, 1544,  576, 472,  560, 490,  554, 470,  584, 466,  578, 472,  584, 1540,  580, 494,  562, 462,  582, 1542,  578, 1546,  586, 1512,  586, 25186,  8508, 4190,  586, 1538,  584, 466,  576, 1546,  586, 464,  580, 470,  552, 1546,  586, 464,  580, 1544,  576, 474,  582, 1542,  580, 470,  584, 1516,  584, 1540,  582, 468,  588, 1536,  584, 466,  578, 1546,  586, 1512,  586, 462,  580, 1544,  578, 1546,  586, 464,  580, 470,  586, 464,  578, 470,  586, 464,  580, 1520,  580, 470,  584, 464,  580, 1544,  576, 1548,  584, 1540,  582};  // PIONEER A55AD827A55AD827
uint32_t address = 0xA51B;
uint32_t command = 0xA51B;
uint64_t data = 0xA55AD827A55AD827;
```

**Power On**

```
Timestamp : 000028.843
Library   : v2.7.8

Protocol  : PIONEER
Code      : 0xA55A58A7A55A58A7 (64 Bits)
uint16_t rawData[203] = {8514, 4210,  554, 1546,  586, 488,  556, 1542,  578, 496,  560, 490,  552, 1546,  586, 488,  556, 1518,  580, 494,  560, 1538,  584, 492,  562, 1536,  584, 1540,  582, 494,  528, 1544,  588, 486,  556, 494,  560, 1536,  584, 490,  554, 1544,  588, 1512,  588, 488,  556, 494,  562, 488,  556, 1542,  578, 498,  558, 1540,  580, 494,  560, 488,  554, 1544,  554, 1544,  588, 1536,  586, 25188,  8506, 4192,  584, 1542,  580, 494,  538, 1536,  586, 488,  554, 494,  560, 1538,  582, 492,  564, 1536,  584, 490,  552, 1520,  578, 496,  558, 1540,  580, 1544,  578, 496,  558, 1540,  582, 494,  562, 488,  534, 1540,  580, 494,  560, 1538,  584, 1540,  580, 494,  560, 488,  556, 494,  560, 1512,  586, 490,  554, 1544,  588, 486,  556, 492,  562, 1536,  584, 1540,  582, 1518,  582, 25192,  8502, 4196,  582, 1544,  588, 486,  558, 1540,  580, 470,  584, 490,  530, 1542,  578, 470,  584, 1540,  582, 468,  586, 1536,  584, 466,  578, 1520,  578, 1546,  586, 464,  580, 1542,  578, 472,  584, 490,  552, 1548,  586, 464,  580, 1520,  580, 1544,  588, 462,  582, 468,  586, 488,  556, 1542,  578, 470,  584, 1514,  584, 466,  578, 472,  584, 1540,  580, 1542,  578, 1546,  586};  // PIONEER A55A58A7A55A58A7
uint32_t address = 0xA51A;
uint32_t command = 0xA51A;
uint64_t data = 0xA55A58A7A55A58A7;

Timestamp : 000036.222
Library   : v2.7.8

Protocol  : PIONEER
Code      : 0xA55A58A7A55A58A7 (64 Bits)
uint16_t rawData[203] = {8512, 4212,  554, 1546,  586, 462,  580, 1544,  578, 472,  582, 466,  588, 1536,  584, 464,  580, 1520,  578, 470,  584, 1540,  582, 468,  586, 1536,  586, 1538,  582, 468,  554, 1546,  586, 464,  580, 468,  586, 1538,  584, 466,  578, 1546,  586, 1512,  586, 464,  580, 468,  586, 464,  580, 1542,  578, 472,  582, 1540,  580, 470,  586, 464,  580, 1544,  554, 1544,  578, 1546,  586, 25188,  8508, 4192,  584, 1540,  580, 470,  552, 1546,  586, 464,  578, 470,  584, 1538,  582, 468,  576, 1546,  586, 464,  578, 1518,  580, 496,  558, 1540,  582, 1542,  578, 472,  584, 1540,  582, 468,  586, 488,  534, 1540,  582, 468,  586, 1536,  584, 1540,  582, 470,  586, 462,  580, 468,  586, 1514,  586, 462,  582, 1542,  580, 470,  584, 492,  552, 1546,  586, 1538,  584, 1514,  584, 25190,  8506, 4192,  584, 1542,  580, 470,  584, 1538,  582, 468,  576, 474,  560, 1538,  582, 468,  586, 1536,  584, 464,  578, 1544,  588, 462,  582, 1518,  582, 1542,  580, 472,  582, 1540,  580, 468,  58 [... cut ...] 4,  582, 1542,  578, 472,  582, 1540,  582, 468,  586, 464,  558, 1540,  582, 468,  586, 1538,  584, 466,  578, 1546,  586, 464,  580, 1518,  580, 1544,  578, 472,  582, 1542,  580, 470,  584, 464,  578, 470,  584, 1538,  582, 1516,  582, 1542,  580, 470,  584, 466,  578, 470,  584, 1538,  584, 1540,  558, 468,  586, 462,  582, 468,  586, 1536,  584, 1540,  582, 1544,  578};  // PIONEER A55A38C7A55A38C7
uint32_t address = 0xA51C;
uint32_t command = 0xA51C;
uint64_t data = 0xA55A38C7A55A38C7;
```

crankyoldgit, 2020-07-21:
Interesting. Can you please capture a volume up or down as well.

t3h-n00b, 2020-07-21:
**Volume Up**

```
Timestamp : 000011.875
Library   : v2.7.8

Protocol  : PIONEER
Code      : 0xA55A50AFA55A50AF (64 Bits)
uint16_t rawData[203] = {8510, 4190,  586, 1538,  582, 468,  554, 1542,  578, 472,  582, 468,  586, 1536,  586, 464,  580, 1544,  588, 462,  582, 1516,  584, 466,  586, 1536,  586, 1538,  582, 468,  588, 1536,  584, 466,  578, 496,  536, 1538,  584, 466,  578, 1546,  586, 488,  554, 470,  586, 464,  580, 470,  584, 1538,  584, 466,  588, 1510,  588, 462,  580, 1542,  580, 1544,  588, 1536,  584, 1514,  584, 25188,  8506, 4194,  584, 1542,  580, 470,  584, 1538,  582, 468,  586, 462,  560, 1540,  582, 468,  586, 1536,  584, 466,  578, 1546,  586, 464,  582, 1518,  580, 1542,  590, 460,  582, 1542,  580, 470,  586, 464,  580, 1544,  588, 462,  582, 1516,  582, 468,  588, 462,  582, 468,  588, 462,  582, 1542,  580, 472,  582, 1540,  582, 468,  586, 1512,  588, 1536,  584, 1540,  582, 1542,  578, 25170,  8514, 4186,  580, 1546,  586, 462,  582, 1542,  578, 470,  584, 464,  578, 1544,  586, 464,  580, 1518,  582, 468,  586, 1538,  582, 466,  588, 1536,  584, 1538,  562, 464,  578, 1544,  588, 462,  582, 468,  586, 1536,  584, 466,  578, 1546,  586, 462,  580, 468,  586, 464,  558, 466,  578, 1546,  586, 464,  580, 1544,  578, 472,  582, 1540,  580, 1518,  580, 1544,  588, 1536,  584};  // PIONEER A55A50AFA55A50AF
uint32_t address = 0xA50A;
uint32_t command = 0xA50A;
uint64_t data = 0xA55A50AFA55A50AF;

Timestamp : 000013.198
Library   : v2.7.8

Protocol  : PIONEER
Code      : 0xA55A50AFA55A50AF (64 Bits)
uint16_t rawData[203] = {8510, 4188,  588, 1538,  584, 466,  554, 1542,  578, 472,  582, 492,  550, 1546,  586, 464,  580, 1544,  578, 472,  582, 1516,  584, 492,  562, 1536,  584, 1540,  582, 466,  588, 1536,  586, 464,  578, 472,  562, 1536,  586, 464,  578, 1546,  586, 462,  582, 494,  560, 488,  554, 494,  560, 1536,  586, 464,  578, 1520,  578, 472,  582, 1540,  582, 1542,  578, 1546,  586, 1512,  586, 25186,  8508, 4192,  586, 1538,  582, 468,  586, 1536,  586, 464,  578, 472,  560, 1536,  584, 466,  578, 1544,  588, 462,  580, 1544,  578, 498,  556, 1516,  584, 1540,  580, 470,  586, 1538,  584, 466,  576, 472,  584, 1540,  580, 470,  586, 1512,  586, 464,  580, 470,  584, 464,  580, 470,  586, 1538,  584, 466,  588, 1536,  586, 466,  578, 1520,  580, 1544,  588, 1536,  584, 1540,  582, 25166,  8506, 4192,  584, 1540,  582, 468,  586, 1538,  584, 468,  576, 472,  582, 1542,  580, 470,  586, 1512,  586, 464,  578, 1544,  578, 472,  582, 1540,  580, 1544,  556, 470,  584, 1538,  582, 468,  586, 488,  556, 1542,  580, 470,  584, 1538,  582, 468,  586, 488,  556, 494,  528, 470,  584, 1540,  580, 468,  586, 1536,  584, 466,  578, 1546,  586, 1512,  586, 1538,  584, 1540,  580};  // PIONEER A55A50AFA55A50AF
uint32_t address = 0xA50A;
uint32_t command = 0xA50A;
uint64_t data = 0xA55A50AFA55A50AF;
```

**Volume Down**
```

Timestamp : 000014.678
Library   : v2.7.8

Protocol  : PIONEER
Code      : 0xA55AD02FA55AD02F (64 Bits)
uint16_t rawData[203] = {8506, 4192,  584, 1542,  580, 470,  562, 1536,  584, 466,  578, 470,  584, 1540,  582, 468,  586, 1536,  584, 466,  578, 1520,  578, 472,  582, 1540,  580, 1544,  578, 472,  582, 1540,  580, 470,  584, 1514,  586, 1538,  582, 468,  588, 1536,  584, 466,  578, 470,  584, 490,  552, 472,  584, 466,  578, 472,  584, 1514,  584, 464,  578, 1544,  588, 1536,  584, 1540,  580, 1518,  582, 25190,  8516, 4184,  582, 1544,  588, 462,  582, 1542,  580, 470,  584, 464,  556, 1542,  580, 470,  584, 1540,  582, 468,  586, 1536,  584, 466,  578, 1520,  578, 1546,  586, 464,  580, 1542,  578, 472,  582, 1540,  580, 1518,  580, 470,  586, 1538,  584, 466,  578, 472,  582, 466,  576, 472,  582, 466,  578, 472,  582, 1540,  580, 470,  584, 1514,  586, 1538,  582, 1540,  580, 1544,  588, 25160,  8512, 4186,  578, 1546,  586, 464,  580, 1544,  588, 462,  582, 468,  586, 1536,  584, 466,  578, 1520,  578, 472,  584, 1540,  582, 468,  586, 1538,  584, 1540,  560, 466,  578, 1546,  586, 464,  580, 1544,  578, 1546,  586, 464,  580, 1518,  582, 494,  560, 464,  578, 470,  584, 464,  578, 470,  584, 464,  580, 1544,  578, 472,  582, 1540,  580, 1518,  580, 1544,  588, 1536,  586};  // PIONEER A55AD02FA55AD02F
uint32_t address = 0xA50B;
uint32_t command = 0xA50B;
uint64_t data = 0xA55AD02FA55AD02F;

Timestamp : 000015.943
Library   : v2.7.8

Protocol  : PIONEER
Code      : 0xA55AD02FA55AD02F (64 Bits)
uint16_t rawData[203] = {8516, 4184,  582, 1544,  588, 486,  534, 1538,  582, 492,  562, 462,  582, 1542,  580, 496,  558, 1540,  582, 494,  562, 1512,  588, 488,  556, 1542,  578, 1546,  586, 488,  554, 1544,  578, 470,  584, 1516,  584, 1540,  580, 494,  560, 1538,  582, 492,  564, 460,  582, 466,  588, 462,  582, 466,  588, 462,  582, 1516,  584, 466,  588, 1536,  586, 1538,  582, 1542,  580, 1520,  578, 25194,  8512, 4188,  588, 1538,  584, 466,  578, 1546,  586, 464,  580, 470,  552, 1546,  586, 464,  580, 1544,  578, 472,  582, 1542,  580, 470,  584, 1514,  584, 1540,  582, 468,  588, 1534,  586, 464,  578, 1544,  588, 1510,  588, 464,  580, 1544,  578, 498,  558, 466,  578, 472,  582, 466,  578, 472,  584, 466,  578, 1546,  586, 464,  580, 1518,  582, 1542,  578, 1546,  586, 1538,  582, 25166,  8506, 4192,  584, 1542,  580, 470,  584, 1540,  582, 468,  586, 462,  580, 1542,  580, 472,  584, 1514,  584, 492,  552, 1546,  586, 488,  554, 1542,  578, 1546,  554, 472,  582, 1540,  580, 470,  584, 1538,  582, 1542,  580, 470,  584, 1514,  584, 466,  578, 470,  5 [... cut ...] 2,  584, 1540,  582, 468,  554, 1546,  586, 462,  580, 468,  586, 1538,  584, 466,  578, 1546,  586, 464,  580, 1518,  580, 470,  584, 1540,  582, 1542,  578, 470,  584, 1540,  582, 468,  586, 462,  558, 1540,  580, 468,  586, 1538,  584, 466,  578, 470,  584, 1540,  580, 1518,  580, 1544,  576, 472,  582, 1542,  580, 470,  584, 1540,  582, 1518,  582, 468,  576, 474,  582};  // PIONEER A55A5AA5A55A53AC
uint32_t address = 0xA55A;
uint32_t command = 0xA5CA;
uint64_t data = 0xA55A5AA5A55A53AC;
```

crankyoldgit, 2020-07-22:
I've had a look at the data you provided.

The expected Pioneer protocol the library is expecting is a pair of _32_ bit data sections that add up to the entire _64_ bit message.
i.e. section `"A"` (_32bits_) + section `"B"` (_32bits_) gives us a message of `"AB"` (_64bits_)

The messages you've collected indicate `"ABA"` (_96bits_) or `"ABABAB"` (_192 bits_), where `A` & `B` are the same _"32bit"_ data section for each respective letter.

Either we've got the protocol horribly wrong (which I doubt) because the Pioneer documents indicate it's the way the library is using it. Or ... Your remote is behaving oddly, or there is something odd with the protocol that we don't know about.

In order for you to get an _"ABA"_ or longer sequence, you are probably going to have to use a `repeat`. (as per my earlier feedback). That is, instead of generating an "AB" message, produce an _"ABAB"_ message, which contains the "ABA" message at the start. Or a `repeat` of `2` to generate the longer ones of _"ABABAB"_
i.e. Try setting the `repeat` parameter to `1` when you send, instead of the default `0`.

There is currently no easy way I can immediately think of for how you can use `sendPioneer()` to generate _just_ an _"ABA"_ style message.

If the a `repeat` of `1` works, then we are all okay probably, but if it doesn't then we may need to add/tweak the protocol substantially.

t3h-n00b, 2020-07-22:
Looks good with a repeat of 1 on the Power On/Off/Toggles

crankyoldgit, 2020-07-22:
Phew

closed 2020-07-26 by crankyoldgit

crankyoldgit, 2020-07-30:
FYI, the aforementioned changes have been included in the latest release (v2.7.9) of the library.

commit 195113db (2020-07-21): Pioneer: Update timings based on user collected data.
* Adjust protocol timings based on the average seen across ten different captures.
* Update unit tests accordingly.
* Updated supported devices.

Fixes #1220
PR #1222 (2020-07-26): Pioneer: Update timings based on user collected data.
* Adjust protocol timings based on the average seen across ten different captures.
* Update unit tests accordingly.
* Updated supported devices.

Fixes #1220
commit 6a5918b7 (2020-07-26): Pioneer: Update timings based on user collected data. (#1222)

* Adjust protocol timings based on the average seen across ten different captures.
* Update unit tests accordingly.
* Updated supported devices.

Fixes #1220

* Add Unit test case to help debug Pioneer Power Toggle message.
Ref: https://github.com/crankyoldgit/IRremoteESP8266/issues/1220#issuecomment-661676014