crankyoldgit/IRremoteESP8266 #1223: in GREE protocol i have transmitting issue

MarkEvens, 2020-07-22:
I am using arduino as componnent with esp-idf framwork and add IRremoteESP8266 with it.
**********************************************************************************************************
Below is my code
**********************************************************************************************************
#include "Arduino.h"
#include "freertos/FreeRTOS.h"
#include "esp_wifi.h"
#include "esp_system.h"
#include "esp_event.h"
#include "esp_event_loop.h"
#include "nvs_flash.h"
#include "driver/gpio.h"

#include <IRremoteESP8266.h>
#include <IRac.h>
#include <IRtext.h>
#include <IRutils.h>

const uint16_t kIrLed = 4;  // The ESP GPIO pin to use that controls the IR LED.
IRac ac(kIrLed);  // Create a A/C object using GPIO to sending messages with.

extern "C" void app_main()
{
	//Serial.begin(115200);
	  //delay(200);
	  ac.next.protocol = decode_type_t(24);
	  ac.next.model = 1;
	  ac.next.mode = stdAc::opmode_t::kCool;
	  ac.next.celsius = true;
	  ac.next.degrees = 25;
	  ac.next.fanspeed = stdAc::fanspeed_t::kMedium;
	  ac.next.swingv = stdAc::swingv_t::kOff;
	  ac.next.swingh = stdAc::swingh_t::kOff;
	  ac.next.light = false;
	  ac.next.beep = false;
	  ac.next.econo = false;
	  ac.next.filter = false;
	  ac.next.turbo = false;
	  ac.next.quiet = false;
	  ac.next.sleep = -1;
	  ac.next.clean = false;
	  ac.next.clock = -1;
	  ac.next.power = false;
    while(1) {
    	ac.sendAc();  // Have the IRac class create and send a message.
    	  //delay(5000);  // Wait 5 seconds.
    	//  Serial.println("BUTTON TRANSMITED..");
        vTaskDelay(5000 / portTICK_PERIOD_MS);
    }
}
****************************************************************************************************
When transmite signal using this code and received it in irrecieveDumpV2 code it shows unknown protocol and raw array of 139
element. but time of all element is changing in everey transmission

I tried all the frequency of esp32 like 160/240Mhz etc.

crankyoldgit, 2020-07-22:
Can you please include those capture dumps from IRrecvDumpV2 so we can see what is going on.

MarkEvens, 2020-07-23:
**************************************************CAPTURE DUMPED************************************************************************
IRrecvDumpV2 is now running and waiting for IR input on Pin 14
Timestamp : 000005.144
Library   : v2.7.4

Protocol  : UNKNOWN
Code      : 0x2DD58E79 (70 Bits)
uint16_t rawData[139] = {8998, 4512,  634, 1596,  634, 536,  632, 536,  634, 536,  612, 554,  610, 1616,  612, 558,  612, 554,  632, 1596,  614, 552,  612, 558,  628, 1606,  614, 552,  612, 560,  634, 534,  634, 536,  612, 552,  612, 558,  634, 536,  634, 534,  632, 538,  610, 556,  614, 552,  610, 558,  632, 536,  632, 536,  634, 536,  612, 566,  614, 1612,  612, 556,  614, 1610,  614, 566,  634, 534,  612, 1620,  612, 556,  634, 7464,  614, 556,  610, 556,  612, 558,  634, 534,  634, 536,  612, 552,  612, 556,  612, 560,  612, 560,  612, 552,  610, 560,  634, 534,  634, 536,  610, 1616,  614, 558,  612, 554,  624, 548,  634, 538,  610, 556,  612, 558,  634, 534,  634, 536,  612, 550,  612, 558,  612, 558,  612, 556,  612, 558,  634, 534,  634, 536,  630, 1596,  610, 1618,  620, 556,  632};  // UNKNOWN 2DD58E79

Timestamp : 000010.263
Library   : v2.7.4

Protocol  : UNKNOWN
Code      : 0x2DD58E79 (70 Bits)
uint16_t rawData[139] = {8996, 4512,  634, 1594,  634, 536,  632, 534,  614, 554,  612, 552,  612, 1614,  632, 536,  634, 538,  634, 1594,  634, 536,  632, 536,  618, 1614,  634, 534,  634, 536,  610, 556,  612, 560,  632, 538,  610, 556,  614, 552,  632, 536,  612, 558,  634, 534,  634, 536,  610, 558,  614, 552,  612, 558,  634, 534,  644, 540,  616, 1614,  612, 556,  614, 1610,  614, 566,  634, 536,  610, 1618,  634, 536,  612, 7488,  614, 554,  610, 556,  610, 560,  634, 534,  634, 536,  612, 552,  612, 556,  612, 560,  612, 560,  612, 552,  612, 558,  634, 534,  634, 536,  610, 1616,  612, 558,  612, 554,  624, 548,  634, 536,  610, 556,  610, 560,  634, 534,  634, 536,  612, 552,  612, 558,  612, 558,  612, 556,  612, 558,  634, 534,  634, 536,  632, 1598,  610, 1616,  620, 556,  636};  // UNKNOWN 2DD58E79

Timestamp : 000015.382
Library   : v2.7.4

Protocol  : UNKNOWN
Code      : 0x2DD58E79 (70 Bits)
uint16_t rawData[139] = {8998, 4512,  634, 1592,  634, 538,  630, 536,  614, 554,  610, 552,  614, 1612,  612, 552,  612, 558,  634, 1596,  612, 552,  612, 558,  624, 1610,  634, 536,  634, 536,  632, 536,  612, 554,  612, 558,  612, 554,  634, 536,  614, 552,  612, 558,  612, 552,  612, 558,  636, 536,  610, 556,  614, 554,  610, 556,  610, 560,  612, 1612,  612, 560,  610, 1620,  632, 544,  634, 536,  632, 1596,  614, 552,  612, 7508,  614, 556,  610, 556,  612, 560,  634, 534,  634, 536,  612, 552,  612, 556,  612, 560,  612, 560,  614, 552,  612, 558,  634, 534,  634, 536,  610, 1614,  612, 560,  612, 554,  624, 550,  634, 536,  610, 556,  612, 558,  636, 534,  634, 536,  610, 552,  612, 558,  612, 558,  612, 556,  612, 558,  634, 534,  634, 536,  632, 1596,  612, 1616,  620, 556,  632};  // UNKNOWN 2DD58E79

Timestamp : 000020.502
Library   : v2.7.4

Protocol  : UNKNOWN
Code      : 0x2DD58E79 (70 Bits)
uint16_t rawData[139] = {8996, 4512,  634, 1594,  634, 536,  632, 536,  612, 554,  612, 552,  612, 1612,  632, 534,  634, 538,  632, 1594,  634, 538,  630, 536,  618, 1614,  634, 534,  636, 534,  610, 556,  612, 560,  634, 536,  610, 556,  612, 552,  634, 534,  612, 558,  634, 534,  634, 536,  610, 558,  614, 552,  612, 558,  636, 534,  642, 540,  616, 1614,  610, 556,  614, 1610,  614, 566,  634, 536,  610, 1618,  634, 536,  610, 7488,  614, 556,  610, 556,  612, 558,  634, 534,  634, 536,  610, 552,  612, 556,  612, 560,  610, 560,  612, 552,  610, 560,  634, 534,  634, 536,  610, 1614,  612, 558,  612, 554,  624, 548,  634, 538,  610, 556,  612, 560,  634, 534,  634, 536,  612, 552,  612, 558,  612, 558,  612, 556,  612, 558,  634, 534,  634, 536,  630, 1596,  612, 1618,  620, 556,  632};  // UNKNOWN 2DD58E79

Timestamp : 000025.622
Library   : v2.7.4

Protocol  : UNKNOWN
Code      : 0x2DD58E79 (70 Bits)
uint16_t rawData[139] = {8998, 4510,  634, 1594,  634, 536,  632, 536,  614, 552,  612, 552,  612, 1612,  614, 552,  612, 558,  634, 1596,  614, 552,  612, 558,  624, 1610,  634, 536,  634, 536,  632, 536,  612, 554,  612, 558,  612, 556,  632, 536,  612, 552,  612, 560,  612, 552,  612, 558,  636, 536,  610, 556,  614, 554,  610, 556,  612, 558,  612, 1612,  612, 558,  612, 1618,  636, 544,  632, 534,  634, 1596,  614, 552,  612, 7506,  614, 556,  610, 556,  612, 560,  634, 534,  634, 536,  610, 552,  612, 556,  612, 560,  610, 560,  614, 552,  610, 560,  634, 534,  634, 536,  610, 1616,  612, 558,  612, 554,  624, 548,  634, 536,  610, 556,  612, 558,  634, 534,  634, 536,  612, 552,  612, 558,  612, 558,  612, 556,  612, 558,  634, 534,  634, 536,  630, 1596,  612, 1618,  620, 556,  632};  // UNKNOWN 2DD58E79

MarkEvens, 2020-07-23:
Received dump by latest liberary

Timestamp : 000017.358
Library   : v2.7.8

Protocol  : UNKNOWN
Code      : 0x2DD58E79 (70 Bits)
uint16_t rawData[139] = {9000, 4514,  634, 1596,  614, 554,  610, 556,  632, 540,  612, 558,  612, 1612,  634, 538,  610, 554,  636, 1594,  612, 554,  630, 540,  628, 1604,  634, 536,  632, 534,  634, 538,  612, 554,  632, 536,  634, 536,  632, 538,  614, 554,  632, 536,  614, 554,  610, 556,  630, 540,  634, 536,  610, 556,  612, 554,  608, 568,  614, 1612,  634, 534,  634, 1592,  612, 566,  634, 534,  612, 1618,  634, 536,  634, 7460,  634, 538,  630, 534,  634, 538,  612, 552,  632, 540,  612, 552,  632, 538,  634, 538,  610, 562,  616, 552,  612, 556,  612, 558,  612, 554,  634, 1596,  614, 554,  610, 558,  626, 548,  634, 536,  610, 556,  614, 554,  610, 556,  632, 538,  636, 534,  634, 536,  612, 556,  632, 534,  634, 538,  630, 538,  612, 554,  612, 1612,  636, 1594,  626, 552,  610};  // UNKNOWN 2DD58E79

Timestamp : 000022.477
Library   : v2.7.8

Protocol  : UNKNOWN
Code      : 0x2DD58E79 (70 Bits)
uint16_t rawData[139] = {8998, 4514,  614, 1610,  612, 558,  610, 556,  630, 540,  612, 558,  610, 1614,  634, 538,  612, 554,  634, 1594,  634, 536,  632, 538,  626, 1606,  614, 554,  610, 556,  630, 540,  634, 536,  610, 558,  612, 552,  610, 558,  636, 534,  632, 538,  610, 556,  614, 554,  610, 558,  634, 534,  632, 536,  634, 536,  612, 568,  612, 1612,  612, 558,  630, 1598,  634, 544,  634, 534,  634, 1596,  632, 538,  612, 7478,  634, 536,  632, 536,  634, 536,  612, 552,  630, 540,  612, 552,  632, 540,  634, 536,  610, 562,  616, 552,  612, 558,  610, 560,  612, 552,  634, 1596,  614, 554,  610, 558,  626, 548,  634, 536,  610, 556,  612, 554,  610, 556,  630, 540,  634, 534,  634, 536,  612, 556,  632, 534,  634, 536,  632, 536,  612, 554,  612, 1612,  636, 1594,  624, 554,  610};  // UNKNOWN 2DD58E79

Timestamp : 000027.595
Library   : v2.7.8

Protocol  : UNKNOWN
Code      : 0x2DD58E79 (70 Bits)
uint16_t rawData[139] = {8996, 4514,  612, 1612,  612, 560,  612, 552,  612, 558,  634, 536,  634, 1592,  634, 536,  610, 558,  632, 1596,  614, 552,  630, 540,  626, 1608,  634, 536,  632, 534,  634, 538,  612, 552,  632, 536,  632, 538,  630, 538,  612, 554,  634, 536,  610, 556,  612, 554,  610, 558,  634, 534,  634, 536,  632, 538,  612, 568,  614, 1610,  612, 558,  634, 1594,  634, 548,  632, 538,  632, 1598,  634, 538,  634, 7454,  612, 554,  612, 556,  612, 560,  634, 536,  632, 536,  610, 556,  614, 554,  610, 558,  610, 568,  630, 538,  634, 534,  634, 534,  612, 552,  632, 1596,  610, 558,  612, 554,  626, 548,  632, 538,  610, 556,  632, 540,  634, 534,  634, 536,  610, 552,  612, 558,  632, 540,  612, 552,  630, 540,  612, 552,  630, 540,  612, 1612,  632, 1596,  626, 552,  612};  // UNKNOWN 2DD58E79

Timestamp : 000032.683
Library   : v2.7.8

Protocol  : UNKNOWN
Code      : 0xDF06F900 (70 Bits)
uint16_t rawData[139] = {8948, 4522,  630, 1596,  632, 536,  604, 676,  544, 556,  612, 556,  590, 1594,  632, 536,  634, 536,  606, 1666,  584, 580,  588, 538,  618, 1614,  634, 534,  632, 584,  552, 608,  592, 538,  632, 578,  614, 514,  606, 558,  658, 512,  634, 536,  632, 536,  606, 558,  606, 560,  658, 508,  658, 514,  602, 562,  602, 584,  634, 1598,  572, 592,  632, 1596,  608, 610,  590, 538,  630, 1648,  554, 560,  630, 7466,  606, 560,  636, 536,  632, 584,  586, 578,  590, 540,  632, 534,  606, 606,  586, 536,  604, 574,  630, 540,  632, 582,  612, 510,  608, 558,  632, 1660,  570, 534,  634, 580,  598, 592,  594, 512,  656, 512,  632, 536,  632, 578,  614, 514,  606, 558,  634, 534,  634, 536,  632, 534,  634, 536,  606, 558,  660, 574,  540, 1622,  658, 1572,  620, 554,  606};  // UNKNOWN DF06F900

Suraj-singh999898, 2020-07-23:
Same Issue 

esp32 IR repeater code written in esp idf with ardunio esp32 core
Pressing GREE remote button. 
Device can't re-transmitter
Showing 
Protocol: UNKNOWN
Code      : 0xEBB83DCE (70 Bits)

crankyoldgit, 2020-07-23:
> it shows unknown protocol and raw array of 139
> element. but time of all element is changing in everey transmission

Okay, with the data, I think I now understand this question.
See the FAQ (https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#Why_is_the_raw_data_for_a_button_or_AC_state_always_different_for_each_capture) for why it's different every time.

As for why it isn't being decoded correctly, that's a separate issue. I'll look into that soon.

crankyoldgit, 2020-07-23:
> Same Issue
> 
> esp32 IR repeater code written in esp idf with Arduino esp32 core
> Pressing GREE remote button.
> Device can't re-transmit
> Showing
> Protocol: UNKNOWN
> Code : 0xEBB83DCE (70 Bits)

@Suraj-singh999898 I am not sure what you are saying here. That doesn't look like the output of any of the examples that come with the library. Can you please be more specific?

Suraj-singh999898, 2020-07-23:
We found the problem 
but still don't know the solution
Gree raw code there is one > 18554 value
sending from esp idf that will receiver >  8818

c

crankyoldgit, 2020-07-23:
> We found the problem
> but still don't know the solution
> Gree raw code there is one > 18554 value
> sending from esp idf that will receiver > 8818

I see no `18544` value in any of the data in this issue. What are you talking about?

Jaydip-Chabhadiya, 2020-07-23:
@crankyoldgit 
I found issue with ESP-IDF with this particular Gree remote
Please have a look at graph which clearly shows there may be some timing issues

Please have a look at this
https://docs.google.com/spreadsheets/d/1lPKVLC3N5xE6vW2tZoPdfKqXvG9_L4Kdl4hh4Pj6f6E/edit?usp=sharing

Hope you will figureout quick solution

crankyoldgit, 2020-07-23:
For starters, this library is only meant to work with the Arduino framework. So, technically you're using it "out of spec".
But I think I know what the problem is related to this:

> ```
> //delay(5000); // Wait 5 seconds.
> // Serial.println("BUTTON TRANSMITED..");
> vTaskDelay(5000 / portTICK_PERIOD_MS);
> ```

I'm guessing you don't have the `delay()` function/call in the plain ESP-IDF.
The `delay()` function gets used for `_delayMicroseconds()` calls where the value is larger than `kMaxAccurateUsecDelay` (`16383`). That will be happening in the Gree protocol (and quite a few others)

So, you've got a few options, either change the code to use your `vTaskDelay()` function instead of `delay()` in:
https://github.com/crankyoldgit/IRremoteESP8266/blob/ad436f76ba8e102bb54c89c6ce15e6cde95b9054/src/IRsend.cpp#L111-L142

Or .. Change `ALLOW_DELAY_CALLS` to `false` here:
https://github.com/crankyoldgit/IRremoteESP8266/blob/ad436f76ba8e102bb54c89c6ce15e6cde95b9054/src/IRremoteESP8266.h#L692-L698
That should cause it to only use the `delayMicroseconds()` call instead of `delay()`.

crankyoldgit, 2020-07-23:
> Or .. Change `ALLOW_DELAY_CALLS` to `false` here:

Or .. compile with the following flag(s): `-DALLOW_DELAY_CALLS=false`

Jaydip-Chabhadiya, 2020-07-23:
@crankyoldgit 
OK Thanks for the quick reply
I'll check and let you know

crankyoldgit, 2020-07-23:
Um @Jaydip-Chabhadiya @Suraj-singh999898 @MarkEvens ... Why are all three of you having this same issue at the same time?
Is this something you are using as a part of some other project? If so, what?

Jaydip-Chabhadiya, 2020-07-23:
@crankyoldgit 
We are working from home on same project due to COVID-19, that's why you see different results

However ALLOW_DELAY_CALLS false works!

Thank you for quick help :+1: 

I really appriciate your help :clap:

crankyoldgit, 2020-07-23:
> We are working from home on same project due to COVID-19, that's why you see different results

What's the project?

Jaydip-Chabhadiya, 2020-07-23:
Its about controlling air conditioner (on/off/fan-mode) in timely manner based on temperature value received by sensor.

closed 2020-07-23 by crankyoldgit