crankyoldgit/IRremoteESP8266 #1230: Inquiring about a flowchart pf processes

Faberdash, 2020-08-05:
Hi David,
I am a computer science undergrad currently working on my bachelor project, in which I am supposed to build a smart IR receiver on an ESP8266 chip. I am using your library, and first of all, I would like to thank you and your collaborators for the amazing effort put in this library.
Secondly, I am required to include a flowchart of processes for any external library I use in my Thesis paper so, is there a chance you have one that I can cite (a simplified version is more than sufficient, I am only concerned with multimedia IR protocols).
I would have prepared one myself, but sadly I am a little out of time as my paper is due this week and the flowchart is the only missing piece. Thanks in advance :)

crankyoldgit, 2020-08-06:
How much detail do you require in the flowchart? Do you have any links to some examples?

crankyoldgit, 2020-08-06:
How's this?
[IRrecv Capture-Decode Flow Chart.pdf](https://github.com/crankyoldgit/IRremoteESP8266/files/5032196/IRrecv.Capture-Decode.Flow.Chart.pdf)

Faberdash, 2020-08-06:
It is sufficient for the time being, I still can make amendments to my paper until my Thesis Defense date mid-September so, I will wait for the feedback from my supervisors and if they demand more details, I'll ask you again if I need any assistance.
I can't thank you enough for such a quick response, It is much appreciated <3

crankyoldgit, 2020-08-07:
[image]
[image]
Uploading the underlying flow charts so I can use them in a Wiki entry

crankyoldgit, 2020-08-07:
And now the wiki page exists: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Overview-of-how-the-library-works

closed 2020-08-07 by crankyoldgit