crankyoldgit/IRremoteESP8266 #1231: Turning off protocols does not reduce library size

brownrb, 2020-08-06:
Hi
I am using ESP32 Dev and the latest Arduino 1.0.4 core with the latest IRemoteESP8266 library
I was trying to reduce the size of the program by turning off all protocols and only enabling NEC

#define _IR_ENABLE_DEFAULT_  false                // turn off all protocols to save space
#define DECODE_NEC           true                 // enable NEC
#include <IRremoteESP8266.h>
#include <IRrecv.h>
#include <IRutils.h>
const uint16_t RECV_PIN = IRPIN;
IRrecv irrecv(RECV_PIN);
decode_results results;

However, i get the same program and data size even if I comment out these two lines
//#define _IR_ENABLE_DEFAULT_  false                // turn off all protocols to save space
//#define DECODE_NEC           true                 // enable NEC

I don't think that should be so. I believe the code space should be significantly less. Am I doing something wrong?

Regards
Robert

crankyoldgit, 2020-08-06:
> Am I doing something wrong?

Yes.
Uncommenting those lines is not how you do it.

You need the following before that section:
```
#define IR_ENABLE_DEFAULT false
#define DECODE_NEC true
```

crankyoldgit, 2020-08-07:
Apologies, your message came across as fairly garbled on my phone's screen. Yes, you are still doing it "wrong", but not as wrong as I thought at first.

Basically, follow these instructions:
https://github.com/crankyoldgit/IRremoteESP8266/blob/857a9eed7e38333734f9f4bd8f070a4703886609/src/IRremoteESP8266.h#L62-L82

You'll need to make those `#define`s in the `IRremoteESP8266.h` file, not in your "own" code. OR you need to compile ALL the code with the appropriate `-D` options as suggested in the code snippet above.

In short, the way you have done it is your `.ino` project file with have those `#define`s but the rest of the library won't. The compilers will be compiling the rest of the library without them, thus resulting in the maximum size.
You have two options for influencing if they are included or not.
1) Change them in `IRremoteESP8266.h` as every core part of the library pulls in the file at some point and will act according to it.
  e.g. Change:
https://github.com/crankyoldgit/IRremoteESP8266/blob/857a9eed7e38333734f9f4bd8f070a4703886609/src/IRremoteESP8266.h#L81
to `#define _IR_ENABLE_DEFAULT_ false`
and
https://github.com/crankyoldgit/IRremoteESP8266/blob/857a9eed7e38333734f9f4bd8f070a4703886609/src/IRremoteESP8266.h#L100
to `#define SEND_NEC true`

**OR**

2) Compile every part of the library's code with `-DIR_ENABLE_DEFAULT=false -DDECODE_NEC=true`. You can do this easily if you're using PlatformIO by modifying the `platformio.ini` file.

crankyoldgit, 2020-08-18:
Friendly chase up. Did you get this working?

brownrb, 2020-08-18:
Hi
Thanks for the help. yes I ended up copying the files and ripping everything out except NEC protocol. Working fine. I left all the copyright etc intact. :-)

closed 2020-08-18 by brownrb