crankyoldgit/IRremoteESP8266 #1233: Expanding support for Daikin 176 protocol

pete571, 2020-08-08:
Hi,
I have Daikin A/C indoor unit FFQ35B8V1B with remote BRC4C151.
After experimenting, I find I can control it with the IRDaikin176 class from the library (using v2.7.8).

But my system supports modes DRY, AUTO, FAN, COOL, and HEAT, whereas the library class seems to only support DRY, FAN, and COOL. Somehow I would like to add the AUTO and HEAT modes to the class.

I used IRrecvDumpv3 to capture the codes when pressing the MODE button on my remote and cycling through the 5 possibilities. The dump is in the attached file.

I think modifying the library is beyond me so I have raised this issue as a feature request and appreciate your feedback. Happy to provide any further information required.

Thanks!
[daikin176 modes.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/5044691/daikin176.modes.txt)

crankyoldgit, 2020-08-08:
Hey @pete571 thanks for providing all the useful data. That was really helpful.
I think I've added support for you additional modes in PR #1235 / Branch https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1233

Can you please download it and us know how it goes? (Note: The values for the operation modes have now changed in that branch)

pete571, 2020-08-09:
Hi,

Thanks, I tried the new code and here is what I observe:

**setMode(0)** - AUTO according to IRrecvDumpV3
This does operate the a/c but according to the IRrecvDumpV3 it always sets the temperature to 17 regardless of what is set using setTemp(). In the Daikin I have, AUTO temperatures can only be set to a limited range. The AUTO temperatures are set using a different display on the remote, not the digital temperature display, but rather a scale that looks like this "H - M - L". On this scale H = 25, H-M = 23, M = 22, M-L = 21, and L =19. SoI am not sure what the a/c does when it receives an IR AUTO command with a temp of 17, but it is probably not what is expected.

These are the commands captured from my remote when using AUTO:
	0x11DA171804001E11DA17180073043000002036002037 (temp on H, 25)
	0x11DA171804001E11DA17180073043000001C36002033 (temp on H-M, 23)
	0x11DA171804001E11DA17180073043000001A36002031 (temp on M, 22)
	0x11DA171804001E11DA1718007304300000183600202F (temp on M-L, 21)
	0x11DA171804001E11DA1718007304300000143600202B (temp on L, 19)

**setMode(1)** - UNKNOWN according to IRrecvDumpV3, but aligns with the HEAT command captured from my remote
This does operate the a/c and it produces hot air by feel.

**setMode(2)
setMode(4)
setMode(5)
setMode(6)** - COOL according to IRrecvDumpV3 and aligns with the COOL command captured from my remote. Mode 2, 4, 5, and 6 seem to send the same code.
This does operate the a/c and it produces cool air by feel.

**setMode(3)**- UNKNOWN according to IRrecvDumpV3 and I cannot align it with any codes from my remote
a/c responds to this, but by feel I cannot tell what it is doing.

**setMode(7)**- UNKNOWN according to IRrecvDumpV3 and I cannot align it with any codes from my remote
a/c responds to this, but by feel I cannot tell what it is doing.

Do these results align with the new operation modes you mention?
Also, seems like the mode to string code needs an update?

Thanks

crankyoldgit, 2020-08-09:
D'oh! I forgot to update the `toString()` mode values, they should be fixed in the latest commit to that branch. Please re-download and test it.

The valid "mode" values for `IRDaikin176::setMode()` are:
https://github.com/crankyoldgit/IRremoteESP8266/blob/d643e04ca3b45335d64f1567eb961bf536ffab48/src/ir_Daikin.h#L349-L353
Anything other than those values should default to Cool (2) mode.

When/where are you calling `setTemp()` with some other value?
Setting a mode of Dry or Heat sets the Temp to 17C.
https://github.com/crankyoldgit/IRremoteESP8266/blob/d643e04ca3b45335d64f1567eb961bf536ffab48/src/ir_Daikin.h#L361

So I'm guessing in your code that you've set the mode to heat or dry, then changed to Auto without doing a `setTemp(xx)` afterwards.

pete571, 2020-08-09:
Ok, I get it, that makes sense.
I tried the new commit and now all looks good.

Thanks for the project and prompt responses!

crankyoldgit, 2020-08-09:
Thanks for the confirmation it's now all working as expected. I'll merge it into the main branch.

closed 2020-08-09 by crankyoldgit

crankyoldgit, 2020-08-31:
The code changes mention have now been included in the newly released v2.7.10 of the library.

PR #1235 (2020-08-09): [Breaking] Daikin176: Change operating mode values.
* More user supplied data has shown the `altmode` was the real mode all along. Change code to reflect that.
* Change mode values to use `kDaikin176Fan` etc.
* Add support for Heat & Auto modes.
* Update unit tests accordingly.
* Update the supported model info.

Fixes #1233
commit fb2793fd (2020-08-09): [Breaking] Daikin176: Change & increase operating mode values. (#1235)

* More user supplied data has shown the `altmode` was the real mode all along. Change code to reflect that.
* Change mode values to use `kDaikin176Fan` etc.
* Add support for Heat & Auto modes.
* Update unit tests accordingly.
* Update the supported model info.

Fixes #1233