crankyoldgit/IRremoteESP8266 #1234: Issue with Swing with MIDEA protocol

paulomanuelp, 2020-08-08:
Hello, 

I have a Daitsu HRV2 that uses the MIDEA protocol. For now all the modes, temperature set,  fan speed and the turn off of swing mode are working we ll. But the command to turn on does not work at all. Could you please help with this issue?

Thanks in advance,
Paulo

### Version/revision of the library used
2.7.9

### Expected behavior

When I exec the following code:
`IRac ac(kIrLed);`
`ac.next.swingv = stdAc::swingv_t::kAuto;`

Is expected that ac swing is turn on.

### Actual behavior

The ac does not respond to the signal (also does not give any beep).

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
**To turn the swing ON:**

Timestamp : 000355.633
Library   : v2.7.9

Protocol  : MIDEA
Code      : 0xA201FFFFFF7C (48 Bits)
Mesg Desc.: Swing(V) Toggle: On, Econo Toggle: Off
uint16_t rawData[199] = {4386, 4408,  542, 1606,  540, 532,  512, 1634,  516, 558,  544, 528,  516, 558,  512, 1634,  542, 530,  490, 584,  544, 528,  486, 588,  514, 558,  542, 532,  510, 560,  514, 560,  486, 1660,  512, 1636,  542, 1604,  514, 1634,  466, 1682,  514, 1632,  544, 1602,  512, 1634,  516, 1632,  542, 1606,  544, 1604,  572, 1574,  514, 1632,  546, 1602,  514, 1632,  544, 1602,  514, 1632,  542, 1606,  512, 1634,  540, 1606,  514, 1632,  542, 1604,  570, 1576,  516, 1632,  516, 1630,  542, 530,  520, 1628,  538, 1608,  546, 1602,  540, 1606,  600, 1546,  572, 502,  544, 530,  514, 5222,  4364, 4406,  546, 528,  544, 1604,  542, 530,  570, 1576,  544, 1602,  570, 1576,  572, 500,  542, 1606,  544, 1602,  544, 1604,  572, 1576,  516, 1632,  602, 1544,  516, 1630,  570, 1576,  544, 530,  542, 532,  544, 528,  542, 530,  516, 556,  570, 502,  540, 532,  544, 530,  514, 556,  546, 526,  574, 500,  514, 558,  542, 528,  542, 530,  544, 528,  546, 526,  552, 520,  544, 530,  544, 528,  570, 502,  546, 528,  572, 500,  516, 558,  542, 530,  572, 502,  572, 1574,  544, 528,  542, 530,  514, 558,  544, 528,  572, 502,  576, 1570,  574, 1574,  544};  // MIDEA A201FFFFFF7C
uint64_t data = 0xA201FFFFFF7C;

Protocol  : MIDEA
Code      : 0xA201FFFFFF7C (48 Bits)
Mesg Desc.: Swing(V) Toggle: On, Econo Toggle: Off
uint16_t rawData[199] = {4414, 4380,  574, 1574,  572, 500,  636, 1510,  544, 530,  518, 554,  546, 526,  518, 1630,  546, 526,  546, 526,  574, 498,  544, 528,  544, 528,  546, 528,  574, 498,  516, 558,  546, 1600,  546, 1602,  544, 1602,  546, 1602,  546, 1578,  566, 1604,  574, 1574,  544, 1602,  548, 1600,  544, 1604,  544, 1604,  576, 1572,  544, 1602,  518, 1630,  574, 1572,  546, 1602,  542, 1604,  544, 1604,  548, 1600,  572, 1576,  544, 1602,  518, 1630,  546, 1602,  602, 1544,  516, 1632,  518, 554,  546, 1602,  574, 1574,  546, 1602,  576, 1570,  546, 1600,  574, 498,  516, 558,  572, 5166,  4422, 4350,  574, 500,  516, 1632,  542, 530,  574, 1574,  572, 1574,  490, 1658,  542, 530,  516, 1630,  544, 1604,  514, 1632,  518, 1630,  544, 1604,  518, 1630,  572, 1576,  570, 1576,  574, 498,  544, 530,  544, 530,  542, 528,  572, 502,  574, 500,  490, 582,  544, 530,  546, 528,  572, 500,  604, 468,  606, 468,  546, 526,  574, 498,  544, 530,  572, 500,  544, 528,  548, 524,  546, 526,  572, 502,  544, 530,  542, 528,  546, 528,  516, 556,  518, 556,  570, 1576,  574, 500,  542, 530,  516, 556,  570, 502,  548, 526,  570, 1576,  574, 1574,  544};  // MIDEA A201FFFFFF7C
uint64_t data = 0xA201FFFFFF7C;

**To turn the swing OFF:**

Timestamp : 000314.898
Library   : v2.7.9

Protocol  : MIDEA
Code      : 0xA201FFFFFF7C (48 Bits)
Mesg Desc.: Swing(V) Toggle: On, Econo Toggle: Off
uint16_t rawData[199] = {4418, 4376,  572, 1574,  604, 470,  606, 1542,  572, 500,  574, 498,  604, 468,  604, 1544,  572, 500,  576, 496,  572, 500,  544, 528,  574, 500,  604, 466,  578, 494,  574, 498,  574, 1572,  636, 1512,  608, 1540,  578, 1568,  550, 1598,  572, 1574,  576, 1572,  578, 1570,  604, 1542,  544, 1602,  604, 1542,  576, 1570,  574, 1574,  602, 1544,  546, 1576,  626, 1544,  576, 1546,  572, 1600,  576, 1570,  572, 1574,  576, 1572,  574, 1572,  576, 1570,  546, 1602,  546, 1600,  576, 496,  574, 1572,  578, 1570,  578, 1568,  604, 1544,  574, 1572,  546, 526,  574, 498,  548, 5190,  4424, 4346,  576, 472,  570, 1600,  574, 498,  546, 1602,  604, 1542,  576, 1570,  576, 496,  574, 1574,  552, 1592,  576, 1572,  576, 1570,  546, 1602,  606, 1540,  546, 1600,  574, 1572,  546, 504,  628, 468,  544, 528,  574, 498,  606, 466,  604, 470,  516, 556,  578, 494,  610, 462,  610, 462,  572, 500,  602, 470,  578, 494,  574, 498,  572, 500,  634, 438,  546, 524,  604, 468,  608, 442,  632, 462,  576, 474,  596, 500,  546, 526,  602, 448,  570, 526,  612, 1536,  580, 492,  638, 434,  610, 462,  636, 436,  576, 496,  602, 1544,  572, 1574,  610};  // MIDEA A201FFFFFF7C
uint64_t data = 0xA201FFFFFF7C;

Timestamp : 000511.720
Library   : v2.7.9

Protocol  : MIDEA
Code      : 0xA201FFFFFF7C (48 Bits)
Mesg Desc.: Swing(V) Toggle: On, Econo Toggle: Off
uint16_t rawData[199] = {4386, 4408,  572, 1574,  546, 528,  572, 1576,  544, 530,  546, 526,  574, 500,  544, 1602,  544, 528,  516, 556,  574, 498,  544, 528,  542, 530,  606, 466,  546, 528,  574, 498,  514, 1632,  574, 1572,  572, 1576,  544, 1602,  544, 1604,  568, 1578,  572, 1574,  542, 1606,  516, 1632,  570, 1576,  516, 1632,  546, 1602,  576, 1572,  518, 1628,  516, 1632,  544, 1604,  546, 1602,  544, 1602,  514, 1634,  544, 1600,  490, 1658,  516, 1632,  572, 1550,  538, 1632,  542, 1604,  544, 530,  516, 1630,  544, 1602,  544, 1602,  602, 1544,  540, 1606,  544, 530,  574, 500,  544, 5170,  4416, 4380,  544, 530,  544, 1602,  542, 530,  572, 1576,  518, 1630,  546, 1602,  516, 556,  544, 1604,  544, 1604,  544, 1602,  546, 1602,  574, 1574,  542, 1604,  572, 1576,  570, 1576,  542, 530,  544, 528,  602, 470,  524, 548,  544, 532,  542, 528,  572, 500,  516, 558,  516, 556,  514, 558,  572, 500,  548, 526,  516, 556,  574, 498,  572, 502,  574, 498,  544, 530,  572, 502,  544, 528,  546, 528,  570, 502,  542, 530,  574, 498,  544, 528,  542, 532,  544, 1602,  546, 528,  544, 530,  572, 500,  542, 532,  572, 500,  572, 1576,  578, 1570,  516};  // MIDEA A201FFFFFF7C
uint64_t data = 0xA201FFFFFF7C;

#### Circuit diagram and hardware used (if applicable)
_Link to an image of the circuit diagram used. Part number of the IR receiver module etc. ESP8266 or ESP32 board type._

### I have followed the steps in the [Troubleshooting Guide]Yes

### Has this library/code previously worked as expected for you?
No

NiKiZe, 2020-08-08:
Please note that it is a toggle, meaning that there is no on or off, only toggle, which you can see in your dump as well since it is the same data in all your dumps.

paulomanuelp, 2020-08-08:
Yes that's strange. When I send the stdAc::swingv_t::kOff command the ac responds with a beep, but when I send stdAc::swingv_t::kAuto the ac does not respond.  I already try just to send stdAc::swingv_t::kOff to turn on and off the swing, but it only disables the swing.
 What should I do next?

NiKiZe, 2020-08-08:
First send the AC state to the AC.
After this send only the swing change, some ACs won't accept for example a command that both changes power, and toggles swing in the same command. Can't say what it is in your case, but while troubleshooting make sure that only one thing changes at once, and after that try to "batch" it, at least that is what I would have tried.

crankyoldgit, 2020-08-08:
As you can see from the data you collected. The remote sends the same command to "toggle" the swing.
The `IRac` class tries to track the setting for you and should only send the toggle message when it thinks it should.
It could be that bit is messed up.

Can you please supply us with the code (and sequence) you are using **and** a full cut-n-paste of the output of the dump program for the same sequence from your real remote. We can then try to work out what might be going wrong?

crankyoldgit, 2020-08-08:
FYI @NiKiZe has made the correct suggests etc. I'll leave this to handle. I just added my $0.02 worth.

paulomanuelp, 2020-08-10:
Hello,

I adapt the IRrecvDumpv3 and add this at end of setup:
```
    Serial.println("Turn on ac");
  ac.next.protocol = decode_type_t::MIDEA;  // Set a protocol to use.
  ac.next.model = 1;  // Some A/Cs have different models. Try just the first.
  ac.next.celsius = true;  // Use Celsius for temp units. False = Fahrenheit 
  
  ac.next.power = true;
  ac.next.mode = stdAc::opmode_t::kFan;
  ac.next.swingv = stdAc::swingv_t::kAuto;

  ac.sendAc();
```

And this to end of loop:

```
  delay(5000);

  Serial.println("Enable Swing");

  ac.next.swingv = stdAc::swingv_t::kAuto;

  ac.sendAc();

  delay(5000);
```

And this is the output I received:

```
Enable Swing
Timestamp : 000026.283
Library   : v2.7.8

Protocol  : MIDEA
Code      : 0xA18448FFFF52 (48 Bits)
Mesg Desc.: Power: On, Mode: 4 (Fan), Celsius: On, Temp: 25C/77F, Fan: 0 (Auto), Sleep: Off, Swing(V) Toggle: Off
uint16_t rawData[199] = {4510, 4468,  544, 1746,  528, 600,  528, 1720,  528, 552,  542, 604,  514, 612,  514, 636,  530, 1724,  530, 1672,  570, 600,  490, 610,  516, 610,  538, 590,  514, 1732,  516, 610,  516, 612,  556, 598,  490, 1756,  512, 614,  530, 572,  516, 1732,  516, 612,  514, 636,  490, 612,  516, 1756,  530, 1718,  530, 1668,  518, 1752,  516, 1730,  516, 1732,  516, 1756,  488, 1734,  516, 1708,  564, 1706,  514, 1732,  516, 1756,  530, 1694,  514, 1732,  516, 1756,  530, 1720,  530, 598,  530, 1696,  492, 656,  528, 1720,  528, 600,  490, 610,  514, 1756,  528, 600,  528, 5614,  4482, 4466,  542, 628,  490, 1732,  514, 636,  530, 1694,  516, 1756,  490, 1732,  516, 1730,  516, 612,  514, 636,  490, 1732,  516, 1732,  514, 1730,  516, 1730,  516, 612,  514, 1730,  516, 1758,  530, 1720,  530, 600,  490, 1756,  530, 1694,  514, 612,  516, 1756,  530, 1720,  530, 1694,  516, 612,  516, 636,  528, 600,  528, 576,  492, 656,  528, 578,  492, 656,  528, 576,  516, 636,  528, 600,  528, 576,  514, 636,  490, 636,  528, 600,  530, 598,  528, 602,  528, 1720,  528, 600,  528, 1694,  514, 636,  530, 1718,  490, 1756,  530, 600,  490, 1730,  516};  // MIDEA A18448FFFF52
uint64_t data = 0xA18448FFFF52;

Enable Swing
Timestamp : 000036.622
Library   : v2.7.8

Protocol  : MIDEA
Code      : 0xA18448FFFF52 (48 Bits)
Mesg Desc.: Power: On, Mode: 4 (Fan), Celsius: On, Temp: 25C/77F, Fan: 0 (Auto), Sleep: Off, Swing(V) Toggle: Off
uint16_t rawData[199] = {4476, 4534,  520, 1732,  550, 576,  518, 1728,  518, 608,  550, 576,  550, 576,  518, 608,  518, 1734,  520, 1726,  520, 608,  520, 606,  518, 584,  550, 572,  552, 1716,  498, 630,  518, 608,  552, 574,  550, 1696,  518, 608,  520, 608,  550, 1694,  520, 608,  518, 606,  520, 608,  520, 1728,  516, 1730,  550, 1696,  550, 1696,  520, 1726,  550, 1696,  552, 1670,  578, 1694,  550, 1696,  518, 1728,  520, 1726,  520, 1728,  550, 1696,  520, 1726,  520, 1728,  518, 1704,  578, 572,  518, 1728,  552, 574,  502, 1744,  520, 582,  552, 594,  520, 1726,  520, 608,  520, 5648,  4462, 4506,  518, 608,  550, 1672,  578, 572,  520, 1728,  550, 1696,  518, 1728,  550, 1696,  550, 576,  520, 608,  550, 1696,  552, 1694,  518, 1706,  550, 1716,  520, 606,  520, 1728,  520, 1706,  550, 1716,  518, 608,  518, 1728,  518, 1728,  520, 608,  520, 1726,  520, 1704,  552, 1716,  518, 608,  520, 606,  550, 576,  518, 608,  520, 608,  518, 610,  552, 574,  520, 608,  550, 576,  510, 616,  520, 604,  520, 608,  518, 608,  520, 608,  552, 576,  550, 576,  520, 1728,  518, 608,  550, 1696,  520, 608,  500, 1746,  520, 1726,  518, 608,  550, 1696,  520};  // MIDEA A18448FFFF52
uint64_t data = 0xA18448FFFF52;
```

It looks like that the the `ac.next.swingv = stdAc::swingv_t::kAuto;` property change is not doing nothing right?

paulomanuelp, 2020-08-10:
Meanwhile in the remote to turn on the swing:

Turn on
```
Timestamp : 000158.752
Library   : v2.7.8

Protocol  : MIDEA
Code      : 0xA201FFFFFF7C (48 Bits)
Mesg Desc.: Swing(V) Toggle: On
uint16_t rawData[199] = {4386, 4406,  518, 1628,  518, 556,  518, 1606,  540, 554,  518, 554,  518, 556,  518, 1628,  518, 554,  518, 554,  518, 556,  516, 532,  540, 556,  516, 554,  518, 556,  518, 554,  518, 1628,  516, 1630,  518, 1630,  516, 1628,  520, 1628,  518, 1628,  518, 1630,  516, 1628,  518, 1628,  518, 1628,  516, 1630,  516, 1630,  516, 1630,  518, 1628,  518, 1630,  516, 1630,  516, 1630,  516, 1630,  516, 1630,  518, 1628,  518, 1630,  516, 1630,  516, 1630,  516, 1630,  516, 1630,  516, 556,  516, 1630,  518, 1628,  518, 1628,  518, 1630,  516, 1630,  518, 556,  516, 556,  516, 5218,  4364, 4404,  518, 556,  516, 1630,  516, 556,  516, 1630,  516, 1630,  516, 1630,  518, 556,  518, 1628,  516, 1630,  518, 1630,  516, 1630,  516, 1630,  516, 1630,  518, 1606,  540, 1606,  540, 556,  518, 554,  518, 554,  516, 556,  518, 556,  516, 556,  516, 556,  516, 554,  518, 556,  516, 556,  516, 556,  516, 556,  516, 554,  518, 556,  516, 556,  516, 554,  518, 554,  518, 554,  518, 556,  516, 556,  516, 556,  516, 556,  516, 556,  516, 554,  518, 532,  540, 1630,  518, 554,  518, 556,  516, 554,  518, 554,  516, 556,  516, 1630,  516, 1630,  518};  // MIDEA A201FFFFFF7C
uint64_t data = 0xA201FFFFFF7C;
```

Turn off:
```
Timestamp : 000236.879
Library   : v2.7.8

Protocol  : MIDEA
Code      : 0xA201FFFFFF7C (48 Bits)
Mesg Desc.: Swing(V) Toggle: On
uint16_t rawData[199] = {4392, 4380,  544, 1602,  546, 528,  542, 1604,  544, 528,  544, 554,  518, 530,  546, 1624,  520, 528,  544, 528,  544, 530,  542, 554,  518, 554,  518, 554,  520, 552,  518, 530,  568, 1576,  544, 1628,  518, 1602,  544, 1628,  518, 1602,  568, 1578,  544, 1626,  520, 1602,  544, 1602,  544, 1602,  544, 1602,  544, 1602,  544, 1602,  544, 1628,  520, 1630,  516, 1628,  518, 1628,  520, 1600,  546, 1628,  518, 1602,  544, 1628,  518, 1628,  518, 1602,  546, 1626,  520, 1602,  544, 554,  518, 1628,  520, 1602,  544, 1626,  520, 1602,  544, 1602,  544, 528,  544, 554,  518, 5190,  4392, 4378,  546, 528,  544, 1626,  520, 528,  544, 1628,  520, 1602,  544, 1602,  544, 528,  570, 1576,  544, 1602,  544, 1626,  520, 1626,  518, 1602,  546, 1602,  570, 1602,  544, 1576,  544, 554,  520, 528,  544, 528,  570, 502,  542, 530,  570, 502,  546, 528,  570, 502,  544, 528,  570, 502,  570, 528,  528, 520,  544, 528,  568, 502,  572, 502,  570, 502,  544, 528,  570, 528,  520, 528,  544, 528,  568, 528,  520, 526,  546, 528,  570, 502,  570, 504,  546, 1600,  544, 552,  546, 504,  542, 528,  570, 502,  546, 528,  544, 1602,  554, 1592,  570};  // MIDEA A201FFFFFF7C
uint64_t data = 0xA201FFFFFF7C;
```
Example 2:

Turn on:
```
Timestamp : 000286.586
Library   : v2.7.8

Protocol  : MIDEA
Code      : 0xA201FFFFFF7C (48 Bits)
Mesg Desc.: Swing(V) Toggle: On
uint16_t rawData[199] = {4442, 4346,  574, 1576,  628, 444,  572, 1576,  544, 526,  572, 500,  568, 504,  546, 1600,  626, 446,  572, 500,  572, 502,  570, 500,  572, 502,  570, 500,  572, 500,  550, 522,  626, 1520,  572, 1574,  572, 1576,  544, 1626,  546, 1576,  570, 1576,  544, 1602,  544, 1602,  544, 1602,  544, 1600,  546, 1628,  520, 1626,  546, 1600,  546, 1576,  572, 1576,  546, 1600,  570, 1576,  572, 1574,  572, 1574,  574, 1574,  574, 1572,  574, 1574,  572, 1574,  572, 1574,  574, 1576,  568, 504,  570, 1574,  572, 1576,  544, 1602,  570, 1576,  572, 1574,  628, 444,  572, 500,  572, 5164,  4422, 4348,  576, 496,  578, 1570,  576, 496,  578, 1568,  578, 1570,  576, 1570,  550, 524,  548, 1598,  520, 1628,  518, 1630,  516, 1630,  516, 1630,  516, 1628,  518, 1630,  516, 1630,  518, 554,  518, 554,  518, 554,  518, 554,  518, 556,  516, 556,  518, 554,  516, 556,  518, 554,  540, 532,  540, 532,  540, 532,  540, 532,  544, 530,  542, 530,  542, 528,  542, 530,  544, 530,  542, 530,  544, 504,  568, 528,  542, 530,  544, 528,  544, 530,  544, 528,  546, 1602,  544, 530,  540, 532,  540, 532,  574, 498,  544, 528,  572, 1574,  544, 1604,  542};  // MIDEA A201FFFFFF7C
uint64_t data = 0xA201FFFFFF7C;
```

Turn off:
```
Timestamp : 000357.770
Library   : v2.7.8

Protocol  : MIDEA
Code      : 0xA201FFFFFF7C (48 Bits)
Mesg Desc.: Swing(V) Toggle: On
uint16_t rawData[199] = {4440, 4350,  572, 1576,  570, 500,  572, 1574,  572, 502,  572, 498,  572, 500,  566, 1582,  572, 500,  572, 502,  572, 500,  572, 502,  572, 502,  572, 500,  574, 500,  572, 500,  572, 1574,  572, 1574,  572, 1576,  570, 1574,  572, 1574,  572, 1574,  570, 1576,  570, 1576,  570, 1578,  570, 1578,  542, 1600,  572, 1576,  570, 1576,  570, 1576,  570, 1576,  570, 1576,  572, 1574,  572, 1574,  572, 1574,  628, 1518,  550, 1598,  548, 1600,  572, 1574,  576, 1572,  578, 1570,  578, 494,  576, 1568,  550, 1596,  520, 1626,  518, 1628,  516, 1630,  518, 556,  516, 556,  518, 5218,  4364, 4406,  516, 554,  518, 1628,  516, 556,  516, 1630,  544, 1602,  518, 1630,  544, 528,  544, 1602,  544, 1602,  544, 1602,  546, 1600,  546, 1600,  546, 1578,  570, 1600,  546, 1602,  544, 528,  570, 502,  544, 528,  542, 530,  546, 528,  544, 528,  574, 498,  546, 528,  574, 472,  568, 528,  604, 470,  544, 526,  546, 528,  544, 526,  546, 526,  576, 498,  546, 526,  574, 498,  546, 528,  576, 496,  544, 504,  568, 526,  574, 498,  544, 528,  546, 526,  544, 1604,  572, 500,  574, 498,  546, 526,  574, 498,  544, 530,  574, 1572,  542, 1604,  546};  // MIDEA A201FFFFFF7C
uint64_t data = 0xA201FFFFFF7C;
```

Looks like that the command is always the same and must be the ac that store the swing status to know if the swing is on or off right?

paulomanuelp, 2020-08-10:
I found that the issue is in the ir_Midea.cpp file:
```
void IRMideaAC::send(const uint16_t repeat) {
  this->checksum();  // Ensure correct checksum before sending.
  _irsend.sendMidea(remote_state, kMideaBits, repeat);
  // Handle toggling the swing if we need to.
  if (_SwingVToggle && !isSwingVToggle()) {
    _irsend.sendMidea(kMideaACToggleSwingV, kMideaBits, repeat);
  }
  _SwingVToggle = false;  // The toggle message has been sent, so reset.
}
```

I create a new method just to enable the swing:
```
void IRMideaAC::sendSwingVToggle(const uint16_t repeat) {
  this->checksum();  // Ensure correct checksum before sending.
  _irsend.sendMidea(kMideaACToggleSwingV, kMideaBits, repeat);
}
```

And in my method I call like this:
```
IRMideaAC ac(kIrLed); 
  ac.begin();
  ac.setSwingVToggle(true);
  Serial.println(ac.toString());
  ac.sendSwingVToggle();
```
And the swing works well in the ac :)

Dump:
```
Timestamp : 000061.659
Library   : v2.7.8

Protocol  : MIDEA
Code      : 0xA201FFFFFF7C (48 Bits)
Mesg Desc.: Swing(V) Toggle: On
uint16_t rawData[199] = {4530, 4462,  524, 1722,  522, 606,  522, 1726,  520, 606,  520, 606,  522, 604,  568, 1682,  522, 606,  518, 606,  520, 606,  498, 628,  496, 634,  552, 570,  496, 630,  496, 630,  522, 1730,  552, 1690,  520, 1726,  550, 1696,  524, 1722,  498, 1750,  522, 1726,  552, 1692,  552, 1690,  522, 1724,  522, 1726,  520, 1726,  520, 1728,  520, 1726,  496, 1750,  550, 1696,  522, 1726,  518, 1728,  498, 1748,  522, 1726,  550, 1696,  520, 1728,  494, 1750,  522, 1724,  522, 1724,  520, 606,  520, 1726,  494, 1754,  520, 1726,  548, 1698,  522, 1726,  522, 604,  522, 606,  518, 5648,  4430, 4538,  520, 608,  518, 1730,  552, 572,  552, 1688,  496, 1750,  522, 1724,  520, 610,  580, 1660,  496, 1750,  518, 1730,  494, 1750,  520, 1728,  552, 1692,  522, 1726,  520, 1726,  522, 608,  550, 570,  548, 580,  520, 606,  552, 570,  518, 608,  548, 578,  522, 608,  522, 600,  520, 606,  496, 630,  496, 632,  496, 632,  610, 512,  498, 628,  520, 606,  548, 582,  520, 606,  520, 608,  496, 630,  518, 608,  550, 574,  520, 608,  468, 660,  496, 632,  548, 1700,  522, 600,  520, 608,  518, 606,  520, 608,  548, 578,  520, 1728,  520, 1728,  548};  // MIDEA A201FFFFFF7C
uint64_t data = 0xA201FFFFFF7C;
```

crankyoldgit, 2020-08-11:
> It looks like that the the `ac.next.swingv = stdAc::swingv_t::kAuto;` property change is not doing nothing right?

Going back to the above line/code (from https://github.com/crankyoldgit/IRremoteESP8266/issues/1234#issuecomment-671362116)

You are not changing the `swingv` setting at all. It's a toggle. So it will only send a "toggle" message when `next.swingv` changes from non-off -> off, and off -> non-off.
If you don't change `next.swingv` after you send it, it won't generate the special swingVtoggle message.

paulomanuelp, 2020-08-13:
Hello @crankyoldgit ,

Could you give an example how to do it?

Best regards,
Paulo

crankyoldgit, 2020-08-14:
Try something like:
```
IRac ac(kIrLed);  // Create a A/C object using GPIO to sending messages with.

// Set up what we want to send.
// See state_t, opmode_t, fanspeed_t, swingv_t, & swingh_t in IRsend.h for
// all the various options.
ac.next.protocol = decode_type_t::MIDEA;  // Set a protocol to use.
ac.next.model = 1;  // Some A/Cs have different models. Try just the first.
ac.next.mode = stdAc::opmode_t::kCool;  // Run in cool mode initially.
ac.next.celsius = true;  // Use Celsius for temp units. False = Fahrenheit
ac.next.degrees = 25;  // 25 degrees.
ac.next.fanspeed = stdAc::fanspeed_t::kMedium;  // Start the fan at medium.
ac.next.swingv = stdAc::swingv_t::kOff;  // Don't swing the fan up or down.
ac.next.swingh = stdAc::swingh_t::kOff;  // Don't swing the fan left or right.
ac.next.light = false;  // Turn off any LED/Lights/Display that we can.
ac.next.beep = false;  // Turn off any beep from the A/C if we can.
ac.next.econo = false;  // Turn off any economy modes if we can.
ac.next.filter = false;  // Turn off any Ion/Mold/Health filters if we can.
ac.next.turbo = false;  // Don't use any turbo/powerful/etc modes.
ac.next.quiet = false;  // Don't use any quiet/silent/etc modes.
ac.next.sleep = -1;  // Don't set any sleep time or modes.
ac.next.clean = false;  // Turn off any Cleaning options if we can.
ac.next.clock = -1;  // Don't set any current time if we can avoid it.
ac.next.power = false;  // Initially start with the unit off.

ac.send();  // Send the current state.

delay(10000);  // wait 10 seconds.
```