crankyoldgit/IRremoteESP8266 #1236: Various IR remotes do not want to cooperate.

ElizabethMorves, 2020-08-10:
I am trying to happily make friends with several IR controllers and I successfully recognize both remotes of the remotes I have. Since they are different NEC and SONY, I cannot use the corresponding methods. The most suitable for me is to use the raw data `irsend.sendRaw (rawData, 71, 38);`  In my search I am getting close raw data values and same button data every time。

```
uint16_t rawData[71] = {9250, 4474,  614, 556,  614, 556,  614, 556,  614, 556,  616, 556,  616, 554,  616, 554,  616, 554,  616, 1632,  616, 1632,  614, 1636,  616, 1632,  616, 1634,  616, 1634,  614, 1634,  616, 1632,  592, 1658,  616, 554,  616, 554,  616, 1632,  614, 1634,  616, 554,  616, 554,  616, 556,  616, 554,  616, 1632,  614, 1634,  614, 556,  614, 556,  592, 1658,  618, 1632,  616, 1632,  616, 39408,  9262, 2210,  618};  // NEC FF9867
uint32_t address = 0x0;
uint32_t command = 0x19;
uint64_t data = 0xFF9867;

uint16_t rawData[71] = {9224, 4498,  612, 558,  590, 580,  640, 532,  614, 556,  638, 532,  614, 556,  616, 554,  614, 556,  614, 1634,  612, 1636,  614, 1636,  614, 1634,  616, 1634,  614, 1634,  616, 1634,  590, 1658,  614, 1636,  616, 556,  614, 556,  614, 1634,  592, 1658,  590, 580,  614, 556,  614, 556,  616, 556,  614, 1634,  614, 1638,  586, 582,  614, 556,  614, 1634,  614, 1636,  590, 1658,  614, 39412,  9264, 2212,  616};  // NEC FF9867
uint32_t address = 0x0;
uint32_t command = 0x19;
uint64_t data = 0xFF9867;

uint16_t rawData[71] = {9224, 4498,  612, 558,  590, 580,  640, 532,  614, 556,  638, 532,  614, 556,  616, 554,  614, 556,  614, 1634,  612, 1636,  614, 1636,  614, 1634,  616, 1634,  614, 1634,  616, 1634,  590, 1658,  614, 1636,  616, 556,  614, 556,  614, 1634,  592, 1658,  590, 580,  614, 556,  614, 556,  616, 556,  614, 1634,  614, 1638,  586, 582,  614, 556,  614, 1634,  614, 1636,  590, 1658,  614, 39412,  9264, 2212,  616};  // NEC FF9867
uint32_t address = 0x0;
uint32_t command = 0x19;
uint64_t data = 0xFF9867;
```

Oscilloscope test as well as camera test LED passes, but the TV doesn't want to turn on. I tried to play with frequency but it was not successful. 
Sorry if I distract you from important matters。
Maybe I am reading the data incorrectly`setProperty("rx").send(resultToSourceCode(&results));` (Homie) or my ESP32 (WROOM) is not suitable for these tasks?

Thx

crankyoldgit, 2020-08-10:
I'm really not sure what you are saying/trying to do here.

e.g. `setProperty("rx").send(resultToSourceCode(&results));` .. <- I have no idea what this is trying/meant to do.

If the captures you got from `IRrecvDumpV2+` are correct:
e.g.
```
uint16_t rawData[71] = {9224, 4498,  612, 558,  590, 580,  640, 532,  614, 556,  638, 532,  614, 556,  616, 554,  614, 556,  614, 1634,  612, 1636,  614, 1636,  614, 1634,  616, 1634,  614, 1634,  616, 1634,  590, 1658,  614, 1636,  616, 556,  614, 556,  614, 1634,  592, 1658,  590, 580,  614, 556,  614, 556,  616, 556,  614, 1634,  614, 1638,  586, 582,  614, 556,  614, 1634,  614, 1636,  590, 1658,  614, 39412,  9264, 2212,  616};  // NEC FF9867
uint32_t address = 0x0;
uint32_t command = 0x19;
uint64_t data = 0xFF9867;
```

Then all you'd need to do is do a `irsend.sendNEC(0xFF9867);` to send the same command.
e.g. see https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRsendDemo/IRsendDemo.ino

ElizabethMorves, 2020-08-10:
But what about the sony remote control? I don’t have so much space to implement every method of every remote control.
`irsend.sendNEC(0xFF9867);` its works for me. Switch / case too many words. I still try to use raw data.
I will write if I succeed. Sorry for my English

あリがとうございます

crankyoldgit, 2020-08-10:
Use https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples/IRrecvDumpV3 to capture the Sony message(s).
Take the `uint64_t data = ...;` line and the size (in bits) from it's output for the Sony command you want. e.g. `0xF50`, and `12 bits` for a power off command for some Sony devices.
You would use it like: `sendSony(0xF50, 12);`

ElizabethMorves, 2020-08-11:
I figured it out, I was transferring the string incorrectly. Thank you for your assistance. I understand when the code goes over the line. it can be very difficult to figure it out.
the string value arrives from mqtt. ///value = String  **{** 9240, ....... 622  **};**  I forgot the brackets =) sorry.
 [128] redundant for me but I already got the result.
```c++
void IR::tx(String value){
  irrecv.disableIRIn();
   char val_c[128];
   value.toCharArray(val_c, 128);
   irsend.sendRaw((uint16_t *)val_c, 128, 38);
  irrecv.enableIRIn();
}
```
I only use sendRaw because I cannot know which remote will be used tomorrow. I had to look inside your code.

crankyoldgit, 2020-08-11:
okay. Seems you've resolved your own problem, so marking the issue closed.

closed 2020-08-11 by crankyoldgit