crankyoldgit/IRremoteESP8266 #1237: Hi, there

strawmanbobi, 2020-08-13:
Actually, this is not an issue...
Nice to meet you.
I am the author of IRext (https://irext.net) , an open source IR remote control solution project. So late for hearing from you and your great project. I am recently thinking of making some research on AC remote code reverse engineering, and might learn from you later, just kicking off and keep contact.
BTW, if there is something helpful to you with IRext, please feel free to let me know.
Sincerely : )

crankyoldgit, 2020-08-14:
Nice to meet you too!

I had a quick look over your site/code. Nothing leapt out at me that was immediately helpful, but I'll keep it in mind.

strawmanbobi, 2020-08-14:
Well, let's keep in touch. Thanks.

closed 2020-08-14 by strawmanbobi