crankyoldgit/IRremoteESP8266 #1238: Voltas AC Remote IR Mapping

manj9501, 2020-08-14:
Hola @crankyoldgit et al.
I have put in a day's work to try and decode my AC's IR Remote mapping and after necessary following the steps given in the two wiki articles explaining how to do so, I've come up with the following results:
```
Voltas AC map (Model no: 122LZF 4011252 Window AC)
(MSBF)
  
byte 0 
        b0 = Horizontal Swing(1 = On, 0 = Off) (Works with all Basic Modes) (not compatible with my AC)
	b7-1 = UNKNOWN, typically 0b1111100
	(Becomes Don't Care with value 0b00110011(0x33) if any other parameter is changed, reason unknown)

byte 1
	b3-0 = Basic Modes
                1000 (8) = Chill 
                0100 (4) = Dry (fixed Fan speed(Low), fixed Temperature(24C))
                0010 (2) = Cool
		0001 (1) = Fan	(Fixed fan speed(Hi), Temperature = Cool mode temperature)
	b4 = UNKNOWN, typically 0
	b7-5 = Fan speed
		001 (1) = Low
		010 (2) = Mid
		100 (4) = Hi
		111 (7) = Auto
		
byte 2
	b2-0 = Vertical Swing (Works with all Basic Modes)
		000 (1) = Off
		111 (7) = On
	b3 = WiFi bit(Toggles between 0 and 1 upon pressing the button labelled 'WiFi') (not compatible with my AC)
	b4 = UNKNOWN, typically 0
	b5 = Turbo mode(1 = On, 0 = Off) (Works only with Chill mode)
	b6 = Sleep mode(1 = On, 0 = Off) (Works only with Chill mode)
	b7 = Power Status(1 = On, 0 = Off)
	
byte 3
       b3-0 =  Temperature
		0000 (0) = 16C
		0001 (1) = 17C
		0010 (2) = 18C
		...
		1101 (13) = 29C
		1110 (14) = 30C
	b5-4 = UNKNOWN, typically 0b01
	b6 = Energy Saver mode (1 = On, 0 = Off) (Works only with Chill mode)
	b7 = TEMP bit, goes HIGH for a few seconds upon pressing TEMP button, then changes back to LOW automatically and remote sends the message(not compatible with my AC)
	
 byte 4 
	b0 = Off timer 24 Hr. bit
		0 = Time is 24 Hr.
		1 = Otherwise
	b7-1 = UNKNOWN, typically 0b0011101
	
byte 5
       b0 = Off timer 24 Hr. bit
		0 = Time is 24 Hr.
		1 = Otherwise
	b6-1 = UNKNOWN, typically 0b011101
	b7 = 12+(byte 7 upper nibble) bit (works in conjuction with byte 7)
		0 = Time is 24 Hr or Time < 12 Hrs. 
		1 = Time =>12 Hr.
		
byte 6 = UNKNOWN, typically 0b00111011(0x3B)

byte 7 
	b3-0 = UNKNOWN, typically 0b0001
	b7-4 = Timer bits (works in conjuction with byte 7)
		0000 (0) = 0 Hr.
		0001 (1) = 1 Hr.
		0010 (2) = 2 Hr.
		...
		1010 (11) = 11 Hr.
		1011 (12) = 12 Hr.
		
byte 8
	b4-0 = UNKNOWN, typically 0b00000
	b5 = Lamp bit(1 = On, 0 = Off) (not compatible with my AC)
	b6 = Off Timer status (1 = Activated, 0 = Deactivated)
	b7 = UNKNOWN, typically 0
	
byte 9 = Check byte
	Calcuted using the following formula:
	checksum = ~(Sum of all preceding bytes)&0xFF
```
My setup is a NodeMCU connected to an IR receiver I salvaged from a digital photo frame.
[image]

A few of the IRrecvDumpV3 outputs after modifying the library are given below:
1. Dry Mode, No Swing, No Lamp, No Timer
```
Library   : v2.7.9

Protocol  : VOLTAS
Code      : 0x338488183B3B3B1100E6 (80 Bits)
uint16_t rawData[161] = {1002, 584,  1000, 586,  1000, 2568,  1002, 2570,  1002, 586,  998, 588,  1000, 2568,  1002, 2570,  1002, 2572,  1002, 584,  1002, 586,  1000, 584,  1000, 586,  1002, 2568,  1004, 584,  1000, 586,  1002, 2568,  1002, 584,  1002, 584,  1004, 584,  1000, 2568,  1002, 586,  1000, 586,  998, 590,  998, 584,  1002, 584,  1000, 586,  1000, 2570,  1002, 2568,  1004, 584,  1000, 584,  1002, 584,  1002, 582,  1004, 584,  1002, 2568,  1002, 2570,  1004, 2570,  1000, 586,  1002, 2568,  1004, 2568,  1006, 584,  1000, 584,  1002, 2568,  1002, 2570,  1002, 2568,  1002, 586,  1002, 2570,  1000, 2570,  1002, 588,  998, 586,  1000, 2568,  1004, 2568,  1004, 2568,  1002, 588,  998, 2570,  1002, 2568,  1004, 586,  1002, 584,  1000, 586,  1000, 2570,  1000, 586,  1000, 584,  1002, 586,  1000, 2568,  1004, 584,  1000, 586,  1000, 586,  1002, 584,  1002, 586,  1000, 586,  1000, 586,  1000, 586,  1000, 2568,  1002, 2568,  1002, 2568,  1004, 586,  1000, 584,  1000, 2570,  1004, 2568,  1004, 584,  1002};  // VOLTAS
uint8_t state[10] = {0x33, 0x84, 0x88, 0x18, 0x3B, 0x3B, 0x3B, 0x11, 0x00, 0xE6};
```
2. Chill mode, no swing, Fan Auto, Turbo ON, No Timer, No lamp
```
Library   : v2.7.9

Protocol  : VOLTAS
Code      : 0x33E8A8183B3B3B110062 (80 Bits)
uint16_t rawData[161] = {1014, 574,  1012, 574,  1012, 2558,  1014, 2558,  1014, 574,  1014, 574,  1012, 2558,  1014, 2558,  1014, 2558,  1014, 2558,  1014, 2556,  1014, 574,  1012, 2558,  1014, 574,  1012, 574,  1012, 574,  1012, 2558,  1014, 572,  1014, 2558,  1014, 574,  1010, 2558,  1014, 574,  1012, 574,  1012, 574,  1012, 574,  1012, 572,  1014, 572,  1012, 2558,  1014, 2556,  1014, 572,  1036, 550,  1036, 550,  1012, 574,  1012, 576,  1010, 2556,  1038, 2534,  1038, 2534,  1016, 572,  1038, 2530,  1014, 2558,  1038, 548,  1012, 574,  1034, 2534,  1016, 2556,  1038, 2534,  1014, 574,  1012, 2558,  1014, 2558,  1014, 574,  1012, 572,  1012, 2558,  1014, 2556,  1014, 2558,  1014, 574,  1012, 2556,  1038, 2534,  1014, 572,  1012, 574,  1012, 574,  1036, 2534,  1012, 574,  1012, 574,  1012, 574,  1036, 2532,  1016, 572,  1014, 572,  1012, 574,  1012, 574,  1012, 572,  1014, 572,  1012, 574,  1012, 572,  1012, 574,  1012, 2556,  1014, 2556,  1014, 574,  1010, 574,  1012, 574,  1012, 2558,  1014, 572,  1014};  // VOLTAS
uint8_t state[10] = {0x33, 0xE8, 0xA8, 0x18, 0x3B, 0x3B, 0x3B, 0x11, 0x00, 0x62};
```

Also I've built an IR sender circuit and can confirm that the AC is responding to the change in parameters, with the correct checksum only.

Can you please confirm if I'm on the right path, so I can proceed towards the coding part, which admittedly, doesn't happen to be my strongest suit.

Voltas Remote pic:
[image]

crankyoldgit, 2020-08-14:
> Can you please confirm if I'm on the right path, so I can proceed towards the coding part, which admittedly, doesn't happen to be my strongest suit.

Yes, it looks like you're on the correct path. Given you've got a steadily increasing value for the temp in Byte 3, I'd say you've got the bit ordering correct. So, congrats! You're well on the way to having this all figured out, or already have enough to make it completely usable.

Feel free to submit a PR with what you've got so far and we can work on getting it as supported as we can.
As I say in the wiki/faq etc, if you do the decoding/reverse engineering of the protocol, I'll typically do the coding for you.
It looks like you've worked out enough of the protocol (inc the all-important checksum) that we can add support for it fairly simply. However, if you want to do the coding, feel free, I won't stop you. ;-)

Things I'll ask in advance:
1) What is the model number of the remote (if any)?
2) Is there a heat mode on the remote? (if so, what code does it produce)
3) Re: `Byte[0]`, does _"any other parameter is changed"_ apply to the entire IR message, or just `Byte[0]`?
4) In your notation/description, is `b0` the LEAST significant bit, or the MOST?
5) Re: `Byte[3] b7`, does the unit still change the temp if a message has a different temp & `b7` is `0`? i.e. do we have to have this flagged if we change the temp? And how does the unit behave if it was always `1`?
6) Re: `Byte[4 & 5], b0`, Can you elaborate on what you mean here? It isn't immediately obvious.

manj9501, 2020-08-14:
> 1. What is the model number of the remote (if any)?

I've probably misplaced the manual for this AC, so I can't really help you with the model number of the remote right now.

> 2. Is there a heat mode on the remote? (if so, what code does it produce)

No, there is no such mode on the remote.

> 3. Re: Byte[0], does "any other parameter is changed" apply to the entire IR message, or just Byte[0]?

It applies to the entire IR message. Peculiar behavior!

> 4. In your notation/description, is b0 the LEAST significant bit, or the MOST?

b0 is the LSB.

> 5. Re: Byte[3] b7, does the unit still change the temp if a message has a different temp & b7 is 0? i.e. do we have to have this flagged if we change the temp? And how does the unit behave if it was always 1?

If we press the TEMP button, Byte[3] b7 is SET but when we change the temperature on the remote, before letting Byte[3] b7 return to RESET, the remote sends Byte[3] with this flag RESET.  
Also I did some research on this TEMP function. It has to do with temperature display on the AC unit showing the ambient temperature for 5 secs before coming back to showing the temperature set using the remote. But as I mentioned, my AC is a rather dumb one and it doesn't show ambient temperature.
So, I suppose it'll take some more research for me to properly answer this question.

> 6. Re: Byte[4 & 5], b0, Can you elaborate on what you mean here? It isn't immediately obvious.

Following table may make it clear:
[image]

manj9501, 2020-08-14:
Coming back to Byte[3] b7, I did some testing by modifying the IRsendDemo. 
I hard coded this bit to SET and played around with the temperature bits.
What I noticed was that the temperature was definitely being updated on my AC unit, despite the bit being SET all the time.

crankyoldgit, 2020-08-15:
> I've probably misplaced the manual for this AC, so I can't really help you with the model number of the remote right now.

No worries. If there isn't one printed on the remote (sometimes in the battery compartment) then we will just do without. ;-)

manj9501, 2020-08-15:
I've looked in the battery compartment too, though without any breakthroughs. 
Once I lay my hands on the manual and the remote model no. does happen to be there, I'll definitely revert back.

Plus, I hope I was able to clear your previous doubts with regards to the state mapping.

crankyoldgit, 2020-08-16:
> I've looked in the battery compartment too, though without any breakthroughs.
> Once I lay my hands on the manual and the remote model no. does happen to be there, I'll definitely revert back.

No rush & not urgent.

> Plus, I hope I was able to clear your previous doubts with regards to the state mapping.

Yep. All good. I won't know more till I start coding. Speaking of which, can you please send what you've got working so far? Preferably in the form of a Pull Request (PR). Once I have it I'll start working on it.

crankyoldgit, 2020-08-16:
Now I've cleaned up & merged your PR, I'll start on the detailed support for the protocol.

crankyoldgit, 2020-08-16:
@manj9501 I'm adding detailed support in branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/detailed_voltas
Feel free to follow along & download + test while I add to it. It's far from finished at the moment, so don't expect much.

manj9501, 2020-08-16:
> Feel free to follow along & download + test while I add to it.

Yes, doing it now!

manj9501, 2020-08-16:
No sooner did I upload the IRrecvDumpV3 example from _detailed_voltas_ branch on to my NodeMCU setup than I discovered the significance of the 'don't expect much' part of your statement.

I encountered a rapidly crashing NodeMCU with the following Serial Log:

```
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Panic IRrecvDumpV3.ino:131 void setup(): Assertion 'irutils::lowLevelSanityCheck()' failed.

>>>stack>>>

ctx: cont
sp: 3fffff30 end: 3fffffc0 offset: 0000
3fffff30:  0000001c 0001c200 00000000 00000000  
3fffff40:  000000fe 00000000 00000000 00000000  
3fffff50:  00000000 00000000 00000000 4024fa30  
3fffff60:  3ffeee94 00000000 3ffeed38 3ffeee54  
3fffff70:  3fffdad0 00000000 3ffeed38 40216f36  
3fffff80:  0001c200 0000001c 00000002 40216f6a  
3fffff90:  12345678 00000000 3ffeed38 40201070  
3fffffa0:  feefeffe feefeffe 3ffeee14 40216af0  
3fffffb0:  feefeffe feefeffe 3ffe84e4 40100f79  
<<<stack<<<

```

crankyoldgit, 2020-08-16:
Bugger. Please comment out the `assert` line at IRrecvDumpV3.ino:131 and try again.

manj9501, 2020-08-16:
Commenting that line out definitely helped and I can confirm that the basic functionality is working as expected.

Attaching some samples of the decoded signals below:

> Protocol  : VOLTAS
> Code      : 0x3328A01C08083B41005C (80 Bits)
> Mesg Desc.: Power: On, Turbo: On, WiFi: Off, Light: Off
> uint16_t rawData[161] = {1014, 574,  1012, 574,  1012, 2556,  1036, 2534,  1038, 548,  1012, 574,  1012, 2558,  1014, 2556,  1014, 574,  1012, 574,  1012, 2558,  1014, 574,  1012, 2558,  1014, 574,  1012, 574,  1012, 572,  1012, 2556,  1014, 574,  1036, 2534,  1014, 574,  1012, 574,  1012, 574,  1012, 574,  1012, 574,  1012, 574,  1012, 574,  1012, 574,  1012, 2558,  1014, 2556,  1038, 2534,  1014, 574,  1012, 574,  1012, 574,  1012, 574,  1012, 574,  1012, 574,  1012, 2558,  1014, 574,  1012, 574,  1010, 574,  1012, 574,  1012, 574,  1012, 574,  1012, 574,  1012, 2558,  1014, 574,  1012, 574,  1012, 572,  1012, 574,  1012, 574,  1012, 2558,  1012, 2558,  1014, 2558,  1014, 572,  1012, 2558,  1014, 2558,  1012, 574,  1012, 2558,  1014, 574,  1012, 574,  1012, 574,  1012, 574,  1012, 574,  1012, 2556,  1014, 574,  1012, 574,  1012, 574,  1012, 574,  1012, 574,  1014, 572,  1012, 574,  1012, 574,  1012, 574,  1012, 2558,  1012, 574,  1012, 2558,  1012, 2558,  1012, 2558,  1014, 574,  1038, 550,  1010};  // VOLTAS
> uint8_t state[10] = {0x33, 0x28, 0xA0, 0x1C, 0x08, 0x08, 0x3B, 0x41, 0x00, 0x5C};
> 
> 
> Timestamp : 000066.157
> Library   : v2.7.9
> 
> Protocol  : VOLTAS
> Code      : 0x3328A81C08083B410054 (80 Bits)
> Mesg Desc.: Power: On, Turbo: On, WiFi: On, Light: Off
> uint16_t rawData[161] = {1012, 576,  1012, 572,  1014, 2558,  1012, 2558,  1014, 574,  1012, 576,  1010, 2558,  1014, 2558,  1012, 576,  1010, 574,  1012, 2558,  1014, 574,  1012, 2558,  1014, 574,  1010, 574,  1012, 574,  1010, 2558,  1014, 574,  1012, 2558,  1038, 550,  1010, 2558,  1014, 574,  1012, 574,  1012, 574,  1010, 574,  1012, 574,  1012, 574,  1036, 2534,  1014, 2558,  1012, 2558,  1014, 574,  1010, 574,  1012, 574,  1010, 574,  1010, 576,  1012, 574,  1010, 2558,  1014, 574,  1012, 576,  1012, 574,  1012, 574,  1012, 574,  1012, 574,  1010, 574,  1012, 2558,  1036, 550,  1012, 574,  1012, 574,  1012, 574,  1012, 574,  1012, 2558,  1012, 2558,  1014, 2560,  1036, 550,  1036, 2534,  1012, 2558,  1012, 576,  1010, 2558,  1036, 552,  1012, 574,  1010, 574,  1012, 574,  1012, 574,  1012, 2558,  1012, 574,  1010, 574,  1010, 574,  1010, 576,  1010, 574,  1010, 574,  1036, 550,  1012, 574,  1010, 574,  1012, 2558,  1012, 574,  1012, 2556,  1012, 576,  1010, 2558,  1012, 576,  1010, 574,  1012};  // VOLTAS
> uint8_t state[10] = {0x33, 0x28, 0xA8, 0x1C, 0x08, 0x08, 0x3B, 0x41, 0x00, 0x54};
> 
> 
> Timestamp : 000067.093
> Library   : v2.7.9
> 
> Protocol  : VOLTAS
> Code      : 0x3328881C08083B410074 (80 Bits)
> Mesg Desc.: Power: On, Turbo: Off, WiFi: On, Light: Off
> uint16_t rawData[161] = {1012, 576,  1010, 574,  1012, 2558,  1012, 2558,  1034, 552,  1010, 574,  1012, 2558,  1012, 2558,  1012, 574,  1012, 574,  1036, 2534,  1014, 574,  1010, 2558,  1012, 574,  1012, 574,  1012, 574,  1012, 2558,  1036, 550,  1010, 574,  1010, 574,  1034, 2534,  1014, 574,  1010, 574,  1010, 574,  1012, 574,  1010, 574,  1012, 574,  1010, 2558,  1014, 2558,  1012, 2558,  1012, 574,  1010, 574,  1012, 574,  1012, 574,  1012, 574,  1012, 574,  1010, 2558,  1014, 574,  1010, 576,  1010, 574,  1010, 574,  1010, 574,  1010, 576,  1010, 574,  1012, 2558,  1014, 574,  1012, 574,  1010, 574,  1010, 574,  1012, 574,  1010, 2558,  1012, 2558,  1012, 2558,  1014, 574,  1012, 2558,  1012, 2558,  1012, 574,  1012, 2558,  1014, 574,  1012, 574,  1012, 574,  1012, 574,  1012, 574,  1012, 2558,  1012, 574,  1012, 574,  1012, 574,  1012, 572,  1012, 574,  1012, 574,  1010, 574,  1012, 574,  1010, 574,  1012, 2558,  1012, 2558,  1012, 2556,  1014, 574,  1010, 2560,  1012, 574,  1012, 574,  1012};  // VOLTAS
> uint8_t state[10] = {0x33, 0x28, 0x88, 0x1C, 0x08, 0x08, 0x3B, 0x41, 0x00, 0x74};
> 
> 
> Timestamp : 000075.465
> Library   : v2.7.9
> 
> Protocol  : VOLTAS
> Code      : 0x3328801C08083B41007C (80 Bits)
> Mesg Desc.: Power: On, Turbo: Off, WiFi: Off, Light: Off
> uint16_t rawData[161] = {1014, 574,  1012, 574,  1012, 2556,  1014, 2556,  1014, 574,  1036, 548,  1012, 2558,  1038, 2534,  1014, 572,  1034, 552,  1010, 2558,  1014, 574,  1014, 2556,  1014, 574,  1012, 574,  1012, 572,  1012, 2558,  1014, 572,  1014, 572,  1012, 574,  1012, 574,  1012, 572,  1012, 574,  1010, 574,  1036, 550,  1012, 572,  1012, 574,  1012, 2556,  1014, 2556,  1038, 2532,  1012, 574,  1012, 572,  1038, 550,  1010, 574,  1012, 574,  1012, 574,  1012, 2556,  1012, 574,  1012, 574,  1034, 552,  1010, 574,  1038, 548,  1012, 574,  1036, 550,  1036, 2534,  1012, 574,  1036, 550,  1012, 574,  1012, 574,  1012, 574,  1010, 2558,  1038, 2534,  1012, 2556,  1040, 548,  1038, 2532,  1014, 2558,  1014, 574,  1012, 2558,  1014, 572,  1012, 574,  1038, 550,  1012, 572,  1012, 574,  1012, 2556,  1040, 548,  1038, 548,  1014, 572,  1014, 574,  1012, 574,  1012, 574,  1012, 574,  1036, 550,  1010, 574,  1038, 2532,  1038, 2532,  1016, 2556,  1014, 2558,  1014, 2558,  1014, 572,  1014, 572,  1012};  // VOLTAS
> uint8_t state[10] = {0x33, 0x28, 0x80, 0x1C, 0x08, 0x08, 0x3B, 0x41, 0x00, 0x7C};
> 
> 
> Timestamp : 000080.964
> Library   : v2.7.9
> 
> Protocol  : VOLTAS
> Code      : 0x3328001C08083B4100FC (80 Bits)
> Mesg Desc.: Power: Off, Turbo: Off, WiFi: Off, Light: Off
[... cut ...]   1038, 550,  1012, 572,  1012, 2558,  1038, 2532,  1014, 2558,  1014, 574,  1012, 2556,  1038, 2532,  1014, 572,  1012, 2558,  1038, 550,  1012, 574,  1012, 574,  1012, 574,  1038, 548,  1038, 2530,  1014, 574,  1012, 574,  1036, 550,  1012, 574,  1012, 572,  1012, 574,  1012, 574,  1012, 574,  1012, 2558,  1014, 2558,  1012, 2558,  1014, 2556,  1014, 2558,  1014, 2558,  1014, 572,  1012, 574,  1010};  // VOLTAS
> uint8_t state[10] = {0x33, 0x28, 0x00, 0x1C, 0x08, 0x08, 0x3B, 0x41, 0x00, 0xFC};

crankyoldgit, 2020-08-16:
Let me know if the descriptions don't match at any time.
Thanks for confirming.
Sorry about the bug.

manj9501, 2020-08-16:
Sure, I'll keep you updated! 
Also, I tested the bugfix of PR #1245, it makes things hunky-dory again!

crankyoldgit, 2020-08-16:
> Also, I tested the bugfix of PR #1245, it makes things hunky-dory again!

Thanks for confirming!

crankyoldgit, 2020-08-17:
@manj9501 What's the difference between "Chill" and "Cool" operation modes?
Normally we have Auto, Cool, Heat, Dry, & Fan as the typical modes.

manj9501, 2020-08-17:
Turns out I was wrong! I did some research and well, the mode I've mistaken for 'Cool' all this while is actually the Heat mode. My bad! 
Guess I should have looked for the manual before embarking on this journey.
Also, since my place gets plenty hot every summer season, it was kind of hard for me to see why anybody here would need a Heat mode. 
I made this mistake even though Heat mode has a Sun as its icon, how so parochial of me!
I wanna reiterate that Heat mode is just the Chill mode without the Turbo, Energy Saver and Sleep functionalities.

crankyoldgit, 2020-08-17:
So, are these values correct then for the modes?
```
const uint8_t kVoltasFan   = 0b0001;  ///< 1
const uint8_t kVoltasHeat  = 0b0010;  ///< 2
const uint8_t kVoltasDry   = 0b0100;  ///< 4
const uint8_t kVoltasCool  = 0b1000;  ///< 8
```

manj9501, 2020-08-17:
Yes, these are the right values.

crankyoldgit, 2020-08-17:
Thanks, your reward is a new update to the branch with more things to test/confirm working etc.
e.g. temp, mode, fan, ...

manj9501, 2020-08-17:
On it, mate!

manj9501, 2020-08-17:
Apart from the fact that the IRrecvDumpV3 is calling Hi fan speed Low and vice versa, everything looks good!

Additionally, I just noticed that the Eco option in the Chill mode makes the temperature fixed at 24C and the fan speed fixed at Mid, until Eco is turned off.
Also, if the A/C was turned off from the remote with the Eco ON, Eco will still be ON, the next time A/C is switched on from the remote.

crankyoldgit, 2020-08-17:
Re: Fan.

That's interesting, I'm using the same values as your initial analysis.
i.e.
```
const uint8_t kVoltasFanLow =  0b001;  ///< 1
const uint8_t kVoltasFanMed =  0b010;  ///< 2
const uint8_t kVoltasFanHigh = 0b100;  ///< 4
const uint8_t kVoltasFanAuto = 0b111;  ///< 7
```

Is your initial analysis incorrect (and my values), or are the numeric numbers being displayed correct, but the text is wrong?

crankyoldgit, 2020-08-17:
> Eco option in the Chill mode

That's `Cool` mode, right?

manj9501, 2020-08-17:
There's a flaw in my initial analysis, I just realized.

Following is the correct mapping:
const uint8_t kVoltasFanHigh =  0b001;  ///< 1
const uint8_t kVoltasFanMed =  0b010;  ///< 2
const uint8_t kVoltasFanLow = 0b100;  ///< 4
const uint8_t kVoltasFanAuto = 0b111;  ///< 7

> That's Cool mode, right?

Yes, it's the Cool mode

crankyoldgit, 2020-08-17:
Oh, and can you please supply me with the hex code (i.e. 20 hex digits) for a message that has **everything** off?
I want it for the _base_ / initial state of the remote when it's reset.

manj9501, 2020-08-17:
> I want it for the base / initial state of the remote when it's reset.

You mean after the batteries have been removed for a sufficiently long time, right?

manj9501, 2020-08-17:
Well, if that's what you meant, here are the 20 hex digits: 332880173B3B3B11004B
Desc: Power: On, Mode: 8 (Cool), Temp: 23C, Fan: 1 (High), Turbo: Off, Econo: Off, WiFi: Off, Light: Off

crankyoldgit, 2020-08-17:
> > I want it for the base / initial state of the remote when it's reset.
> 
> You mean after the batteries have been removed for a sufficiently long time, right?

No. This is for when the library resets the state/creates it's own initial state.

> Desc: Power: On, Mode: 8 (Cool), Temp: 23C, Fan: 1 (High), Turbo: Off, Econo: Off, WiFi: Off, Light: Off

Power off too please.

crankyoldgit, 2020-08-17:
> I want it for the base / initial state of the remote when it's reset.

I just re-read that .. yeah, I wrote that line rather badly. :-(

manj9501, 2020-08-17:
> Power off too please.

0x332800173B3B3B1100CB
Power: Off, Mode: 8 (Cool), Temp: 23C, Fan: 1 (High), Turbo: Off, Econo: Off, WiFi: Off, Light: Off

crankyoldgit, 2020-08-17:
There ya go. Some of the issues fixed and more support (swings).
Download and test away.

manj9501, 2020-08-17:
I haven't tested the sending capabilities of this branch. Was I supposed to test it along with the receive part?

crankyoldgit, 2020-08-17:
With the latest update, sending via the class(es) (and IRMQTTServer etc) might work now, but I wouldn't hold your breath.
i.e. Feed back welcome etc.

manj9501, 2020-08-17:
Something is amiss with the logic of `Swing(H)`. 
It was switched OFF in the first message given below (byte0=F8). But in the second message, it's status is changed to ON, even though it was the fan mode, that was changed from High to Auto.

> Timestamp : 000099.092
> Library   : v2.7.9
> 
> Protocol  : VOLTAS
> Code      : 0xF828801A3B3B3B110083 (80 Bits)
> Mesg Desc.: Power: On, Mode: 8 (Cool), Temp: 26C, Fan: 1 (High), Swing(V): Off, Swing(H): Off, Turbo: Off, Econo: Off, WiFi: Off, Light: Off
> uint8_t state[10] = {0xF8, 0x28, 0x80, 0x1A, 0x3B, 0x3B, 0x3B, 0x11, 0x00, 0x83};
> 
> 
> Timestamp : 000107.755
> Library   : v2.7.9
> 
> Protocol  : VOLTAS
> Code      : 0x33E8801A3B3B3B110088 (80 Bits)
> Mesg Desc.: Power: On, Mode: 8 (Cool), Temp: 26C, Fan: 7 (Auto), Swing(V): Off, Swing(H): On, Turbo: Off, Econo: Off, WiFi: Off, Light: Off
> uint8_t state[10] = {0x33, 0xE8, 0x80, 0x1A, 0x3B, 0x3B, 0x3B, 0x11, 0x00, 0x88};

This behavior is shown every time we change any parameter after switching the `Swing(H)` OFF

manj9501, 2020-08-17:
Yep, something's definitely not right with `Swing(H)`. 
I modified _TurnOnArgoAc.ino_ to check if correct values are being sent via the library:
Below are the settings I coded:
```
  ac.setPower(true);
  ac.setMode(kVoltasDry);
  ac.setSwingH(false);
```
Now since I use the setSwingH(false) option, the expected data in byte0 should be 0x78, but here's the actual data I received on my receiver setup:

> Timestamp : 001217.563
> Library   : v2.7.9
> 
> Protocol  : VOLTAS
> Code      : 0x328480183B3B3B1100EF (80 Bits)
> Mesg Desc.: Power: On, Mode: 4 (Dry), Temp: 24C, Fan: 4 (Low), Swing(V): Off, Swing(H): Off, Turbo: Off, Econo: Off, WiFi: Off, Light: Off
> uint8_t state[10] = {0x32, 0x84, 0x80, 0x18, 0x3B, 0x3B, 0x3B, 0x11, 0x00, 0xEF};

As you can see, byte0 is 0x32, which is anomalous.

crankyoldgit, 2020-08-17:
> byte 0 
>         b0 = Horizontal Swing(1 = On, 0 = Off) (Works with all Basic Modes) (not compatible with my AC)
> 	b7-1 = UNKNOWN, typically 0b1111100
> 	(Becomes Don't Care with value 0b00110011(0x33) if any other parameter is changed, reason unknown)

Implemented as:
```
    // Byte 0
    uint8_t SwingH         :1;
    uint8_t Unknown0       :7;
```

i.e. the LSB of Byte[0] is the state of the SwingH.
Or have I gotten this understanding wrong?
e.g. Or is it that if and only the "top" 7 bits are `0b1111100`, then `b0` sets the SwingH value on or off.
Otherwise it's normally `0x33` in Byte[0]. (We have no idea what the state of SwingH is)

Confused.

manj9501, 2020-08-17:
> Or is it that if and only if the "top" 7 bits are 0b1111100, then b0 sets the SwingH value on or off.
> Otherwise it's normally 0x33 in Byte[0]. (We have no idea what the state of SwingH is)

I feel this describes it better. This would mean introduction of a third 'Unknown' state for SwingH.

Other simpler solution would be to send only either ON(0x78) or OFF(0x79) values for SwingH as byte[0], omitting 0x33 altogether.

crankyoldgit, 2020-08-17:
Thanks for the confirmation. Yes, that would work for sending but for receiving ... Gah! I hate A/C vendors some times. ;-)

I'll have to ponder this for a while to think of how to implement this best.

manj9501, 2020-08-23:
Well, I reckon I haven't been thorough enough at several places during the reverse engineering phase and I guess that's where I've come a gutser ;-)

First off, it was testing whether my AC unit responds to all the buttons on its remote.
Normally, whenever a button is pressed on the remote, the AC responds by beeping.
As it turns out, the AC unit doesn't respond to the Horizontal Swing button at all, even though it responds to several other incompatible functions/buttons such as 'Light', 'Temp', 'WiFi'.

Now to validate the behaviour I've been observing, i.e. the non-responsiveness of the AC unit upon using `setSwingH(true/false)`, I edited the _IRsendDemo.ino_ example and sent the following array of states:
`{0xF9, 0x28, 0x80, 0x1A, 0x18, 0x18, 0x3B, 0x31, 0x00, 0xA8}`
Lo and behold, my AC unit turned a blind eye to this command and didn't respond, even as my receiver setup recognized it.

I then sent the following array of states:
`{0xF8, 0x28, 0x80, 0x1A, 0x18, 0x18, 0x3B, 0x31, 0x00, 0xA9}`
As was the case previously, there was no response.

Finally, to piece together this puzzle completely, I sent the following array
`{0x33, 0x28, 0x80, 0x1A, 0x18, 0x18, 0x3B, 0x31, 0x00, 0x6E}`
and the AC responded.
So, I suppose that's sorted and the code may be working fine too.

My second mistake was not realizing that ON timer would only work when the AC has been turned OFF from the remote.
Yes, you read that right, there's also ON timer functionality. 
I've been pressing the ON timer button when the AC was turned ON from the remote, all this while. What a bummer :-/
I'll update its bit mapping soon.

My sincere apologies for having wasted your precious time, owing to my silly mistakes!

> Oh, also, you did update your sending ESP with the latest version of the branch didn't you?

Yes, I did. That's what solved the issue with `setLight(true)`.

manj9501, 2020-08-23:
I've updated the original mapping to include the ON Timer bit mapping.
Also following image may help:
[image]

> That clearer?

Your explanation is what made me retrace the steps I had followed till then to see what's wrong, so thanks a ton!

crankyoldgit, 2020-08-24:
> As it turns out, the AC unit doesn't respond to the Horizontal Swing button at all, even though it responds to several other incompatible functions/buttons such as 'Light', 'Temp', 'WiFi'.

Okay. Given that info/feedback, what I might do is implement a [state_t::model](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/structstdAc_1_1state__t.html#aa1a57a63b2ea80c1f9c4a1bcf16a4c62) for the class, similar to other protocols/remotes.
That will mean we can lock the swingh functionality behind a different model (or disabled via a specific model) so the class can/can't produce those messages by accident.

Re: Timer stuff
I'll get to it soon, but Timers are at the [low end](https://github.com/crankyoldgit/IRremoteESP8266/issues/1229#issuecomment-677481870) of the priority list.

manj9501, 2020-08-24:
> Given that info/feedback, what I might do is implement a `state_t::model` for the class, similar to other protocols/remotes.

That sounds just about right! Plus I think it'll save you some work if in the future, somebody else reverse engineers some different Voltas model with a similar bit mapping. 

> I'll get to it soon, but Timers are at the low end of the priority list.

I went through the reasons you listed for this, and I couldn't agree more. I myself was thinking along the same lines after I first read this part of your comment.

crankyoldgit, 2020-08-24:
Okay, model support should now be added/working.
basically, you need to:
```
ac.setModel(voltas_ac_remote_model_t::kVoltas122LZF);  // 1
```
and it should disable control of SwingH/prevent it from producing a bad (not recognised) code on your A/C.
```
ac.setModel(voltas_ac_remote_model_t::kVoltasUnknown);  // 0
```
should _enable_ SwingH support.

`kVoltas122LZF`/`1` is the default model.

It should also detect (when it can) if SwingH is being changed on a real remote, and report it as the unknown/`0` model.

Let me know how that goes for you.

manj9501, 2020-08-24:
After testing both the sending and the receiving aspects, I must say that the latest commit is pretty close to what we can call a real success. The model is being decoded correctly, and `setSwingH()` is not sending undesired values without explicitly setting the model to `kVoltasUnknown` first.

Also, there's two more things I missed out on earlier:
1. A constraint related to the Econo option in the Cool mode, which fixes the temperature to 24C and the fan speed to Mid., until the Econo option is deactivated
2. Fan speeds in the Fan mode. Fan mode actually has 3 fan speed settings: Hi, Mid, Lo.

crankyoldgit, 2020-08-25:
> 1. A constraint related to the Econo option in the Cool mode, which fixes the temperature to 24C and the fan speed to Mid., until the Econo option is deactivated

Just confirming, if you're in econo option on etc, you can't choose any other mode? Or if you chose another mode, it turns off econo? (I may implement it as the latter for coding convenience, i.e. econo is only allowed if you're in cool mode, otherwise it's off.)

> 2. Fan speeds in the Fan mode. Fan mode actually has 3 fan speed settings: Hi, Mid, Lo.

So no Auto fan speed mode. Correct?

crankyoldgit, 2020-08-25:
> Fan speeds in the Fan mode. Fan mode actually has 3 fan speed settings: Hi, Mid, Lo.

&

> ```
> 	b3-0 = Basic Modes
>                 1000 (8) = Chill 
>                 0100 (4) = Dry (fixed Fan speed(Low), fixed Temperature(24C))
>                 0010 (2) = Cool
> 		0001 (1) = Fan	(Fixed fan speed(Hi), Temperature = Cool mode temperature)
> ```

Seem at odds with each other. Please clarify which is correct.

crankyoldgit, 2020-08-25:
> My second mistake was not realizing that ON timer would only work when the AC has been turned OFF from the remote.
> Yes, you read that right, there's also ON timer functionality.
> I've been pressing [...]

& 

> I've updated the original mapping to include the ON Timer bit mapping.
> Also following image may help:
> [image]

I'm confused.

How do you tell which timer is active in a message? i.e. the On or the Off timer? Is it "It's an On timer if the Power bit is `0`, and an Off timer if the Power bit is `1`?

Byte[4] `b0` & Byte[5] `b0` seem to both track/mean "The timer is enabled", that's fine. Also if Byte[7] `b3-0` is `0b0000` then it's disabled too.

Byte[4] `b7` tracks the >=12h for the On Timer, and similarly Byte[5] `b7` for the Off Timer.

[Q] If the On timer is engaged, what is _Byte[5] `b7`_'s value during all this? `0`?
Similarly, if the Off timer is  is engaged, what is _Byte[4] `b7`_'s value during all this? `0`?

crankyoldgit, 2020-08-25:
Ah ... I just noticed the Byte[7] bit ranges are different. D'oh!

crankyoldgit, 2020-08-25:
I have a theory about `Byte[4] b0` & `Byte[5] b0`. I think as there are two of them, one is "Off Timer is enabled" & the other "On Timer is enabled".
With your new knowledge, can you please contrive a remote state where _one & only one_ of the timers is active with the remote, to see if those bits are particular to a specific timer?

crankyoldgit, 2020-08-25:
First draft of timer support added. Please test.
After implementing it, I'd wager that `Byte[4] b0` will be "on timer enabled" & `Byte[5] b0` will be "off timer enabled".

manj9501, 2020-08-25:
As I originally stated:
`b6 = Energy Saver mode (1 = On, 0 = Off) (Works only with Chill mode)`
So

> i.e. econo is only allowed if you're in cool mode, otherwise it's off.

is definitely the correct implementation.

> Or if you chose another mode, it turns off econo?

Obviously, yes!

manj9501, 2020-08-25:
> Seem at odds with each other. Please clarify which is correct.

As I mentioned, this happens to be something I didn't notice until yesterday. So

> Fan speeds in the Fan mode. Fan mode actually has 3 fan speed settings: Hi, Mid, Lo.

is the correct one.

crankyoldgit, 2020-08-25:
> > Fan speeds in the Fan mode. Fan mode actually has 3 fan speed settings: Hi, Mid, Lo.
> 
> is the correct one.

Thanks, I'll implement that now.

manj9501, 2020-08-25:
> It's an On timer if the Power bit is 0, and an Off timer if the Power bit is 1?

That may actually be the case. Plus, I had also updated `Byte8` in the mapping, please take a look.

crankyoldgit, 2020-08-25:
Fan speed limitation in Fan mode added.

crankyoldgit, 2020-08-25:
> > It's an On timer if the Power bit is 0, and an Off timer if the Power bit is 1?
> 
> That may actually be the case. Plus, I had also updated `Byte8` to include ON Timer activation status in the mapping, please take a look.

Done/updated accordingly.

manj9501, 2020-08-25:
> With your new knowledge, can you please contrive a remote state where one & only one of the timers is active with the remote, to see if those bits are particular to a specific timer? i.e. to confirm / disprove my theory

Do I still need to supply this information?

crankyoldgit, 2020-08-25:
> > With your new knowledge, can you please contrive a remote state where one & only one of the timers is active with the remote, to see if those bits are particular to a specific timer? i.e. to confirm / disprove my theory
> 
> Do I still need to supply this information?

If you can find proof those two bits can be different from each other, yes please.

manj9501, 2020-08-25:
> If you can find proof those two bits can be different from each other, yes please.

I'll keep an eye out while testing.

[39 more replies]

PR #1248 (2020-08-28): Voltas: Add detailed support for Voltas A/Cs
* Add `IRVoltas` class to handle detailed setting support.
  - Model
  - Power
  - Mode
  - Fan Speed
  - Temperature
  - Turbo (Cool Mode only)
  - Econo (Cool Mode only)
  - Sleep (Cool Mode only)
  - Wifi
  - Light/Lamp
  - SwingV
  - SwingH (Unavailable for Model 122LZF)
  - Off Timer
  - On Timer
* Add support for `VOLTAS` to Common A/C API (`IRac`)
* Unit tests.
* Tested against real received messages, and sending messages to a real device.

Fixes #1238
commit ea792470 (2020-08-28): Voltas: Add detailed support for Voltas A/Cs (#1248)

* Add `IRVoltas` class to handle detailed setting support.
  - Model
  - Power
  - Mode
  - Fan Speed
  - Temperature
  - Turbo (Cool Mode only)
  - Econo (Cool Mode only)
  - Sleep (Cool Mode only)
  - Wifi
  - Light/Lamp
  - SwingV
  - SwingH (Unavailable for Model 122LZF)
  - Off Timer
  - On Timer
* Add support for `VOLTAS` to Common A/C API (`IRac`)
* Unit tests.
* Tested against real received messages, and sending messages to a real device.

Fixes #1238
closed 2020-08-28 by crankyoldgit