crankyoldgit/IRremoteESP8266 #1239: MQTT example pronto code from source code not working

anduz, 2020-08-14:
### Version/revision of the library used
2.7.9

### Expected behavior
Use pronto code through MQTT (mosquitto) with the following code:
25,R1,0000,0081,0000,0030,000E,0027,0028,000C,0028,000C,000E,0026,000E,0026,0028,000C,000E,0026,000E,0026,000E,0026,0028,000C,0028,000C,000E,0026,0028,000C,0028,000C,0028,000C,000E,0026,000E,0026,000E,0026,000E,0026,000E,0026,0028,000C,0028,000C,000E,0026,000E,0284,000E,0026,0028,000C,0028,000C,000E,0026,000E,0026,0028,000C,000E,0026,000E,0026,000E,0026,0028,000C,0028,000C,000E,0026,0028,000C,0028,000C,0028,000C,000E,0026,000E,0026,000E,0026,000E,0026,000E,0026,0028,000C,0028,000C,000E,0026,000E,122E

Even the MQTT message provided with the example code is not working:
25,R1,0000,006E,0022,0002,0155,00AA,0015,0040,0015,0040,0015,0015,0015,0015,0015,0015,0015,0015,0015,0015,0015,0040,0015,0040,0015,0015,0015,0040,0015,0015,0015,0015,0015,0015,0015,0040,0015,0015,0015,0015,0015,0040,0015,0040,0015,0015,0015,0015,0015,0015,0015,0015,0015,0015,0015,0040,0015,0015,0015,0015,0015,0040,0015,0040,0015,0040,0015,0040,0015,0040,0015,0640,0155,0055,0015,0E40

This example code is working as a MQTT message:
30,38000,9000,4500,500,1500,500,750,500,750

Using HTTP is also working:
http://<deviceip>/ir?repeats=1&type=25&code=0000,0081,0000,0030,000E,0027,000E,0026,0028,000C,000E,0026,000E,0026,0028,000C,000E,0026,000E,0026,000E,0026,0028,000C,0028,000C,000E,0026,0028,000C,0028,000C,0028,000C,000E,0026,000E,0026,000E,0026,000E,0026,000E,0026,0028,000C,000E,0026,000E,0026,0028,026A,000E,0026,000E,0026,0028,000C,000E,0026,000E,0026,0028,000C,000E,0026,000E,0026,000E,0026,0028,000C,0028,000C,000E,0026,0028,000C,0028,000C,0028,000C,000E,0026,000E,0026,000E,0026,000E,0026,000E,0026,0028,000C,000E,0026,000E,0026,0028,1211

Tried also this:
#define MQTT_MAX_PACKET_SIZE 4096

### Actual behavior
IR message is not sent.

### Steps to reproduce the behavior
_What can we do to (pref. reliably) repeat what is happening?_

#### Example code used

### Has this library/code previously worked as expected for you?
Not tried before

### Other useful information
Board: D1 mini

crankyoldgit, 2020-08-15:
It is highly likely you have built(compiled) the libraries & example incorrectly. I suggest you use the suggested value of `768` and that you make that modification in the pubsubclient library.
See:
https://github.com/crankyoldgit/IRremoteESP8266/blob/a6cd2acb345aa93a8fc71038683eeb240a9e1792/examples/IRMQTTServer/IRMQTTServer.ino#L38

anduz, 2020-08-15:
It is working now after changing the value directly in pubsubclient.h.
Thank you very much for the help, it was cleary written in the source code
.. .sorry

crankyoldgit, 2020-08-16:
No worries, glad it was a simple fix for you.

closed 2020-08-16 by crankyoldgit