crankyoldgit/IRremoteESP8266 #1241: Metz IR protocol

mtdcr, 2020-08-15:
As of v2.7.9, IRremoteESP8266 doesn't support the protocol used by [Metz](https://metz-ce.de/) for their television sets.

I'd like to implement this protocol, but I can't flash my IR bridge (NX-4519-675) easily. It currently runs Tasmota, which was installed using tuya-convert.

Can anybody recommend a cheap, ready-to-flash-via-serial IR bridge for this task? It doesn't need to look nice, but I'd prefer something that doesn't require any soldering skills.

In the meantime, I'd like to use this issue to collect knowledge about the protocol.

Here's a log from Tasmota 8.4.0 (using a snapshot of IRremoteESP8266 post-2.7.8) with a RM17 remote control unit:
```json
{"IrReceived":{"Protocol":"UNKNOWN","Bits":21,"Hash":"0xB109E9C0","Repeat":0,"RawData":[880,2336,462,1656,460,952,486,1630,462,1654,460,1660,462,948,488,902,486,1656,486,926,486,1630,486,924,488,1630,460,954,460,950,512,1606,460,978,436,1656,486,926,486,1630,458],"RawDataInfo":[41,41,0]}}
{"IrReceived":{"Protocol":"UNKNOWN","Bits":21,"Hash":"0xAA3EAEBB","Repeat":0,"RawData":[906,2312,458,958,458,978,460,1630,488,1632,488,1630,488,952,434,952,460,1656,460,952,486,1632,486,926,486,1630,484,928,458,984,458,1630,488,924,486,1634,488,922,488,1658,440],"RawDataInfo":[41,41,0]}}
{"IrReceived":{"Protocol":"UNKNOWN","Bits":21,"Hash":"0x59033308","Repeat":0,"RawData":[878,2338,458,1660,460,956,484,1632,486,1632,486,1634,484,926,488,922,488,1632,486,924,464,1656,458,1660,460,952,488,920,460,958,484,1630,488,924,486,926,488,1628,488,1632,488],"RawDataInfo":[41,41,0]}}
{"IrReceived":{"Protocol":"UNKNOWN","Bits":21,"Hash":"0x54F2229A","Repeat":0,"RawData":[880,2338,486,928,458,950,486,1630,458,1662,488,1630,462,952,460,952,484,1630,486,926,490,926,486,922,458,952,460,1660,456,958,460,1656,456,1664,462,1656,458,1660,486,924,462],"RawDataInfo":[41,41,0]}}
{"IrReceived":{"Protocol":"UNKNOWN","Bits":21,"Hash":"0x3F33AB99","Repeat":0,"RawData":[904,2312,458,1662,488,926,508,1608,460,1658,458,1660,458,956,488,924,462,1656,464,948,488,924,484,928,462,1654,490,1630,462,946,486,1634,464,1654,462,1656,460,954,460,950,464],"RawDataInfo":[41,41,0]}}
{"IrReceived":{"Protocol":"UNKNOWN","Bits":21,"Hash":"0x03059AFF","Repeat":0,"RawData":[880,2336,486,928,486,926,484,1632,462,1658,464,1656,486,924,460,952,486,1632,482,932,462,950,486,1632,486,926,486,926,486,924,462,1658,462,1658,460,952,462,1654,488,1632,458],"RawDataInfo":[41,41,0]}}
{"IrReceived":{"Protocol":"UNKNOWN","Bits":21,"Hash":"0x3F33AB99","Repeat":0,"RawData":[880,2334,486,1634,486,926,462,1656,460,1660,462,1654,460,954,462,950,488,1628,460,950,484,932,462,950,488,1628,484,1636,486,928,512,1604,462,1656,484,1634,458,952,462,952,464],"RawDataInfo":[41,41,0]}}
{"IrReceived":{"Protocol":"UNKNOWN","Bits":21,"Hash":"0x9FED84D6","Repeat":0,"RawData":[880,2336,456,956,462,950,462,1656,460,1658,462,1658,462,952,484,930,460,1656,458,982,436,952,486,1630,460,954,484,1630,456,958,462,1656,486,1632,460,952,462,1656,462,952,460],"RawDataInfo":[41,41,0]}}
{"IrReceived":{"Protocol":"UNKNOWN","Bits":21,"Hash":"0xA826A111","Repeat":0,"RawData":[876,2342,460,1686,434,948,462,1660,460,1654,462,1656,456,956,486,930,458,1658,484,954,434,952,458,1656,462,978,436,1654,456,956,460,1658,462,1654,462,952,512,1606,464,950,458],"RawDataInfo":[41,41,0]}}
{"IrReceived":{"Protocol":"UNKNOWN","Bits":21,"Hash":"0xA23D127A","Repeat":0,"RawData":[878,2340,458,952,458,954,486,1632,462,1656,462,1654,458,984,436,952,486,1630,462,950,484,956,434,1656,460,1662,480,1636,484,924,488,1630,462,1656,460,950,488,924,488,926,488],"RawDataInfo":[41,41,0]}}
{"IrReceived":{"Protocol":"UNKNOWN","Bits":21,"Hash":"0x7C48B335","Repeat":0,"RawData":[878,2312,482,1636,488,926,460,1656,488,1628,464,1658,462,978,434,950,486,1632,462,948,462,954,484,1632,486,1630,460,1656,460,952,462,1658,462,1654,458,956,460,950,462,952,458],"RawDataInfo":[41,41,0]}}
{"IrReceived":{"Protocol":"UNKNOWN","Bits":21,"Hash":"0xE19898FB","Repeat":0,"RawData":[880,2336,456,956,484,930,462,1652,458,1660,460,1658,464,948,486,926,484,1634,458,950,462,950,460,952,456,956,462,948,472,940,460,1656,460,1660,458,1660,460,1658,458,1660,460],"RawDataInfo":[41,41,0]}}
{"IrReceived":{"Protocol":"UNKNOWN","Bits":21,"Hash":"0x92FB0300","Repeat":0,"RawData":[878,2338,484,1634,462,948,460,1660,486,1632,464,1654,458,952,486,954,460,1630,488,924,484,928,462,942,472,946,458,956,484,928,484,1632,460,1658,488,1630,484,1634,458,1660,462],"RawDataInfo":[41,41,0]}}
{"IrReceived":{"Protocol":"UNKNOWN","Bits":21,"Hash":"0x4852135F","Repeat":0,"RawData":[878,2336,432,980,432,980,456,1662,456,1662,456,1664,456,956,456,954,456,1662,432,980,456,954,432,980,456,1664,458,954,456,956,458,1660,460,1660,430,1688,458,954,456,1662,460],"RawDataInfo":[41,41,0]}}
{"IrReceived":{"Protocol":"UNKNOWN","Bits":21,"Hash":"0xCB8701E4","Repeat":0,"RawData":[880,2334,458,1662,482,930,462,1656,462,1656,460,1658,462,948,462,952,464,1654,484,928,462,950,462,950,462,1658,460,952,486,924,460,1662,462,1654,462,1656,462,950,462,1656,460],"RawDataInfo":[41,41,0]}}
{"IrReceived":{"Protocol":"UNKNOWN","Bits":21,"Hash":"0x4852135F","Repeat":0,"RawData":[880,2334,488,926,486,926,460,1656,460,1658,462,1656,458,956,462,948,488,1630,484,928,462,950,482,928,458,1658,460,956,486,924,460,1658,460,1660,462,1654,458,956,486,1632,488],"RawDataInfo":[41,41,0]}}
{"IrReceived":{"Protocol":"UNKNOWN","Bits":21,"Hash":"0x4B5FAAB9","Repeat":0,"RawData":[876,2338,458,1660,458,954,458,1660,458,1660,456,1662,456,956,432,980,456,956,456,1662,432,980,456,1662,432,978,456,1662,458,1660,432,978,458,1662,454,956,456,1662,456,954,458],"RawDataInfo":[41,41,0]}}
{"IrReceived":{"Protocol":"UNKNOWN","Bits":21,"Hash":"0xF0DAC27E","Repeat":0,"RawData":[902,2314,456,954,458,954,460,1658,458,1658,460,1660,458,954,460,952,458,954,458,1660,458,956,458,1660,458,954,456,1660,458,1660,458,954,456,1660,458,954,458,1660,458,956,454],"RawDataInfo":[41,41,0]}}
{"IrReceived":{"Protocol":"UNKNOWN","Bits":21,"Hash":"0x113E1B7A","Repeat":0,"RawData":[878,2336,462,1656,458,954,458,1660,456,1662,458,1660,460,952,462,950,432,978,460,1660,456,956,458,1660,462,952,460,952,460,1658,432,980,456,1664,458,954,458,1660,460,1658,462],"RawDataInfo":[41,41,0]}}
{"IrReceived":{"Protocol":"UNKNOWN","Bits":21,"Hash":"0xA2128CF5","Repeat":0,"RawData":[876,2338,458,954,458,954,460,1658,458,1660,456,1662,458,954,432,980,460,952,458,1660,458,954,460,1658,456,954,456,956,458,1658,458,954,456,1662,456,956,454,1662,458,1662,456],"RawDataInfo":[41,41,0]}}
{"IrReceived":{"Protocol":"UNKNOWN","Bits":21,"Hash":"0x113E1B7A","Repeat":0,"RawData":[878,2338,460,1656,458,954,432,1688,458,1660,460,1656,458,954,456,956,460,950,486,1632,456,956,462,1656,460,952,490,920,464,1656,488,922,462,1658,460,952,482,1634,462,1658,458],"RawDataInfo":[41,41,0]}}
{"IrReceived":{"Protocol":"UNKNOWN","Bits":21,"Hash":"0xA2128CF5","Repeat":0,"RawData":[880,2334,458,958,460,950,488,1628,460,1658,460,1658,464,946,488,926,486,926,486,1630,488,926,486,1632,458,954,488,924,460,1658,484,928,460,1654,486,928,488,1628,486,1632,458],"RawDataInfo":[41,41,0]}}
{"IrReceived":{"Protocol":"UNKNOWN","Bits":21,"Hash":"0x113E1B7A","Repeat":0,"RawData":[904,2310,456,1664,460,952,488,1630,488,1628,484,1634,460,982,460,952,460,924,486,1630,462,950,462,1658,510,900,488,924,486,1630,464,948,486,1632,484,928,484,1636,462,1654,458],"RawDataInfo":[41,41,0]}}
{"IrReceived":{"Protocol":"UNKNOWN","Bits":21,"Hash":"0xF0DAC27E","Repeat":0,"RawData":[904,2312,460,956,486,924,486,1632,486,1630,460,1660,488,924,460,952,460,952,462,1656,484,926,464,1654,458,954,460,1658,462,1654,458,954,460,1656,460,954,460,1684,436,950,462],"RawDataInfo":[41,41,0]}}
{"IrReceived":{"Protocol":"UNKNOWN","Bits":21,"Hash":"0x04B709BD","Repeat":0,"RawData":[878,2338,460,1658,484,928,462,1654,458,1660,456,1662,486,926,484,928,460,950,486,1632,458,1658,460,956,460,1658,488,1628,486,1632,458,954,468,972,434,1656,488,926,484,926,484],"RawDataInfo":[41,41,0]}}
{"IrReceived":{"Protocol":"UNKNOWN","Bits":21,"Hash":"0xCF714B7F","Repeat":0,"RawData":[878,2338,462,978,458,952,458,1630,462,1656,462,1656,484,930,484,926,484,1632,486,926,486,1632,460,1658,460,1658,460,980,434,978,434,1656,462,950,490,924,462,950,488,1628,458],"RawDataInfo":[41,41,0]}}
{"IrReceived":{"Protocol":"UNKNOWN","Bits":21,"Hash":"0x04B709BD","Repeat":0,"RawData":[904,2310,456,1662,486,954,460,1632,486,1630,458,1662,486,926,462,952,460,952,488,1630,460,1660,462,950,462,1656,514,1604,458,1662,486,928,486,924,488,1658,460,950,462,924,488],"RawDataInfo":[41,41,0]}}
[... cut ...]
I also have a RM16 remote control unit, which is pretty much like the RM17 without PVR buttons. Both are pictured in this manual on page 8: https://www.metz-ce.de/fileadmin/fm-dam/Download/Homepage_Englisch/Bed-Anl_englisch/LCD/610/610_47_2014.pdf

A picture of RM14 and RM17: https://de.m.wikipedia.org/wiki/Datei:Metz-Fernbedienungen.jpg

Manuals (German):
* RM17: https://www.metz-ce.de/fileadmin/fm-dam/Download/TV-Erlebnis/Bed-Anl_deutsch/RM17/610_47_0060.pdf
* RM17-2: https://www.metz-ce.de/fileadmin/fm-dam/Download/TV-Erlebnis/Bed-Anl_deutsch/RM17/613_47_0010.pdf
* RM19: https://www.metz-ce.de/fileadmin/fm-dam/Download/TV-Erlebnis/Bed-Anl_deutsch/RM19/614470022.pdf

crankyoldgit, 2020-08-16:
@mtdcr I've created PR #1242 which should add support for your TV remote's protocol. Can you please download and test out the branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Metz

> I'd like to implement this protocol, but I can't flash my IR bridge (NX-4519-675) easily. It currently runs Tasmota, which was installed using tuya-convert.
>
> Can anybody recommend a cheap, [...]

You should be able to drop that branch in on top of the exist copy in Tasmota and build Tasmota using it. Note: You'll have to turn on that protocol manually, or use one of the build environments for making Tasmota with FULL IR support. Ask @s-hadinger, as he is pretty familiar with it.

As for a "recommended" board, I'd recommend either a Wemos D1 Mini, or a NodeMCU board, and a cheap "breadboard" with some dupont cables, and the recommended IR-LED and NPN transistor. As per https://github.com/crankyoldgit/IRremoteESP8266/wiki#ir-sending

[Using a breadboard](https://www.sciencebuddies.org/science-fair-projects/references/how-to-use-a-breadboard) is pretty simple. No soldering needed.

s-hadinger, 2020-08-16:
David is correct. Just copy this version on top of the previous library in Tasmota and recompile with Tasmota-IR variant. It enables all protocols by default. 

If you can't do it easily I can provide a precompiled version later.

crankyoldgit, 2020-08-16:
Thanks @s-hadinger. This comes up enough that it'd be nice to have a FAQ/doc on Tasmota to point too.
i.e. "Halp! I n33dz da l4t35T IR lib in Tasm0ta! Can haz pls?"

mtdcr, 2020-08-17:
Thank you, it works! That's awesome! I can now control my Metz Ch610 television. Receiving codes from the remotes works as well.

crankyoldgit, 2020-08-18:
Thanks for confirming its working (excluding the address/command bits swap).

crankyoldgit, 2020-08-20:
Ref: https://github.com/crankyoldgit/IRremoteESP8266/pull/1242#discussion_r471853352
@mtdcr I think I've successfully swapped them around in the latest commit to this branch.
Can you please confirm the results are as you expect now?

mtdcr, 2020-08-20:
I can confirm it still works, sending and receiving.

crankyoldgit, 2020-08-20:
Yes, but are the address/command values as you would expect?

crankyoldgit, 2020-08-31:
The code changes mention have now been included in the newly released v2.7.10 of the library.

commit 9737dc21 (2020-08-16): Add support for Metz protocol.
* Add `sendMetz()` & `decodeMetz()`.
* Add `encodeMetz()` to handle creation of valid Metz codes from Address & Command bits (inc toggle bit)
* Unit test coverage based on supplied data.

Note: Protocol is a simple non-A/C one for controlling TVs etc.

Fixes #1241
commit 2c6f814b (2020-08-31): Add support for Metz protocol. (#1242)

* Add `sendMetz()` & `decodeMetz()`.
* Add `encodeMetz()` to handle creation of valid Metz codes from Address & Command bits (inc toggle bit)
* Unit test coverage based on supplied data.

Note: Protocol is a simple non-A/C one for controlling TVs etc.

Fixes #1241