crankyoldgit/IRremoteESP8266 #1249: TurnOn example codes question.

Shalabyer, 2020-08-25:
Library   : v2.7.9

TurnOn example codes differs from each other in so many way which really make sense, but what I'am asking here is about the function "on".
It exists in some like TurnOnMitsubishiAC example and dosn't exist in others like TurnOnFujitsuAC example which used setCmd and in TurnOnMitsubishiHeavyAc which used setPower.
although both of the past two examples have the function "on" in their code in these links [TurnOnMitsubishiHeavyAc](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRMitsubishiHeavy152Ac.html#a5c7aec50b53fdc3af591e077a4a268e4) , [TurnOnFujitsuAC](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRFujitsuAC.html#adcb24818d088c879beb7d76ada332f43).
What I'am asking here can i just use "on" function for all turn on codes.
mainly, can i just use these commands in all  supported ACs
```
  ac.on();
  ac.setFan(2);
  ac.setMode(kCoolixCool);
  ac.setTemp(26);
  ac.send(); 
```
taking in consideration setModel in  Fujitsu, Gree, Hitachi, LG,  Panasonic and Whirlpool.
Otherwise how do i know how to use other ACs specially the ones which have no example code in the library, taking in consideration that the only needed commands are what i just stated. Are they mainly used in all codes?
One last thing , can i use more than one library in the same code like the following:
```
#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ir_Gree.h>
#include <ir_Toshiba.h>
#include <ir_Coolix.h>
const uint16_t kIrLed = 4;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).
IRGreeAC ac(kIrLed);  // Set the GPIO to be used for sending messages.
IRToshibaAC acToshiba(kIrLed);  // Set the GPIO to be used for sending messages.
IRCoolixAC acCoolix(kIrLed);  // Set the GPIO to be used for sending messages.
void setup() {
 Serial.begin(115200);
  delay(200);
}

void loop() {
  Serial.println("Sending IR command to A/C ...");
  ac.begin();
  ac.on();
  ac.setFan(0);
  ac.setMode(kGreeCool);
  ac.setTemp(20);  // 16-30C
  ac.send();
  delay(500);
  ac.off();
  ac.send(); // send a turn off???
  delay(500);
 //----
  Serial.println("Sending IR Toshiba command to A/C ...");
  acToshiba.begin();
  acToshiba.on();
  acToshiba.setFan(1);
  acToshiba.setMode(kToshibaAcCool);
  acToshiba.setTemp(22);  // 16-30C
  acToshiba.send();
  delay(500);
  acToshiba.off();
  acToshiba.send(); // send a turn off???
  delay(500);
//---

 Serial.println("Sending IR Coolix command to A/C ...");
 acCoolix.begin();
  acCoolix.on();
  acCoolix.setFan(2);
  acCoolix.setMode(kCoolixCool);
  acCoolix.setTemp(26);  // 16-30C
  acCoolix.send();
  delay(500);
  acCoolix.off();
  acCoolix.send(); // send a turn off???
  delay(5000);
}
```
Or should i make a delay after "begin()" function ?.

NiKiZe, 2020-08-25:
It probably depends on what your end goal is.
To control a single ac and you know which one it is you can do
```c
  ac.on();
  ac.send(); 
```

With this you will only know that it should be on, but not which state it is in otherwise (temperature, mode etc)
And `.on()` and `.send()` is common for all ac protocol implementations.

With that said, if you want to control multiple devices you should probably take a look on some of the other examples.
https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples/Web-AC-control
https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/CommonAcControl/CommonAcControl.ino

Also, have you read thru [the documentation](https://github.com/crankyoldgit/IRremoteESP8266/wiki) ?

Does this answer your questions?

Shalabyer, 2020-08-25:
Thanks for your replay, I really appreciate that.
taking in consideration the CommonAcControl example which is a great one actually that i didn't look through deeply before,
in model line:
`ac.next.model = 1; `
the number '1' is the default for any ac which only has one mode, but what if it's more than one mode ,
i.e, the following message from IRReciveDump example:

>     Protocol  : FUJITSU_AC
>     Code      : 0x1463001010FC08304001000000008F (120 Bits)
>     Mesg Desc.: Model: 2 (ARDB1), Power: On, Mode: 1 (Cool), Temp: 20C, Fan: 0 (Auto), Command: N/A

that means that i will only use the '2' number only in `ac.next.model = 2; ` without even looking at the model name -ARDB1- am i getting it right?, or dose that number stated in `ac.next.model = 1; ` should depend on something else?
One last thing, I'am really thankful for just a great example like the CommonAcControl one as it saved me alot of time without spending time on looking in every ac condition.

NiKiZe, 2020-08-25:
Look into the model when you need it, for now ignore it, or read the sourcecode for the protocols that uses it (they are not many, and those that do mostly has difference in swing or some such things, again it depends on what your goal is)
A good IDE will have a "find usage" feature ;)

Shalabyer, 2020-08-25:
I'am sure i will not go deeply into swing and such things , iI'm only focusing on mode, fan and temp and sure power on and off.
so i suppose that i will not need to look into the protocol files and such thing, already did when i tried to search for that line `ac.next.model = 1;` and didn't found anything that reference number `2` for model `ARDB1` in FUJITSU_AC protocol when searching in the following files: IRutils.h, IRutils.cpp, IRac.h, IRac.cpp, Fujitsu.h and Fujitsu.cpp .
So my question is which number should i put for model `ARDB1` in FUJITSU_AC protocol in line `ac.next.model = 1;` in Co CmmonAcControl example? should i just follow what I've gotten from the IRReciveDump from the following message ?

> Mesg Desc.: Model: 2 (ARDB1), Power: On, Mode: 1 (Cool), Temp: 20C, Fan: 0 (Auto), Command: N/A

Shalabyer, 2020-08-25:
Alright I've tried it and yes we should use reference number 2 for model ARDB1 as we gotten from IRReciveDump, Thank you .

closed 2020-08-25 by Shalabyer

crankyoldgit, 2020-08-26:
> So my question is which number should i put for model `ARDB1` in FUJITSU_AC protocol in line `ac.next.model = 1;` in Co CmmonAcControl example?

Technically, you should use `fujitsu_ac_remote_model_t::ARDB1` or `ARDB1` instead of `2`, just in case the enum's order gets changed. However, `2` is pretty safe.

https://github.com/crankyoldgit/IRremoteESP8266/blob/ba2d5ca241ffb3963dc3f96097046b7d86bd24bc/src/IRsend.h#L119-L126
https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/IRsend_8h.html#a7204e78a1fe37a819c0b66f87a685dc0

crankyoldgit, 2020-08-26:
> And `.on()` and `.send()` is common for all ac protocol implementations.

\<pedantic nitpick>
@NiKiZe Only `.send()` is common to all ac protocol classes. `.on()` & `.setPower(bool)` is _very_ common, but not all of them have it. Some protocols use a `.setPowerToggle()` instead.
\</pedantic nitpick>