crankyoldgit/IRremoteESP8266 #1250: UNKNOWN Protocols and missing codes when using CommonAcControl in some protocols

Shalabyer, 2020-08-26:
Library   : v2.7.9
### Expected behavior
_What steps did you do and what should it have done?_

e.g.
1. Used CommonAcControl example in one NodeMCU and modified in it.
2. Used another NodeMCU as a receiver with IRrecvDumpV3  to read the message.
2. Tested more than one protocol SHARP_AC, FUJITSU_AC, SAMSUNG_AC and COOLIX which all worked good for me with the same configurations and code, only changing the protocol.

### Actual behavior
_What steps did you do, and what did or didn't actually happen?_

1. Some protocols that stated in the code and also in the additional notes below it didn't work well for me like ARGO, TOTREC, TOSHIBA, FUJITSU_AC, LG2, DAIKIN128, DAIKIN64 and AIRWELL.

#### Output of raw data from IRrecvDumpV3.ino 

Timestamp : 000315.414
Library   : v2.7.9

> 
> Protocol  : TOSHIBA_AC
> Code      : 0xF20D03FC0120040025 (72 Bits)
> Mesg Desc.: Temp: 19C, Power: On, Mode: 4 (Fan), Fan: 0 (Auto), Turbo: Off, Econo: Off
> uint16_t rawData[527] = {4424, 4308,  568, 1628,  568, 1620,  566, 1620,  570, 1616,  566, 510,  568, 508,  568, 1618,  568, 514,  566, 512,  566, 508,  568, 510,  568, 508,  568, 1618,  568, 1618,  570, 508,  568, 1620,  542, 534,  568, 510,  566, 508,  568, 510,  568, 510,  566, 510,  564, 1620,  568, 1618,  568, 1618,  568, 1618,  568, 1618,  568, 1620,  566, 1620,  568, 1616,  568, 508,  568, 512,  566, 508,  566, 510,  568, 510,  568, 508,  566, 510,  566, 510,  566, 508,  568, 1620,  566, 510,  566, 508,  568, 1618,  568, 510,  566, 510,  566, 510,  566, 508,  566, 512,  566, 510,  566, 508,  568, 510,  566, 510,  568, 508,  568, 1618,  568, 508,  568, 510,  568, 510,  568, 510,  566, 510,  566, 510,  566, 510,  566, 510,  566, 510,  566, 510,  566, 510,  566, 510,  568, 1618,  568, 510,  568, 510,  564, 1620,  568, 510,  566, 1618,  568, 7420,  4428, 4282,  568, 1618,  566, 1620,  566, 1620,  568, 1618,  566, 510,  566, 510,  566, 1618,  568, 510,  566, 510,  566, 508,  566, 512,  566, 510,  566, 1620,  568, 1620,  568, 510,  566, 1620,  568, 510,  566, 510,  566, 512,  566, 510,  566, 510,  566, 510,  566, 1618,  568, 1620,  566, 1620,  568, 1620,  568, 1618,  566, 1620,  542, 1644,  566, 1618,  568, 510,  566, 512,  566, 510,  566, 512,  564, 512,  566, 510,  566, 510,  564, 512,  566, 510,  566, 1620,  568, 510,  566, 510,  566, 1620,  568, 510,  566, 512,  566, 512,  566, 510,  564, 512,  566, 512,  566, 510,  564, 512,  564, 512,  564, 512,  564, 1620,  566, 510,  566, 512,  566, 512,  566, 510,  564, 512,  564, 512,  564, 512,  564, 514,  564, 512,  564, 514,  562, 514,  564, 512,  564, 1622,  566, 512,  564, 512,  564, 1620,  566, 512,  564, 1620,  566, 7436,  4400, 4310,  566, 1622,  566, 1620,  568, 1618,  568, 1620,  566, 512,  564, 512,  566, 1620,  566, 512,  564, 512,  564, 512,  564, 512,  564, 512,  564, 1620,  566, 1620,  566, 510,  564, 1622,  566, 512,  566, 510,  564, 514,  564, 514,  562, 514,  562, 538,  538, 514,  562, 1624,  564, 1622,  566, 1620,  566, 1620,  566, 1620,  566, 1620,  566, 1620,  566, 1622,  566, 514,  562, 514,  562, 514,  562, 1620,  564, 514,  562, 538,  540, 536,  538, 538,  538, 1624,  564, 512,  562, 538,  540, 538,  538, 538,  538, 538,  538, 538,  538, 1622,  564, 516,  562, 536,  538, 538,  538, 1622,  564, 514,  562, 538,  538, 538,  538, 1622,  564, 1622,  564, 7420,  4396, 4312,  566, 1620,  566, 1620,  566, 1620,  566, 1622,  566, 512,  564, 514,  562, 1622,  564, 540,  538, 538,  540, 536,  538, 538,  538, 538,  538, 1622,  564, 1622,  564, 538,  538, 1624,  564, 538,  538, 538,  538, 538,  538, 540,  538, 538,  538, 540,  538, 540,  538, 1624,  564, 1622,  564, 1624,  564, 1620,  564, 1622,  564, 1622,  564, 1622,  564, 1622,  564, 540,  538, 538,  538, 538,  536, 1624,  564, 540,  536, 538,  538, 540,  536, 540,  536, 1626,  564, 538,  538, 538,  536, 540,  536, 540,  536, 540,  536, 538,  536, 1624,  562, 540,  536, 538,  538, 540,  536, 1624,  562, 538,  538, 540,  536, 540,  536, 1624,  562, 1624,  562};  // TOSHIBA_AC
> uint8_t state[9] = {0xF2, 0x0D, 0x03, 0xFC, 0x01, 0x20, 0x04, 0x00, 0x25};
> 
> 
> Timestamp : 000320.982
> Library   : v2.7.9
> 
> Protocol  : TOSHIBA_AC
> Code      : 0xF20D03FC0120040025 (72 Bits)
> Mesg Desc.: Temp: 19C, Power: On, Mode: 4 (Fan), Fan: 0 (Auto), Turbo: Off, Econo: Off
> uint16_t rawData[681] = {4414, 4294,  568, 1598,  564, 1644,  544, 1642,  542, 1644,  566, 510,  566, 510,  566, 1618,  570, 512,  568, 510,  568, 510,  568, 508,  566, 510,  568, 1618,  568, 1618,  568, 510,  568, 1618,  568, 508,  568, 510,  592, 484,  568, 508,  568, 510,  568, 510,  566, 1618,  568, 1620,  566, 1620,  568, 1618,  568, 1618,  568, 1618,  568, 1620,  568, 1620,  568, 508,  566, 512,  566, 510,  568, 508,  566, 510,  568, 510,  568, 508,  568, 510,  568, 508,  568, 1618,  566, 512,  566, 510,  566, 1618,  568, 510,  568, 508,  566, 510,  568, 510,  566, 510,  566, 510,  568, 510,  568, 508,  568, 510,  566, 510,  566, 1618,  568, 524,  566, 512,  566, 510,  568, 508,  566, 512,  566, 510,  566, 510,  566, 510,  566, 510,  566, 512,  566, 510,  568, 508,  568, 1618,  566, 510,  568, 508,  566, 1618,  568, 510,  566, 1620,  568, 7418,  4426, 4282,  568, 1620,  568, 1620,  566, 1618,  568, 1618,  568, 510,  568, 508,  566, 1618,  568, 510,  568, 508,  566, 510,  566, 510,  568, 508,  566, 1620,  568, 1618,  568, 510,  566, 1620,  568, 510,  566, 510,  566, 510,  566, 510,  568, 508,  566, 510,  566, 1620,  566, 1620,  566, 1620,  568, 1620,  566, 1620,  568, 1618,  568, 1618,  566, 1620,  566, 510,  566, 512,  566, 510,  566, 510,  566, 512,  564, 512,  564, 512,  566, 510,  566, 510,  566, 1622,  566, 510,  566, 510,  566, 1620,  566, 510,  566, 510,  566, 512,  566, 510,  566, 512,  564, 512,  564, 512,  564, 512,  566, 512,  564, 514,  564, 1620,  566, 510,  566, 512,  564, 512,  564, 512,  562, 514,  564, 514,  562, 514,  562, 514,  564, 514,  562, 516,  562, 514,  562, 514,  562, 1622,  566, 512,  564, 512,  564, 1622,  566, 512,  564, 1622,  566, 7432,  4400, 4310,  568, 1620,  566, 1622,  566, 1620,  566, 1620,  568, 510,  564, 512,  564, 1620,  566, 512,  566, 512,  564, 514,  562, 514,  562, 514,  562, 1622,  566, 1620,  566, 512,  564, 1620,  566, 510,  564, 514,  564, 514,  562, 512,  562, 538,  538, 516,  562, 538,  538, 1624,  566, 1620,  566, 1620,  566, 1622,  566, 1620,  566, 1620,  566, 1620,  568, 1618,  566, 512,  566, 512,  564, 536,  540, 1620,  566, 514,  562, 538,  540, 536,  540, 538,  538, 1622,  564, 514,  562, 538,  538, 538,  538, 536,  538, 538,  538, 538,  538, 1622,  564, 516,  562, 538,  538, 538,  538, 1622,  564, 536,  538, 538,  538, 538,  538, 1622,  564, 1622,  564, 7420,  4396, 4314,  566, 1622,  564, 1620,  566, 1622,  564, 1622,  564, 512,  562, 538,  538, 1622,  564, 538,  538, 538,  538, 538,  538, 538,  538, 540,  536, 1622,  564, 1622,  564, 538,  538, 1624,  564, 536,  538, 538,  538, 538,  538, 538,  538, 538,  538, 538,  538, 538,  536, 1626,  564, 1622,  564, 1622,  564, 1622,  564, 1624,  564, 1622,  564, 1622,  564, 1622,  564, 540,  538, 538,  538, 538,  538, 1624,  564, 538,  536, 540,  536, 540,  536, 540,  536, 1624,  562, 540,  536, 540,  536, 540,  536, 540,  536, 540,  536, 540,  536, 1626,  562, 540,  536, 540,  536, 540,  536, 1624,  562, 540,  536, 540,  536, 540,  536, 1624,  562, 1626,  562, 7732,  2070, 2300,  182, 368,  476, 7928,  580, 636,  550, 1582,  574, 640,  492, 694,  518, 1612,  572, 644,  516, 672,  518, 670,  518, 646,  542, 644,  544, 1610,  544, 670,  518, 1612,  520, 1638,  572, 642,  516, 672,  518, 1610,  520, 1638,  546, 668,  518, 1612,  520, 696,  518, 1612,  520, 694,  520, 670,  518, 1612,  520, 696,  518, 670,  518, 646,  542, 646,  462, 730,  544, 644,  544, 646,  544, 644,  544, 644,  544, 644,  542, 644,  544, 644,  544, 644,  518, 670,  542, 646,  542, 644,  544, 644,  542, 646,  542, 646,  542, 670,  520, 668,  518, 670,  520, 644,  542, 646,  542, 670,  518, 670,  518, 670,  518, 646,  542, 670,  518, 646,  544, 670,  518, 670,  518, 646,  542, 670,  492, 694,  518, 670,  518, 670,  494, 694,  518, 646,  544, 644,  518, 696,  492, 1638,  544, 1612,  518, 694,  518, 1612,  518, 694,  518, 670,  518, 6234,  520};  // TOSHIBA_AC
> uint8_t state[9] = {0xF2, 0x0D, 0x03, 0xFC, 0x01, 0x20, 0x04, 0x00, 0x25};
> 
> 
> Timestamp : 000326.256
> Library   : v2.7.9
> 
> Protocol  : TROTEC
> Code      : 0x123423010000000024 (72 Bits)
> Mesg Desc.: Power: Off, Mode: 3 (Fan), Temp: 19C, Fan: 2 (Medium), Sleep: Off
[... cut ...]
DAIKIN64 OFF is the same as ON code 0xEC25161600004216
AIRWELL OFF code is same as ON  code 0x250500002
```
**IMPORTANT NOTE:** the past errors listed from the default CommonAcControl is gathered from 6 repeats of the code loop and they are all the same (i.e, made sure that these are not just random IR signals,, they are repeated 6 times.)
#### Circuit diagram and hardware used (if applicable)
The sender one is ESP8266-12E and the RX one is NodeMCU

##
### Has this library/code previously worked as expected for you?
No

crankyoldgit, 2020-08-26:
Re: ARGO. It's not the most well supported/tested protocol.
see:
https://github.com/crankyoldgit/IRremoteESP8266/blob/ba2d5ca241ffb3963dc3f96097046b7d86bd24bc/src/ir_Argo.cpp#L457-L458
IIRC someone contributed the send routine and protocol structure/code. I've no ability to verify that actually even works. Can you please supply some raw captures from the actual remote?

> When i tried to put a delay of 5 seconds between protocols (i.e, just before //---------------- line) i got nothing but a stream of unknown signals with of course their delays , so can someone [...]

That sounds plain weird. Can you provide some sample `rawData[]` for those cases (and what code you think produced the message)?

> TOSHIBA OFF  gives the same as ON which is 0xF20D03FC0180810000

Hmmm. Toshiba has an odd-ball power on/off style. Will need to look into your code and how the library reacts further to see what may be going on.

> TROTEC ON and OFF codes are 0x7B0C9C1D  and  0xAEF02725 are readed but with UNKNOWN Protocols
> FUJITSU_AC ON is readed but with UNKNOWN Protocol  0x9297C062
> LG2 is readed as LG with codes 0x8800A2C and 0x88C0051

Will need to look into these separately. Possibly separate issues are warranted for each protocol.

> DAIKIN128 OFF code is the same as ON code 0x164200000000250CA10000000000000B

`DAIKIN128` uses a Power toggle message. so this is expected.

> DAIKIN64 OFF is the same as ON code 0xEC25161600004216

`DAIKIN64` uses a Power toggle message. so this is expected.

> AIRWELL OFF code is same as ON  code 0x250500002

`AIRWELL` uses a Power toggle message. so this is expected.

crankyoldgit, 2020-08-26:
>   ac.next.mode =IRac::strToOpmode("Fan", stdAc::opmode_t::kFan);  // Run in cool mode initially.

Um ... for a lot of these `IRac::strToBlah()`s you can just do:
`ac.next.mode = stdAc::opmode_t::kFan;`

Shalabyer, 2020-08-26:
> IIRC someone contributed the send routine and protocol structure/code. I've no ability to verify that actually even works. Can you please supply some raw captures from the actual remote?
Iam sorry I don't have any of these but i was testing for future use.

> That sounds plain weird. Can you provide some sample rawData[] for those cases (and what code/state/data you think produced the message)?

```
Timestamp : 000008.878
Library   : v2.7.9

Protocol  : UNKNOWN
Code      : 0x9FA3CDF5 (266 Bits)
uint16_t rawData[531] = {98, 262,  132, 18334,  4438, 4288,  542, 1648,  540, 1646,  542, 1644,  542, 1646,  540, 536,  566, 510,  566, 1620,  566, 520,  568, 508,  542, 534,  568, 510,  568, 508,  568, 1618,  568, 1618,  568, 508,  542, 1646,  542, 536,  568, 508,  568, 510,  568, 508,  592, 484,  568, 510,  568, 1618,  566, 1622,  542, 1644,  542, 1646,  566, 1618,  542, 1644,  568, 1618,  544, 1644,  540, 536,  568, 510,  568, 508,  568, 508,  568, 508,  592, 484,  568, 510,  568, 510,  566, 510,  566, 1620,  542, 534,  566, 510,  568, 1618,  568, 508,  568, 508,  568, 510,  568, 508,  568, 512,  566, 510,  644, 428,  570, 510,  568, 510,  568, 508,  568, 1618,  568, 508,  568, 510,  566, 510,  568, 508,  566, 510,  568, 508,  568, 508,  568, 508,  568, 510,  568, 510,  544, 534,  568, 508,  568, 1618,  566, 510,  568, 508,  568, 1618,  606, 472,  568, 1622,  568, 7418,  4430, 4280,  570, 1618,  542, 1644,  568, 1618,  568, 1618,  542, 536,  566, 508,  542, 1644,  568, 510,  568, 510,  568, 508,  568, 510,  568, 508,  568, 1620,  566, 1620,  568, 508,  568, 1620,  568, 510,  568, 508,  568, 510,  568, 510,  568, 508,  568, 510,  568, 1618,  566, 1622,  566, 1620,  566, 1620,  568, 1618,  544, 1644,  568, 1618,  568, 1618,  566, 510,  568, 510,  568, 508,  568, 510,  566, 510,  566, 510,  566, 510,  542, 534,  568, 510,  566, 1620,  568, 510,  566, 510,  566, 1620,  566, 508,  568, 510,  568, 510,  568, 508,  566, 512,  566, 510,  568, 508,  566, 510,  566, 510,  566, 510,  566, 1618,  568, 510,  566, 512,  566, 570,  522, 496,  566, 508,  568, 510,  566, 510,  568, 510,  566, 512,  564, 512,  566, 510,  566, 510,  566, 1620,  566, 510,  566, 510,  566, 1618,  568, 510,  564, 1620,  566, 7436,  4402, 4308,  566, 1620,  568, 1620,  566, 1620,  568, 1618,  568, 510,  566, 510,  566, 1618,  568, 512,  566, 510,  566, 508,  568, 510,  566, 512,  566, 1618,  568, 1618,  568, 510,  566, 1620,  568, 510,  566, 510,  566, 510,  566, 510,  566, 512,  566, 510,  564, 512,  564, 1620,  568, 1618,  566, 1620,  566, 1620,  568, 1620,  566, 1620,  566, 1620,  568, 1620,  566, 512,  566, 510,  566, 508,  566, 1620,  566, 510,  566, 510,  566, 512,  566, 510,  566, 1620,  568, 510,  566, 510,  566, 512,  564, 512,  566, 510,  566, 512,  566, 1620,  566, 512,  566, 512,  564, 512,  566, 1620,  566, 510,  566, 510,  566, 512,  564, 1620,  568, 1618,  566, 7420,  4400, 4308,  568, 1618,  568, 1620,  566, 1620,  566, 1618,  566, 512,  566, 510,  566, 1620,  566, 512,  566, 512,  564, 510,  566, 512,  566, 512,  564, 1620,  566, 1620,  566, 510,  564, 1622,  568, 510,  566, 510,  566, 512,  564, 512,  564, 512,  564, 514,  564, 512,  562, 1622,  566, 1620,  566, 1620,  566, 1620,  568, 1618,  566, 1620,  564, 1620,  566, 1620,  566, 512,  566, 510,  564, 512,  564, 1620,  566, 510,  566, 512,  566, 512,  564, 512,  564, 1622,  566, 512,  564, 514,  562, 514,  564, 512,  564, 512,  564, 512,  562, 1622,  564, 514,  564, 514,  564, 512,  562, 1622,  566, 512,  564, 514,  562, 514,  562, 1622,  566, 1620,  566};  // UNKNOWN 9FA3CDF5

Timestamp : 000014.319
Library   : v2.7.9

Protocol  : UNKNOWN
Code      : 0x19A12C55 (265 Bits)
uint16_t rawData[529] = {94, 33906,  4412, 4284,  520, 1668,  518, 1668,  520, 1666,  518, 1668,  518, 558,  542, 536,  540, 1644,  542, 540,  542, 536,  540, 536,  568, 508,  566, 510,  566, 1620,  542, 1644,  542, 536,  566, 1620,  542, 534,  542, 536,  542, 534,  568, 508,  566, 510,  566, 510,  542, 1644,  542, 1646,  542, 1644,  542, 1644,  540, 1646,  542, 1644,  540, 1646,  518, 1668,  542, 536,  566, 512,  542, 534,  566, 508,  542, 534,  568, 508,  568, 508,  568, 510,  570, 506,  542, 1646,  542, 536,  542, 536,  566, 1620,  542, 536,  542, 532,  566, 510,  542, 534,  568, 510,  566, 510,  568, 510,  568, 508,  568, 508,  568, 510,  566, 1620,  542, 536,  542, 534,  568, 508,  568, 510,  566, 510,  604, 474,  542, 534,  542, 534,  542, 534,  566, 512,  568, 508,  568, 508,  568, 1620,  566, 510,  566, 508,  566, 1620,  566, 510,  542, 1644,  542, 7444,  4428, 4280,  542, 1646,  542, 1644,  542, 1644,  542, 1646,  542, 536,  542, 536,  566, 1618,  568, 510,  566, 508,  568, 510,  542, 534,  566, 510,  542, 1644,  544, 1642,  542, 534,  568, 1620,  568, 508,  542, 534,  542, 534,  566, 510,  568, 508,  568, 508,  568, 1618,  542, 1646,  542, 1644,  542, 1644,  542, 1644,  542, 1644,  542, 1644,  542, 1644,  568, 510,  566, 512,  566, 510,  566, 510,  566, 510,  568, 510,  568, 508,  566, 510,  566, 510,  566, 1620,  568, 510,  566, 510,  566, 1620,  566, 510,  544, 532,  604, 474,  566, 512,  566, 510,  566, 508,  566, 510,  568, 508,  568, 508,  566, 510,  568, 1616,  568, 508,  566, 512,  566, 510,  568, 508,  568, 510,  568, 508,  566, 510,  566, 510,  568, 508,  568, 512,  566, 510,  568, 508,  566, 1618,  566, 510,  566, 510,  566, 1620,  568, 510,  568, 1618,  568, 7432,  4402, 4306,  544, 1644,  566, 1618,  544, 1644,  542, 1644,  568, 508,  566, 510,  542, 1644,  568, 510,  568, 508,  5 [... cut ...]
> Hmmm. Toshiba has an odd-ball power on/off style. Will need to look into your code and how the library reacts further to see what may be going on.

Will it's not my code it's yours i guess :), as i said in these errors i used the default CommonAcControl

> Will need to look into these separately. Possibly separate issues are warranted for each protocol.

Will i may not really need all these other protocols so what mainly i would be pleased if you helped me with is the TOSHIBA one and that 5 second delay problem , as of that 5 second delay problem i may be too afraid that i can't rely on the way that the example uses in my main code.

Shalabyer, 2020-08-26:
> Um ... for a lot of these IRac::strToBlah()s you can just do:
> ac.next.mode = stdAc::opmode_t::kFan;
 Will i was trying the strTo**** functions because i will be mainly use it , i don't think it will really affect the output as they already worked with some protocols that i already tested with the same code like SHARP_AC, FUJITSU_AC, SAMSUNG_AC and COOLIX .

Shalabyer, 2020-08-26:
> Can you please supply some raw captures from the actual remote?

I have a KT-3888 Universal AC Remote which has about 700 AC protocols 
 [Here](https://photos.app.goo.gl/fWykB2f2tYfn948N9) are the supported ACs names, if you know any of these that uses any thing that you need i can try to help you with it's raw of data.

crankyoldgit, 2020-08-26:
> that is the messages readed from the code stated in the issue only adding delay(5000); after each` //------------------` line.
> 

Okay. If you look at the raw data you've provided, you'll see a lot of values that are sub-200 uSeconds. That's a very strong indicator that either a) your IR receiver hardware module is not very good, or b) there is significant IR noise in your environment.

See https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#Help_Im_getting_very_inconsistent_results_when_capturing_an_IR_message_using_a_VS1838b_IR_demodulator
&
https://github.com/crankyoldgit/IRremoteESP8266/issues/1173#issuecomment-640018653

e.g. A sample from your data.
```
uint16_t rawData[529] = {96, 7892,  4416, 4284,  540, 1648,  542, 1644,  518, 1668,  542, 1644,  542, 534,  540, 536,  566, 1620,  566, 514,  566, 512,  568, 508,  568, 508,  568, 508,  568, 1618,  542, 1644,  566, 510,  568, 1618,  568, 510,  566, 510,  568, 508,  568, 508,  568, 508,  568, 508,  568, 1618,  568, 1620,  568, 1618,  542, 1646,  542, 1646,  566, 1618,  566, 1620,  542, 1646,  542, 534,  566, 510,  542, 534,  568, 508,  568, 510,  566, 510,  568, 510,  568, 508,  568, 508,  568, 1618,  542, 536,  568, 508,  568, 1620,  542, 534,  568, 508,  566, 510,  566, 510,  566, 512,  568, 510,  566, 510,  568, 508,  568, 510,  568, 506,  568, 1618,  568, 508,  568, 508,  568, 508,  568, 508,  570, 506,  568, 510,  566, 510,  568, 510,  566, 510,  568, 514,  564, 508,  568, 508,  568, 1618,  568, 508,  568, 510,  568, 1618,  568, 508,  568, 1618,  568, 7418,  4428, 4282,  566, 1620,  568, 1618,  542, 1646,  566, 1620,  568, 508,  542, 534,  568, 1618,  566, 512,  566, 510,  566, 510,  568, 510,  568, 510,  568, 1618,  568, 1618,  568, 510,  566, 1620,  542, 534,  564, 512,  568, 508,  566, 510,  568, 508,  568, 508,  568, 1618,  568, 1620,  568, 1616,  544, 1644,  568, 1618,  566, 1620,  566, 1620,  568, 1618,  600, 474,  566, 512,  568, 508,  566, 510,  568, 510,  568, 508,  566, 510,  566, 510,  566, 510,  566, 1620,  568, 508,  568, 508,  566, 1620,  568, 508,  568, 508,  568, 508,  568, 508,  568, 510,  566, 512,  566, 510,  568, 510,  566, 510,  568, 508,  566, 1620,  566, 510,  568, 510,  566, 510,  566, 510,  566, 508,  566, 512,  566, 510,  566, 508,  568, 510,  566, 510,  566, 510,  566, 510,  582, 1606,  568, 510,  568, 508,  568, 1618,  568, 510,  568, 1618,  568, 7432,  4402, 4306,  568, 1620,  566, 1620,  568, 1618,  542, 1644,  568, 508,  566, 510,  566, 1620,  568, 510,  568, 510,  566, 510,  566, 510,  566, 510,  566, 1618,  568, 1618,  568, 508,  568, 1620,  568, 510,  566, 510,  568, 510,  566, 510,  566, 510,  566, 512,  566, 510,  566, 1620,  568, 1618,  568, 1618,  566, 1620,  568, 1620,  568, 1618,  568, 1620,  566, 1620,  566, 512,  566, 510,  566, 512,  566, 1618,  566, 510,  566, 512,  566, 510,  566, 512,  566, 1618,  570, 510,  566, 512,  540, 536,  566, 510,  566, 512,  564, 512,  564, 1620,  566, 512,  566, 512,  566, 510,  566, 1620,  566, 512,  566, 512,  564, 512,  566, 1618,  566, 1622,  566, 7418,  4400, 4308,  568, 1620,  566, 1620,  568, 1618,  566, 1620,  566, 512,  566, 512,  566, 1620,  566, 512,  566, 510,  566, 512,  564, 512,  564, 512,  564, 1620,  568, 1618,  568, 510,  576, 1614,  580, 502,  564, 510,  566, 512,  564, 512,  566, 510,  578, 502,  568, 508,  564, 1622,  566, 1620,  566, 1620,  566, 1620,  566, 1622,  578, 1610,  566, 1620,  566, 1620,  566, 512,  576, 504,  578, 502,  564, 1618,  574, 506,  564, 514,  564, 514,  564, 574,  516, 1612,  566, 512,  564, 514,  562, 514,  562, 514,  564, 574,  514, 504,  568, 1616,  566, 514,  564, 512,  570, 508,  562, 1622,  572, 506,  562, 514,  562, 538,  538, 1622,  564, 1622,  566};  // UNKNOWN 19A12C55
```

Note the `96` value at the start. That's IR noise the sensor is picking up. It's arriving before the start of the message (i.e. the `4416, 4284, ` values.) This is confusing the decoder(s).

Two ways to fix a lot of the problems you are having with your sensor/environmental issues is to set [max_skip ](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRrecv.html#aeaa5c07a8b46f8fbb982f996cc1f9f4b) to 2-4 etc.

Or adjust the `kTimeout` parameter down to closer to the `15` mSecond value
https://github.com/crankyoldgit/IRremoteESP8266/blob/ba2d5ca241ffb3963dc3f96097046b7d86bd24bc/examples/IRrecvDumpV3/IRrecvDumpV3.ino#L74-L81

See also https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide#receiving-an-ir-signal

Due to how the micro controllers work on normal IR devices, a sent signal from the library will probably be understood better than the super-generic-try-to-do-everything `IRrecv` methods used by this library. In short, a normal device is looking for only one protocol, it can be more robust at decoding that one protocol. This library needs to be less rigorous and more flexible thus by design, can perform sub-optimally compared to real-world dedicated hardware and environments.

> I have a KT-3888 Universal AC Remote which has about 700 AC protocols
Here are the supported ACs names, if you know any of these that uses any thing that you need i can try to help you with it's raw of data.

Thanks for the offer. We prefer data from OEM remotes so we know we are not re-implementing/reverse-engineering someone else's potentially dodgy implementation.

Shalabyer, 2020-08-26:
I'am Really thankful for all your time and answers that's very kind of you , but one last thing I'am sorry i tried and i need your help in it
in such a code how can i toggle power to turn off the AC,

```
 ac.next.protocol = strToDecodeType("AIRWELL");  // Change the protocol used.
      ac.next.power = true;  // We want to turn on the A/C unit.
  ac.next.mode =IRac::strToOpmode("Fan", stdAc::opmode_t::kFan);  // Run in cool mode initially.
  ac.next.degrees = 19;  
  ac.next.model = IRac::strToModel("",1);
  ac.next.fanspeed = IRac::strToFanspeed("Auto",stdAc::fanspeed_t::kAuto);
      Serial.println("Sending a message to turn ON the A/C unit.");
      ac.sendAc();  // Have the IRac class create and send a message.
      delay(5000);  // Wait 5 seconds.
      //ac.next.power = IRac::strToBool("Off",false);  
      Serial.println("Send a message to turn OFF the A/C unit.");
      ac.handleToggles();  //how can i deal it in here the right way
      //ac.sendAc();  // Send the message.
      delay(5000);
```

Also the problem i did stated about the TOSHIBA Protocol , keeping in mind that I'am using the default example and the power is not shifting.

crankyoldgit, 2020-08-27:
> in such a code how can i toggle power to turn off the AC,

You don't need to do anything special other than change the value of `ac.next.power` between sending via `.sendAc();` calls.
The `IRac` class should & **was** doing it all for you.
I think what you're not understanding is that, for A/Cs that use a "toggle" setting, such as Power on `AIRWELL`s is that the `on` and the `off` IR message are exactly the same.
e.g. An `AIRWELL` A/C that is off, when receiving the message `0x250500002` will then turn on, ... and subsequently, if it gets another `0x250500002` message, it will change from on to off, and vice versa. Thus the same message you saw repeated is exactly what the `IRac` class should be producing to do as you originally requested.
```
ac.next.protocol = AIRWELL;
ac.next.power = true;  // on
ac.sendAc();  // Will send 0x250500002. i.e. Toggle the power on the A/C unit. Hopefully Off to On.
ac.next.power = false;  // off
ac.sendAc();  // Will send 0x250500002. i.e. Toggle the power on the A/C unit. Hopefully On to Off.
```

The _Hopefully_ comment is because there is no way the library can know the initial/actual power state of the Airwell A/C. You'd need to get them in sync.

crankyoldgit, 2020-08-27:
> Also the problem i did stated about the TOSHIBA Protocol , keeping in mind that I'am using the default example and the power is not shifting.

I haven't forgotten. I'm looking into that now.

crankyoldgit, 2020-08-27:
I found the bug with Toshiba a/c, it will be fixed with PR #1251

closed 2020-08-27 by crankyoldgit

Shalabyer, 2020-08-27:
> e.g. An AIRWELL A/C that is off, when receiving the message 0x250500002 will then turn on, ... and subsequently, if it gets another 0x250500002 message, it will change from on to off, and vice [...]

Oh i get it, Thanks for your great help and this great library, hope you get the bug fixed easily :)

crankyoldgit, 2020-08-27:
> hope you get the bug fixed easily

Already fixed in the `master` branch.

crankyoldgit, 2020-08-31:
The code changes (Toshiba) mention have now been included in the newly released v2.7.10 of the library.

PR #1251 (2020-08-27): [BUG] TOSHIBA_AC: not turning off when using `IRac` class.
* `setPower()` needs to happen after other functions that can change the operation mode.
* Unit tests added to help debug/confirm the issue has been fixed successfully.

Fixes #1250
commit d0e2d831 (2020-08-27): [BUG] TOSHIBA_AC: not turning off when using `IRac` class. (#1251)

* setPower() needs to happen after other functions that can change the operation mode.
* Unit tests added to help debug/confirm the issue has been fixed successfully.

Fixes #1250