crankyoldgit/IRremoteESP8266 #1252: LG2 no example file

Tweak-NL, 2020-08-27:
Hello, there are some examples for turning on a AC but LG seems to be missing.
There is a cpp file with LG and LG2 support (i think mine is LG2) but would like to see
some examples how to turn it on and off and change temp etc.

Can u please add this?

NiKiZe, 2020-08-27:
I would rather see the number of examples reduced since they are mostly just a copy of one another.

You can most likely take some other example, change the include and protocol name when the ac instance is created. Have you tried this, do you get any specific errors when you try to?

Have you tried to search among the closed issued for anything similar, maybe you can find good responses and examples there as well.

Tweak-NL, 2020-08-27:
Sorry but this is new for me, are the commands to turn them on the same as others? When i look in the CPP file it looks like u need to specify wich remote? And no i dont have errors, because i need to order the IR LED (i only got the nodemcu here) do i need to order the IR receiver to?

so, yes i know how to include something and i know what it does but i dont know "the commands"

NiKiZe, 2020-08-27:
You can also use the common interface. See the example at
https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/CommonAcControl/CommonAcControl.ino

NiKiZe, 2020-08-27:
If you do take another example, do the basic changes, and try to compile it you will get errors if the calls are not the same ;)

You can try to compile it even without the hardware.
If there is any issues then the best is to be able to "record" the original remote. But you can always try with only the transmit circuit.

Tweak-NL, 2020-08-27:
When i checked some examples i searched for: void IRLgAc: in the CPP file.
Now i created this:

#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ir_LG.h>

const uint16_t kIrLed = 4;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).
IRLgAc ac(kIrLed);  // Set the GPIO to be used for sending messages.

void setup() {
  // put your setup code here, to run once:
  ac.begin();
  ac.setModel(AKB75215403);
  Serial.begin(115200);
}

void loop() {
  Serial.println("Sending...");
  
  // Set up what we want to send. See ir_LG.cpp for all the options.
  ac.setPower(true);
  ac.setFan(kLgAcFanAuto);
  ac.setMode(kLgAcCool);
  ac.setTemp(20);

  // Now send the IR signal.
#if SEND_LG
  ac.send();
#else  // SEND_LG
  Serial.println("Can't send because SEND_LG has been disabled.");
#endif  // SEND_LG

  delay(5000);
}

No errors with compiling, i really dont know if i did this right but looks like it. Now need to wait for the IR LED

crankyoldgit, 2020-08-27:
Looks about right to me.

crankyoldgit, 2020-08-28:
@NiKiZe is pretty much correct. There are enough examples already. Adding one for each A/C protocol would result in a mess for us to maintain. So, no, we will not add one for you. However, if you want to write one and maintain it, I'd be happy to create a link to it.
You can see all the options/settings available for LG A/Cs at: https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRLgAc.html

Tweak-NL, 2020-08-28:
Okay, when they placed this AC i looked arround for controlling it with IR. And i found this one.. There is/was a LG file but no examples. So it kinda looked like maybe it's not finished.. But now i understand it should work. Im going to try this code when i receive the hardware. And again, im noob with this =)

closed 2020-08-28 by Tweak-NL

Tweak-NL, 2020-08-29:
If someone finds this question.. The code i posted works fine! Today received the transistor and IR LED and AC is turning on.