crankyoldgit/IRremoteESP8266 #1255: Timing Errors decoding Fujitsu IR

mikyjazz, 2020-08-30:
_(Please use this template for reporting issues. You can delete what ever is not relevant. Giving us this information will help us help you faster. Please also read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions) & [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide). Your problem may already have an answer there.)_

### Version/revision of the library used
_Typically located in the `library.json` & `src/IRremoteESP8266.h` files in the root directory of the library.
e.g. v2.0.0, or 'master' as at 1st of June, 2017. etc._

### Expected behavior
_What steps did you do and what should it have done?_

e.g.
1. Initialise the IRsend class.
2. IRsend.sendFoobar(0xdeadbeef);
3. Foobar branded BBQ turns on and cooks me some ribs.

### Actual behavior
_What steps did you do, and what did or didn't actually happen?_

e.g.
1. Initialise the IRsend class.
2. IRsend.sendFoobar(0xdeadbeef);
3. Foobar BBQ went into Cow(er)-saving mode and fried me a couple of eggs instead.

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
_Include some raw dumps of what the device saw._

### Steps to reproduce the behavior
_What can we do to (pref. reliably) repeat what is happening?_

#### Example code used
_Include all relevant code snippets or links to the actual code files. Tip: [How to quote your code so it is still readable](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code)._

#### Circuit diagram and hardware used (if applicable)
_Link to an image of the circuit diagram used. Part number of the IR receiver module etc. ESP8266 or ESP32 board type._

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
_Yes/No._

### Has this library/code previously worked as expected for you?
_Yes/No. If "Yes", which version last worked for you?_

### Other useful information
_More information is always welcome. Be verbose._

I am trying to emulate a Fujitsu AR-RCE1E remote control - It seems that the transmission works fine, the AC receive the commands...
 When I try to receive the codes I get wrong decoding for about 30 percent of the packets ...
When all goes well the packets are 128 or 56 bits long while when the decoding fails the packets are respectively 129 or 130 bits or 57 or 58 bits long.
Trying to read in RAW mode I get the following evidence:

these are the same packets:  "POWER OFF"

"UNKNOWN",   "Bits":58, "RawData":[3228,1656,394,418,394,418,394,1230,394,416,394,1230,394,416,394,418,394,424,396,1232,394,1232,394,304,508,416,394,416,394,1230,394,1230,396,426,394,310,502,416,394,416,394,306,504,416,396,416,396,416,396,424,396,416,398,304,504,416,394,416,394,1230,394,414,396,416,394,304,516,418,394,304,508,416,394,418,394,1232,394,416,396,416,394,426,394,416,394,1232,394,310,500,418,394,416,396,416,394,416,394,428,392,1232,392,416,394,1230,394,1232,394,1230,396,1232,392,1232,394,1230,396],"RawDataInfo":[115,115,0]}}

"FUJITSU_AC","Bits":56, "RawData":[3226,1656,394,304,508,416,394,1232,394,306,504,1232,394,416,394,416,396,304,516,1232,392,1232,394,416,394,416,394,418,392,1232,552,1074,394,426,394,416,394,416,396,416,396,416,396,414,396,416,394,416,394,426,394,416,394,302,508,416,394,416,394,1232,394,418,392,422,390,424,398,412,394,416,396,416,394,306,504,1 ...    

I could not copy the entire correct packet due to the size of the buffer

Please note the repetition of some bits with different timing

WRONG: 418,394

same position, RIGHT: 304,508

Can you address this problem?
Can you tell me how / where to change the timing of the decoding to carry out tests?
Thank you

crankyoldgit, 2020-08-31:
@mikyjazz Thanks for the report of the issue.

> Output of raw data from IRrecvDumpV2.ino (if applicable)

Per the issue template (quoted above), and the [troubleshooting guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide#receiving-an-ir-signal) there are a large number of issues that could be causing your problem. Too many to list here. As your also running this via Tasmota, there is a LOT of stuff happening on your ESP chip than just the library code. It could be something is happening just at that moment when that part of the message/signal is being received. Hence we suggest/recommend not running anything else while decoding (e.g. Using the `IRrecvDumpV2` or `V3` code examples. That will give the most reliable capture info.

That said, you can tweak the timing values for the Fujitsu protocol here:
https://github.com/crankyoldgit/IRremoteESP8266/blob/3c5137c9586aaad860dd9d268a9b72b06678577a/src/ir_Fujitsu.cpp#L18-L24

You can also adjust the decoding tolerance value (not sure how to do that in/via Tasmota, I'm not sure where it is set in Tasmota (Check with them) but in the underlying (our) library, it's set here:
https://github.com/crankyoldgit/IRremoteESP8266/blob/3c5137c9586aaad860dd9d268a9b72b06678577a/src/IRrecv.h#L35
https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/IRrecv_8h.html#a7884008b3a738dfc7bd8658655e10272
or
https://github.com/crankyoldgit/IRremoteESP8266/blob/3c5137c9586aaad860dd9d268a9b72b06678577a/src/IRrecv.h#L127
https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRrecv.html#aa091c449db70c65fd0221669df7438ea

Try lowering the bit mark value, and/or raising the zero space value till you get something reliable. Or adjust the default tolerance from `25%` to `30` to `35%` etc.

In short due to "something" in your environment, your remote's timings are being received as too far out of spec some times. It's likely something external to the ESP. The timings used are good enough to decode a normal Fujitsu A/C remote 99% of the time (I know, I have one ;-) but not all Fuji remotes are the same, so you may need to tweak it a little for your circumstances.

mikyjazz, 2020-08-31:
Hi, David,
many thanks for the quick and accurate reply.
I confirm (as you will have understood from the attached dumps) that I am using your library within Tasmota; I agree with you that many things can cause the problem and that "a lot" of things happen in the ESP chip during decoding, especially if the Tasmota is running underneath. I will try to vary the timing constants of the signal and the tolerance, eventually I will also try to look at the Tasmota code but I think it is very complex ... I really know these problems very well as I have written a lot of code for the "soft" decoding of signals since the times of the 6502 and Z80 microprocessors even if by now I deal  (unfortunately) with other kinds of software ... The problem in this case appears repetitive with the same values ​​in the error and not random (as for example it could be the occurrence of an interrupt during decoding) ... I'll first try to test outside Tasmota, as you rightly suggested ... I'll let you know if I can fix the problem.
Thanks and congratulations again for the beautiful and useful library.
Best regards
Michele Perniola

crankyoldgit, 2020-08-31:
> The problem in this case appears repetitive with the same values ​​in the error and not random (as for example it could be the occurrence of an interrupt during decoding)

As you can see by the [flow chart of how capture works](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Overview-of-how-the-library-works#capturing-an-ir-message) there is very little going on or work to be done. However, if there is some delay in servicing the interrupt, it will affect the timing values collected.

If, as you say it happens at exactly the same "bit" of the message each time, then it could be the IR remote control device itself. Try new batteries etc. See the troubleshooting checklist for more ideas. Or .. it could just be a problem with _your_ remote, or a design issue.

I'd appreciate the feedback if it is a "only when running Tasmota" issue or not etc. It will improve or lower my confidence level for data collected via that method.

That said, we have had Fuji remotes behave oddly like this in the past for some other users. Maybe they just use cheap components or something? Basically, I'm saying that, sadly you are not alone if you've got a dodgy remote.

As you can see in your data, there is a high degree of variance between the recorded values. Something (external to the library) is causing that. Without an oscilloscope reading on the IR decoder module, I can only guess if it's external to your circuit or if it is something happening _on_ the ESP based on the data so far.

mikyjazz, 2020-09-01:
Hi, David,
in the end I managed to solve the problem, all good for you! It's not a library problem!
First I built a new receiver using a nodemcu card EPS12E and a single IR receiver; flashed with Tasmota it worked 100% without any errors! This obviously made me suspicious and then I tried again with the RBridge I was using: IMEGO W30 (flashed tasmota) I looked at the IR receiver and I noticed that it was an anonymous one so I decided to replace it with the same one I was using for the newly built receiver: a VS1838B.
The situation improved but there were still some errors; looking for the differences between the two circuits I noticed that in the W30 the receiver was powered at 5V while I powered it at 3.3V in my receiver so I proceeded to change the VS1838B power supply bringing it to 3.3V also in the IMEGO W30 and all it worked 100% error free.
At this point I submit to you this thing:
The remote control as I have already told you is an AR-RCE1E which is not present in the library however it is largely decoded correctly.
There are the following keys that are not decoded:

SWING SET FIXED POSITION (any step same code)
{"IrReceived":{"Protocol":"FUJITSU_AC","Bits":56,"Data":"0x0x14630010106C93","Repeat":0,"IRHVAC":{"Vendor":"FUJITSU_AC","Model":1,"Power":"On","Mode":"Auto","Celsius":"On","Temp":16,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}

TIMER SLEEP
{"IrReceived":{"Protocol":"FUJITSU_AC","Bits":128,"Data":"0x0x1463001010FE0930A011002C090020CA","Repeat":0,"IRHVAC":{"Vendor":"FUJITSU_AC","Model":1,"Power":"On","Mode":"Cool","Celsius":"On","Temp":26,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}

TIMER OFF
{"IrReceived":{"Protocol":"FUJITSU_AC","Bits":128,"Data":"0x0x1463001010FE0930A021001E080020C9","Repeat":0,"IRHVAC":{"Vendor":"FUJITSU_AC","Model":1,"Power":"On","Mode":"Cool","Celsius":"On","Temp":26,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}

TIMER ON
{"IrReceived":{"Protocol":"FUJITSU_AC","Bits":128,"Data":"0x0x1463001010FE0930A041001E88072022","Repeat":0,"IRHVAC":{"Vendor":"FUJITSU_AC","Model":1,"Power":"On","Mode":"Cool","Celsius":"On","Temp":26,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}

TIMER CANCEL
{"IrReceived":{"Protocol":"FUJITSU_AC","Bits":128,"Data":"0x0x1463001010FE0930A00100000000200F","Repeat":0,"IRHVAC":{"Vendor":"FUJITSU_AC","Model":1,"Power":"On","Mode":"Cool","Celsius":"On","Temp":26,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}

Do you want to do it? Are you interested?
Do I do it (if you give me any suggestions)?

Cheers
Michele

crankyoldgit, 2020-09-02:
Glad you sorted your issue out. As I indicated, I was pretty sure it _wasn't_ a library issue.

> At this point I submit to you this thing:
> The remote control as I have already told you is an AR-RCE1E which is not present in the library however it is largely decoded correctly.
> There are the [...]

Try decoding it using [`IRrecvDumpV3`](https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples/IRrecvDumpV3) and you will get a lot more information about what the library really thinks about the message/protocol. Tasmota only uses the [`IRac` class interface](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRac.html), which supports only the[ _Common_ A/C modes/settings](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/structstdAc_1_1state__t.html). It will give a lot more information than Tasmota does.

E.g. You might find that it is already fully supported via the[ `IRFujitsuAC` class](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRFujitsuAC.html).

> Do you want to do it? Are you interested?

I can do it, if it proves new/different etc. Yes, but try the above first, so we can tell if it is already supported etc.

> Do I do it (if you give me any suggestions)?

Normally, I'd say sure, have a crack at it yourself, however the `IRFujitsuAC` class is a horrible spaghetti-like monster. So, don't let that be your first addition to the library. ;-)

> (by suggestion I mean the packet encoding / decoding scheme :) )

Help with that is always welcome, just with Fujitsu ... it's complicated by the same protocol shared between several models that do things differently. But let us not get to far ahead of us.

If it is supported, and works, can you tell me what Model numbers it is detected as (i.e. Is it always Model `1`?) and what the Brand/Model numbers are for the remote (AR-RCE1E) **and** the actual A/C ones too. I can then add them to the supported devices list.

mikyjazz, 2020-09-02:
Hi, David,
we assess the situation...
- I have carried out the tests with IRrecvDump3, I have verified that the message kFujitsuAcCmdStepVert is correctly identified and decoded.
- The various "timer" messages are not identified (since they are not defined in the code).
I have completely decoded the position and values ​​of the messages for setting the timer; I realized that it is the remote control (as well as the air conditioner) that keeps memory of the status of the various settings since, as is well known, it retransmits the complete status of the machine each time a command is sent.
This also applies to the timer settings; the remote control has an internal clock so - like the air conditioner - it counts up or back the time and at each new command, if the timer is set, the time is adapted ... in a nutshell if you set "on between a 'now "and then after five minutes you change the temperature, in the status package you will also find" on in 55 minutes "...
This obviously complicates things for a possible implementation; furthermore, in the "protocol" structure a field for "timer" is not defined; there is - at most - the "sleep" field that could be used.
But the problem remains of how to keep track of time; probably the IRBridge can serve many devices at the same time with a single instance so it cannot be done locally; instead it seems obvious to implement the timer function at a higher level, for example in an instance of Home Assistant (as I use) and in fact I don't think it makes sense to do it at a lower level ...
As for the "impulsive" or "toggle" commands, it should be noted the lack of feedback from the air conditioner for which the considerations on the conservation of the state are always valid and also to evaluate the opportunity to implement a similar thing (eg . economy mode or swing position) in the absence of feedback ... About the practical application I was thinking of trying to insert the nodemcu card and the photodevices inside the air conditioner ... he also has a led for the economy mode so I was thinking of the possibility of reading it through an additional phototransistor or by connecting directly to its terminals ... in this way I could at least reliably activate the economy mode, which interests me ...
I guess you may have already made all these considerations, I'm glad to share this thought with you ...
I have the dump of all the messages related to the timer setting, if you ever need them you can ask me and I will send them to you immediately.
The air conditioner I'm working on is a Fujitsu-General mod. ASHG09LLCA / AOHG09LLC
Best wishes
Michele

crankyoldgit, 2020-09-03:
> I have the dump of all the messages related to the timer setting, if you ever need them you can ask me and I will send them to you immediately.

Sure. Upload them to the issue. Or post a link etc.

> The air conditioner I'm working on is a Fujitsu-General mod. ASHG09LLCA / AOHG09LLC
So, is it a Model: 1, or does it get detected as some other model?

Re: Timer.

Yes, it gets complicated fast. Hence why the `IRac` really doesn't attempt to handle them. Their functionality differs wildly between manufacturers.
This library will not attempt to handle "time" automatically for the user. It's beyond it's scope. We do provide the ability to set the value accordingly, but it's not our job to update them.
Also, _timers_ are in a lot of ways, better done external to the protocol. e.g. Have the ESP or the home automation system just send an "on" or "off" message at the appropriate time etc. This allows more functionality than what a pure protocol-only system can offer.

If I understand you correctly, the Fuji _timers_ are a "turn on in XX minutes from now" style. Not a "at 16:05 turn on" message. Correct?

Timers are at the low end of the priority spectrum for support, but if/since you've worked out all the bits, I might as well add them. Possibly best to do/request that in a new issue too.

mikyjazz, 2020-09-03:
Hi, David,
yes, my air conditioner is recognized as model 1 except for the two 56-bit messages which are model 3 (I am attaching the above messages; I looked into the code and I don't think there is any way to distinguish the model correctly in those cases since the command code is the same for everyone and the remote control signature is the same for everyone)
the timer is correctly set with "60 minutes from now" and not with an absolute time as the remote control does not have an internal clock.
I proceed with the implementation of the timer in Home assistant ... I think it is the best solution!
Feel free to contact me if you need more info, thank you very much for your cooperation, see you soon!
Michele

POWERFUL
Code      : 0x14 63 00 10 10 39 C6 (56 Bits)
Mesg Desc.: Model: 3 (ARREB1E), Power: On, Mode: 0 (Auto), Temp: 16C, Fan: 0 (Auto), Clean: Off, Filter: Off, Swing: 0 (Off), Command: Powerful, Outside Quiet: Off

ECONOMY 
Code      : 0x14 63 00 10 10 09 F6 (56 Bits)
Mesg Desc.: Model: 3 (ARREB1E), Power: On, Mode: 0 (Auto), Temp: 16C, Fan: 0 (Auto), Clean: Off, Filter: Off, Swing: 0 (Off), Command: Econo, Outside Quiet: Off

STEP SWING V
Code      : 0x14 63 00 10 10 6C 93 (56 Bits)
Mesg Desc.: Model: 1 (ARRAH2E), Power: On, Mode: 0 (Auto), Temp: 16C, Fan: 0 (Auto), Clean: Off, Filter: Off, Swing: 0 (Off), Command: Step Swing(V)

TIMER SLEEP 2H
Code      : 0x14 63 00 10 10 FE 09 30 A0 10 01 78 08 00 20 7F (128 Bits)
Mesg Desc.: Model: 1 (ARRAH2E), Power: On, Mode: 0 (Auto), Temp: 26C, Fan: 1 (High), Clean: Off, Filter: Off, Swing: 0 (Off), Command: N/A

TIMER SLEEP 3H
Code      : 0x14 63 00 10 10 FE 09 30 A0 10 01 B4 08 00 20 43 (128 Bits)

TIMER SLEEP 5H
Code      : 0x14 63 00 10 10 FE 09 30 A0 10 01 2C 09 00 20 CA (128 Bits)

TIMER SLEEP 7H
Code      : 0x14 63 00 10 10 FE 09 30 A0 10 01 A4 09 00 20 52 (128 Bits)

TIMER SLEEP 9H
Code      : 0x14 63 00 10 10 FE 09 30 A0 10 01 1C 0A 00 20 D9 (128 Bits)

TIMER OFF 0.5H
Code      : 0x14 63 00 10 10 FE 09 30 A0 20 01 1E 08 00 20 C9 (128 Bits)
Mesg Desc.: Model: 1 (ARRAH2E), Power: On, Mode: 0 (Auto), Temp: 26C, Fan: 1 (High), Clean: Off, Filter: Off, Swing: 0 (Off), Command: N/A

TIMER OFF 1H
Code      : 0x14 63 00 10 10 FE 09 30 A0 20 01 3C 08 00 20 AB (128 Bits)

TIMER OFF 1.5H
Code      : 0x14 63 00 10 10 FE 09 30 A0 20 01 5A 08 00 20 8D (128 Bits)

TIMER OFF 2H
Code      : 0x14 63 00 10 10 FE 09 30 A0 20 01 78 08 00 20 6F (128 Bits)

TIMER OFF 2.5H
Code      : 0x14 63 00 10 10 FE 09 30 A0 20 01 96 08 00 20 51 (128 Bits)

TIMER OFF 3H
Code      : 0x14 63 00 10 10 FE 09 30 A0 20 01 B4 08 00 20 33 (128 Bits)

TIMER OFF 3.5H
Code      : 0x14 63 00 10 10 FE 09 30 A0 20 01 D2 08 00 20 15 (128 Bits)

TIMER OFF 4H
Code      : 0x14 63 00 10 10 FE 09 30 A0 20 01 F0 08 00 20 F7 (128 Bits)

TIMER OFF 4.5H
Code      : 0x14 63 00 10 10 FE 09 30 A0 20 01 0E 09 00 20 D8 (128 Bits)

TIMER OFF 5H
Code      : 0x14 63 00 10 10 FE 09 30 A0 20 01 2C 09 00 20 BA (128 Bits)

TIMER OFF 5.5H
Code      : 0x14 63 00 10 10 FE 09 30 A0 20 01 4A 09 00 20 9C (128 Bits)

TIMER OFF 6H
[... 77 lines cut ...]
Code      : 0x14 63 00 10 10 FE 09 30 A0 30 01 00 40 9A 20 05 (128 Bits)

TIMER ON 7.5H
Code      : 0x14 63 00 10 10 FE 09 30 A0 30 01 00 20 9C 20 23 (128 Bits)

TIMER ON 8H
Code      : 0x14 63 00 10 10 FE 09 30 A0 30 01 00 00 9E 20 41 (128 Bits)

TIMER ON 8.5H
Code      : 0x14 63 00 10 10 FE 09 30 A0 30 01 00 E0 9F 20 60 (128 Bits)

TIMER ON 9H
Code      : 0x14 63 00 10 10 FE 09 30 A0 30 01 00 C0 A1 20 7E (128 Bits)

TIMER ON 9.5H
Code      : 0x14 63 00 10 10 FE 09 30 A0 30 01 00 A0 A3 20 9C (128 Bits)

TIMER ON 10H
Code      : 0x14 63 00 10 10 FE 09 30 A0 30 01 00 80 A5 20 BA (128 Bits)

TIMER ON 11H
Code      : 0x14 63 00 10 10 FE 09 30 A0 30 01 00 40 A9 20 F6 (128 Bits)

TIMER ON 12H
Code      : 0x14 63 00 10 10 FE 09 30 A0 30 01 00 00 AD 20 32 (128 Bits)

crankyoldgit, 2020-09-03:
Ah. I misunderstood. I thought you had done the analysis to work out how the bits change to control the timer(s) correctly.

mikyjazz, 2020-09-03:
Yes, I do...

BUFFER(9) -> 0x00  -> TIMER STOP

BUFFER(9) -> 0x10  -> TIMER SLEEP

BUFFER(9) -> 0x20  -> TIMER OFF

BUFFER(9) -> 0x30  -> TIMER ON

FOR TIMER SLEEP and TIMER OFF

TIME (minutes) = (BUFFER(0xc) >> 4) - 0x800 + BUFFER(0xb)

FOR TIMER ON

TIME (minutes) = (BUFFER(0xd) >> 4) - 0x800 + BUFFER(0xc)

mikyjazz, 2020-09-03:
Those positions are already identified in ir_Fujitsu.cpp line 156

    remote_state[7] = 0x30;
    remote_state[8] = (_cmd == kFujitsuAcCmdTurnOn) | (tempByte << 4);
    remote_state[9] = _mode | 0 << 4;  // timer off
    remote_state[10] = _fanSpeed;
    remote_state[11] = 0;  // timerOff values
    remote_state[12] = 0;  // timerOff/On values
    remote_state[13] = 0;  // timerOn values

mikyjazz, 2020-09-04:
Hi David,
opened issue, well done!
Please note,  lower BUFFER(9) 4 bytes are "mode", you must OR...
bye
Michele

crankyoldgit, 2020-09-04:
> You must must in any case set them...

D'oh! I wish I had seen this sooner. I worked this out myself after beating my head against a wall for 30+ mins.

crankyoldgit, 2020-09-05:
Hey @mikyjazz can you please download/test branch https://github.com/crankyoldgit/IRremoteESP8266/tree/FujiTimer / PR #1262 and let me know how it goes for you.

My (unit) tests seem to indicate it should work, but nothing beats the real thing!

closed 2020-09-06 by crankyoldgit

crankyoldgit, 2020-10-04:
FYI, the changes mentioned above have now been included in the new v2.7.11 release of the library.

PR #1262 (2020-09-06): Fujitsu: Add support for timers.
* Add Sleep, On, & Off timer support.
* Remove `this->` from Fujitsu code base.
* Add unit test coverage for Timers.
* Update existing Unit tests.
* Misc other code style cleanup.

Fixes #1261
Fixes #1255
commit 7e84ab7a (2020-09-06): Fujitsu: Add support for timers. (#1262)

* Add Sleep, On, & Off timer support.
* Remove `this->` from Fujitsu code base.
* Add unit test coverage for Timers.
* Update existing Unit tests.
* Misc other code style cleanup.

Fixes #1261
Fixes #1255