crankyoldgit/IRremoteESP8266 #1256: Transcold AC (M1-F-NO-6) not working with any protocol

chandravegeta, 2020-08-31:
Recently started using this library. Not able to control Transcold AC . Protocol is showing unknown. If any protocol which works for this AC is available please share Or let us know how to create drivers for this AC .

crankyoldgit, 2020-08-31:
This is covered in the FAQ and wiki in multiple places.

See:
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#The_libraryexample_code_doesnt_understand_my_Air_Conditioner_remote_
&
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol

chandravegeta, 2020-08-31:
Thanks for sharing the links, please suggest if you have a template to create new driver for that AC. Which can then support your library . 
My basic understanding is that I can get raw data from AC remote . This AC has 24 bits data which has 8bits straight and 8 bit reverse . I have collected header mark,space and bit mark, one and zero space . All these data I have collected it .

It would be help if any templete is there so that with these data I can directly add those data into files.

The existing AC files are not complicated for me, it would be a helpful if there is standard template for this.

chandravegeta, 2020-08-31:
Also , even if i create a new driver for this AC somehow, How do i ensure that this AC still available after any library update from you guys ?   
It will be tedious work for me to manually update my files every time. 

please suggest

NiKiZe, 2020-08-31:
The Wiki have that covered as well https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-IR-protocol
Then there is the maintainers guide as well.

One other option is to complete the data that is requested in the issue template, which seems to have been removed in this case.
If you do post some examples of your raw data from dumpv2+ then the basic protocol can be added fairly quickly, but any PR is accepted as well, again see the documentation and wiki ;)

chandravegeta, 2020-08-31:
This is raw data of Transcold AC

Library   : v2.7.9

Protocol  : UNKNOWN
Code      : 0xC38A8243 (51 Bits)
uint16_t rawData[101] = {5944, 7612,  558, 3556,  556, 3556,  556, 3556,  556, 1526,  554, 3556,  556, 1528,  554, 1526,  556, 3558,  554, 1524,  556, 1528,  556, 1526,  556, 3556,  554, 1528,  556, 3556,  554, 3556,  556, 1528,  554, 1526,  556, 3556,  556, 3556,  554, 1528,  554, 1526,  554, 3558,  554, 1528,  554, 3556,  556, 3556,  556, 1526,  554, 1526,  556, 3556,  554, 3556,  554, 1526,  554, 3556,  556, 1526,  556, 1526,  554, 3558,  554, 1526,  556, 3556,  556, 1526,  556, 3554,  556, 1524,  556, 1526,  556, 3556,  556, 1526,  554, 3556,  556, 1524,  558, 3556,  554, 1526,  556, 3556,  554, 3556,  556, 7514,  556};  // UNKNOWN C38A8243

crankyoldgit, 2020-08-31:
That looks like a _very_ clean signal. Amazingly so!

The "template" you desire can be obtained from following this step: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-IR-protocol#use-the-auto_analyse_raw_datapy-program

> Also , even if i create a new driver for this AC somehow, How do i ensure that this AC still available after any library update from you guys ?
It will be tedious work for me to manually update my files every time.

It (The process) is called a "Git Pull Request" (Or PR for short), do a google search for that term, and it will tell you more on how it works.
e.g.
https://blog.axosoft.com/learning-git-pull-request/#:~:text=You%20make%20local%20code%20changes,changes%20before%20they%20become%20final.

chandravegeta, 2020-08-31:
Thanks for the reply.

As i told the Transcold is an Air conditioner , i need it to be working from Aircon Page where i can control ON/OFF Temp, mode, Fanspeed and swing . In the document for new IR protocol ,the INAX.cpp file is pretty straight forward, it doesn't hold good for Aircon page for AC.

If you have a document which explains where and how you are reading input from Aircon page and then push the bits based on what temp, mode, fanspeed, swing.  this will speed up my process in creating the driver for this AC.

crankyoldgit, 2020-08-31:
[Inax](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Inax.cpp) is NOT an A/C protocol.

If you look at the list of [supported protocols](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/SupportedProtocols.md) it will tell you which ones are A/C protocols. e.g. [Argo](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Argo.cpp).

> If you have a document which explains where and how you are reading input from Aircon page and then push the bits ...

No, there is no documentation really on how to do that, as if/when the library has full support, the `IRMQTTServer` example & the [`IRac`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRac.html) class interface it will _"just work"&trade;_ automatically.

So all you need to do is to **read** and follow the two wiki documents already linked.
i.e.
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-IR-protocol & https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol

They _really_ do provide you with all the info/steps/guides required to get support added and to assist you in reverse-engineering what bits in the messages control what etc.

crankyoldgit, 2020-09-01:
@iamDshetty I've added "basic" protocol support for Transcold in branch https://github.com/crankyoldgit/IRremoteESP8266/tree/Transcold / PR #1258 

Can you please download, test it, & give me feedback please?

The library in that branch should be able to send & decode basic (i.e. Hex codes) Transcold messages.

i.e. You should now be at this [step](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#a-note-on-collecting-data) in the [Adding support for a new AC protocol](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol) wiki page.

Decoding the bit order and what the bits means (reverse-engineering the protocol) is solely up to you now.

iamDshetty, 2020-09-01:
Thanks, 
But yesterday i tried using coolix file and modified for transcold AC. Seems to be working. 

Doing final testing to ensure everything is working as expected. After confirmation will share the files.

Anyhow thanks again for sharing test file. Will test that as well.

crankyoldgit, 2020-09-15:
@iamDshetty How your progress going?

iamDshetty, 2020-09-17:
I have used Coolix files and changed Constant values based on Transcold AC. It worked for me.
 i have created a pull request, not sure if its properly reflecting( Sorry my bad i am new to github). 

Anyhow , attaching the files here. (added new files ir_Transcold.h and ir_Transcold.cpp)
[src.zip](https://github.com/crankyoldgit/IRremoteESP8266/files/5236465/src.zip)

crankyoldgit, 2020-09-19:
> I have used Coolix files and changed Constant values based on Transcold AC. It worked for me.
> i have created a pull request, not sure if its properly reflecting( Sorry my bad i am new to github).

I think you may have created a PR against your own repository, not this one.
 
> Anyhow , attaching the files here. (added new files ir_Transcold.h and ir_Transcold.cpp)
> [src.zip](https://github.com/crankyoldgit/IRremoteESP8266/files/5236465/src.zip)

Thanks, I'll look into incorporating it soon.

crankyoldgit, 2020-09-21:
@iamDshetty Can you please download and try out branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/transcold_detailed

It should be all of your code plus some improvements and fixes.

Let me know how it goes.

iamDshetty, 2020-09-21:
@crankyoldgit  Sorry, i didn't made the PR properly. thanks for helping out.
Will check the changes and let you know.

iamDshetty, 2020-09-21:
@crankyoldgit , Its Working.

Another minor concern, library will bulk up due to additional AC protocols in future. Any plans for that?

crankyoldgit, 2020-09-21:
> @crankyoldgit , Its Working.

Excellent. Thanks for confirming. Did you test send or receiving/decoding, or both?

> Another minor concern, library will bulk up due to additional AC protocols in future. Any plans for that?

There already exists ways to turn of un-needed protocols to save resources. See:
https://github.com/crankyoldgit/IRremoteESP8266/blob/bfa52019fc9aed7ca6c68c4cbcd3b12075249c5f/src/IRremoteESP8266.h#L62-L88

crankyoldgit, 2020-09-21:
@iamDshetty When adding some test cases to verify your code, I've come across what I think are some errors.
e.g.
```
const uint8_t kTranscoldTempOffset = 8;
const uint8_t kTranscoldTempSize = 4;
```
&
```
const uint8_t kTranscoldSensorTempOffset = 8;
const uint8_t kTranscoldSensorTempSize = 4;
```

They both seem to be referring to the same nibble (4 bits). i.e. The 9th to the 12th Least Significant Bits of the message code.
i.e. `remote_state`

What's the story with that? Do you have a bit-by-bit breakdown/description) of the structure of `remote_state` somewhere at all?

iamDshetty, 2020-09-22:
@crankyoldgit 
Tested only SEND.  Decode not tested. it will take me few weeks to get back the remote.

 getTempRaw() only this is required, setSensorTempRaw() not required. 

Bit Calculation can be found in below link:
https://docs.google.com/spreadsheets/d/1qdoyB0FyJm85HPP9oXcfui0n4ztXBFlik6kiNlkO2IM/edit?usp=sharing

crankyoldgit, 2020-09-22:
> @crankyoldgit
> Tested only SEND. Decode not tested. it will take me few weeks to get back the remote.
> 
> getTempRaw() only this is required, setSensorTempRaw() not required.

Oh. What (features) were tested? and what other aspects are not supported/not required/not working etc?

> Bit Calculation can be found in below link:
> https://docs.google.com/spreadsheets/d/1qdoyB0FyJm85HPP9oXcfui0n4ztXBFlik6kiNlkO2IM/edit?usp=sharing

Thanks.
Looking at that data, it looks like only Mode, Fan speed, Power, Temp, & Swing (not sure if it is vertical or horizontal, or something else.) Correct?
i.e. I should strip out support for turbo, clean, light, sleep, sensor, zone follow etc.

iamDshetty, 2020-09-22:
Its an old AC. Tested ON/OFF, Modes, TEMP, FAN Speeds.

These (turbo, clean, light, sleep, sensor, zone follow) are not there in this AC and its not required.
Only these (Mode, Fan speed, Power, Temp, & Swing) buttons are there.

crankyoldgit, 2020-09-22:
Okay, I've removed everything else but those.
Please re-download and test that branch.

closed 2020-09-24 by crankyoldgit

crankyoldgit, 2020-10-04:
FYI, the changes mentioned above have now been included in the new v2.7.11 release of the library.

PR #1278 (2020-09-24): Transcold: Add detailed support.
* Manually merge in the work done by @iamDshetty
* Lots of code clean up to get it working properly.
  - Code style
  - Better constants.
  - Lint
  - Bugs
  - Remove dead/unused code & settings.
* Add unit test for basic features.
* Settings/Features supported:
  - Mode
  - Fan speed
  - Power
  - Temp
  - Swing

* Sending confirmed working in: https://github.com/crankyoldgit/IRremoteESP8266/issues/1256#issuecomment-696055698

Fixes #1256
commit 5e1731af (2020-09-24): Transcold: Add detailed support. (#1278)

* Manually merge in the work done by @iamDshetty
* Lots of code clean up to get it working properly.
  - Code style
  - Better constants.
  - Lint
  - Bugs
  - Remove dead/unused code & settings.
* Add unit test for basic features.
* Settings/Features supported:
  - Mode
  - Fan speed
  - Power
  - Temp
  - Swing

* Sending confirmed working in: https://github.com/crankyoldgit/IRremoteESP8266/issues/1256#issuecomment-696055698

Fixes #1256