crankyoldgit/IRremoteESP8266 #1260: Add functionality for econo mode and Fahrenheit for neoclima

griffisc306, 2020-09-04:
I have most of it worked out. Will fill in details shortly

griffisc306, 2020-09-04:
A/C Model: Soleus Air TTWM1-10-01
Remote: ZCF/TL-05

Econo Mode:
Power On, Fan High, 16C, Econo Mode Off

Library   : v2.7.10

Protocol  : NEOCLIMA
Code      : 0x00000000000D00220020A5F4 (96 Bits)
Mesg Desc.: Power: On, Mode: 1 (Cool), Temp: 16C, Fan: 1 (High), Swing(V): Off, Swing(H): On, Sleep: Off, Turbo: Off, Hold: Off, Ion: Off, Eye: Off, Light: Off, Follow: Off, 8C Heat: Off, Fresh: Off, Button: 13 (UNKNOWN)
uint16_t rawData[197] = {6046, 7412,  538, 594,  548, 556,  540, 588,  510, 596,  508, 600,  474, 598,  540, 590,  544, 560,  540, 588,  540, 592,  508, 574,  504, 598,  536, 566,  546, 560,  540, 544,  502, 598,  540, 588,  542, 562,  512, 594,  502, 576,  504, 598,  510, 598,  538, 566,  482, 596,  508, 624,  538, 568,  544, 592,  478, 598,  480, 624,  544, 558,  538, 568,  506, 626,  482, 628,  502, 600,  510, 594,  51
0, 596,  480, 628,  502, 600,  508, 626,  532, 570,  542, 1656,  512, 572,  502, 1692,  510, 1660,  510, 594,  540, 568,  480, 628,  506, 596,  542, 592,  544, 560,  508, 576,  502, 600,  536, 568,  510, 622,  510, 600,  506, 598,  542, 588,  546, 1626,  544, 564,  482, 624,  508, 626,  508, 1688,  542, 592,  578, 556,  508, 626,  480, 598,  540, 564,  508, 600,  484, 596,  506, 600,  510, 596,  506, 628,  478, 626,  508, 594,  540, 568,  508, 576,  542, 560,  546, 1652,  540, 568,  480, 598,  546, 1652,  542, 566,  508, 1644,  506, 596,  510, 596,  542, 1660,  484, 596,  548, 1652,  540, 594,  550, 584,  508, 1640,  508, 624,  544, 1630,  544, 1630,  508, 1670,  504, 1694,  510, 7424,  542};  // NEOCLIMA
uint8_t state[12] = {0x00, 0x00, 0x00, 0x00, 0x00, 0x0D, 0x00, 0x22, 0x00, 0x20, 0xA5, 0xF4};

Power On, Fan High, 16C, Econo Mode On Pushed "Energy Saver" Button 0x0D State[5]
 
Library   : v2.7.10

Protocol  : NEOCLIMA
Code      : 0x00000010000D00220020A504 (96 Bits)
Mesg Desc.: Power: On, Mode: 1 (Cool), Temp: 16C, Fan: 1 (High), Swing(V): Off, Swing(H): On, Sleep: Off, Turbo: Off, Hold: Off, Ion: Off, Eye: Off, Light: Off, Follow: Off, 8C Heat: Off, Fresh: Off, Button: 13 (UNKNOWN)
uint16_t rawData[197] = {6114, 7402,  538, 594,  542, 534,  506, 596,  506, 594,  548, 556,  548, 554,  548, 556,  536, 540,  506, 622,  508, 594,  548, 556,  548, 582,  546, 584,  546, 556,  540, 538,  506, 596,  510, 620,  548, 556,  548, 582,  548, 582,  548, 556,  542, 536,  506, 596,  510, 594,  548, 582,  548, 582,  546, 582,  548, 584,  546, 1646,  550, 582,  548, 582,  546, 532,  508, 622,  508, 596,  550, 554,  548, 580,  548, 556,  542, 536,  508, 596,  546, 556,  548, 1646,  510, 596,  548, 1620,  540, 1606,  504, 622,  508, 596,  548, 554,  548, 556,  546, 584,  510, 568,  506, 596,  548, 556,  550, 556,  542, 536,  508, 596,  550, 554,  548, 582,  550, 1648,  544, 564,  502, 574,  512, 592,  548, 1648,  546, 558,  550, 532,  528, 602,  506, 622,  514, 590,  536, 572,  544, 534,  506, 596,  548, 558,  510, 594,  548, 584,  538, 568,  506, 598,  546, 558,  550, 558,  540, 1656,  504, 574,  508, 596,  508, 1686,  538, 568,  540, 1612,  502, 600,  546, 558,  546, 1624,  546, 562,  480, 1664,  546, 586,  540, 590,  512, 1688,  542, 590,  510, 624,  504, 576,  506, 624,  542, 592,  536, 7392,  546};  // NEOCLIMA
uint8_t state[12] = {0x00, 0x00, 0x00, 0x10, 0x00, 0x0D, 0x00, 0x22, 0x00, 0x20, 0xA5, 0x04};

The Bit changed is state[3]. states are 0x00 = off and 0x10 = on

Temp button: (C/F)
Power On, Fan High, 16C, Econo Mode Off C/f Button state[5] 0x1E
Library   : v2.7.10

Protocol  : NEOCLIMA
Code      : 0x00000010001E00220020A515 (96 Bits)
Mesg Desc.: Power: On, Mode: 1 (Cool), Temp: 16C, Fan: 1 (High), Swing(V): Off, Swing(H): On, Sleep: Off, Turbo: Off, Hold: Off, Ion: Off, Eye: Off, Light: Off, Follow: Off, 8C Heat: Off, Fresh: Off, Button: 30 (UNKNOWN)
uint16_t rawData[197] = {6072, 7434,  552, 580,  548, 556,  482, 596,  508, 596,  548, 556,  548, 584,  548, 532,  506, 596,  510, 620,  550, 556,  548, 556,  536, 542,  508, 596,  550, 556,  546, 534,  506, 596,  510, 620,  550, 582,  550, 554,  550, 532,  506, 596,  514, 590,  550, 582,  550, 558,  508, 596,  508, 596,  550, 556,  550, 554,  542, 1606,  508, 596,  550, 582,  550, 556,  548, 586,  510, 570,  508, 596,  552, 580,  548, 582,  550, 530,  508, 596,  550, 556,  550, 582,  548, 1602,  504, 1666,  548, 1620,  538, 1660,  540, 596,  548, 584,  510, 598,  506, 624,  514, 592,  546, 584,  540, 594,  512, 570,  506, 596,  548, 582,  550, 558,  508, 600,  508, 1690,  544, 560,  510, 596,  542, 540,  508, 1690,  550, 554,  548, 586,  540, 592,  482, 598,  512, 594,  540, 566,  506, 578,  504, 600,  548, 556,  508, 574,  508, 622,  544, 564,  546, 584,  484, 598,  510, 622,  542, 1632,  542, 594,  504, 576,  506, 1692,  530, 578,  514, 1640,  504, 596,  542, 566,  510, 1662,  484, 600,  510, 1686,  516, 1686,  514, 594,  482, 1666,  538, 568,  512, 1662,  480, 630,  504, 598,  516, 618,  540, 7394,  538};  // NEOCLIMA
uint8_t state[12] = {0x00, 0x00, 0x00, 0x10, 0x00, 0x1E, 0x00, 0x22, 0x00, 0x20, 0xA5, 0x15};

state[7] 0x22 = Celsius. state[7] 0xA2 = Fahrenheit
At min temp, temp bit does not change when changing modes. after that, it changes at different intervals

at 20 C/ 68F
uint8_t state[12] = {0x00, 0x00, 0x00, 0x10, 0x00, 0x1E, 0x00, 0xA2, 0x00, 0x27, 0xA5, 0x9C}; 68F
uint8_t state[12] = {0x00, 0x00, 0x00, 0x10, 0x00, 0x1E, 0x00, 0x22, 0x00, 0x24, 0xA5, 0x19}; 20C

uint8_t state[12] = {0x00, 0x00, 0x00, 0x10, 0x00, 0x02, 0x00, 0xA2, 0x00, 0x2E, 0xA5, 0x87}; 75F
uint8_t state[12] = {0x00, 0x00, 0x00, 0x10, 0x00, 0x1E, 0x00, 0x22, 0x00, 0x28, 0xA5, 0x1D}; 24C

uint8_t state[12] = {0x00, 0x00, 0x00, 0x10, 0x00, 0x02, 0x00, 0xA2, 0x00, 0x36, 0xA5, 0x8F}; 83F
uint8_t state[12] = {0x00, 0x00, 0x00, 0x10, 0x00, 0x1E, 0x00, 0x22, 0x00, 0x2C, 0xA5, 0x21}; 28C

MAX Temp: 
uint8_t state[12] = {0x00, 0x00, 0x00, 0x10, 0x00, 0x02, 0x00, 0xA2, 0x00, 0x3D, 0xA5, 0x96}; 90F
uint8_t state[12] = {0x00, 0x00, 0x00, 0x10, 0x00, 0x1E, 0x00, 0x22, 0x00, 0x30, 0xA5, 0x25};32C

let me know if there any other info you could use

crankyoldgit, 2020-09-05:
Thanks for all of that. I've added support for `Econo` in branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/More_Neoclima

I'll add the Fahrenheit stuff soon.

crankyoldgit, 2020-09-05:
@griffisc306 If you want to try the aforementioned [branch ](https://github.com/crankyoldgit/IRremoteESP8266/tree/More_Neoclima)(inc PR #1265 ) it should now support the Economy & native Fahrenheit settings. Can you please test it out and let me know how you go?

griffisc306, 2020-09-05:
My goodness thank you for such a quick response. I should have it tested either tonight or tomorrow.

griffisc306, 2020-09-06:
This worked perfectly!! Thank you again.

crankyoldgit, 2020-09-06:
Thanks for confirming.

closed 2020-09-06 by crankyoldgit

crankyoldgit, 2020-10-04:
FYI, the changes mentioned above have now been included in the new v2.7.11 release of the library.

commit bfc057b4 (2020-09-05): Neoclima: Add Fahrenheit options.
* Add methods to use native Fahrenheit settings.
* Support the C/F button on the remote.
* Update/extend the unit tests accordingly.

Fixes #1260
PR #1265 (2020-09-06): Neoclima: Add Economy & Fahrenheit support
* Update supported models for this protocol.
* Economy and Fahrenheit support added.
* Unit tests updated/added.
* `IRac` support updated.
* Remove `this->` from `ir_Neoclima.cpp`

Fixes #1260
commit aff9cc31 (2020-09-06): Neoclima: Add Economy & Fahrenheit support (#1265)

* Update supported models for this protocol.
* Economy and Fahrenheit support added.
* Unit tests updated/added.
* `IRac` support updated.
* Remove `this->` from `ir_Neoclima.cpp`

Fixes #1260