crankyoldgit/IRremoteESP8266 #1261: Fujitsu Timer support

crankyoldgit, 2020-09-04:
Yes, I do...

BUFFER(9) -> 0x00  -> TIMER STOP

BUFFER(9) -> 0x10  -> TIMER SLEEP

BUFFER(9) -> 0x20  -> TIMER OFF

BUFFER(9) -> 0x30  -> TIMER ON

FOR TIMER SLEEP and TIMER OFF

TIME (minutes) = (BUFFER(0xc)  &07 ) <<  0x8 + BUFFER(0xb) 

FOR TIMER ON

TIME (minutes) =(BUFFER(0xd)  & 7f) << 0x8 + BUFFER(0xc) >> 0x4

ok... stop editing :D :D :D

_Originally posted by @mikyjazz in https://github.com/crankyoldgit/IRremoteESP8266/issues/1255#issuecomment-686460057_

crankyoldgit, 2020-09-04:
> TIMER ON 12H
Code : 0x14 63 00 10 10 FE 09 30 A0 30 01 00 00 AD 20 32 (128 Bits)

@mikyjazz is 12H the maximum value the remote can set for any of the timers?

mikyjazz, 2020-09-04:
it is the maximum value allowed from the remote control...
Don't know what happens if you set greater value by software...

crankyoldgit, 2020-09-04:
Hey @mikyjazz can you please download/test branch https://github.com/crankyoldgit/IRremoteESP8266/tree/FujiTimer / PR #1262 and let me know how it goes for you.

My (unit) tests seem to indicate it should work, but nothing beats the real thing!

mikyjazz, 2020-09-05:
I'll try asap

mikyjazz, 2020-09-06:
Hi David,
the commands seem to work even if I haven't really been with the stopwatch to check the right shutdown and start times ;) Unfortunately, on the air conditioner you don't have the feedback so you can't really know if the timer is set at the right time if you're not there to check with the stopwatch :)
I rewrote a few lines of code (at line 155) ; it seems cleaner and more controlled to me. what do you think about it?
About the list of supported models
// Brand: Fujitsu General, Model: AOHG09LLC A / C (ARRAH2E)
is the outdoor unit ...
Let me know, see you soon
Michele

  if (fullCmd) {  // long codes
    uint8_t tempByte = _temp - kFujitsuAcMinTemp;
    // Nr. of bytes in the message after this byte.
    remote_state[6] = _state_length - 7;

    uint8_t timerType = getTimerType();
    remote_state[7] = 0x30;
    remote_state[8] = (_cmd == kFujitsuAcCmdTurnOn) | (tempByte << 4);
    remote_state[9] = _mode | (timerType << 4);
    remote_state[10] = _fanSpeed;

    // Set the On/Off/Sleep timer Nr of mins.
    switch (timerType)
    {
      case kFujitsuAcSleepTimer:
      case kFujitsuAcOffTimer:
        remote_state[11] = getOffSleepTimer() & 0xFF;
        remote_state[12] = ((getOffSleepTimer() >> 8) & 0x07) | 0x08;
        remote_state[13] = 0;
        break;
      case kFujitsuAcOnTimer:
        remote_state[11] = 0;
        remote_state[12] = ((getOnTimer() & 0x0F) << 4);
        remote_state[13] = ((getOnTimer() >> 4) & 0x3F) | 0x80;
      default:
        remote_state[11] = remote_state[12] = remote_state[13] = 0;
        break;
    }

    switch (_model) {

crankyoldgit, 2020-09-06:
Thanks for confirming it works.
I agree, your code is easier to understand, however one of the goals of the library is to be as small as possible/practical whilst still being readable. You version will likely produce more code and thus be slightly larger in it's flash footprint. We are under pressure from projects such as ESPEasy and Tasmota to save bytes where we can. So, alas, I won't be using your suggested code.

FYI, your code has some redundant stuff in it. Not sure if the compiler will/would optimise it out.
e.g.
`remote_state[11] = getOffSleepTimer() & 0xFF;` -> `remote_state[11] = getOffSleepTimer();`  No need to mask as it's being shoved into a `uint8_t`
&
`remote_state[12] = ((getOnTimer() & 0x0F) << 4);` -> `remote_state[12] = getOnTimer() << 4;`  Ditto.

closed 2020-09-06 by crankyoldgit

mikyjazz, 2020-09-07:
Hi David,  I agree with you about code size issues, however, please consider, for example, using the switch to set the data correctly, for example the setbit (3 ...) (7 ...) according to the needs (of the timer type) and not in every chance because unexpected results can always occur ...
uint8_t is not a native type and the behavior of a compiler may be different; all variable size adjustments should be made explicitly :)
I am attaching a link ...
http://www-users.math.umn.edu/~arnold//disasters/ariane.html#:~:text=Specifically%20a%2064%20bit%20floating,and%20thus%20the%20conversion%20failed.

Ok :) good job anyway! Thank you! 
Michele

crankyoldgit, 2020-10-04:
FYI, the changes mentioned above have now been included in the new v2.7.11 release of the library.

PR #1262 (2020-09-06): Fujitsu: Add support for timers.
* Add Sleep, On, & Off timer support.
* Remove `this->` from Fujitsu code base.
* Add unit test coverage for Timers.
* Update existing Unit tests.
* Misc other code style cleanup.

Fixes #1261
Fixes #1255
commit 7e84ab7a (2020-09-06): Fujitsu: Add support for timers. (#1262)

* Add Sleep, On, & Off timer support.
* Remove `this->` from Fujitsu code base.
* Add unit test coverage for Timers.
* Update existing Unit tests.
* Misc other code style cleanup.

Fixes #1261
Fixes #1255