crankyoldgit/IRremoteESP8266 #1267: How to get AC remote model using IRrecveDumpV2

MarkEvens, 2020-09-09:
I am trying to decode protocol using this example so i get protocol perfectly but don't know how to get remote model number, There is any function to get model number. your help is greatfull.

crankyoldgit, 2020-09-09:
If the library knows about a model number for the given protocol, it will print it out (as best it can) on the serial port with the rest of the information (e.g. IRrecvDumpV2 or V3)
Not all protocols support model numbers. Model numbers within a protocol are _made-up_ constructs by this library when trying to deal with slightly different aspects/operations of certain remotes & a/c's .

i.e. There is no `model number` buried in part of the message.

**_If and only if_** a protocol supported by this library _has support for different models_, then there is a `ac.getModel()` function/method in the applicable class.

These are the only protocols with _models_ the library has:
https://github.com/crankyoldgit/IRremoteESP8266/blob/7e84ab7a197da02dfc7e5b22ef6dc479783c7acc/src/IRsend.h#L119-L167

crankyoldgit, 2020-09-10:
Just to be clear(er), Model Numbers does/means nothing on most protocols. If IRrecvDumpV2+ doesn't list a model number, then it doesn't matter what the Model number is set to, because it isn't used.

closed 2020-09-10 by crankyoldgit