crankyoldgit/IRremoteESP8266 #1268: Multiple receiver objects do not work seperately

bilkosem, 2020-09-10:
I am currently using version 2.5.6. 

I need to create multiple objects of IRrecv class and receive IR signals from both of these objects. However when I define 2 IRrecv object the last one overwrites the first one and it force me to use only 1 IR receiver at a time.

For example I defined 2 IRrecv object on pins 5 and 12 of ESP8266 Generic
```
IRrecv irrecv_1(5, kCaptureBufferSize, kTimeout, true);
IRrecv irrecv_2(12, kCaptureBufferSize, kTimeout, true);
```
when I send IR signals to pin 5 I condition returns false and code only gets into the "else", but when I send IR signals to pin 12, this if statement returns true and code gets into the "if".
```
if (irrecv_1.decode(&results)) {}
else{}
```

I check the IRrecv.h and IRrecv.cpp. The reason is probably these lines in IRrecv.cpp:
```
volatile irparams_t irparams;
irparams_t *irparams_save;  // A copy of the interrupt state while decoding.
```
This `irparams` object does not belong to the class IRrecv. So for each instance of IRrecv overwrites the `irparams` and even if I define 3 different IRrecv object, when I tried to read input from them, I will only read input from the pin of the last IRrecv object.

I tried to use this variable as a attribute of IRrecv, but coud not make it yet.

crankyoldgit, 2020-09-10:
See the FAQ item on this topic: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#Does_this_library_support_multiple_distinct_IR_capture_sources

NiKiZe, 2020-09-10:
If one where to ignore limited resources, for example would the ESP32 be able to handle this?
If so would it be possible to support this via a compile time flag?

Not suggesting it being made, just thinking out loud if it is possible.

crankyoldgit, 2020-09-10:
One of the problems is that it requires a separate interrupt service routine per input. Thus it cost IRAM which is in very short supply. AFAIK there is no way to dynamically allocated that at run time, it has to be done at compile time. Which would result in a horrible mess to implement.

**IF** there was some way of knowing dynamically which gpio caused the interrupt, then we could have a shared routine and allow multiple receivers at run time. I tried looking into it, there is no simple way.

In short, IRAM is too limited a resource on the ESP family to waste for a option that is only **very** rarely used/requested.

crankyoldgit, 2020-09-10:
In addition, the [read_timeout()](https://github.com/crankyoldgit/IRremoteESP8266/blob/7e84ab7a197da02dfc7e5b22ef6dc479783c7acc/src/IRrecv.cpp#L76) timer interrupt would need to be duplicated. There are a total of 4 timers on the ESP32, and I think only 1 (not sure) on the ESP8266. They too (inc function) would need to be duplicated.

I've tried to make it happen/work before and quickly devolved into ugly code, sub-optimal scenarios and compile-time options to this correctly.

For the rare occasion that people need dual+ receivers, it's for handling multiple directions. That can be achieved by circuitry means. i.e. Tying the IR module data lines together and just not knowing which module received the signal, and it all works.
e.g. It covers probably 90% of the already rare cases where people want/need multiple IR receiver modules.

If people are wanting to use an IR recv module for say, independent rooms in a house, and will have exactly the same brand/protocol remote in each room, to control a specific devices in a room. i.e. be able to tell which room the signal originated in. Then the above solution won't work off-the-shelf.

They would be better served spending $5+ and getting a separate ESP module, or creating latch circuit(s) to signal/indicate to the ESP which IR module got the signal which is beyond the scope of this project/library to do or provide that.

bilkosem, 2020-09-17:
thank you for the response. I'm closing the issue but I will update you If I manage to make it work stably,

closed 2020-09-17 by bilkosem