crankyoldgit/IRremoteESP8266 #1277: Support for Samsung AR12TXEAAWKNEU

xorbital, 2020-09-21:
### Version/revision of the library used
Latest dev (bfa5201)

### Actual behavior
I'm trying to use this library, more specifically the MQTT server example with my AC, however this fails.
It works for the most part with the ControlSamsungAC example, however using the mqttserver doesn't work. The reason is that (I've confirmed this with ESPEasy) some of the commands seem to differ.
Example: Sending "Cool, 22°C, Fan Auto" works, but sending "Cool, 22°C, Fan Auto, Beep on" seems to break the protocol, since the A/C doesn't do anything when sending this command.

The model is a Samsung AR12TXEAAWKNEU.

#### Circuit diagram and hardware used (if applicable)
The one from the wiki with a 1 Ohm resistor added in series with the LED, works.

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
_Yes._

### Other useful information
It's a bit tricky. Have a look at the remote:
[image]
There are a lot of buttons, but those actually do nothing on my unit:

- Purify
- Indirect / Direct

This is intended as stated in the manual for the A/C, seems like other models have different feature combinations then.
I've captured a lot of IR codes with the arduino-heatpumpIR library and decoded the signal as far as I could, is this applicable to this library as well? If not, I'll create some dumps with your tools, which ones would make the most sense?
This is what I could get, it is the command for Cooling, 16°C, Windfree (Fan Auto):
`02,92,0F,00,00,00,F0,01,D2,FE,7B,00,11,F0`
```
02 - ??
92
  9 - Quiet mode: 8: quiet / special, 9: normal
  2 - ?
0F - ??
00
  0 - 0: normal, 1: indirect, 2: direct
  0 - ??
00 - ??
00
  0 - Beep, 0: off, 2: on
  0 - ?
F0 - ??
01 - ??
D2 - Checksum 
FE
  F - Swing: A: vertical, B: horizontal, C: vertical+horizontal, F: off
  E - Not sure; F:swing, E: no swing, also E: 17-30°C, F: 16°C
7B
  7 - options? 7: normal, 6: Display off, E: Clean
  B - B: Windfree, 1: normal, 7: Fast, F: ECO
00
  0 - temperature (0: 16°C bis E: 30°C), always 8 in Fan Mode, when selecting "Fast" the last set temperature is sent
  0 - ?? 0: (normal), [1: purify] 2: clean
11
  1 - Mode; 1: Cool, 2: Dry, 3: Fan, 4: Heat, 8: Auto
  1 - Fanspeed; 1: Auto, 5: Low, 9: Medium, B: High, F: Turbo
F0 - ??
```
Some bytes change without making too much sense to me, like `F` after the checksum controls the swing, but `E` is also there when the temp. is set to 16°C, weird. 
Some combinations of commands are invalid, like setting a high fan speed in dry mode. Also, as you might see there's horizontal swing in this unit as well.
Turning the unit on or off results in a longer chain off commands, which include timer information, haven't taken a look at that yet.

Let me know what additional information I can provide, I'll have a look in a day :+1:

crankyoldgit, 2020-09-21:
> I've captured a lot of IR codes with the arduino-heatpumpIR library and decoded the signal as far as I could, is this applicable to this library as well? If not, I'll create some dumps with your tools, which ones would make the most sense?

Dunno. Maybe. Please use data captures using our tools. e.g. IRrecvDumpV2 or V3.

Samsung's protocol has been tricky for us to work out & support. There are certainly aspects of it we don't understand.

Our existing "Beep" byte is in _byte[13]_ (the last byte), where as you report it as in _byte[5]_, yet your temperature share the same byte as in our support. That is _byte[11]_

This indicates where there is a good overlap of the existing model support, but they do differ.
We are going to need to work out how we can detect the model difference from a raw incoming IR signal some how.

For reference, here is the Issue/thread for adding Samsung A/C support (#505) .. as you'll see, it's a mess.

If you can express your finding & data in a Google Spreadsheet, that'd be great/handy/helpful to get it supported.
See https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol

xorbital, 2020-09-22:
Thank you for your quick answer :slightly_smiling_face: 
I've pointed my remote at the IR receiver, but I could only pick up UNKNOWN codes, no matter the mode.
I've played around with the DumpV2 and DumpV3, changed `kTimeout` and `kMinUnknownSize`, however no luck sadly. 

After a bit more testing it seems like the beep command is not even persistent, and just toggles the state of it in the A/C.

I took a look at #505, yeah, looks like it :slightly_frowning_face: 

Since I only got "UNKNOWN" as a result, is this of any use? I can expand it if it is: https://cryptpad.fr/sheet/#/2/sheet/view/e1PSfhwjfTGCAPbse4h28RjvvG+YCiLgH3SxhxTNFQQ/

crankyoldgit, 2020-09-24:
@xorbital After looking at the raw data (e.g. Row 5 of your spreadsheet) and doing a lot of debugging, I was able to find out that your "raw data" is terribly out of spec/tolerance for the library to match it by default.
I was able to get it to match at a value of 40%. e.g. `irrecv.setTolerance(40);` or change:
https://github.com/crankyoldgit/IRremoteESP8266/blob/bfa52019fc9aed7ca6c68c4cbcd3b12075249c5f/src/IRrecv.h#L35

See https://github.com/crankyoldgit/IRremoteESP8266/commit/f3b4d7585e2b6f10ffe3258ed31da464571df1d8 for more details.

There are a number of things that could cause this (in no particular order):
a) Low batteries in the remote. (Possible)
b) Environmental noise. (Unlikely)
c) Your A/C remote is just old/out of spec/poor etc. (Possible)
d) The hardware IR demodulator you are using is not of sufficient quality. (Likely)
e) You're running something else on the ESP chip that is bogging it down. (Unlikely)
f) The distance you're capturing it at is too short/long etc. (Possible)

Anyway, there is nothing in the library that needs fixing. The protocol isn't a new variant, and it does "match" with a lot of coercion.

You've at least got a software-only settings change solution to your issue. Good luck.

xorbital, 2020-09-24:
@crankyoldgit Thank you very much!

a) Probably not, the remote works across the whole room and it doesn't show a battery warning.
b) Unlikely as you said, I've tried receiving while it was inside a box, so completely dark, didn't get anything different
c) The A/C is one of the most recent models, highlighted by the fact that it can only use the glorious smartthings cloud instead of a local API. Maybe they did something crazy to the IR protocol..
d) Yeah, this could be it. I'll order a new one and report back. Any suggestions?
e) Not that I know of, I can swap it for a ESP32 for testing and report back.
f) Alright, I'll try some distance changes, guess I buy a new demodulator first.

In the meantime I will try receiving codes with the tolerance change you've suggested. There are probably still issues with the data since sending is inconsistent.

xorbital, 2020-09-24:
So, decoding is still not great, setting the tolerance to 40-55% helps, but power on and power off commands can't be decoded 95% of the time. However, since the other commands work most of the time, I've found the output to be identical to what https://github.com/ToniA/Raw-IR-decoder-for-Arduino gives me, interestingly that works, it can't decode all commands, but spit out the hex values of the bytes themselves.
It's probably some weird hardware issue on my side, since this runs on an arduino (same cheap IR receiver), I'll test further. However, I can confirm that the data in the first post is correct, dumpv3 and rawirdecode output the same for a command like the one mentioned in the first post.
Still, I have to take a look at the stuff that's sent when pressing the power button, here's an example of "Cool, 22°C, Fan Auto":
```
02,92,0F,00,00,00,F0,01,D2,FE,71,60,11,F0
```
This is the stuff that's sent when pressing the button to turn it off now:
```
02,B2,0F,00,00,00,C0,01,D2,0F,00,00,00,00,01,F2,FE,71,60,11,C0
```
And turning it back on:
```
02,92,0F,00,00,00,F0,01,D2,0F,00,00,00,00,01,D2,FE,71,60,11,F0
```
Taking a closer look:
"Normal": 02,92,0F,00,00,00,F0,01,D2,**FE,71,60,11**,F0
Power off: 02,B2,0F,00,00,00,C0,01,D2,0F,00,00,00,00,01,F2,**FE,71,60,11**,C0
Power on: 02,92,0F,00,00,00,F0,01,D2,0F,00,00,00,00,01,D2,**FE,71,60,11**,F0

The longer commands also get sent when setting a timer, here's the same setting with a timer of 0.5h to turn the A/C off:
```
02,92,0F,00,00,00,F0,01,A2,3F,00,00,04,00,01,D2,FE,71,60,11,F0
```
Increasing the timer one step at a time:

02,92,0F,00,00,00,F0,01,B2,**8**F,0**0**,00,04,00,01,D2,FE,71,60,11,F0
02,92,0F,00,00,00,F0,01,92,**B**F,0**0**,00,04,00,01,D2,FE,71,60,11,F0
02,92,0F,00,00,00,F0,01,B2,**0**F,0**1**,00,04,00,01,D2,FE,71,60,11,F0
02,92,0F,00,00,00,F0,01,92,**3**F,0**1**,00,04,00,01,D2,FE,71,60,11,F0
02,92,0F,00,00,00,F0,01,A2,**8**F,0**1**,00,04,00,01,D2,FE,71,60,11,F0

Can this library work with those codes? If so, I'll continue decoding everything into a spreadsheet :slightly_smiling_face:

crankyoldgit, 2020-09-24:
Glad you got it some-what working.

SamsungAC, as I indicated earlier, is a mess. It is one of, if not _the_ protocol we've had most trouble with.

To answer some of your questions:

Yes, it can handle the "extended"/long codes.
e.g.

```
uint8_t long_state[kSamsungAcExtendedStateLength] = [0x02, 0x92, 0x0F, 0x00, 0x00, 0x00, 0xF0, 0x01, 0xA2, 0x8F, 0x01, 0x 00, 0x04, 0x00, 0x01, 0xD2, 0xFE, 0x71, 0x60, 0x11, 0xF0];
irsend.sendSamsungAC(long_state, kSamsungAcExtendedStateLength);
```

There is limited support for these "extended" messages in the [`IRSamsungAC`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRSamsungAc.html) class.

The biggest hold-back from the library working with the "extended" messages (i.e. Power on/off, & timer etc) is we've been unable to consistently determine/replicate the checksum calculation for them. The "normal" ones, we've got worked out mostly. Extended, well, I've wasted far too many hours and had no success. Solve that issue, and we can look at those messages.

The library uses custom "known good" values for sending the on & off extended messages at present when sending via the class method.
See:
https://github.com/crankyoldgit/IRremoteESP8266/blob/bfa52019fc9aed7ca6c68c4cbcd3b12075249c5f/src/ir_Samsung.cpp#L383-L407

crankyoldgit, 2020-09-24:
> b) Unlikely as you said, I've tried receiving while it was inside a box, so completely dark, didn't get anything different

Unless the box is quite big, it was probably too close. About 1 meter is probably optimal. Less than 30cm and I've seen problems with some demodulators.

> d) Yeah, this could be it. I'll order a new one and report back. Any suggestions?

**Don't** get a _VS1838b_. See/read: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#help-im-getting-very-inconsistent-results-when-capturing-an-ir-message-using-a-vs1838b-ir-demodulator

Read the discussions there in. See also: https://www.analysir.com/blog/2014/12/08/infrared-receiver-showdown-tsop34438-vs-vs1838b-winner-revealed/

xorbital, 2020-09-24:
> The library uses custom "known good" values for sending the on & off extended messages at present when sending via the class method.

Ah that makes sense, since I've seen my A/C turn on to 24°C before switching to the set temperature.
What I've also noticed is that the "Power On" message actually isn't necessary for my unit (don't know if that's true for other ones as well?)
So just sending the short commands also turns on the A/C, no matter what.
Turns out the one I have actually is a VS1838B, although I always used 5V for it. While I wait for a better one to arrive, it seems to work consistent with my setup now.

I'll take a look at the checksum calculation soon, in the meantime I'll create a spreadsheet with all the other commands I could decode so far.

Thank you for your time :+1:

crankyoldgit, 2020-09-24:
> Turns out the one I have actually is a VS1838B, although I always used 5V for it. While I wait for a better one to arrive, it seems to work consistent with my setup now.

Be careful if you use it on an ESP32. 5V in via a GPIO can kill the ESP. The 8266 can take 3.3/5V input. The 32 can't.

xorbital, 2020-09-24:
Yes, I read that in the FAQ. Again, decoding works with the arduino for now, sorry for wasting your time with that.
Still, the main issue is that sending doesn't work since the data is different, so I'll continue capturing that.

xorbital, 2020-09-24:
I think that's all: https://cryptpad.fr/sheet/#/2/sheet/view/REURkHbGuQdU+wQkBcwKBXmccftjIB+Tyzin34EXs-g/

Gonna take a look at the long codes and the checksum issues next.

crankyoldgit, 2020-09-27:
> EDIT: Not sure about Power on/off, however this unit can be powered on and off without long messages.

Ooo. That's an interesting development! Can you supply a `kSamsungAcStateLength` (i.e. not extended) hex/state code for some other users to try to see if the shorter "off" code works for them too?

xorbital, 2020-09-27:
> Ooo. That's an interesting development! Can you supply a `kSamsungAcStateLength` (i.e. not extended) hex/state code for some other users to try to see if the shorter "off" code works for them too?
I'll have a look into it! The trick is to cover the LED of the remote, turn it on, uncover it and then set it to the setting you want. This sends out a short signal, but it turns on the A/C.
Similarly, if you turn off the remote while covering it, you'll block the long signal, however if you then set a timer (also sends out long signal) and then unset it, it will send out a short command that will turn it off. This might be possible with other units as well.

crankyoldgit, 2020-10-04:
FYI, the changes mentioned above (i.e. the additional unit tests) have now been included in the new v2.7.11 release of the library.

xorbital, 2020-10-04:
Nice. Still probably due to my shitty IR decoder. I'll run a few sending tests now so that I can finish up the short codes.

xorbital, 2020-10-04:
@crankyoldgit I've run a few tests, sending codes and receiving. I can confirm this is how my A/C works with short codes: https://cryptpad.fr/sheet/#/2/sheet/view/REURkHbGuQdU+wQkBcwKBXmccftjIB+Tyzin34EXs-g/

The trick to get the short codes for turning the A/C off is to:

- Turn the A/C off with the remote
- Set a timer on the remote (e.g. on in 0.5h)
- Change the settings to which the A/C should wake up to (e.g. reduce the temperature)

This sends out a short code which will turn off the A/C. Some features are ignored when sending the power off command, like fan speed and swing, but since the unit is off I don't think that matters.
Again, for my unit, you don't need long codes at all. Maybe someone with a different model can test this behaviour with their own unit as well. The long codes are used for timers only and the "good' sleep" function, which regulates the temperature while sleeping. Neat idea, doesn't work well though, so I wouldn't really miss it.

Is there anything else I should provide?

crankyoldgit, 2021-08-08:
When PR #1554 is merged, we should be able to use custom extended/long Samsung messages. So we me able re-visit this issue soon.

crankyoldgit, 2021-08-12:
So, PR #1554 has been merged, and PR #1562 is waiting for user testing/merging (It enables extended messages again) we can probably recommence work on this again.
I had a quick look at the spreadsheet again. Can you summarise what features I need to look at to add/fix etc to catch me up?

xorbital, 2021-08-12:
Hey, thank you for getting back :)

It's been a while, but I have yet to decode the timer and "good sleep" stuff.
Some of the feature combinations, e.g. swing + cool + quiet didn't work back then, also the combinations of vertical / horizontal swing didn't work.
Windfree was also not working.
I can take another look at it first though.

Should I try uncovering issues with the current samsung code or rather decode the rest of the features?

crankyoldgit, 2021-08-12:
> Should I try uncovering issues with the current samsung code or rather decode the rest of the features?

That's really your call. My _slight_ preference is testing the changes in PR #1562 before they are merged.

xorbital, 2021-08-29:
Thanks for the update, I'll have a look into it as soon as possible, don't have that much time right now 😞

xorbital, 2021-09-10:
Alright, so I've tried the MQTT server and sent out a few commands via the web UI, as well as the ControlSamsungAC example, sadly the swing doesn't work on either one properly nor can I turn off the unit via either of them.

Seems like my heatpump is just different enough :disappointed: 

I'll finish decoding the long codes, including timers and the goodsleep-feature.
Also, I'm using a different IR receiver instead of my good ol' VS1838b.

crankyoldgit, 2021-09-10:
> I'm keeping track of the long codes in https://cryptpad.fr/sheet/#/2/sheet/view/Mhqn80CETnO6ZZRVVLEWwz1ovPV7b8oOP3TbUUHE4mA/

Thanks. So far I see only one long/extended code in the spreadsheet. Is that correct?
Could you please add/label the nibbles at the top of the sheet with their nibble or byte position? (Start at `0` please)
Also include the `state[]` line as well. That saves me form having to convert it for the tests I will need to do.
e.g.
```
       |      Byte 0     |      Byte 0     |   ...   | Action Description
       +--------+--------+--------+--------+-- ... --+-------------------
Nibble |   0    |    2   |    B   |   2    |   ...   | Pressed "Good Sleep" to activate it. Was in "Cool, 22C, Fan Low" before.
       +--------+--------+--------+--------+-- ... --+-------------------
```

> Alright, so I've tried the MQTT server and sent out a few commands via the web UI, as well as the ControlSamsungAC example, sadly the swing doesn't work on either one properly nor can I turn off the unit via either of them.

I'll need the modified example code,  and the MQTT commands (and displayed state before and after the command) to debug those.

xorbital, 2021-09-19:
I've recorded quite a few codes now, although the combinations are endless.
I might need to expand the section with the combined timers if the few I recorded aren't sufficient.
The longer 21 byte codes are **only** required for timers, and the only interesting one is the Good Sleep Timer, otherwise the shorter 14 byte codes are enough for controling the device.
A few observations:
- The device has the following modes: Auto, Cool, Dry, Fan and Heat.
- You can only use "direct", "indirect", "quiet", "fast" or "eco" when running in Cool or Heat mode.
- You can't use Windfree in Heat or Auto mode.
- In some modes, the Fan Speed is locked at Auto Speed.
- You can't combine the swing functionality with Windfree, direct nor indirect.
- SwingH, SwingV or SwingHV are the only swing modes supported, you can't tell the device to point the air stream in a specific location, only to swing up/down and/or left/right
- You can turn on the device with any valid power on code, and turn it off with a short one too.

Although I've been using a new IR sensor and verified each receive, some codes may still be faulty, although they look okay now.

File is uploaded to https://cryptpad.fr/sheet/#/2/sheet/view/35lwyMKs3jw55HrPgEVjA0ZS1unOBA75oO6FIiWpiYU/
And attached as well.

Is this formatting okay? Or should I record them differently?
[Samsung_codes.xlsx](https://github.com/crankyoldgit/IRremoteESP8266/files/7192863/Samsung_codes.xlsx)

crankyoldgit, 2021-09-20:
> The longer 21 byte codes are **only** required for timers, and the only interesting one is the Good Sleep Timer, otherwise the shorter 14 byte codes are enough for controling the device.

Except the Off (& On, I think) messages are _21 byte codes_ if I recall correctly.

I'll take a look at your sheet shortly.

xorbital, 2021-09-20:
Oh yeah, forgot to mention, used https://github.com/ToniA/Raw-IR-decoder-for-Arduino with an Arduino and - most of the time - it spit out 5 Bytes of `0`s before the actual code, however my unit doesn't care wether I send those 0s in front or not.

crankyoldgit, 2021-09-22:
I'll try to get to it at or by the weekend.

crankyoldgit, 2021-09-29:
@xorbital Sorry for taking so long. Life's been busy lately.

I've looked at the spreadsheet (the web one & xlsx file)

For the "Good" Sleep, I need you to do the analysis (or point out where the analysis is) as to which bits and bytes are that controls/affects the sleep operation, and how the values are stored etc. e.g. https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#value-ranges

What I've found useful in my own sheets & others is:
* To "Freeze" the top few rows etc, which identify why byte etc each column is.
* Move the descriptions of the state/action etc to the left most column, and similarly freeze that. That's so it's always visible what the row is etc.
* Colour code the cells that change for each function. e.g. pink for "Good Sleep"; blue, for power control; green for checksum etc etc.
* Add the "function" of the byte/nibble. Under the "byte[]" values and in the Freeze rows etc. So it can be easily seen.
* Put a legend/table saying what each colour means. Perhaps a note saying how the bits are used in a `state[]` byte etc. eg. Byte[10] is the "Geek Factor", it uses `bits 0-5` to store the integer. e.g. `0b00000` (0) is Geek is completely off, `0b10101` (21) is the maximum "Full Geek" mode. Geek Factor numbers range from 0 to 21. Or ... `0` means 0 mins, `1` means 30 mins. `2` is 60 mins, etc up to 12hrs (24 (dec) `0x18` hex)

I need that (later) sort of detail to be able to write the code for it.

FYI, this is the breakdown we have for the existing 14 & 21 byte versions of the messages:
https://github.com/crankyoldgit/IRremoteESP8266/blob/ca44fa33bbe872c8b646e6c7be4b270968268bd7/src/ir_Samsung.h#L44-L145

I'm not expecting you to document it like the code, nor work out every detail, just the bits you need for the timer(s) you want.
However, it should help point out where the checksums are so you can know to safely ignore those changing bits etc.

crankyoldgit, 2021-10-11:
Friendly ping. Chasing up on that extra info/breakdown etc.

xorbital, 2021-10-16:
I'll try having a look at it soon, thanks for your replies :) @IceOnly does sending my codes work for you as well without adding the stuff you've posted?

IceOnly, 2021-10-16:
Yes. But i need the long message to power on/off my device. 
I found this workaround in the source 
https://github.com/crankyoldgit/IRremoteESP8266/issues/604#issuecomment-475020036 this code would be  obsolete with the implementation of SamsungAC v2.

IceOnly, 2021-10-16:
[IR.Codes.Samsung.xlsx](https://github.com/crankyoldgit/IRremoteESP8266/files/7358348/IR.Codes.Samsung.xlsx)
@xorbital are your sure with the swing set half byte? I recoreded some data from my remote to determine the checksum algorithm. This field looks different for my remote :(

IceOnly, 2021-10-16:
OK, my remote is following this rule: https://github.com/crankyoldgit/IRremoteESP8266/issues/1538#issuecomment-894645947
If your remote realy save the swing set in the first half of Byte 9 i will open an new issue.

If not I think we have now everything to implement this specific samsung_ac ir protocol.

xorbital, 2021-10-17:
Sorry for taking so long, @crankyoldgit 
I've re-done the whole process for the short codes just now, using the template you've proposed
If the layout is alright, I'll repeat the process for the long codes.
Web: https://cryptpad.fr/sheet/#/2/sheet/view/qBPhJyWI0v639S4Q7b14A2Q-ON8CarfHeT89SGICktY/
Using a custom rawirdecode told me that the checksum matches for some commands, so I've highlighted the byte as well
[Samsung_Decode.xlsx](https://github.com/crankyoldgit/IRremoteESP8266/files/7360030/Samsung_Decode.xlsx)
.

crankyoldgit, 2021-10-19:
> Sorry for taking so long, @crankyoldgit I've re-done the whole process for the short codes just now, using the template you've proposed If the layout is alright, I'll repeat the process for the long [...]

I can probably work with that (cryptpad link).
Is anything out of place in your findings from:
https://github.com/crankyoldgit/IRremoteESP8266/blob/9f79eddef1ef9264d91b50f2eae56a277b22ce45/src/ir_Samsung.h#L44-L90

If so, what?

xorbital, 2021-11-03:
Here's the updated spreadsheet (new sheet with long codes inside).
Every time the unit is powered on or off, a long code gets sent, each time a setting (e.g. temp, mode etc.) gets changed while the unit is running, a short code is sent.
The combination of on-/off-Timers are a little tricky, I've captured a few of them. If you need more, let me know, it's quite an annoying process to capture them all.
Cryptpad: https://cryptpad.fr/sheet/#/2/sheet/view/r9k8pmELYEjLyC71cD7EsThEYgKGLJygREZ5pVfNkS8/
Attach: 
[Samsung_Decode.xlsx](https://github.com/crankyoldgit/IRremoteESP8266/files/7468687/Samsung_Decode.xlsx)

crankyoldgit, 2021-11-04:
@xorbital Thanks for that, it's insightful.
I was aware that "normal" messages which contain typical settings are the "short" codes. And power + timers etc were in the "long" codes.

FYI, I think your spreadsheet is incorrect in the "checksum" locations. You're missing the first checksum (which you've labelled "Pwr") and you're off by half a byte on the others.
See:
https://github.com/crankyoldgit/IRremoteESP8266/blob/21f26e04837f3ea0b0b76ec837e901b1953ee9f1/src/ir_Samsung.h#L97-L141

(Thinking aloud here!)
Now, looking at the bit's you've identified for the timers/sleep etc., they map to:
```c++
    // Byte 9
    uint8_t Sum2Upper  :4;
    uint8_t OffTimePt1 :4;
    // Byte 10
    uint8_t OffTimePt2 :4;
    uint8_t OnTimePt1  :4;
    // Byte 11
    uint8_t OnTimePt2  :4;
    uint8_t            :4;
    // Byte 12
    uint8_t OffTimePt3     :1;
    uint8_t OnTimerEnable  :1;
    uint8_t OffTimerEnable :1;
    uint8_t Sleep          :1;
    uint8_t OnTimePt3      :1;
    uint8_t                :3;
```

`OnTimePt3` and `OffTimePt3` seem to indicate a "day" / 24h.

Joining Pt2 + Pt1 together gives us:
`0x03` == `0b00000011` (`0b0000` & `0b0011`) == 3 == 0.5h
`0x08` == `0b00001000` (`0b0000` & `0b1000`) == 8== 1h
`0x0B` == `0b00001011` (`0b0000` & `0b1011`) == 11== 1.5h
`0xA0` == `0b10100000` (`0b1010` & `0b0000`) == 160== 20h
`0xA8` == `0b10101000` (`0b1010` & `0b1000`) == 168== 21h

So the pattern seems to be:
The 5 Most Significant Bits gives the number of hours. e.g. from `0xA8` we get the top 5 being `0b10101` == 21hrs.
The 3 Least Significant Bits gives the number of 10 minute increments. e.g. from `0x0B` we get the lower 3 being `0b011` == 3 (lots of 10 mins) == 30mins _plus_ the top 5 being `0b00001` == 1h, so 1h30m total.
i.e.
The joined "byte" **divided** by 8 gives us the number of hours; The remainder (mod 8) gives us the number of 10 mins in addition to the hours. e.g.
**0xA8:** `(168 / 8) * 60 + (168 % 8) * 10 = 1260 + 0 = 1260 mins` which is 21 hours.
**0x0B:** `(11 / 8) * 60 + (11 % 8) * 10 = 60 + 30 = 90 mins` which is 1.5 hours.

Does that match what you're seeing?

xorbital, 2021-11-05:
Thank you for the fast and detailed analysis! :)
Yes, I might be wrong about the checksums, the irdecoder from ToniA sometimes detected a long code successfully and told me the checksum matches, so I assumed it'd be this byte.
Your proposed pattern makes sense 👍  I thought about it like +3=30min, +8=60min, +11=90min added to an even number of hours, but I guess I'm not used to crunching binary numbers 😄 
I could capture some more codes if that would help to confirm the pattern.

crankyoldgit, 2021-11-05:
> I could capture some more codes if that would help to confirm the pattern.

No need yet, as long as you don't find any discrepancies in your data to that formula. if you do, then yes. please report the case.
I'll code something up soon.

crankyoldgit, 2021-11-05:
Actually, it would be helpful if you set a timer, send the message. Then wait over 10 mins, and see if you can get a different value. I.e. if the remote can send time values with a ten minute resolution, if they have to be in 30 min increments.

xorbital, 2021-11-05:
Ah, you think it might update the unit in that case?
I don't think so since the timers work without the remote and since the remote can only be set to .5h increments, but I'll give it a few tries.

crankyoldgit, 2021-11-05:
The best way I can think to test it would be to set an on timer to 30 mins, wait 11 mins, then set an off timer for 60 mins.
In theory the off timer message will include the previous on timer settings. It shouldn't be an on timer settings of 30 in the second message.

Just a guess

xorbital, 2021-11-05:
Ah, I understand. I'll give it a try now

xorbital, 2021-11-05:
Well, looks like you know more about my unit than I do, next you'll probably discover a hidden water dispenser.
Sent the 1h off timer at :45 and :54 while the 0.5h on timer was active
```
12:33:12.367 -> 0,2,B,2,0,F,0,0,0,0,0,0,C,0,0,1,A,2,0,F,3,0,0,0,0,2,0,0,0,1,0,2,F,F,7,1,4,0,1,1,C,0
12:45:43.011 -> 0,2,B,2,0,F,0,0,0,0,0,0,C,0,0,1,9,2,8,F,1,0,0,0,0,6,0,0,0,1,0,2,F,F,7,1,4,0,1,1,C,0
12:54:27.598 -> 0,2,B,2,0,F,0,0,0,0,0,0,C,0,0,1,A,2,8,F,0,0,0,0,0,6,0,0,0,1,0,2,F,F,7,1,4,0,1,1,C,0
```

csv only:
```
0,2,B,2,0,F,0,0,0,0,0,0,C,0,0,1,A,2,0,F,3,0,0,0,0,2,0,0,0,1,0,2,F,F,7,1,4,0,1,1,C,0
0,2,B,2,0,F,0,0,0,0,0,0,C,0,0,1,9,2,8,F,1,0,0,0,0,6,0,0,0,1,0,2,F,F,7,1,4,0,1,1,C,0
0,2,B,2,0,F,0,0,0,0,0,0,C,0,0,1,A,2,8,F,0,0,0,0,0,6,0,0,0,1,0,2,F,F,7,1,4,0,1,1,C,0
```

crankyoldgit, 2021-11-06:
@xorbital I've created branch https://github.com/crankyoldgit/IRremoteESP8266/tree/samsung_timers / PR #1662 which should add the On & Off timers. Please download and test etc. Let me know how it goes.
I'll add the Sleep stuff in a later commit to that branch.
It's not super covered with unit tests yet, but it probably works. Should work for decoding via `IRrecvDumpV3` etc.
Sending is currently untested.

crankyoldgit, 2021-11-06:
@xorbital Can you please supply your remote model number if you can find it?

crankyoldgit, 2021-11-06:
And now that branch should also have sleep timer support. Please test etc etc.

I think that's it. Is there anything else I needed to add?

xorbital, 2021-11-06:
> @xorbital Can you please supply your remote model number if you can find it?

The model number is "DB96 24901C", seems like there are multiple variants with and without windfree mode, that's the one I have.

Thank you very much for your time, I'll be testing in a few hours.

crankyoldgit, 2021-11-06:
> The model number is "DB96 24901C"

Thanks. Added.

xorbital, 2021-11-06:
> And now that branch should also have sleep timer support. Please test etc etc.
> 
> I think that's it. Is there anything else I needed to add?

First of all, thank you very much for your time. One thing I'd like you to add is a way to donate as a way to give back :D 
Anyway, from my testing:
- Horizontal Swing doesn't seem to be supported in the Samsung protocol implementation, is that correct? IRrecvDumpV3 doesn't pick it up. Vertical Swing only gets picked up when not combined with horizontal Swing.
- Beep: On is swapped with the Filter Reset one, (Only sending "Filter Reset" from the remote causes Beep: On to show up) that's probably because of bad data on my end, I'll investigate further.

- All timers and everything else works :)

crankyoldgit, 2021-11-06:
> First of all, thank you very much for your time. One thing I'd like you to add is a way to donate as a way to give back :D

Haha. I'm fine with no donations at this point.
You can buy me a beverage if ever we meet.

> Anyway, from my testing:
> 
> * Horizontal Swing doesn't seem to be supported in the Samsung protocol implementation, is that correct? IRrecvDumpV3 doesn't pick it up. Vertical Swing only gets picked up when not combined with horizontal Swing.

Correct. No one has reverse engineered that bit of had that setting... Yet.

> * Beep: On is swapped with the Filter Reset one, (Only sending "Filter Reset" from the remote causes Beep: On to show up) that's probably because of bad data on my end, I'll investigate further.

It could be a model difference or poor earlier analysis. Dunno.

> * All timers and everything else works :)

Awesome! Thanks for the verification!

closed 2021-11-06 by crankyoldgit

xorbital, 2021-11-07:
> > * Horizontal Swing doesn't seem to be supported in the Samsung protocol implementation, is that correct? IRrecvDumpV3 doesn't pick it up. Vertical Swing only gets picked up when not combined with horizontal Swing.
> 
> Correct. No one has reverse engineered that bit of had that setting... Yet.

Since for my unit it's just setting F-Off, A-Vertical, B-Horizontal, C-Both in the first Nibble of Byte 9 / 16 I hope this can be resolved at some point - but I guess there's more to it than that?
> 
> > * Beep: On is swapped with the Filter Reset one, (Only sending "Filter Reset" from the remote causes Beep: On to show up) that's probably because of bad data on my end, I'll investigate further.
> 
> It could be a model difference or poor earlier analysis. Dunno.

Hm, I've just confirmed my original capture, and the conclusion was correct - the very last nibble sends the "beep toggle" or "filter reset" command, 4=Beep, 2=Filter Reset

crankyoldgit, 2021-11-09:
> Since for my unit it's just setting F-Off, A-Vertical, B-Horizontal, C-Both in the first Nibble of Byte 9 / 16 I hope this can be resolved at some point - but I guess there's more to it than that?

Probably not. Create a new issue and I'll do something about it.

I'll try to look back at the original history of the beep setting/analysis to see if I can find something.

crankyoldgit, 2021-11-09:
Your horizontal swing should now be supported in a new branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/samsung_swing / PR #1667 

Please confirm if it works as you expect.

xorbital, 2021-11-10:
The new PR works for setting the Swing Mode, thank you very much, I'm now testing the new Eco Setting.

Two things aren't working properly, if you want I can create a new issue for each instead.

- I didn't notice it earlier, the sleep timer isn't set properly, sending `Power: On, Mode: 4 (Heat), Temp: 21C, Fan: 0 (Auto), Swing(V): Off, Swing(H): Off, Beep: Off, Clean: Off, Quiet: Off, Powerful: Off, Breeze: Off, Light: On, Ion: Off, Sleep Timer: 08:00` results in:

Original Remote: 
```
0x02820F000010F001A20F04000C0001D2FE715041F0
```
IRremoteESP8266:
```
0x02920F000000F001A20F04000C0001D2FE715041F0
```
They both get decoded to the same command string, but the A/C only sets the sleep timer for the first one.
Seems like Byte 1 and 5 need to be different.

- The "Beep" Setting isn't set correctly for my unit:

Original Remote (since "Beep" is a toggle command, it can't be sent while turning on the unit):
```
0x02920F000000F001C2FE715041F4 
```
IRremoteESP8266:
```
0x02920F000000F001D20F0000000001C2FE715041F2
```
Seems like the very last nibble is causing issues. I've checked the codes sent by the remote again, and confirmed that the last nibble normally is 0, 2 when sending the "Filter Reset" command and 4 when sending the "Beep" command.

I've also noticed that the [Tasmota Docs](https://tasmota.github.io/docs/Tasmota-IR/#sending-irhvac-commands) (at the bottom of that section) state that the sleep timer is specified in seconds. However, the library is expecting this value to be in minutes.
I don't know if that's just an issue for the Samsung Protocol or maybe a documentation issue on their end.

crankyoldgit, 2021-11-10:
Yep, new issue for both please. This issue has gone on far to long and too much history. It will be easier for people to track later in separate issues.

crankyoldgit, 2021-11-10:
FYI, when reporting the code for a A/C message., please use the `state[]` code from a normal `IRrecvDumpV3` prog's output. It means I don't have to convert stuff by hand for the code/testing data.

crankyoldgit, 2021-11-10:
The PR #1667 branch should now be able to construct the same sleep message as the remote generated.

commit d860a19e (2021-11-06): SamsungAc: Add sleep timer support.
* Handle interactions between on, off, & sleep timers.
* Add `setSleepTimer()` & `getSleepTimer()`.
* Add sleep support to SamsungAc in `IRac`.
* Add & extend Unit tests accordingly.
* Change some parameters/variable names to better suiting names.
* Update supported devices.

Fixes #1277
PR #1662 (2021-11-06): [SamsungAc] Add support for On, Off, & Sleep Timers
* Merge Powerful & Breeze bit settings.
* Add basics for setting and decoding On, Off, & Sleep timers in extended messages.
* Unit testing for timer functions.
* Remove optional checksum calcs from `send()` & `sendExtended()` since checksums now work fine.
* Improve logic for when an Extended Message needs to be sent.
* Add `sleep` support for SamsungAc to `IRac`.
* Change some parameters/variable names to better suiting names.
* Update supported devices.

Fixes #1277
commit 01c92ca6 (2021-11-06): [SamsungAc] Add support for On, Off, & Sleep Timers (#1662)

* Merge Powerful & Breeze bit settings.
* Add basics for setting and decoding On, Off, & Sleep timers in extended messages.
* Unit testing for timer functions.
* Remove optional checksum calcs from `send()` & `sendExtended()` since checksums now work fine.
* Improve logic for when an Extended Message needs to be sent.
* Add `sleep` support for SamsungAc to `IRac`.
* Change some parameters/variable names to better suiting names.
* Update supported devices.

Fixes #1277