crankyoldgit/IRremoteESP8266 #1279: Building a decoder

gokul12051997, 2020-09-23:
@crankyoldgit I'm trying to write a decoder based on your foot tiles. I'm using a voltas remote for writing the decoder.
@NiKiZe  I'm trying to write my own decoder. When I observed the Ir burst I found only 2 pulses one with 1.6ms and another with 3.5ms which kept of repeating a unique way to form a 195ms burst. So I started writing by recoding the occurrence of those two pulses and there by formed the 10byte hex data which matched @crankyoldgit decoded results. 
The issue I'm facing right now is that, if don't disturb the system for like 10mins and then send the Ir burst, the decoded data is completely wrong.
Another senenario which I faced is like if I bring a mobile phone in between the remote and the receiver, the decoded result is completely wrong again, which I thought was because of interference and not having a start bit in the voltas ir burst.

crankyoldgit, 2020-09-23:
Are you using this library? or are you writing your own completely from scratch?

> The issue I'm facing right now is that, if don't disturb the system for like 10mins and then send the Ir burst, the decoded data is completely wrong.

This sounds like your "system" is picking up IR noise and not discarding it appropriately. The hardware IR module will occasionally pick up random IR noise that matches a 940nm IR source modulated at 38kHz so it starts reporting a pulse.

This library discards IR signals reported by the hardware IR module if it's only seen a few pulses within X many milli-seconds.
e.g. If you've only got 3-4 pulses inside of 90ms, than that's probably NOT a real IR message, and it should probably "reset" back to assuming is seen/heard no pulses.

gokul12051997, 2020-09-23:
@crankyoldgit I'm writing from the scratch. I've tapped the ir burst in an oscilloscope and found that the pulse train is made up of 1.6ms and 3.5ms. I find those pulses using gpio interrupt and record the data. The decoded data is perfectly matching your decoded values. 
As you mentioned I'm having struggles in eliminating the noises. It would be of great help if you guide me in the correct path, right now I'm hard struck at the point.
Also please let me know if I'm wrong in my approach.

crankyoldgit, 2020-09-23:
For starters, why don't you use the library. It will decode it for you. ;-)

But, assuming you want to do it all from scratch, and that you **only** want to decode this **one** protocol, then I would recommend a [finite-state machine](https://en.wikipedia.org/wiki/Finite-state_machine) approach.

e.g.

[S0] Reset the data, elapsed time timer, & nr_of_bits_so_far.; Wait for the GPIO pin to Rise/Edge trigger up/goes HIGH etc. (ie. Start looking for a bit mark.)
|
V
[S1] Wait unto the GPIO goes to LOW; Save the elapsed timer time in useconds. Reset the timer time. If the time elapsed is NOT approx the bit mark time, go to [S0]. Otherwise, we've found a bit mark.
|
V
[S2] If the nr_of_bits == to the expected number of bits, Go to [S4]
|
V
[S3] Wait for the GPIO pin to Rise/Edge trigger up/goes HIGH etc. Wait unto the GPIO goes to LOW; Save the elapsed timer time in useconds. Reset the timer time. Shift the data to the left. Nr_of_bits_so_far++; If the saved time is approx the Zero Space time, go to [S1]; If the saved time is approx the One Space time, Add 1 to data, and go to [S1]. Otherwise, go to [S0].

[S4] Wait 100,000 usecs (a guess). If there is was no GPIO goes HIGH in that time, we've finished/matched etc. Report the Data.
|
V
[S0] i.e. Listen for a new code again.

gokul12051997, 2020-09-24:
@crankyoldgit I've followed your state machine and still I'm facing the same problem. Is it ok if I mail you the snippet of the code, so that you can pit point the exact location where I've went wrong.

crankyoldgit, 2020-09-24:
[Q] Have you got a `delay(n)` or `yield()` call in your loop? The ESP will watchdog reset if that isn't called every few seconds.

> I've followed your state machine and still I'm facing the same problem. 

The state machine was off the top of my head, so there could be bugs in it. i.e. Completely untested & untried.

> Is it ok if I mail you the snippet of the code, so that you can pit point the exact location where I've went wrong.

Paste the code here. However, this isn't _stackoverflow_, so no promises. :-)
see/read:
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#i-want-to-change-the-example-code-to-make-it-do-something-different-but-i-dont-know-how-please-help-me
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide#things-we-are-unlikely-to-help-you-with

i.e. It's not exactly anything to do with this library, especially when it already works in this library. ;-)

I do however, suggest you add plenty of debugging `Serial.println()`s to your code to see what's going on etc. (just don't put them in your interrupt routines if you are using them.)

gokul12051997, 2020-09-24:
@crankyoldgit I'm pretty new to this environment and I was very interested in knowing your algorithm. Since I found it difficult to decipher, I started to write my own algorithm with was almost similar to the state machine you mentioned. At the same time I'm not asking you to help me write, since because your a pro I thought you would be able to help with the algorithm just by having a small glance at my snippet. I can paste it here. No issues intended : )

crankyoldgit, 2020-09-24:
The library itself doesn't use a simple state machine, because it's trying to match approx 90 protocols. A _simple_ state machine approach doesn't exist/scale for that. ;-)

That state-machine algorithm/approach is bespoke to that one protocol.

This is a simplification of how the library works for capture/decoding.
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Overview-of-how-the-library-works

gokul12051997, 2020-09-24:
@interrupt void gpio_c(void)  //rising edge interrupt
{
	if(pol==0)
      {
        TIM3_CNTRH=0;
	TIM3_CNTRL=0;
        k=0;
	i=9;
        pol=1;
	tp=0;
        for (l=0;l<=9;l++)
	{
	 hex[l]=0;
	}
        
        TIM3_CR1 |= (1 << TIM3_CR1_CEN);  //6.4us per tick

      }
else
  {
 if(pol==1)
	{
		TIM3_CR1 &=~(1 << TIM3_CR1_CEN); 
		v1=(uint16_t)((TIM3_CNTRH <<8)+(TIM3_CNTRL));
		TIM3_CNTRH=0;
		TIM3_CNTRL=0;
			
		if(v1>545 && v1 <567)  //time stamp 1
		{
		hex[i]|=(1UL<<(7-k)); 
		}
		else if(v1>248 && v1 <255)  //time stamp 0
	        {
	        hex[i] &=~(1UL <<(7-k));
		}
	
		k++;

        if(k==8 && i>0)
		{
		   i--;
		   k=0;
		}
		
	if(i==0 && k==8)
		{
		  pol=0;
		  flag=1;
		  tp=1;
		}
		
		
	 if(tp==0)
	 {
		TIM3_CR1 |= (1 << TIM3_CR1_CEN);
		}
	}
	}
}

main()
{
  while(1)
 {
   if(flag==1)
	{
	 flag=0;
	 TIM3_CR1 &=~(1 << TIM3_CR1_CEN);
        }
}
}

Here is my snippet. Incase you find something wrong, Please let me know. @crankyoldgit Thank you for guiding me the entire time. I was pretty dumb not knowing about the multiple protocols.

crankyoldgit, 2020-09-24:
> 		if(v1>545 && v1 <567)  //time stamp 1
> 		{
> 		hex[i]|=(1UL<<(7-k)); 
> 		}
> 		else if(v1>248 && v1 <255)  //time stamp 0
> 	        {
> 	        hex[i] &=~(1UL <<(7-k));
> 		}

To be honest, your code is hard to read with out comments, or meaningful variable names.
However, what I can't obviously see is an `else` clause at the end of the above, which says if it isn't one of the expect values, then reset back to the `pol = 0` state.

i.e. If the time interval isn't one of the expected lengths, you should reset back to the beginning as it isn't a valid signal.

e.g.
```
if(v1>545 && v1 <567)  //time stamp 1
		{
		hex[i]|=(1UL<<(7-k)); 
		}
		else if(v1>248 && v1 <255)  //time stamp 0
	        {
	        hex[i] &=~(1UL <<(7-k));
		}
                else  // An unexpected time interval was recorded. Therefor the message isn't valid.
                   pol = 0;  // Reset the state machine.
                   return;
                }
```

Also, your algorithm seems like it would over write it's result if you don't pause the state machine till `flag` is set back to `0`.

e.g.
```
if(pol==0)
      {
        TIM3_CNTRH=0;
	TIM3_CNTRL=0;
        k=0;
	i=9;
        if (flag == 0)
        {
           return;  // Stay in the `pol = 0` state.
        }
	tp=0;
        for (l=0;l<=9;l++)
	{
	 hex[l]=0;
	}
      
        TIM3_CR1 |= (1 << TIM3_CR1_CEN);  //6.4us per tick
        pol=1;  // Move to the next state.
      }
```

gokul12051997, 2020-09-24:
@crankyoldgit thank you spending you time on this snippet. I've included the else condition and noticed a weird phenomena. It always takes the else path and keeps resetting the machine, whereas if I remove the else condition it's working with the same integrity problem. 

Once again thanks for investing your time on this issue : )

crankyoldgit, 2020-09-25:
I think you need to get some debug information happening so you can work out what's going on.

gokul12051997, 2020-09-26:
@crankyoldgit I've sorted out most of the issues just leaving one hanging. When ever the controller picks up a unknown signal, it kinda ends ups in erroneous state. What i've made is to simply disable and enable interrupt to work back normally again. But I don't think that as the perfect fix but it works!! Thank you for all your help.

crankyoldgit, 2020-09-27:
You've probably got a bug in your state machine logic. i.e. It's probably not resetting to the initial state when it should.
Hence, you probably need to output some debugging/logging info about what the current state is (and some of your variables)

I didn't look into your "8 bit" construction of a "hex" byte too deeply, as there were more obvious issues, but I think your problem is probably in that space. e.g. You're probably getting a random noise "bit" and it's throwing your other assumptions off.

Also, I'm not familiar with the timer/counter mechanism you are using. It looks like you've only got 16 bits of precession/information for your microsecond timer/counter. That means it will wrap or max out after 65+ ms. (65,535 usecs), that could also be a source of your problems.

Anyway, I think we are done with this issue, especially as far out of scope it is. ;-) Marking it closed for now. Good luck.

closed 2020-09-27 by crankyoldgit