crankyoldgit/IRremoteESP8266 #1282: Hacktoberfest 2020! Free T-Shirt!

crankyoldgit, 2020-09-25:
## Hacktoberfest
It's (almost) October and that means it's Hacktoberfest time [again](https://github.com/crankyoldgit/IRremoteESP8266/issues/932)! #hacktoberfest
See https://hacktoberfest.digitalocean.com/ for details.

This year, to get a free T-Shirt, you need to [register](https://hacktoberfest.digitalocean.com/register) & submit 4 PRs during October (1-31). They don't have to be in this project. See the link for the rules. You can [register](https://hacktoberfest.digitalocean.com/register) now!

Here are just some suggestions for things that would improve this project/library to help you get started:

## Easy:
* Fix some typos. I'm a horrendous speller and have shocking grammar. There are plenty to find and fix.
* Improve some of the documentation.

## Medium:
* Add some [doxygen](https://www.doxygen.nl/index.html) description comments to functions that don't have them. There are plenty of examples in the code where I've done this. Use that as a template for any function etc that doesn't have a description at the start of it.
* Improve/fix some of the code style issues. See our [Code Lint And Style guide](https://github.com/markszabo/IRremoteESP8266/wiki/Library-Maintainers-Guide#code-lint-and-style) and the [Google C++ Style Guide](https://google.github.io/styleguide/cppguide.html) for things to look for. I guarantee there are plenty to fix/correct. ;-)
* Improve our code examples. e.g. Add comments, tips or features you think folk might need when first using the library.
* Create/add [README.md](https://en.m.wikipedia.org/wiki/README)s for the code examples or other directories in the library. [Further reading](https://help.github.com/en/articles/about-readmes).
* Know a non-English language? Consider adding or improving [internationalisation (i18n)](https://en.wikipedia.org/wiki/Internationalization_and_localization) support. See [src/locale](../tree/master/src/locale) for details.

## Hard:
* Write some more unit tests or add more unit test cases.
* Find & **fix** some bugs.
* Make the Example code prettier/easier to read/look better.
* Improve our Travis tests.

## Awesome!!!1!one!:
* Add support for a new protocol. ([It's even easier now!](../wiki/Adding-support-for-a-new-IR-protocol))
* Create unit tests for some of the protocols with out any!

Go crazy and contribute to open-source!

crankyoldgit, 2020-10-04:
@siriuslzx Can you please check your Hacktoberfest profile to see if your #1288 counted?

guieiras, 2020-10-16:
@crankyoldgit Can I work on a pt-BR i18n locale?

crankyoldgit, 2020-10-16:
> Can I work on a pt-BR i18n locale?

@guieiras of course you can. Go for it

guieiras, 2020-10-16:
> > Can I work on a pt-BR i18n locale?
> 
> @guieiras of course you can. Go for it

Great! Working on it.

crankyoldgit, 2020-10-27:
Friendly reminder, there are only several days left in #Hacktoberfest 
Get those PRs in!

crankyoldgit, 2020-10-31:
Last chance for this year!

crankyoldgit, 2020-11-04:
It's over. Until next year!

closed 2020-11-04 by crankyoldgit