crankyoldgit/IRremoteESP8266 #1298: LG AC strange behavior

adrianofoschi, 2020-10-14:
### Version/revision of the library used
The version included in the latest tasmota-ircustom.

### Expected behavior
I'm trying to use tasmota for my LG Air conditioning (Remote controller model AKB74515420).

An example "cool" command

```
irhvac {"Vendor":"LG","Model":2,"Mode":"Cool","Power":"On","Celsius":"On","Temp":18,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}
```
It starts.
It should stop if I send a power off command.

### Actual behavior
My IRblaster works ALWAYS when the AC is turned off and I send a command, but it doesn't work when it is already turned on.

Example:
1. Send "cool" command via my adapter, the AC starts
2. I wait about 5 or 10 minutes
3. Send "turn off" command  (or probably any other command), it does not emit the beep and doesn't work, but tasmota console logs regular transmission.

Example turn off command:

```
irhvac {"Vendor":"LG","Model":2,"Mode":"Off","Power":"Off","Celsius":"On","Temp":15,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}
```

### What brand/model IR demodulator are you using?
I built an IR blaster using a SonOFF Basic as there http://www.asknoone.com/ir-blaster-with-tasmota/

Initially I used the receiver to catch the remote ir codes, next I discovered "irhvac" in Tasmota and actually I'm using it to send commands via tasmota transmitter console.

**Some important notice:**
a) The turn off command works within the first 10/20 seconds after the turn on.
b) The turn off command do "beep" when the climate is already turned off.
c) Sometimes it works unaxpectatly

**Conclusion: a command doesn't work 9 times out of 10 if the AC is at full operating speed.**

NiKiZe, 2020-10-14:
From a quick read of the LG code it does some mangling of temp when handling power state
Could you try to send "Mode":"Off" Without "Temp" setting? not sure how tasmota does this,
If Temp is set after power state was set to off, then internally it will be set to on (non zero temp).
However you can not set a zero temp on it's own, it needs to be excluded, or you have to be able to set power after temp

We might be able to improve this by checking power state when temp is set and keep that in memory until power is set to on, but that code will be a bit of a clunky mess.

For reference https://github.com/crankyoldgit/IRremoteESP8266/issues/1008#issuecomment-570763580

crankyoldgit, 2020-10-15:
@adrianofoschi Can you please download and test PR #1299 / Branch https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1298
I've added a _**potential**_ fix in there. As you haven't provided enough info (data captures) and because it's third party software (Tasmota) we really don't have a lot go to track down the issue.

If this works, great, let us know. If it doesn't, let us know, and we'll ask you for more information (or to log an issue with Tasmota) like capturing (via IRrecvDumpV2+) what IR signal/message Tasmota is producing and what your _real_ remote produces in the same setup/settings/etc.

adrianofoschi, 2020-10-15:
Initially I used a sonoff receiver with tasmota to capture the code from original remote controller. (I have 2 sonoffs, one receiver and one transmitter :) )

So actually I'm already trying with the code that I catched with "poweroff". The code is:

`{"IrReceived":{"Protocol":"LG2","Bits":28,"Data":"0x88C0051","DataLSB":"0x1031008A","Repeat":0}}`

Sometimes I see some other protocol "unknown" signals.
So In tasmota I do to turn off the AC:

`irsend {"IrReceived":{"Protocol":"LG2","Bits":28,"Data":"0x88C0051","DataLSB":"0x1031008A","Repeat":0}}`

The behavior is the same that I described in the first post.
I hope this can be usefull...

adrianofoschi, 2020-10-15:
I traced the command:
`irhvac {"Vendor":"LG","Model":2,"Mode":"Off","Power":"Off","Celsius":"On","Temp":15,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}
`
with my tasmota receiver and it produce exactly: 

`{"Protocol":"LG2","Bits":28,"Data":"0x88C0051","DataLSB":"0x1031008A","Repeat":0}`

So the code produces by tasmota (88C0051) seems to be right.

crankyoldgit, 2020-10-15:
Thanks for the extra info.

As far as I can tell from looking at the code and the data you have provided, the library (and tasmota) is producing and sending the correct "off" message.

If the A/C isn't responding as expected when you send `0x88C0051` via `LG2` then I think you have a hardware/environmental issue.

> The important thing is that the problem is not only with the turn off command. But it doesn't work with any command when it is already turned on.

Can you capture & describe the settings on the real A/C remote for something that is a command when it is already turned on?

i.e. How certain are you that it is the `LG2` protocol (i.e. "Vendor":"LG","Model":2)?

adrianofoschi, 2020-10-15:
I provide you a full test with all signals catched with tasmota IR receiver from original remote controller step by step.

1. Turn ON, Mode: Cool, Temp: 18, FanSpeed: Max
`10:07:17 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"LG2","Bits":28,"Data":"0x8800347","DataLSB":"0x1001C0E2","Repeat":0,"IRHVAC":{"Vendor":"LG","Model":2,"Mode":"Cool","Power":"On","Celsius":"On","Temp":18,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}`

2. Change Temp to 19
`10:09:36 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"LG2","Bits":28,"Data":"0x8808440","DataLSB":"0x10012102","Repeat":0,"IRHVAC":{"Vendor":"LG","Model":2,"Mode":"Cool","Power":"On","Celsius":"On","Temp":19,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}`

3. Change FanSpeed to auto
`10:11:05 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"LG2","Bits":28,"Data":"0x8808451","DataLSB":"0x1001218A","Repeat":0,"IRHVAC":{"Vendor":"LG","Model":2,"Mode":"Cool","Power":"On","Celsius":"On","Temp":19,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}`

4. Enable JetMode (temp returned automatically at 18 on remote)
`10:12:39 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":3,"Hash":"0x25AE7EE0","Repeat":0}}`
5. Disable JetMode
`10:13:12 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":32,"Hash":"0x23060DFB","Repeat":0}} `
6. Change Mode to auto (I cannot change temp on remote)
`10:16:10 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"LG2","Bits":28,"Data":"0x880B746","DataLSB":"0x1001ED62","Repeat":0,"IRHVAC":{"Vendor":"LG","Model":2,"Mode":"Auto","Power":"On","Celsius":"On","Temp":22,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}`

7. Change Mode to dry (I cannot change temp on remote)
`10:16:40 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"LG2","Bits":28,"Data":"0x8809902","DataLSB":"0x10019940","Repeat":0,"IRHVAC":{"Vendor":"LG","Model":2,"Mode":"Dry","Power":"On","Celsius":"On","Temp":24,"FanSpeed":"Min","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}`

8. Change Mode to heat (temp go automatically to 30)
`10:18:16 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"LG2","Bits":28,"Data":"0x880CF4F","DataLSB":"0x1001F3F2","Repeat":0,"IRHVAC":{"Vendor":"LG","Model":2,"Mode":"Heat","Power":"On","Celsius":"On","Temp":30,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}`

...I wait for preheating

9. Change Temp to 29
`10:25:42 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"LG2","Bits":28,"Data":"0x880CE4E","DataLSB":"0x10017372","Repeat":0,"IRHVAC":{"Vendor":"LG","Model":2,"Mode":"Heat","Power":"On","Celsius":"On","Temp":29,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}`

10. Power off AC
`10:26:34 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"LG2","Bits":28,"Data":"0x88C0051","DataLSB":"0x1031008A","Repeat":0,"IRHVAC":{"Vendor":"LG","Model":2,"Mode":"Off","Power":"Off","Celsius":"On","Temp":15,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}`

Now I'm trying to send the same codes from ir transmitter but after step 1 it stops to work.
It works only from power state off...
I don't know if the protocol is right, it is the protocol that tasmota gave me with receiver.

GiampaMan, 2020-10-15:
Hi!
I was thinking on opening a new issue but could be better to explain my problem here because is really similar. I have an ESP32 using this library and is great! Problem is that sometimes when I send a command to A/C it does nothing. The weird part of this issue is that sometimes happens and sometimes no, and I send the raw code of the LG air. The other thing is that it only fails on my LG airs, if I use Panasonic everything works great!
Is this an issue with ESP32? I remember to use a long time ago the ESP8266 (which I stopped using due my project RAM use, so I started to use ESP32) and don't remember to have this problems.
Maybe someone could help me. Thanks a lot!

Regards,
Giampaolo

adrianofoschi, 2020-10-15:
I did another test:

1. I captured the remote controller signal with tasmota receiver. The result is:

`16:56:19 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"LG2","Bits":28,"Data":"0x8804E46","DataLSB":"0x10017262","Repeat":0,"IRHVAC":{"Vendor":"LG","Model":2,"Mode":"Heat","Power":"On","Celsius":"On","Temp":29,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}`

2. Next I sent the captured signal from tasmota trasmitter to tasmota receiver to verify it is same, using irhvac command:

`irhvac {"Vendor":"LG","Model":2,"Mode":"Heat","Power":"On","Celsius":"On","Temp":29,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}`

the captured signal is the same:

`17:00:14 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"LG2","Bits":28,"Data":"0x8804E46","DataLSB":"0x10017262","Repeat":0,"IRHVAC":{"Vendor":"LG","Model":2,"Mode":"Heat","Power":"On","Celsius":"On","Temp":29,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
`

so the signal in tasmota seems to be right...I finished the ideas...

crankyoldgit, 2020-10-15:
@adrianofoschi Thanks for all the data.

@GiampaMan There should be no difference between sending on an ESP8266 verses a ESP32.

@adrianofoschi @GiampaMan I think at this stage I'm going to need the `rawData` for your various messages that don't work.
i.e. a _raw_ capture from the real remote.
I think the issue _might_ be in the `repeat` part of the message which was never verified as correct.
see:
https://github.com/crankyoldgit/IRremoteESP8266/blob/ecfc47239dd4f91c980b3a16feb243a88aabe435/src/ir_LG.cpp#L116-L118

I would really prefer the **full** text output from IRrecvDumpV2 or V3, instead of the raw data captured from Tasmota as I know the it will do a more accurate & complete job for this issue. However, if all you can do is use tasmota to capture the raw output, that will have to do.

@adrianofoschi When you are trying to reproduce the message via the `"Protocol":"LG2","Bits":28,"Data":"0x8804E46"` method  **DO NOT** use `"Repeat":0`, instead, use `"Repeat":1` or not at all. The protocol does have a mandatory repeat as far as I know, setting it to `0` may cause issues like you're describing. Unless Tasmota is doing something bizarre via their IRHVAC method, the manditory repeat will be included there by default.

Things like that are one of the reasons why we don't like to debug stuff from other projects, and require people to use the native example code etc so we have a better understanding of what is actually going on.

crankyoldgit, 2020-10-16:
@adrianofoschi Can you please download and test PR #1300 / Branch https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1298
I've fixed a _**potential**_ issue with repeats. It might be affecting you and be the cause of the problem. It might not.
It is just a guess.

adrianofoschi, 2020-10-16:
@crankyoldgit Thanks you for your support.
So if I understand correctly, the IRHVAC is a tasmota command that send always Repeat: 0 in each message, but on the IRRemoteESP8266 side this value it is not verified.

What I'm going to do today is:

- try to send command via "irsend" with Repeat:1 or without Repeat attribute
- try to replace IRRemoteESP8266 version with yours that includes the pr
- try to obtain the raw data from tasmota, if not I will buy another board to do it (nodemcu? You can suggest me something)

Thanks you again

crankyoldgit, 2020-10-16:
> * try to send command via "irsend" with Repeat:1 or without Repeat attribute

The last part of that probably won't work. But it's worth a try. The branch from that PR should address that problem.

> * try to obtain the raw data from tasmota. Anyway I will buy another board to do it with arduino? (nodemcu? You can suggest me something)

There is a way to get Tasmota to report the raw IR data. Others have done it, so you shouldn't need a new board (just yet).
A simple google & search yields: https://tasmota.github.io/docs/Commands/#setoption58
You may need to recompile Tasmota with a different (50+ms) timeout value for IR capture. Not sure how Tasmota sets that value. Maybe @s-hadinger can tell you how. It may be as simple as editing this in our library.
https://github.com/crankyoldgit/IRremoteESP8266/blob/ecfc47239dd4f91c980b3a16feb243a88aabe435/src/IRrecv.h#L39-L49

As for boards, any ESP8266 or ESP32 dev board should suffice. I've personally found the Wemos D1 Mini's to be reliable, cheap, and plentiful.
If you are not moving the IR receiver from the sonoff to the new board, I strongly suggest you **don't** buy one with a [VS1838b](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#Help_Im_getting_very_inconsistent_results_when_capturing_an_IR_message_using_a_VS1838b_IR_demodulator) part.

s-hadinger, 2020-10-16:
Hi,
`IrSend1` is equivalent to `IrSend ..."Repeat":0`. And `IrSend` is the same as `IrSend1`. I hope it clarifies.

To get raw data, use `SetOption58 1`.
I recently added a compressed format so that it doesn't overflow the MQTT buffer, and it's all explained here: https://tasmota.github.io/docs/IRSend-RAW-Encoding/

Use this online tool to convert to the old raw format: https://tasmota.hadinger.fr/util

crankyoldgit, 2020-10-16:
@s-hadinger 
> `IrSend1` is equivalent to `IrSend ..."Repeat":0`. And `IrSend` is the same as `IrSend1`. I hope it clarifies.

So is there no way to specify the number of repeats via the Tasmota interface?

> I recently added a compressed format so that it doesn't overflow the MQTT buffer, and it's all explained here: https://tasmota.github.io/docs/IRSend-RAW-Encoding/
Use this online tool to convert to the old raw format: https://tasmota.hadinger.fr/util

Rather than re-invent the wheel, do you have c++ code for this perchance? I can add it to IRrecvDumpV2+ etc.

> Edit: we only set irrecv->setUnknownThreshold(Settings.param[P_IR_UNKNOW_THRESHOLD]);
Is it the time-out you referring to? If so it can be changed with SetOption38 and defaults to IR_RCV_MIN_UNKNOWN_SIZE which is 6.

No, the timeout I'm referring to is "how long before I give up listening for more pulses in a message?"
e.g. `kTimeoutMs` The default value I referenced above, or the `timeout` value set by when creating the [`IRrecv` object](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRrecv.html#a3bb1bcc1c1a3184294dd35c8f6f758b1)

adrianofoschi, 2020-10-16:
Hi, I'm going to post latest test logs, including raw data with 15ms and 50ms timeouts.
In this test I use tasmota-ir instead tasmota-customir so you can notice that the codes are different from [my previous tests](https://github.com/crankyoldgit/IRremoteESP8266/issues/1298#issuecomment-709305602). 
These codes seems better becouse they work always (maybe) from turned off state.

**Test summary:**

1. **TEST 1:** send signals using original remote controller and catching with my receiver (tasmota-ir with default 15 ms timeout)
2. **TEST 2:** send signals (catched in TEST 1) using my transmitter and caching with my receiver (tasmota-ir with default 15 ms timeout)
3. **TEST 3:** send signals using original remote controller and catching with my receiver (tasmota-ir with default 50 ms timeout)
4. **TEST 4:** send signals (catched in TEST 3) using my transmitter and caching with my receiver (tasmota-ir with default 50 ms timeout)

**TEST 1: catch test using LG original remote controller and tasmota-ir with TIMEOUT 15ms**
1.1. RC: press power to turn on, conditions: heat, temp 29, fan max
`10:09:34 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"LG2","Bits":28,"Data":"0x8804E46","DataLSB":"0x10017262","Repeat":0,"IRHVAC":{"Vendor":"LG","Model":2,"Mode":"Heat","Power":"On","Celsius":"On","Temp":29,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1},"RawData":[3096,9926,482,1578,508,548,488,548,460,560,506,1584,486,564,486,556,486,554,484,550,460,580,462,574,486,564,486,548,484,1592,484,550,460,572,486,1608,486,1608,484,1600,460,580,486,556,460,1608,484,550,486,548,486,556,462,1616,486,1600,484,558,460],"RawDataInfo":[59,59,0]}}`

1.2. wait 5 minutes

1.3. press power to turn off. But it does not turn off, it "beeps" and seems to set again the previous conditions (heat, temp 29, fan max)
`10:14:07 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"LG2","Bits":28,"Data":"0x8804E46","DataLSB":"0x10017262","Repeat":0,"IRHVAC":{"Vendor":"LG","Model":2,"Mode":"Heat","Power":"On","Celsius":"On","Temp":29,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1},"RawData":[3124,9902,514,1576,488,546,486,548,488,546,486,1592,488,546,486,564,486,564,486,556,484,556,486,548,486,546,486,556,486,1600,482,554,482,550,486,1606,486,1588,504,1600,486,556,486,554,486,1584,486,548,462,572,486,556,486,1590,460,1624,486,554,462],"RawDataInfo":[59,59,0]}}`

1.4. press power again and it turns off
`10:14:12 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"LG2","Bits":28,"Data":"0x88C0051","DataLSB":"0x1031008A","Repeat":0,"IRHVAC":{"Vendor":"LG","Model":2,"Mode":"Off","Power":"Off","Celsius":"On","Temp":15,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1},"RawData":[3150,9880,512,1576,486,556,488,546,488,546,486,1586,506,548,486,556,486,554,484,1610,486,1570,480,582,486,556,462,570,504,560,458,574,484,550,486,556,486,548,486,556,486,550,482,548,486,1582,486,556,486,1598,488,554,488,556,486,554,486,1590,486],"RawDataInfo":[59,59,0]}}`

1.5. repeat from (1) then press power to turn on
`10:16:40 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"LG2","Bits":28,"Data":"0x8804E46","DataLSB":"0x10017262","Repeat":0,"IRHVAC":{"Vendor":"LG","Model":2,"Mode":"Heat","Power":"On","Celsius":"On","Temp":29,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1},"RawData":[3126,9894,482,1608,486,556,482,568,486,556,462,1602,506,548,486,556,486,556,486,556,486,544,506,556,488,564,486,548,486,1592,486,548,486,548,462,1622,488,1578,480,1616,460,574,462,572,462,1624,488,554,488,554,486,548,486,1600,486,1584,486,548,462],"RawDataInfo":[59,59,0]}}`

1.6. wait 5 minutes

1.7. press power to turn off, this time it turns off without press again
`10:21:38 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"LG2","Bits":28,"Data":"0x88C0051","DataLSB":"0x1031008A","Repeat":0,"IRHVAC":{"Vendor":"LG","Model":2,"Mode":"Off","Power":"Off","Celsius":"On","Temp":15,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1},"RawData":[3152,9892,484,1606,460,582,460,582,486,546,486,1584,486,556,488,554,486,548,486,1590,486,1582,486,548,486,548,486,556,460,574,486,564,486,556,486,564,486,556,488,554,488,546,484,546,482,1614,486,556,488,1598,486,556,486,556,486,548,486,1598,486],"RawDataInfo":[59,59,0]}}`

....
**TEST 2: repeat catch TEST 1 with my transmitter**

2.1. send data to turn on from from my transmitter
`irsend {"Protocol":"LG2","Bits":28,"Data":"0x8804E46","DataLSB":"0x10017262","Repeat":0}`
catched << `10:28:40 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"LG2","Bits":28,"Data":"0x8804E46","DataLSB":"0x10017262","Repeat":0,"IRHVAC":{"Vendor":"LG","Model":2,"Mode":"Heat","Power":"On","Celsius":"On","Temp":29,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1},"RawData":[3202,9864,556,1606,550,558,550,558,548,558,548,1608,550,558,550,558,550,558,550,558,548,560,548,558,574,534,548,558,550,1 [... cut ...] mp":15,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1},"RawData":[3176,9892,560,1602,550,558,548,558,550,558,550,1608,548,558,548,558,550,558,550,1608,550,1606,550,558,550,558,548,558,552,554,550,558,548,558,548,558,550,558,548,558,550,558,548,558,550,1608,548,558,548,1610,548,560,548,558,548,558,550,1616,548],"RawDataInfo":[59,59,0]}}`

Tell me if you need more data or tests.
Thanks you

s-hadinger, 2020-10-16:
> So is there no way to specify the number of repeats via the Tasmota interface?

Actually you can use either `IrSend<x> {...}` or `IrSend {"Repeat":<y>,...}`. It's just that y=x-1. I.e. `IrSend2 ...` means send 2 times, meaning repeat 1 time or `"Repeat":1`. I know it's confusing but that was the only way to stay backwards compatible and still surface the Repeat argument in JSON.

> No, the timeout I'm referring to is "how long before I give up listening for more pulses in a message?"
> e.g. `kTimeoutMs` The default value I referenced above, or the `timeout` value set by when [...]

Then we're using the default value, and there is no provision to change it via configuration.

> Rather than re-invent the wheel, do you have c++ code for this perchance? I can add it to IRrecvDumpV2+ etc.

Sure. The code to parse it is there: https://github.com/arendst/Tasmota/blob/b48f77104862e9298f3a640fd1bc09d4e42d1475/tasmota/xdrv_05_irremote_full.ino#L649-L681

adrianofoschi, 2020-10-16:
I did a new test with tasmota 9.0.0.x recompiled with [PR #1300](https://github.com/crankyoldgit/IRremoteESP8266/pull/1300) . I tried with `irsend`,`irsend1`,`irsend2`, `irhvac` it doesn't work...

**1. remote controller power on**
`15:38:48 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"LG2","Bits":28,"Data":"0x8804E46","DataLSB":"0x10017262","Repeat":0,"IRHVAC":{"Vendor":"LG","Model":2,"Mode":"Heat","Power":"On","Celsius":"On","Temp":29,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1},"RawData":"+3160-9890+540-1550+510-520EfEfE-1545+530-525E-515HfHeHhEfEfEhE-1575EiEfE-1580E-1560HkEeHeHmEiEfEhEgHkEe+505","RawDataInfo":[59,59,0]}}`
uncompressed raw: `3160,9890,540,1550,510,520,510,520,510,520,510,1545,530,525,510,515,530,520,530,510,530,530,510,520,510,520,510,530,510,1575,510,525,510,520,510,1580,510,1560,530,1575,510,510,530,510,530,1560,510,525,510,520,510,530,510,1545,530,1575,510,510,505`

**2. remote controller power off**
`15:41:25 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"LG2","Bits":28,"Data":"0x88C0051","DataLSB":"0x1031008A","Repeat":0,"IRHVAC":{"Vendor":"LG","Model":2,"Mode":"Off","Power":"Off","Celsius":"On","Temp":15,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1},"RawData":"+3160-9880+520-1550C-545+500cC-535F-1570FeFcCgF-1555ChFgFgFcCgF-530CcCjCcCcCgFjC-1560CcC-1585FeFcCgFlF","RawDataInfo":[59,59,0]}}`

uncompressed raw:
`3160,9880,520,1550,520,545,500,520,520,535,500,1570,500,545,500,520,520,535,500,1555,520,1570,500,535,500,535,500,520,520,535,500,530,520,520,520,530,520,520,520,520,520,535,500,530,520,1560,520,520,520,1585,500,545,500,520,520,535,500,1585,500`
    
**3. my transmitter power on**
`15:46:11 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"LG2","Bits":28,"Data":"0x8804E46","DataLSB":"0x10017262","Repeat":0,"IRHVAC":{"Vendor":"LG","Model":2,"Mode":"Heat","Power":"On","Celsius":"On","Temp":29,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1},"RawData":"+3200-9880+565-1600C-545Ce+560eC-1595CeFeFeCeCeFeCeCeFgFeFeCgFgCgCeFeFgFeCeFeFgCgF-550F","RawDataInfo":[59,59,0]}}`
    
uncompressed raw:
`3200,9880,565,1600,565,545,565,545,560,545,565,1595,565,545,560,545,560,545,565,545,565,545,560,545,565,545,565,545,560,1595,560,545,560,545,565,1595,560,1595,565,1595,565,545,560,545,560,1595,560,545,565,545,560,545,560,1595,565,1595,560,550,560`
    
**4. my transmitter power off**
`15:48:41 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"LG2","Bits":28,"Data":"0x88C0051","DataLSB":"0x1031008A","Repeat":0,"IRHVAC":{"Vendor":"LG","Model":2,"Mode":"Off","Power":"Off","Celsius":"On","Temp":15,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1},"RawData":"+3200-9875+590-1575+565-545EfEfE-1595+560f+570-540HfHgEgEfEfHfC-520HfHfEfEfHfEfHfHgHfHgHfHfHfH-1600E","RawDataInfo":[59,59,0]}}`
    
uncompressed raw:
`3200,9875,590,1575,565,545,565,545,565,545,565,1595,560,545,570,540,560,545,560,1595,565,1595,565,545,565,545,560,545,590,520,560,545,560,545,565,545,565,545,560,545,565,545,560,545,560,1595,560,545,560,1595,560,545,560,545,560,545,560,1600,565`

GiampaMan, 2020-10-16:
Hi @crankyoldgit !
Thanks for the response. I'm not sending it via sendLG, I'm sending it trough sendRaw. If is RAW it's not necessary to repeat right? Because the repeat is included on the RAW. Another thing I would like to comment is that sometimes it works (I mean when sending the raw, the A/C do what's expected) but in other situations the A/C not receives the command so it acts like I never sent the RAW. As you told me, here is the output when receiving the original control signal (In this example 24°C):

```
Timestamp : 000000.000
Protocol  : UNKNOWN
Code      : 0x774756F3 (30 Bits)
Library   : v2.7.8

Raw Timing[59]:
   +  3448, -  9752,    +   656, -  1436,    +   572, -   446,    +   532, -   516,
   +   568, -   494,    +   630, -  1432,    +   604, -   436,    +   604, -   426,
   +   632, -   404,    +   572, -   458,    +   628, -   404,    +   574, -   456,
   +   606, -   430,    +   576, -  1328,    +   676, -   514,    +   516, -   508,
   +   626, -   432,    +   542, -  1520,    +   548, -   488,    +   572, -   458,
   +   542, -  1520,    +   510, -   516,    +   568, -   488,    +   604, -   428, 
   +   596, -   434,    +   630, -   400,    +   578, -   458,    +   514, -   516,
   +   574, -  1490,    +   520

uint16_t rawData[59] = {3448, 9752,  656, 1436,  572, 446,  532, 516,  568, 494,  630, 1432,  604, 436,  604, 426,  632, 404,  572, 458,  628, 404,  574, 456,  606, 430,  576, 1328,  676, 514,  516, 508,  626, 432,  542, 1520,  548, 488,  572, 458,  542, 1520,  510, 516,  568, 488,  604, 428,  596, 434,  630, 400,  578, 458,  514, 516,  574, 1490,  520};  // UNKNOWN 774756F3       
```

Thanks a lot for the help!
Regards,
Giampaolo

crankyoldgit, 2020-10-16:
@GiampaMan Thanks for that. You should probably try the latest version of the library (2.7.11) but I think that's a strong confirmation that there is no "repeat" part of the message.

Oh wait, can you change this line:
https://github.com/crankyoldgit/IRremoteESP8266/blob/3b5bd54ae5486aae31849a2968a037a0e955cb8b/examples/IRrecvDumpV2/IRrecvDumpV2.ino#L102
to `1` or similar.
And try again? This will catch if there is a repeat message and it's super short, and it happens later than 50ms after the first message.

crankyoldgit, 2020-10-16:
> I did a new test with tasmota 9.0.0.x (compressed raw data) recompiled with [PR #1300](https://github.com/crankyoldgit/IRremoteESP8266/pull/1300) . I tried with `irsend`,`irsend1`,`irsend2`, `irhvac` it doesn't work...

I'm running out of ideas as to what could be going on. @GiampaMan provided data that seems to back up there is no "repeat" part of the message. So that _might_ be a red-herring.
I'll try tweaking the timings of the LG2 protocol to match the data you've collected so far, but you can test/simulate that by try sending the "raw" version of a message. If replaying the _raw_ message doesn't work, that leaves a modulation frequency/duty cycle mismatch (very unlikely but not impossible) or there is some tiny & delayed repeat message that we are missing/ignoring.

Other than that .. I'm stumped for ideas. We may need to have someone with access to an oscilloscope & the remote to try to find out exactly what is going on in order to reproduce it.

GiampaMan, 2020-10-17:
Hi @crankyoldgit !
I changed the line you tell me. My output is this:

```
Timestamp : 000000.000
Protocol  : UNKNOWN
Code      : 0x42D5D51C (30 Bits)
Library   : v2.7.8

Raw Timing[59]:
   +  3460, -  9736,    +   594, -  1532,    +   572, -   442,    +   596, -   460,
   +   576, -   456,    +   512, -  1546,    +   570, -   488,    +   604, -   426,
   +   604, -   426,    +   548, -   484,    +   572, -   464,    +   572, -   458,
   +   574, -   458,    +   574, -  1498,    +   578, -   458,    +   562, -   462,
   +   594, -   460,    +   570, -  1492,    +   598, -   432,    +   576, -   458,
   +   542, -  1520,    +   578, -   458,    +   600, -   430,    +   574, -   458, 
   +   578, -   456,    +   604, -   354,    +   648, -   458,    +   596, -   460,
   +   544, -  1534,    +   544

uint16_t rawData[59] = {3460, 9736,  594, 1532,  572, 442,  596, 460,  576, 456,  512, 1546,  570, 488,  604, 426,  604, 426,  548, 484,  572, 464,  572, 458,  574, 458,  574, 1498,  578, 458,  562, 462,  594, 460,  570, 1492,  598, 432,  576, 458,  542, 1520,  578, 458,  600, 430,  574, 458,  578, 456,  604, 354,  648, 458,  596, 460,  544, 1534,  544};  // UNKNOWN 42D5D51C   
```

I will also update the library and send back to you the output with the new version.

Have a nice day!
Regards,
Giampaolo

GiampaMan, 2020-10-17:
@crankyoldgit  I forgot to tell you something.
Until this moment, I tried to send the raw I just recorded In the last comment I send you here. My A/C is not doing anything. The weird part of the issue is that in the past tests I did, sometimes it works for a long time and then just the A/C does nothing and stops to work correctly for another long time and then magically works again. Now I'm downloading new version of your library to tell you. Thanks a lot for the interest on helping!

Giampaolo

crankyoldgit, 2020-10-17:
> Timestamp : 000000.000

I find it odd that the timestamp is always `000000.000` .. it should be changing with each message. (it's the time since boot)

Assuming there was no short "unknown" etc message immediately following that. It's probably safe to assume there is no mandatory repeat message.
The only other thing to try is to set/change:
https://github.com/crankyoldgit/IRremoteESP8266/blob/3b5bd54ae5486aae31849a2968a037a0e955cb8b/examples/IRrecvDumpV2/IRrecvDumpV2.ino#L72
to
`const uint8_t kTimeout = kMaxTimeoutMs;`
You might as well change the `15` one as well, just to be sure.

and try capturing another usual LG A/C RC message.

crankyoldgit, 2020-10-17:
> @crankyoldgit I forgot to tell you something.
> Until this moment, I tried to send the raw I just recorded In the last comment I send you here. My A/C is not doing anything. The weird part of the [...]

Just to confirm, your sending circuit works fine for other IR devices? Correct? i.e. We don't have a sending hardware issue at all.
e.g. You're using a transistor as per the wiki so the IR LED gets the enough current etc.

adrianofoschi, 2020-10-17:
> @crankyoldgit I forgot to tell you something.
> Until this moment, I tried to send the raw I just recorded In the last comment I send you here. My A/C is not doing anything. The weird part of the [...]

Like you I tried to send the raw catched using irsend and nothing happens. Exactly the same when the AC is turned on.

adrianofoschi, 2020-10-17:
> Just to confirm, your sending circuit works fine for other IR devices? Correct? i.e. We don't have a sending hardware issue at all.
> e.g. You're using a transistor as per the wiki so the IR LED gets [...]

I am using my adapter for LG TV and it works...

crankyoldgit, 2020-10-18:
> I am using my adapter for LG TV and it works...

Yes. I am aware your setup works, that question etc was more directed at @GiampaMan as it was quoting his reply.

GiampaMan, 2020-10-18:
Hi @crankyoldgit !

Sorry for the delays on answering. I was doing some tests with new library (the update) and I will tell you ASAP how is going. I don't know if is coincidence, but now is a bit better. The behavior before this change was that the device sometimes works and sometimes not work, but for periods of long time for both situations.

In the other hand something that is happening now and also before this bug, is that sometimes I send an IR code and the A/C don't receive it, but the next send it works great. In this case not for longs periods of time like I was experiencing before.

As I told you a bit ago, I don't know if is coincidence so I will keep testing and let you know ASAP.

To answer your question about my circuit, it's working great, it has the right power and also a long range of distance.

Thanks a lot for the help! I will let you know so you and @adrianofoschi can take a look to my results.

Have both a great day!
Giampaolo

GiampaMan, 2020-10-18:
> Like you I tried to send the raw catched using irsend and nothing happens. Exactly the same when the AC is turned on.

Yes! Exactly the same. I was experiencing that. I'm doing some tests now to check. How are you going, something better?

crankyoldgit, 2020-10-18:
For the record, here is a quick analysis based on @adrianofoschi's `1.7` vs `2.7` timings.
v2.7.11: Library defaults
- kHdrMark   = **3200**
- kHdrSpace  = **9850**
- kBitMark   = **500**
- kOneSpace  = **1600**
- kZeroSpace = **550**

`1.7`: Real remote averaged
- kHdrMark   = **3152**  (_-48us_)
- kHdrSpace  = **9892** (_+42us_)
- kBitMark   = **483** (_-17us_)
- kOneSpace  = **1596** (_-4us_)
- kZeroSpace = **556** (_+6us_)

`2.7:` Tasmota averaged
- kHdrMark   = **3126** (_-74us_)
- kHdrSpace  = **9896** (_+46us_)
- kBitMark   = **531** (_+31us_)
- kOneSpace  = **1548** (_-52us_)
- kZeroSpace = **509** (_-41us_)

Sending same code (LG2 0x88C0051) via IRMQTTServer and capturing via IRrecvDumpV2 (i.e. My hardware):
- kHdrMark   = **3196** (_-4us_)
- kHdrSpace  = **9858** (_+8us_)
- kBitMark   = **582** (+82us)
- kOneSpace  = **1580** (_-20us_)
- kZeroSpace = **521** (_-39us_)

adrianofoschi, 2020-10-18:
> I find it odd that the timestamp is always `000000.000` .. it should be changing with each message. (it's the time since boot)
>
> Assuming there was no short "unknown" etc message immediately [...]

I am going to test again with this patch

crankyoldgit, 2020-10-18:
> * kBitMark   = **582** (+82us)

So the above value in my testing is the most wrong/unexpected/out-of-range.
I traced it to a timing value mistake in `IRsend::sendLG2()` which was adding an extra _50us_ to that. Which may have pushed it over the protocols expected range. I've further lowered it by another _20us_ to try to get it to better replicate the real remote timings.

I've also refactored & cleaned up most of `IRrecv::decodeLG()` to make it more readable, modern, more reliable, and probably smaller.

So ... hopefully, the fix is in PR #1304 / (new branch) https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1298_timing

Please download that and try it out. Let me know etc.

Note: @GiampaMan None of this will affect your raw replay/captures. I have no idea why your raw replays are not working.

What I have noticed is Tasmota does burden the ESP chip a bit more (based on the supplied data) thus could further be interfering with the subtle timings in the protocol. That, I can't fix.

adrianofoschi, 2020-10-18:
> So the above value in my testing is the most wrong/unexpected/out-of-range.
> I traced it to a timing value mistake in `IRsend::sendLG2()` which was adding an extra _50us_ to that. Which may have [...]

I updated tasmota with IRremoteESP8266 PR #1304 on transmitter and receiver too. 
Unfortunately it still doesn't work...it still works the first time only when AC is off. I tried to send the signal as raw, data or irhvac...
The difference I notice is that now _sometimes_ the receiver logs a tons of "UNKNOWN" signals, about one for a second.

power on recorded from remote controller on PR #1304 
`22:51:00 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"LG2","Bits":28,"Data":"0x8804E46","DataLSB":"0x10017262","Repeat":0,"IRHVAC":{"Vendor":"LG","Model":2,"Mode":"Heat","Power":"On","Celsius":"On","Temp":29,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1},"RawData":"+3105-9920+550-1540+520-510EfEfE-1530+545fE-505C-500HjHeEfEfEeE-1560EfEfE-1570E-1550H-1565EjHjHmEfEfEeEgHnEjH","RawDataInfo":[59,59,0]}}`

power off recorded from remote controller on PR #1304 
`22:51:05 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"LG2","Bits":28,"Data":"0x88C0051","DataLSB":"0x1031008A","Repeat":0,"IRHVAC":{"Vendor":"LG","Model":2,"Mode":"Off","Power":"Off","Celsius":"On","Temp":15,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1},"RawData":"+3100-9920+540-1535+545-515+520-505EgG-1540EfGgG-510G-1565G-1550EhEgGjCgGgG-500EgGjGgGjGjGlGjG-1570GjGmEjGnG","RawDataInfo":[59,59,0]}}`

power on recorded from my tasmota transmitter
`23:04:52 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"LG2","Bits":28,"Data":"0x8804E46","DataLSB":"0x10017262","Repeat":0,"IRHVAC":{"Vendor":"LG","Model":2,"Mode":"Heat","Power":"On","Celsius":"On","Temp":29,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1},"RawData":"+3185-9945+470-1655+420-590+445-595ChG-1620+465-570+495-545CkC-600+440-575JkO-625OpO-1645OhJ-580O-1670OiC-1650Eh+460nGrOkOhOnOtOuOpO","RawDataInfo":[59,59,0]}}`

power off recorded from my tasmota transmitter
`23:05:39 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"LG2","Bits":28,"Data":"0x88C0051","DataLSB":"0x1031008A","Repeat":0,"IRHVAC":{"Vendor":"LG","Model":2,"Mode":"Off","Power":"Off","Celsius":"On","Temp":15,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1},"RawData":"+3120-9970+465-1630+470-570Ef+445-595E-1620Gh+440-600JhG-1650JlJkCfEfGkCk+420-620JfJhJkCfCfClJkGiJkCfJkJ-1675J","RawDataInfo":[59,59,0]}}`

**I did another interesting test**...
I sent the poweron command and the AC "beeps". I sent the same command again after 1 second and it "beeps". I sent the same command again one time per second for 30 seconds and it "beeps" always.
Finally I wait about 10 seconds, it gone in pre-heating, I sent the command again and it stops to beep.
Probably there is something that we don't know, becouse seems that the AC changes his state when it is fully operational...
Obviously the poweroff command too works in the first seconds...

Thanks you David

crankyoldgit, 2020-10-19:
FYI, while we are doing this testing/debugging stuff, can you please use the non-compressed raw data output. It means I can actually read it. :)

I've pushed another change to that latest branch to change the duty cycle of the IR led from 50% (default) to 33%.
It's a long shot, and I don't have high hopes for it, but it's worth a try. Let me know how it goes. The duty cycle is one thing you can't change when sending "raw" data. It's locked at 50%. Given your "raw" replays are also not working, it could be it. Or as we've said, there is something else going on we don't understand.

Also, please pass on any UNKNOWN messages you get around the time of the message transmissions. They could be something we don't expect.

adrianofoschi, 2020-10-19:
> FYI, while we are doing this testing/debugging stuff, can you please use the non-compressed raw data output. It means I can actually read it. :)
>
> I've pushed another change to that latest branch to [...]

Hi David, sorry the latest tasmota use compressed data output Idk if it is disabled. I will convert the output for you with the proper online tool. I edited the latest test adding the uncompressed data.
It still doesn't work:

**CATCH SIGNAL FROM REMOTE**
- power on remote controller
`07:29:25 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"LG2","Bits":28,"Data":"0x8804E46","DataLSB":"0x10017262","Repeat":0,"IRHVAC":{"Vendor":"LG","Model":2,"Mode":"Heat","Power":"On","Celsius":"On","Temp":29,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1},"RawData":"+3120-9880+535-1550+540-515+495eG-525+520-1575G-530IhIhIeG-550GeGkIeG-1560IeGeG-1600GjI-1565IhIhIjGeGeGlGmI-1570IhI","RawDataInfo":[59,59,0]}}`
uncompressed raw:
`3120,9880,535,1550,540,515,495,540,495,525,520,1575,495,530,520,525,520,525,520,540,495,550,495,540,495,530,520,540,495,1560,520,540,495,540,495,1600,495,1575,520,1565,520,525,520,525,520,1575,495,540,495,540,495,550,495,1560,520,1570,520,525,520`

- power off remote controller
`07:30:46 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"LG2","Bits":28,"Data":"0x88C0051","DataLSB":"0x1031008A","Repeat":0,"IRHVAC":{"Vendor":"LG","Model":2,"Mode":"Off","Power":"Off","Celsius":"On","Temp":15,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1},"RawData":"+3175-9905+510-1560+480-580+485-550G-565G-1570+505-555GlGhG-1600G-1590+460fGhGh+490-560O-575GlOrGlGhOfO-570O-1610GiO-1625GhGhGlGtG","RawDataInfo":[59,59,0]}}`
uncompressed raw:
`3175,9905,510,1560,480,580,485,550,485,565,485,1570,505,555,485,555,485,550,485,1600,485,1590,460,580,485,550,485,550,490,560,460,575,485,555,460,575,485,555,485,550,460,580,460,570,460,1610,485,565,460,1625,485,550,485,550,485,555,485,1610,485`

**CATCH SIGNAL FROM TRASMITTER**
**Power on test input raw, raw encoded, protocol, irhvac:**
- `irsend 0,3120,9880,535,1550,540,515,495,540,495,525,520,1575,495,530,520,525,520,525,520,540,495,550,495,540,495,530,520,540,495,1560,520,540,495,540,495,1600,495,1575,520,1565,520,525,520,525,520,1575,495,540,495,540,495,550,495,1560,520,1570,520,525,520`
output:
`no one signal registered by receiver and AC does not beep`
-  `irsend 0,+3120-9880+535-1550+540-515+495eG-525+520-1575G-530IhIhIeG-550GeGkIeG-1560IeGeG-1600GjI-1565IhIhIjGeGeGlGmI-1570IhI`
output:
`no one signal registered by receiver and AC does not beep`
- `irsend {"Protocol":"LG2","Bits":28,"Data":"0x8804E46","DataLSB":"0x10017262","Repeat":0}`
output:
`08:06:34 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"LG2","Bits":28,"Data":"0x8804E46","DataLSB":"0x10017262","Repeat":0,"IRHVAC":{"Vendor":"LG","Model":2,"Mode":"Heat","Power":"On","Celsius":"On","Temp":29,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1},"RawData":"+3185-9935+450-1650+470-570EfEf+445-1645EfEfG-590G-595Ef+440jGjEfGhGiGjE-1620GhGhEfEfGhEfGj+465fMlElE-575E","RawDataInfo":[59,59,0]}}`
uncompressed raw:
`3185,9935,450,1650,470,570,470,570,470,570,445,1645,470,570,470,570,445,590,445,595,470,570,440,595,445,595,470,570,445,1645,445,590,445,595,470,1620,445,1645,445,1645,470,570,470,570,445,1645,470,570,445,595,465,570,465,1620,470,1620,470,575,470`
- `irhvac {"Vendor":"LG","Model":2,"Mode":"Heat","Power":"On","Celsius":"On","Temp":29,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}`
output:
`08:07:39 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"LG2","Bits":28,"Data":"0x8804E46","DataLSB":"0x10017262","Repeat":0,"IRHVAC":{"Vendor":"LG","Model":2,"Mode":"Heat","Power":"On","Celsius":"On","Temp":29,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1},"RawData":"+3185-9940+470-1630C-570+445-595FgF-1645CeCeCeF-590CeCeCeFgF-1640+450gFgFhC-1620ClFgFgClFgFgCeFhClF-600C","RawDataInfo":[59,59,0]}}`
uncompressed raw:
`3185,9940,470,1630,470,570,445,595,445,595,445,1645,470,570,470,570,470,570,445,590,470,570,470,570,470,570,445,595,445,1640,450,595,445,595,445,1645,470,1620,470,1620,445,595,445,595,470,1620,445,595,445,595,470,570,445,1645,470,1620,445,600,470`

**Power off test input raw, raw encoded, protocol, irhvac:**
- `irsend 0,3175,9905,510,1560,480,580,485,550,485,565,485,1570,505,555,485,555,485,550,485,1600,485,1590,460,580,485,550,485,550,490,560,460,575,485,555,460,575,485,555,485,550,460,580,460,570,460,1610,485,565,460,1625,485,550,485,550,485,555,485,1610,485`
output:
`no one signal registered and AC does not beep`

- `irsend 0,+3175-9905+510-1560+480-580+485-550G-565G-1570+505-555GlGhG-1600G-1590+460fGhGh+490-560O-575GlOrGlGhOfO-570O-1610GiO-1625GhGhGlGtG`
output:
`no one signal registered and AC does not beep`

- `irsend {"Protocol":"LG2","Bits":28,"Data":"0x88C0051","DataLSB":"0x1031008A","Repeat":0}`
output:
[... cut ...] "Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1},"RawData":"+3185-9940+450-1655+470-570EfEf+445-1645EfG-595GiG-1640E-1620GiGiEfGiGiEfEfGiGiGi+440iEkEfEkGiGiE-565E-1625E","RawDataInfo":[59,59,0]}}`
uncompressed raw:
`3185,9940,450,1655,470,570,470,570,470,570,445,1645,470,570,445,595,445,595,445,1640,470,1620,445,595,445,595,470,570,445,595,445,595,470,570,470,570,445,595,445,595,445,595,440,595,470,1620,470,570,470,1620,445,595,445,595,470,565,470,1625,470`

crankyoldgit, 2020-10-19:
Thanks for the conversion.

As I indicated earlier, using the raw sending method does not allow the duty cycle to be tweaked. You will have to send via the IRHVAC or LG2 protocol methods in the latest updated branch to test it out.

adrianofoschi, 2020-10-19:
> Thanks for the conversion.
> 
> As I indicated earlier, using the raw sending method does not allow the duty cycle to be tweaked. You will have to send via the IRHVAC or LG2 protocol methods in the latest updated branch to test it out.

OK sorry. As you can see I used ALSO irhvac in my latest tests.

crankyoldgit, 2020-10-19:
> OK sorry. As you can see I used ALSO irhvac in my latest tests.

I saw. But it's not clear to me what, if anything, worked or not. The one of the many mysteries here is why the original remote doesn't work all the time either.

Dumb question/suggestions: You have checked the batteries in the remote, and there isn't a strong heat source/sunlight in the room etc?

adrianofoschi, 2020-10-19:
> I saw. But it's not clear to me what, if anything, worked or not. The one of the many mysteries here is why the original remote doesn't work all the time either.
>
> Dumb question/suggestions: You have [...]

The actual behavior is the same as when I opened the issue:

poweron with
`irhvac {"Vendor":"LG","Model":2,"Mode":"Heat","Power":"On","Celsius":"On","Temp":29,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}`

The AC starts. From this moment no one command works. I sent the command and it does not beep or react in any way.
I tried 2 AC in different rooms and I tried with the AC at my office too (same model).
In the office it worked only for 2 hours some days ago...

I could change the battery of the remote controller but how can be this related to the issue?

adrianofoschi, 2020-10-19:
I recorded a video to clarify the behavior:

1. poweron using irhvac
2. poweroff using irhvac, does not beep and does not turn off the AC
3. poweroff using remote controller. The remote was in standby so I need to press 2 times to power off the AC. The first time is necessary to wake up the remote, the second time to poweroff the AC.

The behavior is the same if I try to send any another command at the step 2.

https://www.youtube.com/watch?v=H9V_0Pck7og

adrianofoschi, 2020-10-20:
The latest test I did:

1. power on using remote controller
2. wait only 2 seconds
3. power off using irhvac

and it works....it seems really that the AC expects something different after X seconds from the turn on

adrianofoschi, 2020-10-20:
Xiaomi remote signals:

power on
```
08:17:14 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":36,"Hash":"0x56F39498","Repeat":0,"RawData":"+8970-4510+510-620+485-1750E-1725+535-570+530-595+505kEdCg+480dMgJgLiLfMdLgEdMfMdL-600MdMkL-645MkLgMkH-1720M-1775L-1700MfCgMfLkL-41290+8920-2280J","RawDataInfo":[71,71,0]}}

8970,4510,510,620,485,1750,485,1725,535,570,530,595,505,595,485,620,510,1725,480,620,480,1725,530,1725,505,570,505,1750,480,620,505,1725,485,620,480,1750,480,620,505,600,480,620,480,595,505,645,480,595,505,1725,480,595,535,1720,480,1775,505,1700,480,1750,510,1725,480,1750,505,595,505,41290,8920,2280,530
```

power off (2 signal, first time it doesn't turn off, only beeps)
```
08:17:58 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x616A6996","DataLSB":"0x86569669","Repeat":0,"RawData":"+8945-4535+510-595+535-1700C-1750+485d+530d+505-570IkC-1720E-590H-1745HlC-615HnH-620CfCoHdCgHgHp+480-1725JpCkIrHlCdIkCnHpClClCdJ-41260A-2280J","RawDataInfo":[71,71,0]}}

8945,4535,510,595,535,1700,510,1750,485,595,530,595,505,570,530,570,510,1720,535,590,485,1745,485,1720,510,615,485,1745,485,620,510,1700,510,615,485,595,510,1750,485,1750,485,620,480,1725,505,620,510,570,530,1725,485,1720,510,595,530,570,510,1745,485,620,510,1720,510,1720,510,595,505,41260,8945,2280,505
```

```
08:18:07 MQT: tele/presa_libera/RESULT = {"IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x616A817E","DataLSB":"0x8656817E","Repeat":0,"RawData":"+8970-4515+510-595C-1720+530-1700C-590ChCdCh+535-1695IhCeC-1725C-570+505kFlFgC-620CgCdCdCdCnChCd+485eC-615OkFeO-1750OkFkCeOdI-41280+8950-2250F","RawDataInfo":[71,71,0]}}

8970,4515,510,595,510,1720,530,1700,510,590,510,590,510,595,510,590,535,1695,535,590,510,1720,510,1725,510,570,505,1725,530,570,530,1700,510,620,510,1700,510,595,510,595,510,595,510,620,510,590,510,595,485,1720,510,615,485,1725,530,1720,485,1750,485,1725,530,1725,510,1720,485,595,535,41280,8950,2250,530
```

If I send the latest poweroff signal when it is turned off, it turns on ...
Anyway I'm going to buy a nodeMCU a create a new adapter to avoid hardware problems

crankyoldgit, 2020-10-20:
Your latest update indicates the Xiaomi remote is sending an NEC code to turn it off. Yet, the data you've reported from the original remote is an LG2 0x88C0051 message.

The fact that you both seem to report that it sometimes ignores the actual remote may mean it's some expected behaviour of the A/C. e.g. It wants to avoid being turned off & on etc rapidly.

To test your theory of a sequence number in the message etc, try just recording the remote, and see if it sends a different code for powering it off etc. If the code number remains the same every time, then the message should be the same.

adrianofoschi, 2020-10-20:
> Your latest update indicates the Xiaomi remote is sending an NEC code to turn it off. Yet, the data you've reported from the original remote is an LG2 0x88C0051 message.
>
> The fact that you both seem [...]

The problem is that any command doesn't work after X seconds, not only the power off.

> 
> To test your theory of a sequence number in the message etc, try just recording the remote, and see if it sends a different code for powering it off etc. If the code number remains the same every time, then the message should be the same.

The turn off code is always the same 0x88C0051. But could be a bug in the receiving? I mean that I see 0x88C0051 but really it is a different code.

crankyoldgit, 2020-10-20:
> > The fact that you both seem to report that it sometimes ignores the actual remote may mean it's some expected behaviour of the A/C. e.g. It wants to avoid being turned off & on etc rapidly.
> 
> The problem is that any command doesn't work after X seconds, not only the power off.

Then it is sounding more and more like its an issue with the A/C unit's design etc.
Not with the library.
 
> The turn off code is always the same 0x88C0051. But could be a bug in the receiving? I mean that I see 0x88C0051 but really it is a different code.

Extremely unlikely. Especially given the value changes for other messages.

adrianofoschi, 2020-10-20:
> Then it is sounding more and more like its an issue with the A/C unit's design etc.
> Not with the library.

do you think is not solvable ?

> 
> > The turn off code is always the same 0x88C0051. But could be a bug in the receiving? I mean that I see 0x88C0051 but really it is a different code.
> 
> Extremely unlikely. Especially given the value changes for other messages.

GiampaMan, 2020-10-20:
Hi @adrianofoschi and @crankyoldgit !

Sorry for delays on my response, I was been testing, and unfortunately confirmed that the bug is still there. I had seen @adrianofoschi video with the issue and I think we are experiencing different weird behaviors. I'm waiting a period of time so the bug happens again and I can record a video so you can see what's happening here. I want to let you know that I'm not using Tasmota, I'm using normal Arduino framework with this library. Also I tried to use the last one modification you did, but the same behavior is there (the one on PR #1304 ). I can record a video so you can see the bug and I hope it can help to solve this issue.

Have a great day and thanks for all!
Giampaolo

PD: I will send the video ASAP

adrianofoschi, 2020-10-20:
> Hi @adrianofoschi and @crankyoldgit !
>
> Sorry for delays on my response, I was been testing, and unfortunately confirmed that the bug is still there. I had seen @adrianofoschi video with the issue [...]

I bought a nodeMCU that will arrive tomorrow. Then I will try again with PR #1304 with Tasmota and Arduino.
Regarding Arduino I ask you to help me with the setup or if you can provide me a working code example to run (I never used arduino).

Thanks you @GiampaMan and @crankyoldgit

crankyoldgit, 2020-10-21:
> do you think is not solvable ?

If the actual IR remote can't do it, there is not a lot we can do. All we are doing in emulating the same message the IR remote is sending.

crankyoldgit, 2020-10-21:
> I bought a nodeMCU that will arrive tomorrow. Then I will try again with PR #1304 with Tasmota and Arduino.
> Regarding Arduino I ask you to help me with the setup or if you can provide me a working code example to run (I never used arduino).

For "decoding" an IR message, please use: [IRrecvDumpV2](https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples/IRrecvDumpV2) or [V3](https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples/IRrecvDumpV2)

For "sending", [IRMQTTServer](https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples/IRMQTTServer) is probably the easiest to operate/change etc, but more complex to initially setup.
Using one of the "TurnOnBlahAC.ino" examples offers the highest functionality, but requires you to be able to do some basic coding. eg. https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples/TurnOnToshibaAC

Then there is a middle ground of: https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/CommonAcControl/CommonAcControl.ino
Again, some simple changes will be needed for your setup etc.

crankyoldgit, 2020-10-21:
> I'm waiting a period of time so the bug happens again and I can record a video so you can see what's happening here.

@GiampaMan I admit I am getting confused between the two of you. Can you please re-explain what your issue is and exactly what you're doing that is not behaving as expected?
e.g. supply a simplified .ino file of what you're doing that reproduces the problem.

adrianofoschi, 2020-10-25:
Hi! I have a good news. 
As I announced I rebuilt the IRblaster circuit using a nodeMCU instead a sonoff basic and all **it works as expected using tasmota-ir 8.5.1 (no PR, no recompiling)**.

Some data:

```
#nodemcu test on
17:45:58 RSL: tele/tasmota_14B9B5/RESULT = {"IrReceived":{"Protocol":"LG2","Bits":28,"Data":"0x8804E46","DataLSB":"0x10017262","Repeat":0,"IRHVAC":{"Vendor":"LG","Model":2,"Mode":"Heat","Power":"On","Celsius":"On","Temp":29,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1},"RawData":[3074,9928,566,1516,522,536,500,534,500,520,522,1572,500,528,520,522,520,520,520,536,498,544,498,536,498,528,520,536,498,1556,520,538,498,536,498,1596,498,1574,518,1566,518,524,518,524,542,1550,496,538,520,514,520,522,520,1534,544,1542,542,500,544],"RawDataInfo":[59,59,0]}}

#nodemcu test off
17:47:36 RSL: tele/tasmota_14B9B5/RESULT = {"IrReceived":{"Protocol":"LG2","Bits":28,"Data":"0x88C0051","DataLSB":"0x1031008A","Repeat":0,"IRHVAC":{"Vendor":"LG","Model":2,"Mode":"Off","Power":"Off","Celsius":"On","Temp":15,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1},"RawData":[3102,9930,562,1504,568,498,546,474,568,498,546,1516,568,488,546,488,546,490,546,1524,546,1532,546,474,568,490,546,498,544,478,564,476,566,490,544,498,546,474,566,490,546,474,568,490,544,1532,546,476,568,1526,566,498,544,490,546,474,566,1534,544],"RawDataInfo":[59,59,0]}}
```

The only difference with sonoff-basic is that the distance. I need to put it nearest the AC.
Then in my case was only an hardware bug.
Thanks you

crankyoldgit, 2020-10-25:
> and all **it works as expected**

Huzzah! That's really good news, because as I said, I was out of ideas.

I'm going to close this issue now. @GiampaMan for your problem, can you create a new issue please?

closed 2020-10-25 by crankyoldgit

adrianofoschi, 2020-10-25:
I did more hardware tests and I discover that probably there was not a problem with the sonoff basic but **the problem was the resistor value in the circuit**. I was using a 100kohm resistor as described in the major irblaster scheme. I replaced it with 10kohm resistor, put the blaster nearest the AC and it works perfectly.