crankyoldgit/IRremoteESP8266 #1305: Problem with coolix protocol

joelzavala, 2020-10-21:
Hi, I have an AC prime but when I decode the signal it tells me it's coolix.

I am analyzing the returned hexadecimal and I see that it has problems to recognize the data with some modes and temperatures.

For example, if I set it to fan mode and turn it up, I set certain temperatures, it changes to dry and doesn't show the temperature and gives me a fan speed that it doesn't have in dry mode;

Temperature 21, fan mode

Protocol: COOLIX
Code: 0xB29FE4 (24 Bits)
Decoded: Power: On, Mode: 4 (Fan), Fan: 4 (Min), Follow Zone: Off, Sensor Temperature: Off
uint16_t rawData[199] = {4446, 4520, 530, 1636, 504, 590, 530, 1636, 532, 1624, 504, 602, 504, 592, 504, 1660, 504, 600, 532, 572, 504, 1652, 506, 600, 502, 602, 504, 1652, 504, 1652, 530, 572, 532, 1616, 532, 1634, 504, 592, 530, 582, 532, 1618, 502, 1662, 504, 1652, 530, 1626, 506, 1650, 502, 594, 502, 1662, 478, 1680, 532, 572, 504, 590, 506, 610, 502, 600, 504, 592, 506, 1650, 506, 1660, 530, 1624, 530, 574, 502, 594, 502, 1662, 506, 592, 504, 610, 504, 600, 502, 594, 530, 584, 504, 1644, 528, 1636, 532, 564, 504, 1660, 530, 1626, 528, 5750, 4400, 4522, 504, 1654, 504, 600, 504, 1654, 504, 1662, 504, 592, 504, 600, 504, 1652, 504, 610, 504, 600, 504, 1654, 530, 572, 530, 568, 504, 1660, 530, 1628, 504, 600, 504, 1652, 506, 1644, 560, 552, 504, 600, 532, 1624, 504, 1646, 532, 1634, 504, 1652, 508, 1642, 506, 608, 532, 1616, 532, 1634, 530, 566, 504, 608, 504, 600, 532, 572, 504, 600, 532, 1616, 506, 1652, 532, 1634, 504, 592, 504, 600, 504, 1654, 504, 600, 504, 608, 506, 592, 502, 612, 530, 574, 504, 1652, 532, 1616, 530, 582, 502, 1646, 506, 1658, 532}; // COOLIX B29FE4
uint64_t data = 0xB29FE4;

Temperature 22, fan mode (dry mode with fan speed 4 does not exist)

Protocol: COOLIX
Code: 0xB29FF4 (24 Bits)
Decryption: Power: On, Mode: 1 (Dry), Fan: 4 (Min), Temperature: 30C, Follow Zone Off, Sensor Temperature: Off
uint16_t rawData[199] = {4446, 4520, 506, 1660, 502, 594, 504, 1658, 530, 1626, 500, 604, 532, 564, 530, 1634, 476, 628, 504, 600, 504, 1652, 504, 600, 504, 600, 506, 1650, 530, 1626, 530, 574, 528, 1620, 504, 1662, 506, 590, 504, 608, 504, 1644, 502, 1662, 502, 1654, 532, 1626, 528, 1626, 504, 594, 500, 1664, 504, 1654, 502, 602, 528, 568, 502, 612, 502, 602, 502, 594, 502, 1654, 530, 1634, 560, 1598, 532, 1624, 530, 574, 506, 1652, 504, 600, 502, 602, 502, 602, 478, 626, 504, 602, 504, 592, 528, 1636, 506, 590, 504, 1662, 502, 1654, 504, 5774, 4400, 4522, 502, 1656, 504, 600, 478, 1680, 506, 1658, 506, 592, 502, 602, 502, 1654, 502, 612, 476, 626, 504, 1654, 506, 600, 504, 592, 500, 1664, 530, 1628, 504, 602, 502, 1654, 504, 1644, 502, 612, 528, 576, 530, 1626, 506, 1644, 504, 1662, 504, 1654, 504, 1646, 504, 610, 530, 1618, 502, 1662, 504, 592, 504, 608, 504, 602, 504, 600, 532, 574, 504, 1644, 532, 1626, 504, 1660, 506, 1652, 530, 576, 502, 1656, 504, 600, 504, 600, 506, 602, 504, 600, 504, 592, 530, 582, 504, 1646, 502, 610, 502, 1644, 504, 1662, 506}; // COOLIX B29FF4
uint64_t data = 0xB29FF4;

I would like to know if I can decode it as prime
I am using irDumpV3

Thank you in advance

NiKiZe, 2020-10-21:
Could you report the full range of temperature (record the data and post it) as well as on and off. Also give us the exact model of AC and remote control as requested in the issue template.

You can disable all protocols or specific ones if you want to, however SupportedProtocols.md does not list your brand, and IR does not really confirm to "one protocol per brand"

joelzavala, 2020-10-21:
this AC don´t have 16.

AC: prime cmprc122-W

17 - 30

Timestamp : 016096.816
                      Library   : v2.7.11

Protocol  : COOLIX
                  Code      : 0xB29FE4 (24 Bits)
                                                Mesg Desc.: Power: On, Mode: 4 (Fan), Fan: 4 (Min), Zone Follow: Off, Sensor Temp: Off
uint16_t rawData[199] = {4450, 4526,  500, 1668,  504, 594,  502, 1666,  502, 1656,  526, 578,  504, 596,  474, 1692,  502, 606,  502, 602,  500, 1658,  506, 600,  478, 630,  476, 1682,  476, 1682,  504, 604,  502, 1648,  476, 1690,  478, 620,  474, 664,  480, 1648,  474, 1692,  500, 1656,  502, 1658,  476, 1682,  476, 622,  504, 1688,  480, 1656,  476, 628,  506, 594,  504, 612,  476, 630,  504, 594,  502, 1656,  474, 1692,  476, 1680,  504, 604,  476, 620,  502, 1666,  502, 596,  504, 634,  478, 628,  478, 594,  476, 638,  504, 1646,  474, 1694,  500, 598,  500, 1668,  502, 1656,  476, 5808,  4428, 4504,  476, 1682,  476, 630,  476, 1682,  502, 1666,  476, 622,  478, 630,  502, 1656,  502, 612,  502, 602,  530, 1630,  528, 576,  504, 594,  476, 1692,  504, 1656,  476, 630,  502, 1658,  476, 1676,  502, 612,  502, 604,  502, 1656,  476, 1676,  500, 1666,  504, 1656,  476, 1676,  504, 610,  502, 1648,  474, 1692,  500, 598,  504, 612,  502, 602,  502, 604,  504, 604,  476, 1674,  504, 1656,  504, 1666,  476, 622,  502, 604,  476, 1684,  502, 602,  504, 610,  504, 594,  476, 638,  504, 602,  476, 1684,  500, 1648,  504, 612,  474, 1676,  504, 1664,  478};  // COOLIX B29FE4
     uint64_t data = 0xB29FE4;

Timestamp : 016098.038
                      Library   : v2.7.11

Protocol  : COOLIX
                  Code      : 0xB29FF4 (24 Bits)
                                                Mesg Desc.: Power: On, Mode: 1 (Dry), Fan: 4 (Min), Temp: 30C, Zone Follow: Off, Sensor Temp: Off
uint16_t rawData[199] = {4432, 4528,  502, 1666,  476, 622,  504, 1664,  502, 1656,  504, 602,  476, 622,  476, 1692,  476, 630,  474, 630,  500, 1658,  500, 604,  504, 604,  500, 1656,  502, 1658,  476, 630,  474, 1674,  502, 1666,  502, 596,  476, 638,  476, 1676,  474, 1694,  474, 1684,  504, 1656,  502, 1658,  476, 622,  476, 1692,  502, 1658,  502, 602,  504, 596,  498, 614,  476, 632,  500, 620,  504, 1630,  476, 1716,  478, 1656,  504, 1658,  472, 632,  476, 1706,  478, 628,  478, 604,  504, 628,  476, 630,  478, 628,  478, 620,  478, 1666,  474, 646,  478, 1666,  504, 1678,  452, 5812,  4402, 4530,  502, 1682,  478, 628,  478, 1684,  478, 1666,  502, 598,  474, 654,  478, 1682,  452, 664,  476, 630,  452, 1706,  452, 656,  448, 648,  478, 1690,  450, 1708,  478, 630,  476, 1684,  476, 1674,  478, 638,  448, 656,  476, 1684,  460, 1668,  474, 1716,  450, 1710,  476, 1676,  450, 664,  450, 1702,  476, 1692,  474, 622,  474, 640,  450, 656,  476, 630,  448, 658,  476, 1674,  478, 1682,  450, 1718,  448, 1710,  450, 654,  450, 1712,  476, 630,  450, 656,  450, 656,  450, 656,  474, 622,  450, 666,  448, 1702,  450, 664,  450, 1700,  474, 1694,  450};  // COOLIX B29FF4
     uint64_t data = 0xB29FF4;

Timestamp : 016099.156
                      Library   : v2.7.11

Protocol  : COOLIX
                  Code      : 0xB29FF4 (24 Bits)
                                                Mesg Desc.: Power: On, Mode: 1 (Dry), Fan: 4 (Min), Temp: 30C, Zone Follow: Off, Sensor Temp: Off
uint16_t rawData[199] = {4452, 4528,  504, 1664,  504, 592,  502, 1664,  476, 1682,  476, 630,  482, 616,  476, 1692,  478, 628,  504, 604,  502, 1654,  504, 602,  504, 602,  502, 1658,  476, 1684,  474, 632,  476, 1676,  476, 1692,  476, 620,  502, 614,  476, 1674,  476, 1692,  476, 1684,  502, 1656,  502, 1658,  502, 596,  504, 1664,  502, 1656,  476, 632,  474, 622,  476, 638,  476, 630,  478, 646,  476, 1658,  502, 1666,  504, 1654,  476, 1684,  476, 630,  502, 1658,  476, 628,  506, 602,  502, 602,  478, 628,  502, 604,  476, 622,  500, 1666,  476, 620,  504, 1666,  476, 1684,  476, 5808,  4406, 4528,  476, 1684,  504, 604,  474, 1684,  474, 1694,  530, 566,  504, 628,  480, 1656,  502, 612,  504, 626,  480, 1654,  476, 628,  504, 598,  472, 1694,  476, 1684,  502, 604,  502, 1656,  502, 1648,  502, 612,  504, 602,  476, 1684,  502, 1650,  474, 1692,  502, 1658,  502, 1650,  500, 612,  504, 1650,  528, 1640,  502, 594,  476, 640,  502, 628,  478, 602,  476, 630,  476, 1674,  502, 1656,  476, 1694,  500, 1660,  474, 630,  504, 1656,  502, 602,  504, 604,  474, 656,  478, 604,  476, 646,  506, 586,  502, 1648,  478, 638,  504, 1646,  504, 1666,  502};  // COOLIX B29FF4
     uint64_t data = 0xB29FF4;

Timestamp : 016100.323
                      Library   : v2.7.11

Protocol  : COOLIX
                  Code      : 0xB29FE4 (24 Bits)
                                                Mesg Desc.: Power: On, Mode: 4 (Fan), Fan: 4 (Min), Zone Follow: Off, Sensor Temp: Off
uint16_t rawData[199] = {4432, 4528,  502, 1666,  500, 596,  502, 1664,  474, 1684,  500, 606,  476, 622,  478, 1690,  478, 628,  530, 576,  502, 1658,  476, 630,  472, 632,  478, 1682,  502, 1660,  474, 628,  502, 1648,  476, 1692,  504, 592,  476, 638,  504, 1646,  476, 1692,  476, 1684,  478, 1682,  502, 1658,  476, 620,  478, 1692,  476, 1684,  500, 602,  476, 622,  530, 586,  502, 604,  480, 618,  476, 1684,  502, 1666,  502, 1658,  502, 606,  504, 594,  502, 1688,  506, 568,  476, 664,  480, 626,  478, 594,  452, 662,  478, 1674,  502, 1692,  478, 594,  476, 1694,  476, 1684,  476, 5808,  4432, 4502,  504, 1656,  502, 604,  506, 1656,  474, 1692,  504, 596,  474,  [... cut ...] , 1656,  504, 602,  478, 620,  476, 1692,  476, 1684,  502, 604,  478, 1682,  476, 1676,  502, 612,  502, 602,  478, 1682,  504, 1648,  532, 1636,  506, 1656,  502, 1650,  502, 614,  476, 1674,  502, 1666,  502, 594,  476, 638,  502, 604,  502, 606,  476, 630,  504, 1648,  476, 1686,  502, 1664,  476, 1686,  502, 602,  478, 1684,  504, 602,  504, 602,  478, 628,  478, 630,  502, 594,  476, 640,  500, 1650,  478, 638,  502, 1648,  476, 1692,  476};  // COOLIX B29FF4
     uint64_t data = 0xB29FF4;

joelzavala, 2020-10-21:
And...

OFF

Timestamp : 025987.753
                      Library   : v2.7.11

Protocol  : COOLIX
                  Code      : 0xB27BE0 (24 Bits)
                                                Mesg Desc.: Power: Off
uint16_t rawData[199] = {4450, 4526,  504, 1662,  504, 594,  502, 1666,  504, 1654,  500, 604,  476, 620,  504, 1664,  476, 628,  502, 602,  504, 1654,  504, 602,  502, 602,  476, 1682,  502, 1656,  506, 598,  504, 1646,  476, 638,  478, 1672,  504, 1664,  504, 1656,  502, 1656,  502, 604,  476, 1682,  502, 1656,  504, 1654,  502, 602,  478, 622,  502, 610,  504, 602,  504, 1656,  504, 600,  476, 622,  478, 1680,  504, 1664,  504, 1656,  504, 602,  504, 592,  506, 600,  476, 638,  478, 620,  478, 636,  476, 630,  504, 602,  506, 1654,  476, 1682,  478, 1674,  478, 1690,  502, 1654,  478, 5806,  4406, 4528,  476, 1684,  474, 630,  502, 1656,  502, 1664,  504, 594,  476, 630,  502, 1656,  502, 612,  502, 604,  502, 1656,  476, 630,  504, 594,  476, 1690,  476, 1684,  502, 604,  476, 1682,  502, 604,  502, 1656,  502, 1648,  502, 1666,  476, 1682,  504, 602,  504, 1656,  500, 1650,  476, 1692,  476, 622,  500, 614,  502, 604,  502, 602,  476, 1684,  476, 630,  502, 604,  500, 1648,  504, 1656,  476, 1692,  476, 622,  502, 604,  502, 612,  500, 604,  502, 604,  476, 632,  474, 630,  476, 630,  474, 1682,  502, 1658,  476, 1684,  502, 1648,  502, 1666,  474};  // COOLIX B27BE0
     uint64_t data = 0xB27BE0;

ON

Timestamp : 025989.131
                      Library   : v2.7.11

Protocol  : COOLIX
                  Code      : 0xB29FF4 (24 Bits)
                                                Mesg Desc.: Power: On, Mode: 1 (Dry), Fan: 4 (Min), Temp: 30C, Zone Follow: Off, Sensor Temp: Off
uint16_t rawData[199] = {4450, 4526,  504, 1664,  502, 594,  504, 1664,  502, 1656,  504, 602,  476, 622,  500, 1666,  502, 602,  476, 630,  504, 1652,  506, 600,  506, 602,  476, 1682,  504, 1654,  532, 574,  502, 1646,  502, 1666,  506, 590,  500, 614,  504, 1646,  504, 1662,  502, 1658,  502, 1656,  532, 1628,  502, 594,  502, 1666,  502, 1656,  504, 600,  478, 622,  502, 612,  476, 628,  478, 620,  476, 1682,  502, 1664,  506, 1654,  530, 1630,  532, 572,  478, 1682,  502, 602,  504, 602,  476, 628,  506, 600,  504, 602,  502, 594,  474, 1692,  478, 620,  478, 1690,  502, 1658,  476, 5806,  4406, 4528,  504, 1656,  474, 632,  504, 1654,  504, 1662,  530, 566,  476, 630,  502, 1656,  504, 610,  502, 604,  504, 1654,  504, 602,  476, 622,  476, 1690,  478, 1682,  504, 600,  504, 1656,  502, 1648,  504, 610,  504, 602,  478, 1682,  476, 1674,  502, 1666,  500, 1658,  504, 1646,  504, 610,  502, 1648,  502, 1666,  476, 622,  502, 614,  502, 602,  504, 600,  476, 628,  478, 1674,  476, 1682,  504, 1664,  502, 1656,  532, 576,  476, 1682,  530, 576,  504, 602,  504, 602,  504, 602,  506, 592,  504, 612,  504, 1648,  502, 612,  504, 1648,  476, 1692,  478};  // COOLIX B29FF4
     uint64_t data = 0xB29FF4;

crankyoldgit, 2020-10-22:
Can you please supply what `Code:`/`data` you get for setting of _"Power On, 25C, Mode: Auto, Fan: Auto, Zone Follow: Off, Sensor Temp: Ignore."_ from the original remote?

After a quick look, it seems your device is using the same message protocol, but a different information structure.
i.e. There doesn't look like there is a way for us to differentiate it based on the message timing values, but we'll need to look for some identifying pattern in the `code`/`data`, hence if I get the code for that I might be able to compare it to a known value and see what, if anything is different.

NiKiZe, 2020-10-22:
Does the remote have a display? do you have a picture of the remote?
We only see 2-3 different messages, One possiblity is that it is a "dumb" remote that only sends keypress and not state.
If that is true then a press on the same button would always give you the same code, and other buttons would give a different code.

crankyoldgit, 2020-10-22:
> We only see 2-3 different messages, One possiblity is that it is a "dumb" remote that only sends keypress and not state.

Well spotted @NiKiZe! I completely missed that. Yeah. That is very odd/unusual for an A/C remote.

joelzavala, 2020-10-22:
Yes, it has a display my control, but the strange thing is that if the temperature changes on the display but as you say, the code remains

I have a universal remote control, but if it works with my AC

joelzavala, 2020-10-22:
Here it is in temperature 25, mode Auto, fan speed auto

Timestamp : 002502.001
                      Library : v2.7.11

Protocol : COOLIX
                  Code : 0xB21F88 (24 Bits)
                                                Mesg Off: Power: On, Mode: 2 (Auto), Fan: 0 (Auto0), Temp: 28C, Zone Follow: Off, Sensor Temp: Off
uint16_t rawData[199] = {4448, 4518, 506, 1658, 506, 590, 480, 1686, 504, 1652, 504, 600, 478, 616, 506, 1658, 506, 598, 504, 600, 506, 1648, 506, 598, 504, 598, 478, 1678, 504, 1650, 480, 624, 506, 1642, 504, 608, 480, 624, 480, 624, 478, 1656, 528, 1650, 504, 1644,  502, 1662, 504, 1652, 504, 1644, 506, 1658, 478, 1678, 506, 598, 504, 592, 504, 608, 504, 600, 504, 590, 506, 1648, 506, 598, 480, 634,  504, 600, 504, 1652, 504, 598, 506, 598, 508, 596, 506, 600, 506, 1650, 504, 1652, 506, 1640, 504, 608, 506, 1642, 506, 1658, 506, 1650, 504, 5750, 4422, 4520, 506, 1650, 480, 602, 530, 1648, 480, 1684, 506, 590, 504, 600, 502, 1652, 504, 608, 504, 600, 504, 1630,  530, 596, 506, 590, 504, 1660, 506, 1650, 504, 598, 506, 1630, 528, 598, 504, 600, 480, 624, 480, 1676, 506, 1652, 504, 1652, 506, 1642,  482, 1682, 480, 1676, 480, 1668, 478, 1686, 504, 590, 504, 608, 504, 600, 480, 624, 480, 624, 480, 1668, 504, 610, 504, 598, 506, 598,  480, 1676, 480, 626, 478, 626, 480, 624, 480, 626, 478, 1678, 506, 1652, 504, 1650, 480, 624, 480, 1676, 504, 1644, 478, 1686, 480};  // COOLIX B21F88
     uint64_t data = 0xB21F88;

Translated with www.DeepL.com/Translator (free version)

crankyoldgit, 2020-10-22:
> I have a universal remote control (avaly va-cr1000u) , but if it works with my AC

Is that the device you've been recording, or the original IR remote?
As @NiKiZe has indicated, you're data indicates it's not sending out the "full" internal state of the A/C in the message that we can see. i.e. Each message where you changed the temp (17-30C etc) should be a different code. So it is unlikely we can add detailed (constructing new modes, temps, & settings etc) support for your variant of the A/C.

joelzavala, 2020-10-23:
Yes, with the universal control I am recording the data,
but this is the one that works with my climate

crankyoldgit, 2020-10-23:
Do you have access to the OEM remote at all?

joelzavala, 2020-10-23:
Yes, I have the original control, is that what you mean?

crankyoldgit, 2020-10-24:
Yes. Please use the original controller when testing & collecting data. We could be implementing someone else's bugs when we use a third-party controller.

I should add an FAQ on that. :-/
'

joelzavala, 2020-10-24:
Sure, give me a few minutes for get the data :)

crankyoldgit, 2020-10-24:
Also, if you can, include a pic of it too. :-)

crankyoldgit, 2020-10-24:
Added it to the FAQ https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#i-want-to-support-a-new-device-but-all-i-have-is-an-all-in-one-learning-or-app-driven-ir-remote-device-for-sending-messages

:-)

crankyoldgit, 2020-10-27:
> Sure, give me a few minutes for get the data :)

@joelzavala  Friendly reminder. ;-)

crankyoldgit, 2020-10-31:
> > Sure, give me a few minutes for get the data :)
> 
> @joelzavala Friendly reminder. ;-)

Another friendly reminder!

joelzavala, 2020-11-01:
Hi.

Temperature data from 17- 30

Timestamp : 007532.995
                      Library   : v2.7.11

Protocol  : COOLIX
                  Code      : 0xB23F00 (24 Bits)
                                                Mesg Desc.: Power: On, Mode: 0 (Cool), Fan: 1 (Max), Temp: 17C, Zone Follow: Off, Sensor Temp: Off
uint16_t rawData[199] = {4438, 4288,  640, 1544,  592, 494,  592, 1574,  622, 1522,  614, 492,  618, 466,  594, 1568,  596, 490,  594, 492,  594, 1572,  620, 464,  590, 494,  594, 1568,  620, 1542,  592, 494,  588, 1580,  594, 494,  622, 462,  592, 1574,  616, 1548,  622, 1540,  624, 1538,  594, 1570,  622, 1542,  598, 1572,  598, 1566,  622, 464,  628, 458,  596, 488,  600, 486,  592, 492,  654, 432,  596, 490,  594, 490,  594, 492,  624, 460,  622, 464,  596, 488,  594, 492,  594, 492,  622, 1542,  624, 1540,  622, 1542,  620, 1544,  622, 1542,  624, 1538,  592, 1572,  624, 1540,  468, 5252,  4418, 4296,  622, 1542,  596, 488,  592, 1576,  620, 1544,  622, 462,  624, 462,  624, 1538,  620, 444,  616, 492,  592, 1576,  592, 492,  588, 496,  622, 1540,  592, 1570,  594, 492,  620, 1550,  622, 464,  620, 464,  624, 1542,  622, 1542,  592, 1572,  594, 1568,  594, 1568,  594, 1572,  594, 1574,  618, 1546,  590, 494,  620, 466,  586, 498,  594, 490,  590, 494,  592, 494,  594, 492,  598, 486,  594, 492,  622, 462,  594, 490,  618, 444,  614, 494,  626, 460,  594, 1572,  622, 1542,  618, 1544,  592, 1570,  620, 1542,  592, 1572,  620, 1542,  624, 1542,  492};  // COOLIX B23F00
     uint64_t data = 0xB23F00;

Timestamp : 007536.186
                      Library   : v2.7.11

Protocol  : COOLIX
                  Code      : 0xB23F10 (24 Bits)
                                                Mesg Desc.: Power: On, Mode: 0 (Cool), Fan: 1 (Max), Temp: 18C, Zone Follow: Off, Sensor Temp: Off
uint16_t rawData[199] = {4290, 4436,  626, 1538,  620, 464,  592, 1574,  592, 1570,  590, 496,  590, 494,  592, 1570,  620, 466,  594, 492,  594, 1572,  624, 460,  596, 488,  594, 1568,  596, 1566,  592, 494,  598, 1570,  592, 472,  610, 496,  590, 1576,  620, 1542,  592, 1550,  614, 1570,  626, 1514,  616, 1570,  594, 1576,  620, 1542,  624, 460,  592, 492,  620, 464,  596, 490,  594, 490,  618, 468,  590, 496,  594, 490,  594, 490,  596, 1566,  620, 464,  620, 464,  592, 492,  592, 494,  622, 1548,  624, 1538,  596, 1566,  594, 470,  614, 1574,  616, 1546,  594, 1570,  596, 1568,  492, 5230,  4436, 4296,  592, 1570,  622, 462,  596, 1572,  590, 1574,  594, 490,  590, 494,  618, 1544,  624, 462,  592, 474,  614, 1574,  624, 462,  624, 460,  594, 1568,  622, 1540,  594, 490,  616, 1552,  620, 468,  590, 494,  592, 1574,  588, 1574,  622, 1540,  620, 1544,  624, 1538,  620, 1544,  596, 1572,  596, 1566,  594, 490,  620, 464,  590, 494,  594, 490,  596, 490,  592, 494,  624, 462,  626, 458,  624, 438,  610, 1574,  624, 460,  592, 492,  622, 464,  592, 494,  590, 1578,  622, 1540,  592, 1572,  622, 462,  590, 1576,  594, 1568,  628, 1536,  624, 1540,  470};  // COOLIX B23F10
     uint64_t data = 0xB23F10;

Timestamp : 007537.669
                      Library   : v2.7.11

Protocol  : COOLIX
                  Code      : 0xB23F30 (24 Bits)
                                                Mesg Desc.: Power: On, Mode: 0 (Cool), Fan: 1 (Max), Temp: 19C, Zone Follow: Off, Sensor Temp: Off
uint16_t rawData[199] = {4436, 4282,  648, 1518,  640, 466,  592, 1574,  624, 1538,  626, 458,  620, 466,  622, 1540,  592, 494,  624, 442,  608, 1580,  626, 456,  596, 490,  596, 1566,  594, 1550,  648, 434,  642, 1550,  620, 466,  590, 494,  592, 1574,  596, 1566,  588, 1574,  594, 1570,  622, 1540,  618, 1546,  618, 1550,  620, 1542,  594, 490,  594, 490,  620, 444,  612, 494,  590, 494,  592, 494,  592, 494,  622, 462,  620, 1546,  620, 1542,  618, 466,  594, 490,  620, 464,  590, 496,  620, 1548,  620, 1542,  620, 464,  588, 496,  592, 1570,  596, 1566,  596, 1566,  618, 1544,  470, 5254,  4434, 4300,  622, 1542,  622, 462,  596, 1570,  618, 1544,  594, 490,  590, 494,  592, 1570,  590, 496,  624, 462,  594, 1572,  622, 464,  592, 492,  620, 1542,  624, 1538,  618, 464,  622, 1546,  628, 460,  620, 466,  624, 1520,  642, 1544,  596, 1568,  626, 1536,  594, 1568,  624, 1540,  622, 1548,  622, 1540,  622, 462,  590, 472,  608, 498,  592, 494,  594, 490,  622, 466,  596, 490,  592, 492,  592, 1574,  622, 1540,  592, 492,  592, 492,  592, 492,  622, 464,  596, 1572,  596, 1566,  624, 462,  620, 464,  598, 1564,  620, 1540,  622, 1540,  618, 1546,  494};  // COOLIX B23F30
     uint64_t data = 0xB23F30;

Timestamp : 007539.134
                      Library   : v2.7.11

Protocol  : COOLIX
                  Code      : 0xB23F20 (24 Bits)
                                                Mesg Desc.: Power: On, Mode: 0 (Cool), Fan: 1 (Max), Temp: 20C, Zone Follow: Off, Sensor Temp: Off
uint16_t rawData[199] = {4436, 4300,  622, 1540,  622, 462,  620, 1546,  620, 1542,  624, 460,  592, 492,  594, 1568,  620, 466,  620, 466,  594, 1572,  594, 490,  600, 486,  592, 1570,  596, 1566,  622, 464,  596, 1574,  622, 462,  588, 496,  592, 1576,  598, 1564,  596, 1546,  616, 1568,  596, 1566,  618, 1546,  624, 1544,  622, 1542,  592, 492,  626, 458,  622, 462,  594, 490,  596, 490,  592, 494,  592, 494,  594, 490,  590, 1576,  622, 462,  594, 490,  594, 492,  618, 466,  594, 470,  644, 1544,  622, 1538,  592, 492,  592, 1572,  624, 1540,  622, 1540,  624, 1538,  626, 1538,  492, 5250,  4412, 4300,  590, 1572,  624, 460,  596, 1570,  626, 1536,  622, 464,  [... cut ...] , 1570,  626, 458,  622, 462,  624, 1538,  594, 1568,  592, 494,  590, 1576,  592, 496,  594, 490,  594, 1572,  592, 1572,  592, 1548,  644, 1540,  622, 1542,  618, 1546,  622, 1548,  596, 1566,  622, 462,  594, 490,  626, 460,  590, 494,  620, 464,  618, 468,  588, 1558,  612, 492,  596, 1570,  624, 1540,  596, 488,  598, 486,  594, 490,  624, 442,  640, 468,  594, 1568,  590, 494,  590, 494,  594, 1570,  620, 1542,  622, 1540,  594, 1572,  468};  // COOLIX B23FB0
     uint64_t data = 0xB23FB0;

crankyoldgit, 2020-11-01:
Okay, thanks for the data. So this data seems to be a) different from your 3rd party remote, and b) more like what we are/were expecting for the Coolix protocol. i.e. When given the original messages from the original remote, the library seems to be behaving as normal/expected.

Have you tried using `COOLIX`/ the [`IRCoolixAC` class](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRCoolixAC.html) to control your A/C device?

joelzavala, 2020-11-01:
no, I would have to try it

crankyoldgit, 2020-11-03:
I would suggest you try it via `COOLIX` using either IRMQTTServer (in the examples) or the IRhvac interface on Tasmota.

crankyoldgit, 2020-11-07:
> no, I would have to try it

@joelzavala Any success or update?

crankyoldgit, 2020-11-24:
ping.

crankyoldgit, 2020-11-26:
Closing due to lack of activity. Update this issue and I'll re-open it.

closed 2020-11-26 by crankyoldgit