crankyoldgit/IRremoteESP8266 #1306: Help needed for encoding ZSP-IR-B / ZSP-IR-W IR-Codes

linuxlurak, 2020-10-22:
I'm running a tasmota-ir device and try to control a motorized "elite screens"-screen with a ZSP-IR-B/W (black or white).

Elite screens is using three simple codes, but i don't know how to get them running / how to encode them...
here is the link to the pdf: https://elitescreens.com/kcfinder/upload/files/FAQ/ir_commands.pdf
It says:
[image]

So binary codes are:
```
Up: 111100000001
Stop: 111100000010
Down: 111100000100
```
What to do with them?

Thanks for any help.

NiKiZe, 2020-10-22:
More info is needed to create codes from documentation, for example encoding, and mark and space length. A quicker way might be to capture existing remote with the DumpV2+ sketch

crankyoldgit, 2020-10-22:
Please read/follow: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-IR-protocol

linuxlurak, 2020-10-24:
Thanks for nudging me to the right way.

I got this serial output:
UP:
```
rotocol=UNKNOWN Data=0x2D8CB141 (32 bits)
Timing[63]: 
     +1250, - 450     +1250, - 450     +1250, - 500     +1200, - 500
     + 450, -1250     + 450, -1200     + 500, -1200     + 500, -1200
     + 500, -1200     + 500, -1200     +1250, - 450     + 500, -1200
     + 450, -1250     +1250, - 450     + 450, -1250     +1250, - 450
     +1250, - 450     +1250, - 450     + 500, -1200     +1250, - 450
     +1250, - 450     +1250, - 450     + 500, -1200     +1250, - 450
     +1250, - 450     + 500, -1200     +1250, - 450     + 500, -1200
     +1250, - 450     + 500, -1200     + 500, -1200     +1250
unsigned int  rawData[63] = {1250,450, 1250,450, 1250,500, 1200,500, 450,1250, 450,1200, 500,1200, 500,1200, 500,1200, 500,1200, 1250,450, 500,1200, 450,1250, 1250,450, 450,1250, 1250,450, 1250,450, 1250,450, 500,1200, 1250,450, 1250,450, 1250,450, 500,1200, 1250,450, 1250,450, 500,1200, 1250,450, 500,1200, 1250,450, 500,1200, 500,1200, 1250};  // Protocol=UNKNOWN Data=0x2D8CB141
Pronto Hex: 0000 006D 0020 0000 0030 0011 0030 0011 0030 0013 002E 0013 0011 0030 0011 002E 0013 002E 0013 002E 0013 002E 0013 002E 0030 0011 0013 002E 0011 0030 0030 0011 0011 0030 0030 0011 0030 0011 0030 0011 0013 002E 0030 0011 0030 0011 0030 0011 0013 002E 0030 0011 0030 0011 0013 002E 0030 0011 0013 002E 0030 0011 0013 002E 0013 002E 0030 0806 
```
DOWN:
```
Protocol=UNKNOWN Data=0x4277EAC1 (32 bits)
Timing[63]: 
     +1300, - 450     +1250, - 450     +1250, - 450     +1250, - 450
     + 450, -1250     + 500, -1200     + 450, -1250     + 450, -1250
     + 450, -1250     +1200, - 500     + 450, -1200     + 500, -1200
     + 500, -1200     +1250, - 450     + 500, -1200     +1250, - 450
     +1250, - 450     +1250, - 500     +1200, - 500     + 450, -1250
     +1250, - 450     +1250, - 450     +1250, - 450     + 450, -1250
     +1250, - 450     +1250, - 450     + 450, -1250     + 450, -1250
     +1250, - 450     + 450, -1250     + 450, -1250     +1200
unsigned int  rawData[63] = {1300,450, 1250,450, 1250,450, 1250,450, 450,1250, 500,1200, 450,1250, 450,1250, 450,1250, 1200,500, 450,1200, 500,1200, 500,1200, 1250,450, 500,1200, 1250,450, 1250,450, 1250,500, 1200,500, 450,1250, 1250,450, 1250,450, 1250,450, 450,1250, 1250,450, 1250,450, 450,1250, 450,1250, 1250,450, 450,1250, 450,1250, 1200};  // Protocol=UNKNOWN Data=0x4277EAC1
Pronto Hex: 0000 006D 0020 0000 0032 0011 0030 0011 0030 0011 0030 0011 0011 0030 0013 002E 0011 0030 0011 0030 0011 0030 002E 0013 0011 002E 0013 002E 0013 002E 0030 0011 0013 002E 0030 0011 0030 0011 0030 0013 002E 0013 0011 0030 0030 0011 0030 0011 0030 0011 0011 0030 0030 0011 0030 0011 0011 0030 0011 0030 0030 0011 0011 0030 0011 0030 002E 0806 
```

STOP:```

Protocol=UNKNOWN Data=0x29054E (32 bits)
Timing[63]: 
     +1300, - 450     +1250, - 450     +1250, - 450     +1250, - 450
     + 500, -1200     + 450, -1250     + 450, -1250     + 450, -1250
     + 450, -1200     + 500, -1200     + 500, -1200     +1250, - 450
     + 500, -1200     +1250, - 450     + 500, -1200     +1250, - 450
     +1250, - 500     +1200, - 450     + 500, -1200     + 500, -1200
     +1250, - 450     +1250, - 450     + 500, -1200     + 500, -1200
     + 500, -1200     +1250, - 450     +1250, - 450     +1250, - 500
     +1200, - 450     + 500, -1200     + 500, -1200     + 500
unsigned int  rawData[63] = {1300,450, 1250,450, 1250,450, 1250,450, 500,1200, 450,1250, 450,1250, 450,1250, 450,1200, 500,1200, 500,1200, 1250,450, 500,1200, 1250,450, 500,1200, 1250,450, 1250,500, 1200,450, 500,1200, 500,1200, 1250,450, 1250,450, 500,1200, 500,1200, 500,1200, 1250,450, 1250,450, 1250,500, 1200,450, 500,1200, 500,1200, 500};  // Protocol=UNKNOWN Data=0x29054E
Pronto Hex: 0000 006D 0020 0000 0032 0011 0030 0011 0030 0011 0030 0011 0013 002E 0011 0030 0011 0030 0011 0030 0011 002E 0013 002E 0013 002E 0030 0011 0013 002E 0030 0011 0013 002E 0030 0011 0030 0013 002E 0011 0013 002E 0013 002E 0030 0011 0030 0011 0013 002E 0013 002E 0013 002E 0030 0011 0030 0011 0030 0013 002E 0011 0013 002E 0013 002E 0013 0806 
```

Now, as you can guess, I don't know how to proceed...

crankyoldgit, 2020-10-24:
Can you please use _our_ library and _our_ IRrecvDumpV2(orV3) to do the captures? That we will know if the protocol is already supported or not.

linuxlurak, 2020-10-24:
Thanks again for this hint. Now the results I got with IRrecvDumpV3 from this repo:
UP:
```
Zeitstempel : 000341.596
Bibliothek   : v2.7.11

Protocol  : UNKNOWN
Code      : 0x24750A78 (64 Bits)
uint16_t rawData[127] = {1276, 486,  1216, 488,  1224, 482,  1220, 486,  444, 1222,  480, 1242,  450, 1244,  448, 1224,  478, 1216,  474, 1220,  1254, 482,  448, 1218,  474, 1248,  1228, 454,  476, 1218,  1246, 490,  1254, 454,  1250, 456,  472, 1194,  1250, 458,  1276, 458,  1244, 462,  480, 1186,  1246, 464,  1280, 456,  474, 1192,  1252, 482,  448, 1220,  1244, 462,  500, 1194,  476, 1218,  1278, 29686,  1276, 432,  1282, 424,  1278, 428,  1274, 460,  482, 1186,  474, 1222,  482, 1214,  478, 1220,  472, 1224,  478, 1216,  1280, 428,  502, 1192,  478, 1216,  1280, 456,  474, 1192,  1282, 428,  1274, 434,  1280, 426,  504, 1190,  1274, 432,  1280, 426,  1276, 430,  500, 1194,  1282, 428,  1274, 432,  508, 1186,  1278, 430,  500, 1194,  1282, 452,  478, 1190,  502, 1194,  1282};  // UNKNOWN 24750A78
```

DOWN:
```
Zeitstempel : 000387.910
Bibliothek   : v2.7.11

Protocol  : UNKNOWN
Code      : 0xAA98093C (64 Bits)
uint16_t rawData[127] = {1312, 422,  1282, 450,  1252, 454,  1248, 458,  474, 1194,  508, 1188,  506, 1190,  502, 1198,  506, 1188,  1276, 458,  472, 1196,  508, 1188,  504, 1190,  1274, 458,  482, 1186,  1280, 456,  1256, 450,  1252, 426,  1276, 430,  500, 1194,  1282, 424,  1278, 428,  1274, 432,  510, 1188,  1276, 430,  1284, 422,  508, 1188,  504, 1190,  1274, 432,  508, 1186,  506, 1188,  1276, 29688,  1284, 422,  1280, 426,  1276, 428,  1274, 432,  510, 1184,  506, 1188,  504, 1192,  500, 1198,  504, 1190,  1274, 432,  510, 1186,  506, 1188,  504, 1192,  1284, 450,  482, 1186,  1278, 458,  1254, 452,  1252, 454,  1248, 458,  472, 1196,  1280, 454,  1250, 456,  1246, 460,  480, 1192,  1274, 458,  1254, 452,  480, 1188,  504, 1192,  1284, 450,  480, 1188,  504, 1190,  1276};  // UNKNOWN AA98093C
```

STOP:
```
Protocol  : UNKNOWN
Code      : 0x148A4DFF (64 Bits)
uint16_t rawData[127] = {1278, 456,  1248, 486,  1226, 452,  1252, 454,  474, 1220,  474, 1222,  480, 1214,  476, 1222,  472, 1224,  478, 1216,  476, 1220,  1256, 480,  450, 1216,  1248, 488,  442, 1222,  1252, 486,  1226, 478,  1224, 482,  448, 1220,  472, 1222,  1254, 482,  1220, 486,  444, 1222,  480, 1218,  474, 1220,  1254, 482,  1222, 484,  1218, 488,  1224, 482,  450, 1218,  474, 1220,  470, 30482,  1246, 460,  1274, 460,  1220, 456,  1246, 488,  452, 1214,  478, 1218,  474, 1220,  470, 1228,  476, 1220,  472, 1222,  480, 1214,  1248, 486,  476, 1190,  1274, 460,  480, 1186,  1278, 460,  1252, 426,  1276, 456,  474, 1194,  478, 1216,  1280, 456,  1246, 460,  482, 1186,  474, 1224,  478, 1216,  1282, 454,  1248, 458,  1244, 462,  1252, 456,  474, 1192,  480, 1216,  476};  // UNKNOWN 148A4DFF
```

crankyoldgit, 2020-10-25:
@linuxlurak Can you please download & test PR #1310 / branch https://github.com/crankyoldgit/IRremoteESP8266/tree/EliteScreens and let me know how it goes?

Unfortunately, I can't easily see how to covert the documented bit patterns into the `sendElitescreens()` hex codes etc, so for the meantime you will have to use the 32 bit values, rather than the 12 values their documentation suggests.
e.g. STOP: (from the manual) `0b111100000010`== `0xF02`; (decodes as) `0b1111111010100011001110000111` == `0xFEA3387`;
i.e.
```
sendElitescreens(0xFEA3387);  // STOP button.
```

crankyoldgit, 2020-10-25:
FYI:
https://github.com/crankyoldgit/IRremoteESP8266/blob/5799f6f31ff21a4b6da0ca096f472fcaf5c81d6c/src/ir_EliteScreens.cpp#L15-L18

linuxlurak, 2020-10-25:
I downloaded EliteScreen Branch and compiled 

It gives this result:
```
Zeitstempel : 000054.054
Bibliothek   : v2.7.11

Protocol  : ELITESCREENS
Code      : 0xFBA1136 (32 Bits)
uint16_t rawData[127] = {1274, 460,  1254, 452,  1250, 484,  1218, 488,  442, 1224,  480, 1214,  476, 1218,  474, 1224,  478, 1216,  1248, 460,  470, 1224,  478, 1216,  476, 1218,  1246, 460,  480, 1216,  1250, 460,  1254, 480,  1224, 484,  1218, 488,  442, 1224,  1252, 454,  1248, 458,  1254, 452,  478, 1218,  1246, 488,  1226, 482,  448, 1246,  446, 1220,  1256, 452,  478, 1218,  474, 1220,  1246, 29718,  1252, 454,  1248, 458,  1254, 480,  1224, 482,  448, 1246,  446, 1220,  470, 1224,  478, 1220,  472, 1222,  1252, 454,  476, 1220,  474, 1220,  470, 1224,  1252, 456,  474, 1220,  1254, 454,  1248, 460,  1254, 478,  1224, 482,  448, 1218,  1246, 460,  1254, 454,  1248, 486,  446, 1224,  1250, 456,  1246, 460,  470, 1224,  478, 1216,  1250, 458,  472, 1222,  470, 1224,  1250};  // ELITESCREENS FBA1136
uint64_t data = 0xFBA1136;
_==> Pressing DOWN-Button_

Zeitstempel : 000059.230
Bibliothek   : v2.7.11

Protocol  : ELITESCREENS
Code      : 0xFEA3387 (32 Bits)
uint16_t rawData[191] = {1274, 460,  1252, 454,  1250, 458,  1246, 460,  470, 1226,  478, 1216,  476, 1244,  448, 1224,  478, 1216,  476, 1220,  472, 1222,  1252, 456,  474, 1220,  1246, 462,  478, 1216,  1250, 460,  1254, 452,  1250, 484,  446, 1222,  470, 1222,  1252, 456,  1248, 458,  472, 1250,  454, 1244,  448, 1220,  1254, 452,  1250, 458,  1246, 460,  1252, 454,  476, 1218,  474, 1220,  472, 30480,  1252, 456,  1246, 460,  1254, 452,  1250, 456,  474, 1220,  472, 1222,  482, 1214,  478, 1220,  472, 1250,  452, 1216,  478, 1218,  1246, 462,  480, 1214,  1250, 458,  472, 1222,  1254, 458,  1246, 460,  1254, 478,  450, 1218,  474, 1218,  1248, 460,  1252, 482,  448, 1220,  472, 1224,  478, 1242,  1222, 458,  1254, 452,  1250, 456,  1246, 486,  444, 1250,  452, 1214,  478, 30474,  1256, 452,  1252, 456,  1246, 460,  1254, 478,  452, 1216,  476, 1218,  476, 1220,  472, 1226,  476, 1220,  472, 1222,  482, 1240,  1224, 484,  446, 1222,  1254, 454,  476, 1218,  1248, 462,  1250, 456,  1248, 486,  444, 1224,  478, 1216,  1250, 458,  1244, 488,  454, 1242,  450, 1220,  472, 1224,  1252, 456,  1248, 460,  1254, 480,  1224, 482,  448, 1220,  472, 1222,  470};  // ELITESCREENS FEA3387
uint64_t data = 0xFEA3387;
_==> Pressing STOP-Button_

Zeitstempel : 000062.969
Bibliothek   : v2.7.11

Protocol  : ELITESCREENS
Code      : 0xFEA3387 (32 Bits)
uint16_t rawData[127] = {1252, 482,  1220, 486,  1216, 490,  1224, 456,  496, 1252,  418, 1248,  444, 1278,  424, 1276,  418, 1278,  498, 1170,  446, 1248,  1218, 488,  452, 1268,  1196, 510,  420, 1276,  1200, 482,  1220, 514,  1200, 508,  422, 1272,  420, 1276,  1188, 518,  1216, 490,  420, 1274,  418, 1254,  448, 1272,  1192, 514,  1198, 508,  1196, 510,  1192, 488,  442, 1254,  450, 1246,  446, 30506,  1246, 460,  1220, 484,  1218, 516,  1198, 508,  422, 1274,  418, 1276,  416, 1280,  424, 1248,  476, 1220,  472, 1222,  448, 1272,  1192, 516,  426, 1242,  1222, 512,  418, 1276,  1198, 510,  1192, 514,  1200, 534,  418, 1252,  440, 1254,  1190, 518,  1196, 510,  420, 1248,  444, 1254,  448, 1246,  1218, 488,  1224, 508,  1194, 538,  1164, 488,  442, 1254,  450, 1246,  444};  // ELITESCREENS FEA3387
uint64_t data = 0xFEA3387;
_==> Pressing STOP-Button_

Zeitstempel : 000067.373
Bibliothek   : v2.7.11

Protocol  : ELITESCREENS
Code      : 0xFDA2256 (32 Bits)
uint16_t rawData[127] = {1280, 456,  1248, 458,  1244, 462,  1252, 482,  448, 1244,  448, 1246,  446, 1222,  480, 1220,  472, 1222,  480, 1214,  1250, 484,  446, 1220,  472, 1224,  1252, 484,  446, 1220,  1256, 456,  1246, 460,  1254, 478,  450, 1214,  1250, 460,  1244, 488,  1224, 482,  448, 1244,  1220, 464,  1248, 484,  446, 1220,  1254, 480,  450, 1216,  1248, 460,  470, 1222,  480, 1214,  1250, 29714,  1256, 452,  1250, 456,  1248, 458,  1244, 490,  452, 1214,  478, 1218,  474, 1220,  472, 1228,  476, 1218,  474, 1222,  1254, 482,  448, 1218,  474, 1220,  1254, 480,  450, 1216,  1248, 464,  1250, 456,  1246, 488,  444, 1222,  1254, 456,  1248, 458,  1244, 490,  452, 1214,  1250, 462,  1252, 482,  448, 1216,  1248, 488,  442, 1224,  1252, 456,  474, 1220,  472, 1222,  1254};  // ELITESCREENS FDA2256
uint64_t data = 0xFDA2256;
_==> Pressing UP-Button_
```

Is this, what you asked for to test? It seems to decode the received codes very well.

Do you know how I can test the raw code directly with tasmota-ir (that is built on your code)? I guess it will take a while till tasmota-ir releases a new version that includes this update.

crankyoldgit, 2020-10-25:
> Is this what you asked for to test? It seems to decode the received codes very well.

Excellent. Yes. And I'd like you to confirm you can send with it too. e.g. `sendElitescreens();` etc.

> Do you know how I can test the raw code directly with tasmota-ir (that is built on your code)? I guess it will take a while till tasmota-ir releases a new version that includes this update.

Yes, you can send raw via Tasmota. I don't know their format off the top of my head for that. You'll have to ask them.

They (Tasmota) tend to track this library fairly quickly. So, less than a month or so.
In the meantime, you can replace the version included with Tasmota by copying that branch into the Tasmota `lib` over the one there. It should allow you to use the non-raw version.

linuxlurak, 2020-10-25:
Thanks a lot for your support. I figured out the working config. For the sake of completeness here is the yaml-config for home assistant in combination with a tasmota-ir using raw IR codes:

```
elitescreen_down:
  alias: elitescreen down
  sequence:
  - service: mqtt.publish
    data:
      topic: [CHANGE TO CORRECT TOPIC eg. "cmnd/tasmota_40E0B0/IRsend" without quotes]
      payload: '0,1274,460,1254,452,1250,484,1218,488,442,1224,480,1214,476,1218,474,1224,478,1216,1248,460,470,1224,478,1216,476,1218,1246,460,480,1216,1250,460,1254,480,1224,484,1218,488,442,1224,1252,454,1248,458,1254,452,478,1218,1246,488,1226,482,448,1246,446,1220,1256,452,478,1218,474,1220,1246,29718,1252,454,1248,458,1254,480,1224,482,448,1246,446,1220,470,1224,478,1220,472,1222,1252,454,476,1220,474,1220,470,1224,1252,456,474,1220,1254,454,1248,460,1254,478,1224,482,448,1218,1246,460,1254,454,1248,486,446,1224,1250,456,1246,460,470,1224,478,1216,1250,458,472,1222,470,1224,1250'
      qos: 2
  mode: single
elitescreen_up:
  alias: elitescreen up
  sequence:
  - service: mqtt.publish
    data:
      topic: [CHANGE TO CORRECT TOPIC eg. "cmnd/tasmota_40E0B0/IRsend" without quotes]
      payload: '0,1280,456,1248,458,1244,462,1252,482,448,1244,448,1246,446,1222,480,1220,472,1222,480,1214,1250,484,446,1220,472,1224,1252,484,446,1220,1256,456,1246,460,1254,478,450,1214,1250,460,1244,488,1224,482,448,1244,1220,464,1248,484,446,1220,1254,480,450,1216,1248,460,470,1222,480,1214,1250,29714,1256,452,1250,456,1248,458,1244,490,452,1214,478,1218,474,1220,472,1228,476,1218,474,1222,1254,482,448,1218,474,1220,1254,480,450,1216,1248,464,1250,456,1246,488,444,1222,1254,456,1248,458,1244,490,452,1214,1250,462,1252,482,448,1216,1248,488,442,1224,1252,456,474,1220,472,1222,1254'
      qos: 2
  mode: single
elitescreen_stop:
  alias: elitescreen stop
  sequence:
  - service: mqtt.publish
    data:
      payload: '0,1252,482,1220,486,1216,490,1224,456,496,1252,418,1248,444,1278,424,1276,418,1278,498,1170,446,1248,1218,488,452,1268,1196,510,420,1276,1200,482,1220,514,1200,508,422,1272,420,1276,1188,518,1216,490,420,1274,418,1254,448,1272,1192,514,1198,508,1196,510,1192,488,442,1254,450,1246,446,30506,1246,460,1220,484,1218,516,1198,508,422,1274,418,1276,416,1280,424,1248,476,1220,472,1222,448,1272,1192,516,426,1242,1222,512,418,1276,1198,510,1192,514,1200,534,418,1252,440,1254,1190,518,1196,510,420,1248,444,1254,448,1246,1218,488,1224,508,1194,538,1164,488,442,1254,450,1246,444'
      topic: [CHANGE TO CORRECT TOPIC eg. "cmnd/tasmota_40E0B0/IRsend" without quotes]
      qos: 2
  mode: single
```
I had to add the frequency by adding a leading 0 to the raw data. This 0 represents the default frequency of 38kHz. And as you can see, for raw IRSend you have just to send the raw data, no big JSON-structure or anything.

PR #1310 (2020-10-26): Add support for Elite Screens protocol.
* A constant bit time 32-bit protocol with a single repeat.
* Add `sendElitescreens()` & `decodeElitescreens()`
* Unit test coverage including real & synthetic examples.

Fixes #1306
commit b77b0e1a (2020-10-26): Add support for Elite Screens protocol. (#1310)

* Add support for Elite Screens protocol.
* A constant bit time 32-bit protocol with a single repeat.
* Add `sendElitescreens()` & `decodeElitescreens()`
* Unit test coverage including real & synthetic examples.
* decoding confirmed to be working.
See https://github.com/crankyoldgit/IRremoteESP8266/issues/1306#issuecomment-716120736

Fixes #1306
closed 2020-10-25 by linuxlurak