crankyoldgit/IRremoteESP8266 #1309: Sharp CRMC-A705JBEZ Support

burnt-past, 2020-10-24:
Hey Guys,

Looking for some help to add this remote to the library.  It is as the title suggests a Sharp CRMC-A705JBEZ Remote :)

Here is the link to the google sheet I put together: https://docs.google.com/spreadsheets/d/1Lvkk90cMGtSOWkd8jEdlw8cklFcfK6tJ8rm4FIKi-kA/edit?usp=sharing

Currently running lib v2.7.11

Let me know if you need anything else.

Thanks in advance!

crankyoldgit, 2020-10-25:
Thanks. It's not 100% clear what you're asking for/what the data in the sheet says.
e.g. Are the things in red working/correct, or just what is changing?

Can you be clearer what is working fine, and what is not working fine for this particular model? i.e. Please be more verbose.

Also:

- Is it being detected as `SHARP_AC` etc?
- Can you provide a bigger picture of the remote. It's too hard to read the buttons etc.
- Are there only Cool, Dry, and Auto (Fan?) modes for this device?

As far as I can tell, it seems only the fan speeds need to be tweaked and econo equals light. Is that all that needs to be changed/fixed?

burnt-past, 2020-10-25:
Sorry I didn't mention it, yes the red marked test are the changing values.

   • The IR Dump does recognize it as SHARP_AC.
   • I have added a larger photo to the Google Sheet.
   • The modes are Cool, Dry, Fan Only, and Exhaust Fan. There is no auto mode, there is only an auto fan setting.

The values coming from the library don't seem to control the AC at all. I have tried sending raw data as well, but it seems the library modifies it before sending it? Nothing in the current Sharp Class seems to make my AC do anything.

crankyoldgit, 2020-10-25:
> Sorry I didn't mention it, yes the red marked test are the changing values.
> 
> • The IR Dump does recognize it as SHARP_AC.
> • I have added a larger photo to the Google Sheet.
> • The modes are Cool, Dry, Fan Only, and Exhaust Fan. There is no auto mode, there is only an auto fan setting.

Thanks for that.

> The values coming from the library don't seem to control the AC at all. I have tried sending raw data as well, but it seems the library modifies it before sending it? Nothing in the current Sharp Class seems to make my AC do anything.

Can you please post example code for what you mean here?
`sendRaw()` does not modify it before sending. If that isn't working, you've probably got some other fundamental problems.

burnt-past, 2020-10-25:
> Can you please post example code for what you mean here?
sendRaw() does not modify it before sending. If that isn't working, you've probably got some other fundamental problems.

Yeah, my bad. Started a brand new C++ file and `setRaw()` seems to be working now, not sure what went wrong there...

This does work:
```
#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ir_Sharp.h>

const uint16_t kIrLed = D8; // ESP8266 GPIO pin to use. Recommended: 4 (D2).
IRSharpAc ac(kIrLed);      // Set the GPIO used for sending messages.

const uint8_t on[13] = {0xAA, 0x5A, 0xCF, 0x10, 0xD1, 0x11, 0x22, 0x00, 0x08, 0x80, 0x00, 0xF0, 0xF1};

void setup()
{
  ac.begin();
  ac.setRaw(on);
  ac.send();
}

void loop() {}
```

This does not:
```
#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ir_Sharp.h>

const uint16_t kIrLed = D8; // ESP8266 GPIO pin to use. Recommended: 4 (D2).
IRSharpAc ac(kIrLed);      // Set the GPIO used for sending messages.

void setup()
{
  ac.begin();
  ac.on();
  ac.send();
}

void loop() {}
```

burnt-past, 2020-10-25:
Awesome! Looks like the remote is now getting correctly identified in the Dump program.

crankyoldgit, 2020-10-26:
Thanks. It still has a way to go though to support your device.

burnt-past, 2020-10-26:
I'm just super glad you even taking the time to do so! Thanks a lot!

crankyoldgit, 2020-10-27:
@burnt-past Can you please download, try out, & report on PR #1313 / Branch https://github.com/crankyoldgit/IRremoteESP8266/tree/Sharp_CRMC-A705JBEZ

Both sending and decoding please.
I think it covers all the things you listed in the spreadsheet.

burnt-past, 2020-10-29:
Receive, seems to be fully functional.

Send, seems to not be functioning as expected. It could be that I am using `setModel()` incorrectly?

```
#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ir_Sharp.h>

const uint16_t kIrLed = D8;
IRSharpAc ac(kIrLed);

void setup()
{
  ac.begin();
  ac.setModel(A705);
  ac.setTemp(20);
  ac.send();
}

void loop() {}
```

This is what is being read by the the IR dump.
```
Protocol  : SHARP_AC
Code      : 0xAA5ACF1000012000088000F001 (104 Bits)
Mesg Desc.: Model: 1 (A907), Power: Off, Mode: 0 (Auto), Temp: 15C, Fan: 2 (Auto), Turbo: Off, Swing(V) Toggle: Off, Ion: Off, Econo: -, Clean: Off
uint16_t rawData[211] = {3788, 1946,  504, 472,  504, 1370,  506, 472,  506, 1368,  508, 472,  506, 1370,  504, 474,  506, 1370,  506, 472,  506, 1372,  504, 470,  506, 1372,  528, 1348,  518, 458,  506, 1370,  516, 464,  526, 1348,  506, 1372,  506, 1370,  506, 1372,  506, 470,  530, 446,  506, 1370,  506, 1372,  506, 470,  530, 448,  506, 470,  506, 472,  504, 1370,  530, 448,  506, 470,  506, 472,  530, 446,  506, 472,  508, 468,  506, 470,  506, 472,  506, 470,  506, 470,  506, 472,  506, 1372,  506, 470,  506, 470,  506, 470,  506, 470,  506, 472,  528, 448,  506, 472,  526, 450, 
 506, 470,  506, 472,  504, 472,  506, 470,  506, 1372,  506, 470,  530, 448,  508, 468,  506, 472,  506, 470,  506, 472,  506, 470,  504, 472,  506, 470,  506, 472,  528, 446,  506, 470,  506, 470,  506, 1370,  506, 470,  506, 470,  506, 470,  506, 472,  506, 472,  506, 470,  506, 472,  504, 470,  506, 470,  506, 472,  530, 446,  504, 1372,  506, 470,  506, 472,  506, 472,  506, 470,  504, 472,  504, 472,  506, 470,  506, 472,  504, 472,  506, 472,  522, 454,  504, 472,  504, 1372,  506, 1370,  506, 1370,  506, 1372,  506, 1370,  506, 472,  504, 472,  506, 472,  504, 472,  506, 470,  
506, 470,  506, 472,  506};  // SHARP_AC
uint8_t state[13] = {0xAA, 0x5A, 0xCF, 0x10, 0x00, 0x01, 0x20, 0x00, 0x08, 0x80, 0x00, 0xF0, 0x01};
```

crankyoldgit, 2020-10-29:
Thanks for the feedback.
What would be/is the equivalent _real_ corresponding message from the remote?

crankyoldgit, 2020-10-31:
> Thanks for the feedback.
> What would be/is the equivalent _real_ corresponding message from the remote?

@burnt-past reminder. I can't proceed/fix it without this data.

burnt-past, 2020-10-31:
@crankyoldgit I apologise. I am out of town for Halloween weekend and I do not have hardware with me. I will update you on monday

crankyoldgit, 2020-10-31:
No worries.

burnt-past, 2020-11-04:
Hey, sorry for the late response. 

It would seem that the default mode "AUTO" is not a parameter my remote has. Going back to the test code I used above, it seems that the temperature is not changed by what I had set.

Here is the expected message:

```
Protocol  : SHARP_AC
Code      : 0xAA5ACF10D5117200088000F0E1 (104 Bits)
Mesg Desc.: Model: 2 (A705), Power: On, Mode: 2 (Cool), Temp: 20C, Fan: 7 (High), Turbo: Off, Swing(V) Toggle: Off, Ion: Off, Econo: -, Clean: Off
uint16_t rawData[211] = {3792, 1892,  468, 472,  468, 1440,  448, 494,  448, 1438,  472, 468,  448, 1416,  470, 494,  470, 1394,  468, 494,  448, 1416,  472, 470,  468, 1420,  468, 1418,  468, 468,  494, 1392,  470, 492,  446, 1418,  470, 1418,  468, 1418,  472, 1420,  466, 470,  494, 444,  472, 1416,  496, 1390,  496, 470,  472, 466,  446, 494,  446, 470,  470, 1416,  496, 468,  448, 490,  448, 490,  448, 1416,  470, 472,  466, 1442,  448, 468,  468, 1442,  446, 472,  468, 1418,  470, 1416,  472, 1418,  486, 452,  494, 466,  422, 498,  468, 1440,  448, 470,  494, 468,  446, 470,  470, 492,  448, 1418,  468, 470,  472, 490,  470, 1396,  470, 1440,  448, 1416,  472, 490,  448, 470,  468, 496,  444, 494,  448, 468,  470, 492,  448, 492,  446, 472,  468, 470,  470, 472,  466, 474,  494, 444,  492, 1418,  470, 446,  470, 470,  468, 472,  468, 492,  446, 470,  468, 472,  466, 472,  468, 472,  470, 470,  468, 470,  468, 492,  474, 1390,  470, 462,  502, 466,  446, 474,  468, 468,  472, 472,  468, 470,  468, 492,  446, 472,  494, 466,  446, 472,  468, 492,  472, 466,  448, 1416,  496, 1416,  446, 1420,  468, 1418,  470, 1420,  468, 468,  470, 492,  448, 492,  446, 494,  446, 1442,  446, 1422,  466, 1422,  466};  // SHARP_AC
uint8_t state[13] = {0xAA, 0x5A, 0xCF, 0x10, 0xD5, 0x11, 0x72, 0x00, 0x08, 0x80, 0x00, 0xF0, 0xE1};
```

crankyoldgit, 2020-11-04:
Okay, looking at the reported data from the remote, the state the remote is sending is different from your example code. The are NOT the same state at all. Not even close.

e.g. The state you've provided is "On", (your example code doesn't set "On" at all e.g. `ac.on()`). The default is "Off" for the inital state.
Next, you've got the Fan set to "High" (`7`) from the remote, the default is "Auto" for the Fan. You haven't set that.
Finally, the mode from your remote is "Cool" (`2`), the default for the class is `0`. Which is Fan mode on the `A705`, and Auto on `A907`. See:
```
const uint8_t kSharpAcAuto =                             0b00;  // A907 only
const uint8_t kSharpAcFan =                              0b00;  // A705 only
```

You're correct that there is no "Auto" mode when in `A705` mode. The default is `Fan` mode in `A705` mode. That mode locks the temp. i.e. You need to set the mode first to something that accepts different temperatures.

So, try this:
```
  ac.begin();
  ac.setModel(sharp_ac_remote_model_t::A705);
  ac.setMode(kSharpAcCool);  // 2
  ac.setTemp(20);
  ac.setFan(kSharpAcFanMax);  // 7
  ac.on();
  ac.send();
```

That should hopefully give you roughly the same message as you captured from the remote.

Unfortunately, there is no reliable way to determine the model when the _captured_ state is in "Auto" mode for an A907 or "Fan" for an A705. They appear exactly the same.

burnt-past, 2020-11-05:
Whoops sorry about the mismatch there. Here is what I am getting now with the correct settings on my remote.

```
Protocol  : SHARP_AC  
Code      : 0xAA5ACF10D5117200088000F0E1 (104 Bits)
Mesg Desc.: Model: 2 (A705), Power: On, Mode: 2 (Cool), Temp: 20C, Fan: 7 (High), Turbo: Off, Swing(V) Toggle: Off, Ion: Off, Econo: -, Clean: Off
uint16_t rawData[211] = {3822, 1884,  478, 462,  476, 1410,  478, 462,  476, 1410,  476, 464,  476, 1410,  476, 464,  476, 1410,  476, 464,  476, 1410,  478, 462,  476, 1412,  478, 1410,  476, 464,  474, 1412,  478, 464,  476, 1410,  478, 1410,  476, 1410,  478, 1410,  476, 464,  478, 462,  456, 1432,  478, 1410,  476, 464,  456, 484,  458, 482,  480, 460,  478, 1408,  480, 460,  480, 460,  484, 456,  480, 1406,  482, 458,  484, 1404,  484, 456,  484, 1402,  478, 462,  510, 1376,  480, 1408,  492, 1396,  484, 456,  484, 454,  514, 426,  510, 1376,  482, 458,  514, 426,  512, 426,  514, 426,  512, 1374,  512, 428,  508, 432,  510, 1376,  484, 1404,  508, 1378,  510, 430,  510, 428,  510, 430,  508, 430,  508, 432,  484, 454,  510, 432,  510, 428,  484, 456,  512, 428,  510, 430,  512, 428,  512, 1374,  486, 456,  510, 428,  484, 456,  510, 428,  484, 456,  486, 452,  512, 426,  514, 426,  510, 430,  512, 428,  510, 430,  510, 1376,  484, 456,  512, 428,  512, 428,  510, 430,  512, 426,  512, 426,  484, 454,  510, 430,  514, 426,  516, 422,  512, 428,  508, 430,  508, 1378,  484, 1402,  512, 1376,  484, 1404,  484, 1404,  482, 458,  482, 458,  482, 458,  480, 458,  482, 1404,  482, 1406,  458, 1430,  456};  // SHARP_AC
uint8_t state[13] = {0xAA, 0x5A, 0xCF, 0x10, 0xD5, 0x11, 0x72, 0x00, 0x08, 0x80, 0x00, 0xF0, 0xE1};
```

This is the message sent from your comment.

>   ac.begin();
>   ac.setModel(sharp_ac_remote_model_t::A705);
>   ac.setMode(kSharpAcCool);  // 2
>   ac.setTemp(20);
>   ac.setFan(kSharpAcFanMax);  // 7
>   ac.on();
>   ac.send();

```
Protocol  : SHARP_AC
Code      : 0xAA5ACF10D5317200088000F0C1 (104 Bits)
Mesg Desc.: Model: 2 (A705), Power: On, Mode: 2 (Cool), Temp: 20C, Fan: 7 (High), Turbo: Off, Swing(V) Toggle: Off, Ion: Off, Econo: -, Clean: Off
uint16_t rawData[211] = {3806, 1928,  476, 500,  474, 1400,  504, 476,  502, 1376,  498, 474,  504, 1372,  502, 474,  476, 1398,  490, 490,  476, 1398,  504, 474,  504, 1372,  504, 1372,  504, 474,  500, 1374,  504, 472,  504, 1376,  502, 1372,  504, 1374,  502, 1376,  500, 476,  474, 500,  500, 1374,  506, 1374,  502, 474,  502, 474,  500, 472,  504, 472,  502, 1376,  506, 494,  476, 478,  502, 474,  500, 1376,  500, 476,  476, 1400,  500, 476,  500, 1376,  500, 478,  502, 1372,  476, 1402,  484, 1396,  504, 474,  500, 472,  502, 476,  502, 1372,  476, 1402,  504, 474,  474, 504,  472, 500,  504, 1370,  504, 474,  504, 474,  502, 1372,  498, 1378,  500, 1376,  504, 474,  502, 474,  502, 474,  500, 474,  506, 470,  502, 476,  476, 500,  504, 472,  476, 502,  500, 476,  502, 474,  502, 474,  500, 1374,  504, 474,  496, 478,  500, 480,  496, 480,  504, 468,  480, 502,  502, 472,  498, 478,  498, 480,  474, 502,  500, 478,  496, 1380,  500, 478,  500, 474,  502, 476,  502, 474,  502, 472,  498, 478,  498, 478,  500, 476,  502, 474,  498, 478,  498, 476,  496, 480,  500, 1374,  500, 1374,  502, 1374,  474, 1402,  498, 1380,  498, 476,  500, 478,  498, 478,  472, 506,  500, 474,  498, 1380,  498, 1382,  474};  // SHARP_AC
uint8_t state[13] = {0xAA, 0x5A, 0xCF, 0x10, 0xD5, 0x31, 0x72, 0x00, 0x08, 0x80, 0x00, 0xF0, 0xC1};
```

Looks like the sixth byte wasn't matching, which the power state.

I was able to get the AC to turn on by using `ac.setPower(true, false);` Which works perfectly! I will just have to set that if I am ever unsure of its state.

crankyoldgit, 2020-11-06:
> I was able to get the AC to turn on by using `ac.setPower(true, false);` Which works perfectly! I will just have to set that if I am ever unsure of its state.

Ah yes. My bad. I should have set that parameter myself and mentioned it. That is done automatically in the[ `IRac` class](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRac.html).
See:
https://github.com/crankyoldgit/IRremoteESP8266/blob/f1576855de2c7da8bf5163ae3cb53c1b075c4ee2/src/IRac.cpp#L1691

crankyoldgit, 2020-11-07:
@burnt-past So, how are we going with this issue. I'm not sure there is much left for us to do. Or am I missing something? Do you concur?

burnt-past, 2020-11-07:
Yes, all good. Marking as closed.

commit a8eec563 (2020-10-27): SharpAc: Report fan speeds correctly for A705 models.

Fixes #1309
PR #1313 (2020-11-09): SharpAc: Add model support for A705
* Support models `A907` & `A705`
* Remove `this->` from Sharp.
* Update unit tests accordingly.
* Update common A/C api.

Fixes #1309
commit a74eff11 (2020-11-09): SharpAc: Add model support for A705 (#1313)

* Support models `A907` & `A705`
* Remove `this->` from Sharp.
* Update unit tests accordingly.
* Update common A/C api.

Fixes #1309
closed 2020-11-07 by burnt-past