crankyoldgit/IRremoteESP8266 #1315: IRremoteESP8266 use with Esphome

ahmetem, 2020-10-28:
i want to  use this library with esphome . 
Esphome has many protocol for ac but VESTEL_AC not.
i want to use VESTEL_AC .
can this add esphome ?

taxx, 2020-10-28:
If VESTEL_AC is part of IRremoteESP8266 _(which it [seems to be](https://github.com/crankyoldgit/IRremoteESP8266/search?q=VESTEL_AC))_ then you can use this [method described here](https://github.com/esphome/feature-requests/issues/444#issuecomment-548166019) in ESPHome repo to get it working (probably).

NiKiZe, 2020-10-28:
If you want updates to be done to some other library or project you should use there, not here.

I will close this since you are asking for help with something that is outside of this lib, This is in the realm of https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#I_want_to_change_the_example_code_to_make_it_do_something_different_but_I_dont_know_how_Please_help_me

Another reason is because this issue was created and totally ignoring the template - I'm considering creating a Label for "ignoring issue template" and closing any such issues from now on.

closed 2020-10-28 by NiKiZe

wojciechczyz, 2025-05-29:
Possibly this project will help:
https://github.com/mistic100/ESPHome-IRremoteESP8266